raspberrypi/linux #5706: Reading values from SPS30 sensor leads to Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: sps30_read_raw

oniongarlic, 2023-11-08:
### Describe the bug

Trying to read specific values from sps30 (https://sensirion.com/products/catalog/SPS30/) sensor leads to kernel panic:

[  105.317088] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: sps30_read_raw+0x1c0/0x1c0 [sps30]

Reading from in_massconcentration_pm1_input and in_massconcentration_pm2p5_input works fine,
panic on in_massconcentration_pm4_input

### Steps to reproduce the behaviour

Configure Sensirion sps30 sensor on i2c, try to read sensor values.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)
Linux raspberrypi 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

### Logs

[ 1340.212930] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: sps30_read_raw+0x1c0/0x1c0 [sps30]
[ 1340.224134] CPU: 1 PID: 920 Comm: cat Tainted: G         C         6.1.0-rpi6-rpi-v8 #1  Debian 1:6.1.58-1+rpt2
[ 1340.234433] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[ 1340.240383] Call trace:
[ 1340.242881]  dump_backtrace.part.0+0xec/0x100
[ 1340.247338]  show_stack+0x20/0x30
[ 1340.250726]  dump_stack_lvl+0x88/0xb4
[ 1340.254472]  dump_stack+0x18/0x34
[ 1340.257860]  panic+0x1a0/0x370
[ 1340.260984]  __stack_chk_fail+0x28/0x30
[ 1340.264904]  sps30_trigger_handler+0x0/0xdc [sps30]
[ 1340.269897]  iio_read_channel_info+0xe0/0xf4 [industrialio]
[ 1340.275632]  dev_attr_show+0x28/0x64
[ 1340.279293]  sysfs_kf_seq_show+0x94/0x110
[ 1340.283388]  kernfs_seq_show+0x34/0x40
[ 1340.287223]  seq_read_iter+0x16c/0x45c
[ 1340.291054]  kernfs_fop_read_iter+0x148/0x1b4
[ 1340.295508]  vfs_read+0x214/0x2c0
[ 1340.298898]  ksys_read+0x78/0x10c
[ 1340.302284]  __arm64_sys_read+0x24/0x30
[ 1340.306202]  invoke_syscall+0x50/0x120
[ 1340.310036]  el0_svc_common.constprop.0+0x68/0x124
[ 1340.314932]  do_el0_svc+0x34/0xd0
[ 1340.318324]  el0_svc+0x30/0x9c
[ 1340.321446]  el0t_64_sync_handler+0xf4/0x120
[ 1340.325808]  el0t_64_sync+0x18c/0x190
[ 1340.332137] SMP: stopping secondary CPUs
[ 1340.338700] Kernel Offset: 0x16e2800000 from 0xffffffc008000000
[ 1340.346353] PHYS_OFFSET: 0x0
[ 1340.350587] CPU features: 0x80000,2013c080,0000421b
[ 1340.356847] Memory Limit: none
[ 1340.361237] ---[ end Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: sps30_read_raw+0x1c0/0x1c0 [sps30] ]---

oniongarlic, 2024-03-13:
Upgraded to 6.6.20, still panics.

pelwell, 2024-03-13:
This is unlikely to be a Pi-specific problem, in that the error suggests that something is writing off the end of a structure on the stack, causing stack corruption. @tduszynski, listed as the maintainer, might have some suggestions.

tduszynski, 2024-03-14:
There may be many reasons for this failure, ranging from misbehaving sensor to regression in driver itself. Note that driver was written for the initial release of that sensor and given it supports firmware upgrade driver may no longer be even compatible with devices which are shipped now. 

BTW, have you tried with triggered buffer and uart?

That said, I'll try to verify this using device I have, on condition it still operates properly. But that may require some patience from you because I'm swamped with work.

pelwell, 2024-03-14:
Thank you for responding and considering it. I had a look through the code and the cause was not at all obvious.

oniongarlic, 2024-03-14:
Thanks for looking into it. As mentioned the sensor does work fine with an older kernel, 5.10.103.

tduszynski, 2024-03-15:
Okay, missed that comment. If that is the case and you already have setup next to you, to speed problem resolution up, just do a bisection which should lead you to a culprit.

oniongarlic, 2024-06-11:
I think something was done incorrectly when adding support for the serial interface type was done in 5.14 ( https://kernelnewbies.org/Linux_5.14#Industrial_I.2FO_.28iio.29 ), the buffer handling in i2c measurement reading seems a bit odd to me.