raspberrypi/linux #5709: Update to kernel 6.1.x breaks recording/streaming from HifiBerry

rmens, 2023-11-10:
### Describe the bug

I have a Hifiberry DAC ADC that I use to stream audio with ffmpeg. This worked great with `2023-02-21-raspios-bullseye-arm64-lite`. After doing an OS update and going from kernel `5.15.84` to kernel `6.1.0` the streaming stopped working with ALSA errors and ffmpeg can't read from the device any more. The ffmpeg version stayed the same so that's not it.

### Steps to reproduce the behaviour

- Install `2023-02-21-raspios-bullseye-arm64-lite` and don't upgrade, ensure you have kernel `5.15.84-v8+`
- Stream with the command `ffmpeg -f alsa -channels 2 -sample_rate 48000 -hide_banner -re -y -i default:CARD=sndrpihifiberry /tmp/test.wav` 
- Everything works fine
- Now do a full os update via apt
- Try recording again with kernel `6.1.0` 
- It can't sample the Hifiberry and the recording will fail

### Device (s)

Raspberry Pi 4 Mod. B

### System

- Raspberry Pi reference 2023-10-10
- version 03dc77429335caee083e22ddc8eec09c07f12a7a (clean) (release) (start)
- Linux raspberrypi 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

### Logs

Kernel 5.15

```
root@raspberrypi:~# ffmpeg -f alsa -channels 2 -sample_rate 48000 -hide_banner -re -y -i default:CARD=sndrpihifiberry /tmp/test.wav
Input #0, alsa, from 'default:CARD=sndrpihifiberry':
  Duration: N/A, start: 1699647361.315337, bitrate: 1536 kb/s
    Stream #0:0: Audio: pcm_s16le, 48000 Hz, stereo, s16, 1536 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (pcm_s16le (native) -> pcm_s16le (native))
Press [q] to stop, [?] for help
Output #0, wav, to '/tmp/test.wav':
  Metadata:
    ISFT            : Lavf58.45.100
    Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 48000 Hz, stereo, s16, 1536 kb/s
    Metadata:
      encoder         : Lavc58.91.100 pcm_s16le
size=    2048kB time=00:00:11.06 bitrate=1516.4kbits/s speed=0.999x   
```

Kernel 6.1.0-rpi6-rpi-v8

```
root@raspberrypi:/# ffmpeg -f alsa -channels 2 -sample_rate 48000 -hide_banner -re -y -i default:CARD=sndrpihifiberry /tmp/test.wav
Input #0, alsa, from 'default:CARD=sndrpihifiberry':
  Duration: N/A, start: 1699647622.339270, bitrate: 1536 kb/s
    Stream #0:0: Audio: pcm_s16le, 48000 Hz, stereo, s16, 1536 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (pcm_s16le (native) -> pcm_s16le (native))
Press [q] to stop, [?] for help
Output #0, wav, to '/tmp/test.wav':
  Metadata:
    ISFT            : Lavf58.45.100
    Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 48000 Hz, stereo, s16, 1536 kb/s
    Metadata:
      encoder         : Lavc58.91.100 pcm_s16le
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 17, current: 8; changing to 17. This may result in incorrect timestamps in the output file.
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 17, current: 14; changing to 17. This may result in incorrect timestamps in the output file.
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 19, current: 16; changing to 19. This may result in incorrect timestamps in the output file.
[alsa @ 0x5588de7020] ALSA buffer xrun.e=1574.5kbits/s speed=0.0117x    
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 137736, current: 137730; changing to 137736. This may result in incorrect timestamps in the output file.
[alsa @ 0x5588de7020] ALSA buffer xrun.e=  34.6kbits/s speed=0.524x    
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 275487, current: 275452; changing to 275487. This may result in incorrect timestamps in the output file.
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 275487, current: 275469; changing to 275487. This may result in incorrect timestamps in the output file.
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 275487, current: 275479; changing to 275487. This may result in incorrect timestamps in the output file.
[alsa @ 0x5588de7020] ALSA buffer xrun.e=  25.1kbits/s speed=0.716x    
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 413310, current: 413299; changing to 413310. This may result in incorrect timestamps in the output file.
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 413310, current: 413293; changing to 413310. This may result in incorrect timestamps in the output file.
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 413310, current: 413300; changing to 413310. This may result in incorrect timestamps in the output file.
[wav @ 0x5588e05150] Non-monotonous DTS in output stream 0:0; previous: 413310, current: 413307; changing to 413310. This may result in incorrect timestamps in the output file.
size=      20kB time=00:00:08.61 bitrate=  18.8kbits/s speed=0.947x    
```

### Additional context

This is reproduced on the exact same Pi, with the exact same hardware, the exact same audio input and the exact same ffmpeg version. I could also reproduce this with another card: the HiFiBerry Digi+ I/O.

Also tried this with Ubuntu which has the same problem. Ubuntu 22.04 LTS has kernel 5.15 which works. Ubuntu 23.10 has kernel 6.1.x and has the same problem.

When installing `2023-02-21-raspios-bullseye-arm64-lite` and holding back the `raspberry pi-kernel` package in apt but upgrading everything else, it keeps working. I'm pretty sure this is kernel related.

moodeaudio, 2023-11-12:
Hi, I happen to have a DAC+ ADC and a quick test showed no issues at least on kernel 6.1.21 / Bullseye. Sometimes having a success case can help.

Pipeline 
`iMac -> Headphone out -> [ DAC+ ADC analog in -> alsaloop -> DAC+ ADC RCA out ] -> Powered speakers.`

Here's some console output.
```
pi@moode:~ $ uname -a
Linux moode 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

pi@moode:~ $ cat /etc/debian_version 
11.8

pi@moode:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplusadc], device 0: HiFiBerry DAC+ADC HiFi multicodec-0 [HiFiBerry DAC+ADC HiFi multicodec-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

pi@moode:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplusadc], device 0: HiFiBerry DAC+ADC HiFi multicodec-0 [HiFiBerry DAC+ADC HiFi multicodec-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

pi@moode:~ $ pgrep -l alsaloop
9781 alsaloop

pi@moode:~ $ dpkg -l | grep raspberrypi
ii  libraspberrypi-bin                   1:2+git20230322~143557+9d5250f-1 arm64        Miscellaneous Raspberry Pi utilities
ii  libraspberrypi-dev                   1:2+git20230322~143557+9d5250f-1 arm64        Libraries for the Raspberry Pi's VideoCore IV (headers)
ii  libraspberrypi-doc                   1:2+git20230322~143557+9d5250f-1 all          Libraries for the Raspberry Pi's VideoCore IV (docs)
ii  libraspberrypi0:arm64                1:2+git20230322~143557+9d5250f-1 arm64        Libraries for the Raspberry Pi's VideoCore IV
ii  raspberrypi-archive-keyring          2021.1.1+rpt1                    all          GnuPG archive keys of the Raspberry Pi OS archive
hi  raspberrypi-bootloader               1:1.20230405-1                   arm64        Raspberry Pi bootloader
hi  raspberrypi-kernel                   1:1.20230405-1                   arm64        Raspberry Pi bootloader
ii  raspberrypi-net-mods                 1.3.4                            all          Network configuration for the Raspberry Pi UI
ii  raspberrypi-sys-mods                 20230510~bullseye                arm64        System tweaks for the Raspberry Pi
```

dcabri, 2023-11-12:
Hi, I can concur and I am also seeing exactly the behaviour described above, also for kernel 6.1.21-v8+.

dcabri, 2023-11-12:
Yes, we have the same experience.
It’s with direct alsa access where things  go wrong. When you use the explained  above ffmpeg, you will notice it goes wrong…

rmens, 2023-11-12:
@moodeaudio Would you mind trying to reproduce my FFmpeg example instead of al ALSA loop?

moodeaudio, 2023-11-12:
Sure, no prob.
Using your ffmpeg example cmd I get exact same results as you did.

As a data point the arecord command below worked fine with the DAC+ ADC and MPD was able to play the generated wav file.
```
pi@moode:~ $ arecord -c 2 -f S16_LE -r 48000 -t wav --duration=60 ./test.wav
Recording WAVE './test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
```
This suggests the ALSA plumbing is prolly ok. Bullseye ffmpeg is 4.3.6 while Bookworm includes newer version 5.1.3. Maybe this version is ok with 6.1 kernel? We are still at the beginning of Bookworm builds and so I'm not able to test much yet.

dcabri, 2023-11-13:
Hi,
On Raspberry Pi 4 latest update/upgrade, I recompiled FFmpeg using the script provided by
https://github.com/NRCHKB/nrchkb-ffmpeg-build, but adding -- ALSA module using:

```
# Install alsa
installalsa() {
    cd ~ || {
        echo "cd failed, aborting at alsa-lib:01"
        exit 1
    }
    echo
    echo " ---------------------------------------------------------"
    echo " |                                                       |"
    echo " |            Building/Installing alsa-lib               |"
    echo " |                                                       |"
    echo " ---------------------------------------------------------"
    echo

    wget "https://www.alsa-project.org/files/pub/lib/alsa-lib-1.2.9.tar.bz2"
    tar xjf alsa-lib-1.2.9.tar.bz2
    cd alsa-lib-1.2.9/
    ./configure --host=arm-unknown-linux-gnueabi --prefix=$PREFIX
    make
    sudo make install
    checkForError
}
```

After this, I get a compiled ffmpeg binary, but when I use that one, I get identical result:
```
pi% ./ffmpeg -f alsa -channels 2 -sample_rate 48000 -re -y -i 'hw:CARD=sndrpihifiberry,DEV=0' /tmp/test.wav
ffmpeg version N-112741-gcd7b352c53 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 10 (Debian 10.2.1-6)
  configuration: --extra-cflags=-I/usr/local/include --extra-ldflags=-L/usr/local/lib --extra-libs='-lpthread -lm -latomic' --prefix=/usr/local --arch=arm64 --target-os=linux --enable-nonfree --enable-gpl --enable-hardcoded-tables --enable-gmp --enable-libaom --enable-libass --enable-libdav1d --enable-libdrm --enable-libjack --enable-libfdk-aac --enable-libfreetype --enable-libkvazaar --enable-libmp3lame --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopus --enable-librtmp --enable-libsnappy --enable-libsoxr --enable-libssh --enable-libtheora --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --disable-mmal --enable-version3 --enable-pthreads --enable-libzimg --enable-openssl
  libavutil      58. 32.100 / 58. 32.100
  libavcodec     60. 33.100 / 60. 33.100
  libavformat    60. 17.100 / 60. 17.100
  libavdevice    60.  4.100 / 60.  4.100
  libavfilter     9. 13.100 /  9. 13.100
  libswscale      7.  6.100 /  7.  6.100
  libswresample   4. 13.100 /  4. 13.100
  libpostproc    57.  4.100 / 57.  4.100
[aist#0:0/pcm_s16le @ 0x55cad88750] Guessed Channel Layout: stereo
Input #0, alsa, from 'hw:CARD=sndrpihifiberry,DEV=0':
  Duration: N/A, start: 1699872507.679754, bitrate: 1536 kb/s
  Stream #0:0: Audio: pcm_s16le, 48000 Hz, 2 channels, s16, 1536 kb/s
Stream mapping:
  Stream #0:0 -> #0:0 (pcm_s16le (native) -> pcm_s16le (native))
Press [q] to stop, [?] for help
Output #0, wav, to '/tmp/test.wav':
  Metadata:
    ISFT            : Lavf60.17.100
  Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 48000 Hz, stereo, s16, 1536 kb/s
    Metadata:
      encoder         : Lavc60.33.100 pcm_s16le
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 59, current: 50; changing to 59. This may result in incorrect timestamps in the output file.
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 78, current: 75; changing to 78. This may result in incorrect timestamps in the output file.
[alsa @ 0x55cad78f00] ALSA buffer xrun.e=1512.7kbits/s speed=0.0117x
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 137426, current: 137384; changing to 137426. This may result in incorrect timestamps in the output file.
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 137426, current: 137404; changing to 137426. This may result in incorrect timestamps in the output file.
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 137426, current: 137415; changing to 137426. This may result in incorrect timestamps in the output file.
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 137426, current: 137423; changing to 137426. This may result in incorrect timestamps in the output file.
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 137457, current: 137454; changing to 137457. This may result in incorrect timestamps in the output file.
[alsa @ 0x55cad78f00] ALSA buffer xrun.e=  33.1kbits/s speed=0.523x
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 274744, current: 274734; changing to 274744. This may result in incorrect timestamps in the output file.
[wav @ 0x55cad7a990] Non-monotonic DTS in output stream 0:0; previous: 274744, current: 274731; changing to 274744. This may result in incorrect timestamps in the output file.
[out#0/wav @ 0x55cad88350] video:0kB audio:13kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.604089%
size=      13kB time=00:00:05.72 bitrate=  18.1kbits/s speed=0.937x
Exiting normally, received signal 2.
```

dcabri, 2023-11-13:
This is my script for compilation: 
```
#!/bin/bash

    echo
    echo " ---------------------------------------------------------"
    echo " |                                                   x.x |"
    echo " |           FFmpeg Build Script (Debian)                |"
    echo " |                                                       |"
    echo " ---------------------------------------------------------"
    echo

chkupdate() {
    echo
    echo " ---------------------------------------------------------"
    echo " |              Debian System update                     |"
    echo " ---------------------------------------------------------"
    echo
    sudo apt update -y && sudo apt upgrade -y
}
# chkupdate

installdeps() {
    echo
    echo " ---------------------------------------------------------"
    echo " |               Installing Dependencies                 |"
    echo " ---------------------------------------------------------"
    echo " |                      packages                         |"
    echo " ---------------------------------------------------------"
    sudo apt-get -y install \
	pkg-config \
	autoconf \
	automake \
	build-essential \
	cmake \
	doxygen \
	libtool \
	libtool-bin \
	git \
	wget \
	make \
	g++ \
	gcc \
	nasm \
	yasm \
	graphviz \
	imagemagick \
	meson \
	libtool \
	python3-dev \
	python3-pip \
	texinfo
}
# installdeps

installmods() {
    echo
    echo " ---------------------------------------------------------"
    echo " |                      modules                          |"
    echo " ---------------------------------------------------------"
    sudo apt-get -y install \
        libasound2-dev \
        libass-dev \
        libavcodec-dev \
        libavdevice-dev \
        libavfilter-dev \
        libavformat-dev \
        libavutil-dev \
        libfreetype6-dev \
        libgmp-dev \
[... 160 lines cut ...]
    --enable-libsoxr \
    --enable-libssh \
    --enable-libtheora \
    --enable-libvorbis \
    --enable-libvpx \
    --enable-libzimg \
    --enable-libwebp \
    --enable-libx264 \
    --enable-libx265 \
    --enable-libxml2 \
    --disable-mmal \
    --disable-omx \
    --disable-omx-rpi \
    --target-os=linux \
    --enable-pthreads \
    --enable-openssl \
  && make -j$(nproc) \
  && sudo make install

# Removed:
#   --enable-gnutls      : use openssl or this, but not both
#   --enable-gmp         : gives compilation errors
#   --enable-libopenjpeg : gives compilation errors
#   --arch=armhf         : compiler assembler errors
```

rmens, 2023-11-13:
Thank you for the script @dcabri. I have compiled it by hand and can confirm the issue with the latest FFmpeg on Bullseye 64-bit on a Pi 4. With the latest FFmpeg master it still only works on kernel `5.15.84` but kernel `6.1.21` gives the same error as mentioned above.

nodemand, 2023-11-15:
I have a HifiBerry DAC+ ADC Pro and am using ALSA through PortAudio in my application which works with a USB sound card but cannot record nor playback anything through the HifiBerry.  ALSA lists the device and play works. PortAudio only show the HifiBerry after setting the environment variable PA_ALSA_PLUGHW=1. But when I select the HifiBerry as the device to use for recording and/or playback I get no sound at all! I'm on Linux 6.1.0-rpi6-rpi-2712 aarch64.

pelwell, 2023-12-04:
#5747 is purely Pi 5-specific, and @dcabri is using a Pi 4.

I have to admit to being skeptical that a kernel bump would cause an issue that affected ffmpeg but not arecord (one of several reasons this hasn't received much attention yet), but that indeed appears to be the case. Starting with the February 2023 Bullseye release and rpi-updating to the most recent 6.1 kernel gives the same results using an old IQaudIO Pi-Codec+ board (which acts like a large form-factor rpi-codeczero board).

Test commands:
```
$ sudo alsactl restore -f Pi-Codec/Codec_Zero_StereoMIC_record_and_HP_playback.state
$ arecord --format S16_LE -c2 --rate 48000 arecord1.wav
$ ffmpeg -f alsa -channels 2 -sample_rate 48000 -hide_banner -re -y -i hw:CARD=IQaudIOCODEC ffmpeg1.wav
# Upgrade and repeat...
```

nodemand, 2023-12-04:
> #5747 is purely Pi 5-specific, and @dcabri is using a Pi 4.
> 
🙄oops. Got ya!

dcabri, 2023-12-18:
Small update: ffmpeg is unable to read from the Alsa input device 'hw:CARD=sndrpihifiberry,DEV=0', however ffmpeg is able to write to the Alsa output device 'hw:CARD=C2,DEV=0'. Seems writing is easier than reading :)

bluethumb, 2024-01-21:
Not that it might matter but I have a DacBerry board I try to get running on a pi400. I setup it up and 
I can see the card : tlvaudioCODEC and all the card's features but no control in alsamixer and no sound out 
But I get a lot of errors :
 ASoC: error at soc_component_read_no_lock on tlv320aic3x.1-0018 for register:

I used the default .dtbo and /lib files

regmap_i2c             16384  1 snd_soc_tlv320aic3x_i2c
snd_soc_tlv320aic3x    77824  1 snd_soc_tlv320aic3x_i2c
snd_soc_core          270336  5 snd_soc_bcm2835_i2s,vc4,snd_soc_hdmi_codec,snd_soc_dacberry400,snd_soc_tlv320aic3x
snd_pcm               139264  8 snd_soc_bcm2835_i2s,snd_bcm2835,snd_soc_hdmi_codec,snd_soc_dacberry400,snd_soc_tlv320aic3x,snd_compress,snd_soc_core,snd_pcm_dmaengine

Kernel:  Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

pelwell, 2024-01-21:
Include a reasonable number of complete error messages, not just a fragment.

rmens, 2024-03-31:
Just ran rpi-update, it updated the kernel to `6.6.22-v8-16k+` and that seems to solve my original issue. There wasn't much movement in this ticket, but something might have gotten changed that fixed recording/streaming from a Hifiberry via ffmpeg.

rmens, 2024-06-15:
I apologize for my earlier conclusion; it was premature. The issue is solved when testing with kernel 6.6 on a Raspberry Pi 5, but it still persists on a Raspberry Pi 4. Could you please reopen this issue @popcornmix?

dcabri, 2024-09-21:
The issue is still persisting on 6.6.51-v8-16k+ ... I thought I'd let this be known, so this thread doesn't die...

rmens, 2024-09-21:
Can confirm. I've been holding back the kernel on my Pi 4 ever since opening this issue. It's not very future proof.

pelwell, 2024-09-24:
Well, I've found the culprit: 300689fb04b3f23c1ac1abfe960b48ec414df597, which first appeared in Linux 5.18.

Prior to this commit, the minimum PERIOD_BYTES value was 256, and with this is drops to 16. This isn't a problem for arecord which chooses a period of quarter the buffer size, which equates to 32KB). However, ffmpeg chooses the minimum possible, only falling back to a quarter buffer if no minimum is set (https://github.com/FFmpeg/FFmpeg/blob/master/libavdevice/alsa.c#L260).

There should be a way of overriding that minimum somewhere in the kernel's sound subsystem - it's just a case of finding the best hook to hang it on.

pelwell, 2024-09-25:
#6386 is a small patch that restores the minimum of 256. In about 40 minutes, once the autobuilds have completed, you will be able to install a trial kernel using `sudo rpi-update pulls/6386`.

rmens, 2024-09-27:
Thanks. I have access to test hardware after the weekend. Will test with the trial kernel then.

rmens, 2024-09-29:
Tested with `sudo rpi-update pulls/6386` and that makes it work again on a Pi 4!

rmens, 2024-10-11:
Just tested with a fresh Raspberry Pi OS 12 install that contains kernel 6.6.51+rpt-rpi-v8 after updating.

It works flawlessly now on a Pi 4 and 5. Closing this. Many thanks!

closed 2024-10-11 by rmens