raspberrypi/linux #5710: GPIO 20 and 21 not working as pull-up

andi34, 2023-11-13:
### Describe the bug

On my OpenSource Photobooth software were using the GPIO to trigger different actions. I am not sure if it's a kernel or a firmware thing, but buttons connected to GPIO 20 and 21 don't work any more.

### Steps to reproduce the behaviour

Remotebuzzer Server scrip
https://github.com/PhotoboothProject/photobooth/blob/dev/assets/js/remotebuzzer-server.js

GPIO defined as PU via /boot/config.txt
https://github.com/PhotoboothProject/photobooth/blob/dev/install-photobooth.sh#L799

### Device (s)

Raspberry Pi 4 Mod. B

### System

PiOS bookworm

pelwell, 2023-11-13:
Which OS are you running? If you have raspi-gpio or pinctrl, what does `raspi-gpio get 20-21` (or `pinctrl get 20-21`) report?

pelwell, 2023-11-13:
And if you are on Bookworm, what do get from:
```
$ grep gpio /boot/config.txt /boot/firmware/config.txt
```
?

andi34, 2023-11-13:
> Which OS are you running? If you have raspi-gpio or pinctrl, what does `raspi-gpio get 20-21` (or `pinctrl get 20-21`) report?

[image]

> And if you are on Bookworm, what do get from:
> 
> ```
> $ grep gpio /boot/config.txt /boot/firmware/config.txt
> ```
> 
> ?

Will post asap

pelwell, 2023-11-13:
A small but significant change with Bookworm is that the boot/FAT partition is mounted on /boot/firmware, not /boot. Images come with a symbolic link from /boot/config.txt to /boot/firmware/config.txt, but unfortunately it is quite easy for a link like that to end up being replaced by a copy of the real file, after which edits to it have no effect.

andi34, 2023-11-14:
Thanks for your answer! That might be the reason.
https://github.com/PhotoboothProject/photobooth/pull/465/files
Guess this should fix it.

closed 2023-11-14 by andi34