raspberrypi/linux #5712: Modbus Packet Reading

SoUtHeaST41, 2023-11-15:
### Describe the bug

It is known that the data I need to read is 9926 bytes of data, and my hardware and software communication protocols are serial and modbus ASCII. modbus ASCII protocol does not seem to be able to read such a large amount of data at one time, can I take a packet reading approach, or do you have any better suggestions?

### Steps to reproduce the behaviour

I'm trying to read data from my Thermopile Infrared Array, its size is 80622+6. the software protocol is modbus ascii, the hardware protocol is uart.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi OS

pelwell, 2023-11-15:
> modbus ASCII protocol does not seem to be able to read such a large amount of data at one time

How does this failure manifest itself? An error? Hanging indefinitely?

Don't assume that we understand your application, or even what modbus is.

Do provide some sample code demonstrating your problem.

SoUtHeaST41, 2023-11-15:
> > modbus ASCII protocol does not seem to be able to read such a large amount of data at one time
> 
> How does this failure manifest itself? An error? Hanging indefinitely?
> 
> Don't assume that we understand your application, or even what modbus is.
> 
> Do provide some sample code demonstrating your problem.

My code:
```
import serial
import time

try:
    ser = serial.Serial(
        port='/dev/ttyUSB0',
        baudrate=115200,
        stopbits=1,
        parity='N',
        bytesize=8,
        timeout=1
    )

    while True:
        request_message = "51 75 65 72 79 54 0D 0A"
        request_bytes = bytes.fromhex(request_message.replace(" ", ""))
        print("Sent data:", request_bytes)
        ser.write(request_bytes)

        time.sleep(20)

        expected_length = 80 * 62 * 2 + 6
        response_bytes = b''

        while len(response_bytes) < expected_length:
            remaining_bytes = expected_length - len(response_bytes)
            print("Before reading response")
            try:
                chunk = ser.read(remaining_bytes)
                len_response_bytes = len(response_bytes)
                print("读到的数据:", chunk)
                print("读到的数据的长度:", len_response_bytes)

            except Exception as e:
                print(f"Error reading response:{e}")
            print("After reading response")
            response_bytes += chunk

        print("Received data (hex):", response_bytes)
        response_hex = ' '.join(format(byte, '02X') for byte in response_bytes)
        print("Received data (hex):", response_hex)

        time.sleep(5)

except Exception as e:
    print(f"Error: {e}")
finally:
    ser.close()
```

After running, the terminal displays:
```
Sent data: b'QueryT\r\n'
Before reading response
读到的数据: b''
读到的数据的长度: 0
After reading response
Before reading response
读到的数据: b''
读到的数据的长度: 0
After reading response
Before reading response
读到的数据: b''
读到的数据的长度: 0
After reading response
Before reading response
读到的数据: b''
读到的数据的长度: 0
... ...

```

pelwell, 2023-11-15:
I don't see any indication in the output that the device is responding at all. Why do you think this is a Linux problem and not a hardware/wiring problem?

SoUtHeaST41, 2023-11-15:
> I don't see any indication in the output that the device is responding at all. Why do you think this is a Linux problem and not a hardware/wiring problem?

Because if I'm change the request_message into another which response data is 12 bytes, it runs well, so I think this is a length problem.

pelwell, 2023-11-15:
What happens if you use the UART on GPIOs 14 & 15 (pins 8 & 10) instead?

pelwell, 2023-11-15:
It works for me on Pi 4 running the latest updated Bookworm image, using a Debug Probe as a USB serial port or going via header pins 6 & 8.

SoUtHeaST41, 2023-11-16:
> What happens if you use the UART on GPIOs 14 & 15 (pins 8 & 10) instead?

thank you for your answer. It seems that the uart port on my sensor is bad, because when I connet it to the gpio14  and 15 on raspiberry Pi, it even can't receive the shorter response that I have succeefully recieved on the uart convert to USB briadge before. So I intend that it is a problem of out of range of a frame of data for the modbus ASCII communication protocol.

pelwell, 2023-11-16:
I'm closing the issue as it doesn't seem like a problem with Linux (or even the Pi).

closed 2023-11-16 by pelwell