raspberrypi/linux #5715: tags/stable_20231030: CONFIG_DRM_DISPLAY_CONNECTOR=m fails to build

srcshelton, 2023-11-16:
### Describe the bug

With `CONFIG_DRM_DISPLAY_CONNECTOR=m` or `CONFIG_DRM_DISPLAY_CONNECTOR=y`:

```
  CHK     include/generated/autoksyms.h
  MODPOST Module.symvers - due to: modules.order
ERROR: modpost: "drm_atomic_helper_bridge_duplicate_state" [drivers/gpu/drm/bridge/display-connector.ko] undefined!
ERROR: modpost: "drm_atomic_helper_bridge_destroy_state" [drivers/gpu/drm/bridge/display-connector.ko] undefined!
ERROR: modpost: "drm_atomic_helper_bridge_reset" [drivers/gpu/drm/bridge/display-connector.ko] undefined!
make[3]: *** [/usr/src/linux-6.1.58-raspberrypi/scripts/Makefile.modpost:126: Module.symvers] Error 1
make[2]: *** [/usr/src/linux-6.1.58-raspberrypi/Makefile:1966: modpost] Error 2
make[1]: *** [/usr/src/linux-6.1.58-raspberrypi/Makefile:369: __build_one_by_one] Error 2
make: *** [Makefile:238: __sub-make] Error 2
```

```
  AR      drivers/of/built-in.a - due to: drivers/of/base.o drivers/of/device.o drivers/of/property.o drivers/of/dynamic.o drivers/of/irq.o
  AR      drivers/built-in.a - due to: drivers/gpio/built-in.a drivers/video/built-in.a drivers/clk/built-in.a drivers/gpu/built-in.a drivers/base/built-in.a drivers/dma-buf/built-in.a drivers/i2c/built-in.a drivers/of/built-in.a
  AR      built-in.a - due to: kernel/built-in.a mm/built-in.a fs/built-in.a lib/built-in.a drivers/built-in.a
  CHK     include/generated/autoksyms.h
  AR      vmlinux.a - due to: built-in.a
  LD      vmlinux.o - due to: vmlinux.a
  OBJCOPY modules.builtin.modinfo - due to: vmlinux.o
  GEN     modules.builtin - due to: modules.builtin.modinfo
  GEN     .vmlinux.objs - due to: vmlinux.a
  MODPOST Module.symvers - due to: modules.order vmlinux.o .vmlinux.objs
  CC      .vmlinux.export.o - due to: .vmlinux.export.c
  UPD     include/generated/utsversion.h
  CC      init/version-timestamp.o - due to: include/generated/utsversion.h
  LD      .tmp_vmlinux.kallsyms1
ld: drivers/gpu/drm/bridge/display-connector.o:(.rodata+0x68): undefined reference to `drm_atomic_helper_bridge_duplicate_state'
ld: drivers/gpu/drm/bridge/display-connector.o:(.rodata+0x70): undefined reference to `drm_atomic_helper_bridge_destroy_state'
ld: drivers/gpu/drm/bridge/display-connector.o:(.rodata+0x90): undefined reference to `drm_atomic_helper_bridge_reset'
make[3]: *** [/usr/src/linux-6.1.58-raspberrypi/scripts/Makefile.vmlinux:34: vmlinux] Error 1
make[2]: *** [/usr/src/linux-6.1.58-raspberrypi/Makefile:1255: vmlinux] Error 2
make[1]: *** [/usr/src/linux-6.1.58-raspberrypi/Makefile:369: __build_one_by_one] Error 2
make: *** [Makefile:238: __sub-make] Error 2
```

… it seems that some dependency is missing when 'Display connector support' is enabled in any way.

Setting `CONFIG_DRM_DISPLAY_CONNECTOR=y` requires `CONFIG_DRM=y`, but the second of the above failure-cases still happens.

The setting `DRM_DISPLAY_HELPER` isn't exposed to the user, but perhaps this needs an additional entry to ensure that it is selected if `CONFIG_DRM_DISPLAY_CONNECTOR` is set?

### Steps to reproduce the behaviour

Set `CONFIG_DRM_DISPLAY_CONNECTOR` to any valid value other than `n` in kernel `.config` and rebuild.

Compilation is only successful if `CONFIG_DRM_DISPLAY_CONNECTOR=n` (or there's another undeclared dependency which I don't have selected).

### Device (s)

Raspberry Pi CM4

### System

Hardware is not running a Debian-derived OS;

```
Mar 17 2023 10:50:56 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_cd)
```

```
Linux <hostname> 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

6by9, 2023-11-16:
`DRM_DISPLAY_CONNECTOR=m` is in our default defconfigs eg https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/configs/bcm2711_defconfig#L978 as the compatible vga-connector is used in the vc4-kms-vga666-overlay.dts overlay.

The fact that `DRM_DISPLAY_HELPER` is selected by `DRM_VC4` would mask that omission on any Pi platform, and I would have expected it to be set by any other rendering hardware. Realistically this shouldn't be an issue.

All those Kconfig options for DRM are from mainline Linux, so any fix needs to be authored upstream. It needs a little more investigation before sending that form of patch.

srcshelton, 2023-11-16:
Ahh - this kernel build was broadly intended for headless CM4 use, so I really should simply disable `DRM_DISPLAY_CONNECTOR` in any case… but at the same time I thought it was worth noting that this is a potentially valid configuration (in that `make nconfig`, etc. allows it!) but which doesn't currently compile.

6by9, 2023-11-16:
Looking at the history, 7cd70656d1285b79c001f041a017fcfee4292ff9 appears to have added the use of the drm_atomic_helper_bridge_* functions, but failed to add the `select DRM_DISPLAY_HELPER`. You're the first to notice in the almost exactly 2 years since it was committed.

That looks like a fairly clear cut case of needing a Fixes patch. I am in the process of sorting a load of patches for dri-devel, but as it's such a weird configuration I'm afraid I don't think it warrants creating the patch in this kernel tree first. If it is a real issue to you, please create a PR to fix it up.

srcshelton, 2023-11-16:
I'm happy to have a bash at a patch to do this, but I'll wait for confirmation that such a larger change would be welcomed before looking into it?

6by9, 2023-11-16:
`CONFIG_VC4` supports HDMI audio. There's no option in the driver to disable just the audio side, therefore it has a dependency on `SND` and `SND_SOC`. Adding an option to disable HDMI audio is unlikely to be worth it, hence the dependency. I only mentioned `VC4` as it happens to enable `DRM_KMS_HELPER`.

Ah, I'd made a typo earlier as I'd copied it from your report - it's `DRM_KMS_HELPER' that needs to be enabled, not `DRM_DISPLAY_HELPER`

A `make allnoconfig`, enable `ARCH_BCM2835`, `DRM` and `DRM_DISPLAY_CONNECTOR`, and build does fail.
```
diff --git a/drivers/gpu/drm/bridge/Kconfig b/drivers/gpu/drm/bridge/Kconfig
index 57946d80b02d..f6c0cddc6434 100644
--- a/drivers/gpu/drm/bridge/Kconfig
+++ b/drivers/gpu/drm/bridge/Kconfig
@@ -67,6 +67,7 @@ config DRM_CROS_EC_ANX7688
 config DRM_DISPLAY_CONNECTOR
        tristate "Display connector support"
        depends on OF
+       select DRM_KMS_HELPER
        help
          Driver for display connectors with support for DDC and hot-plug
          detection. Most display controllers handle display connectors
```
fixes it.

6by9, 2023-11-17:
I was creating a PR for DRM related stuff, so stuck the diff in as part of #5716

srcshelton, 2023-11-17:
👍

Separately, rather than having `CONFIG_VC4` hidden unless `CONFIG_SND` and `CONFIG_SND_SOC` are already selected, would it make sense to always show `CONFIG_VC4` and have it _select_ `CONFIG_SND` and `CONFIG_SND_SOC` when chosen?

6by9, 2023-11-17:
Kconfig dependencies are a nightmare to ensure you avoid circular dependencies, so I'm not making any significant changes to the options on `VC4`.