raspberrypi/linux #5719: Can't build Debian packages when using bcm2712_defconfig

andrewluebke, 2023-11-18:
### Describe the bug

When I compile the kernel either on a RPi 5 or on my cross-compile machine and use the bcm2712_defconfig file make bindeb-pkg fails with the following error:

sh ./scripts/package/mkdebian
dpkg-buildpackage -r"fakeroot -u" -a$(cat debian/arch)  -b -nc -uc
dpkg-buildpackage: warning:     debian/changelog(l1): version '6.1.62-v8_16k+-4' is invalid: version number contains illegal character '_'
LINE: linux-upstream (6.1.62-v8_16k+-4) bookworm; urgency=low
dpkg-buildpackage: info: source package linux-upstream
dpkg-buildpackage: info: source version unknown
dpkg-buildpackage: error: version number does not start with digit
make[1]: *** [scripts/Makefile.package:86: bindeb-pkg] Error 255
make: *** [Makefile:1656: bindeb-pkg] Error 2

### Steps to reproduce the behaviour

export ARCH=arm64
export KERNEL=kernel8
export CROSS_COMPILE=/opt/aarch64/bin/aarch64-linux-gnu- 
make bcm2712_defconfig
make Image modules dtbs
make bindeb-pkg

### Device (s)

Other

### System

System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

Linux RPi-5 6.1.0-rpi6-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
Revision        : d04170
Serial          : 3fecfc21f7c0de29
Model           : Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0x0
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

Videocore information
---------------------

2023/10/30 16:45:10 
Copyright (c) 2012 Broadcom
version 30de0ba5 (release) (embedded)

vc_gencmd_read_response returned -1
error=1 error_msg="Command not registered"

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             3949904       0   3949904   0% /dev
tmpfs             824288    5264    819024   1% /run
/dev/sda2      961499068 9593404 903077572   2% /
tmpfs            4121360       0   4121360   0% /dev/shm
tmpfs               5120      48      5072   1% /run/lock
/dev/sda1         522232   62292    459940  12% /boot/firmware
tmpfs             824272       0    824272   0% /run/user/1000

Filename                                Type            Size            Used            Priority
[... 1107 lines cut ...]
[    5.355559] vc4-drm axi:gpu: bcm2712_iommu_of_xlate: MMU 1000005200.iommu
[    5.362309] Console: switching to colour dummy device 80x25
[    5.373608] vc4-drm axi:gpu: bound 107c580000.hvs (ops vc4_hvs_ops [vc4])
[    5.402783] Registered IR keymap rc-cec
[    5.402849] rc rc0: vc4-hdmi-0 as /devices/platform/soc/107c701400.hdmi/rc/rc0
[    5.402907] input: vc4-hdmi-0 as /devices/platform/soc/107c701400.hdmi/rc/rc0/input4
[    5.412439] vc4-drm axi:gpu: bound 107c701400.hdmi (ops vc4_hdmi_ops [vc4])
[    5.438755] Registered IR keymap rc-cec
[    5.438822] rc rc1: vc4-hdmi-1 as /devices/platform/soc/107c706400.hdmi/rc/rc1
[    5.438884] input: vc4-hdmi-1 as /devices/platform/soc/107c706400.hdmi/rc/rc1/input5
[    5.462164] vc4-drm axi:gpu: bound 107c706400.hdmi (ops vc4_hdmi_ops [vc4])
[    5.462357] vc4-drm axi:gpu: bound 107c500000.mop (ops vc4_txp_ops [vc4])
[    5.462455] vc4-drm axi:gpu: bound 107c501000.moplet (ops vc4_txp_ops [vc4])
[    5.462560] vc4-drm axi:gpu: bound 107c410000.pixelvalve (ops vc4_crtc_ops [vc4])
[    5.462660] vc4-drm axi:gpu: bound 107c411000.pixelvalve (ops vc4_crtc_ops [vc4])
[    5.548171] [drm] Initialized vc4 0.0.0 20140616 for axi:gpu on minor 1
[    5.603612] Console: switching to colour frame buffer device 240x67
[    5.621869] vc4-drm axi:gpu: [drm] fb0: vc4drmfb frame buffer device
[    5.948485] Adding 102384k swap on /var/swap.  Priority:-2 extents:1 across:102384k SSFS
[    6.260766] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    6.260777] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    6.263226] pps pps0: new PPS source ptp0
[    6.263289] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[   10.357370] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   10.357389] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

pelwell, 2023-11-21:
It's annoying that Debian would impose their own naming requirements on a kernel - the kernel itself has no such restrictions. The reason behind the use of an underscore was to make clear the grouping, with dashes separating the local version string from everything else, but as you've discovered not everyone agrees.

I can't think of any negative consequence of changing, except for a temporary duplication of module directories.

See https://github.com/raspberrypi/linux/pull/5729.

XECDesign, 2023-11-21:
> It's annoying that Debian would impose their own naming requirements on a kernel - the kernel itself has no such restrictions. The reason behind the use of an underscore was to make clear the [...]

It's not so much that debian is imposing any requirements here, at least not directly. But Debian packages do use `_` as a delimeter between the package name, so it can't be a part of the package name or version. The script that generates this package is not a part of Debian and actually comes with the kernel source. I think it would be more accurate to say that the script is doing the wrong thing by putting the string there without substituting the character with something else.

This probably doesn't matter and just changing LOCALVERSION is the simpler workaround. I'll just need to remember this happened next time I update our packages, because I think this is hardcoded in a few places.

andrewluebke, 2023-12-08:
OK, thanks guys, this seems to be resolved in the current commit so I'll close it.

closed 2023-12-08 by andrewluebke