raspberrypi/linux #5721: Global Shutter Camera not working on CAM0 - RPi 5 4GB

szakalit, 2023-11-19:
### Describe the bug

Im trying to connect two GSC cameras to RPI5 and camera connected to CAM0 doesn't work.
After swapping the cameras in place, the situation is identical.
When i connect GSC to CAM1 and Camera Module 3 Wide NoIR to CAM0 its working fine.

In dmesg its showing error:
**[    4.864117] imx296 6-001a: 8-bit write to 0x3000 failed: -121
[    4.864128] imx296 6-001a: failed to get sensor out of standby (-121)
[    4.864364] imx296: probe of 6-001a failed with error -121**

### Steps to reproduce the behaviour

Connect Global Shutter Camera to CAM0 on RPi 5 

### Device (s)

Other

### System

jakub@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fb56ad562991cf3ae5c96ab50983e1deeaefc7b6, stage4
jakub@raspberrypi:~ $ vcgencmd version
2023/10/18 18:30:17
Copyright (c) 2012 Broadcom
version c2da2ae7 (release) (embedded)
jakub@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

### Logs

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.1.0-rpi6-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0                          , GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001a000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f9fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] On node 0, zone DMA: 1536 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 29 pages/cpu s79144 r8192 d31448 u118784
[    0.000000] pcpu-alloc: s79144 r8192 d31448 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1030656
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_                          alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:A8:91:D3 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0                          x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=13a028dd-02 rootfstype=ext4 fsck.repair=yes rootwai                          t quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=PL
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[... 549 lines cut ...]
[    9.378048] Bluetooth: RFCOMM TTY layer initialized
[    9.378065] Bluetooth: RFCOMM socket layer initialized
[    9.378077] Bluetooth: RFCOMM ver 1.11
[   10.473099] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

jakub@raspberrypi:~ $ sudo i2cdetect -y 4
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
jakub@raspberrypi:~ $ sudo i2cdetect -y 6
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

szakalit, 2023-11-22:
After checking the signals with a logic analyser, it appears that the enable signal for CAM0 is missing, although this is strange because the enable signal is correct at the beginning when detecting the camera.
In the attached screenshots you can see that the enable signal is missing when trying to wake up with the IMX296_CTRL00 command and this is the reason for the error.

CAM1 - the enable signal is present:
[image]
CAM1 - IMX296_CTRL00 command sent successfully:
[image]
CAM0 - the enable signal is mising:
[image]
CAM0 - IMX296_CTRL00 command sent fault:
[image]

6by9, 2023-11-22:
Should have been fixed last week - https://github.com/raspberrypi/linux/commit/6137fb168c08bd8c41c8421bf26f09ed29479f08

closed 2023-11-22 by szakalit