raspberrypi/linux #5722: OV5647 dtoverlay cam0 missing VANA-supply

ArduCAM, 2023-11-20:
### Describe the bug

When setting the ov5647 dtoverlay parameter cam0, VANA-supply does not use the correct overlay.

https://github.com/raspberrypi/linux/blob/1c4c7647c8514dc6c8bf843a8bf69732437e6b98/arch/arm/boot/dts/overlays/ov5647-overlay.dts#L75

### Steps to reproduce the behaviour

add `dtoverlay=ov5647,cam0,vcm` in /boot/config.txt

### Device (s)

Other

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

2023/10/18 18:30:17 
Copyright (c) 2012 Broadcom
version c2da2ae7 (release) (embedded)

Linux dion 6.1.58-v8_16k+ #1 SMP PREEMPT Wed Nov  1 17:19:10 UTC 2023 aarch64 GNU/Linux

pelwell, 2023-11-20:
I'm assuming we need to add the following to the `cam0` override:
```
    <&vcm>,"VANA-supply:0=",<&cam0_reg>,
```
Wouldn't a PR have been quicker?

pelwell, 2023-11-20:
https://github.com/raspberrypi/linux/pull/5723

pelwell, 2023-11-20:
Fixed by #5723.

closed 2023-11-20 by pelwell

ArduCAM, 2023-11-21:
> I'm assuming we need to add the following to the `cam0` override:
> 
> ```
>     <&vcm>,"VANA-supply:0=",<&cam0_reg>,
> ```
> 
> Wouldn't a PR have been quicker?

I'm worried I'll miss something and mess it up. XD