raspberrypi/linux #5724: I2S audio DAC (hifiberry-dacplus) not working on Raspberry Pi 5

KonstaT, 2023-11-20:
### Describe the bug

I2S audio DAC (hifiberry-dacplus) is not working on Raspberry Pi 5.

### Steps to reproduce the behaviour

Enabled the DAC by adding/modifying following lines in config.txt:
```
dtparam=i2s=on
dtoverlay=hifiberry-dacplus
#dtparam=audio=on
```

Audio DAC is not listed as an ALSA device in `cat /proc/asound/cards`.

Relevant error in `dmesg` output:
```
[    7.123353] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_fmt on 1f000a4000.i2s: -22
[    7.123610] snd-rpi-hifiberry-dacplus soc:sound: snd_soc_register_card() failed: -22
[    7.123618] snd-rpi-hifiberry-dacplus: probe of soc:sound failed with error -22
```

```
uname -a
Linux raspberrypi 6.1.0-rpi6-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
```

If I switch the same sdcard and the DAC to Raspberry Pi 4, it works as expected.

```
cat /proc/asound/cards
 0 [vc4hdmi0       ]: vc4-hdmi - vc4-hdmi-0
                      vc4-hdmi-0
 1 [vc4hdmi1       ]: vc4-hdmi - vc4-hdmi-1
                      vc4-hdmi-1
 2 [sndrpihifiberry]: HifiberryDacp - snd_rpi_hifiberry_dacplus
                      snd_rpi_hifiberry_dacplus
```

```
uname -a
Linux raspberrypi 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
```

I can also reproduce the same behaviour/issue with my 6.1.61 Android kernel where the Pi kernel is merged at https://github.com/raspberrypi/linux/tree/6137fb168c08bd8c41c8421bf26f09ed29479f08.

### Device (s)

Other

### System

2023-10-10-raspios-bookworm-arm64
```
sudo apt update
sudo apt full-upgrade
```

pelwell, 2023-11-20:
I think that PR #5725 may solve the problem for you. Wait about an hour for the auto-builds to complete, then run `sudo rpi-update pulls/5725`.

Thanks for reporting this. There have been a number of sound-related issues reported recently, but this is the first for which I have the affected card. I hope the same patch will solve at least some of the other issues.

KonstaT, 2023-11-20:
Thanks. Unfortunately it doesn't seem to fix the issue for me. I compiled a kernel with the above patch and still facing the same error.

KonstaT, 2023-11-20:
On Raspberry Pi OS after `sudo rpi-update pulls/5725`:

```
$ dmesg -l err,warn
[    0.040731] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    0.341948] F2FS not supported on PAGE_SIZE(16384) != 4096
[    5.806444] rpivid_hevc: module is from the staging directory, the quality is unknown, you have been warned.
[    5.928950] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,5-model-b.bin failed with error -2
[    5.975826] hci_uart_bcm serial0-0: supply vbat not found, using dummy regulator
[    5.975917] hci_uart_bcm serial0-0: supply vddio not found, using dummy regulator
[    6.390576] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_fmt on 1f000a4000.i2s: -22
[    6.390787] snd-rpi-hifiberry-dacplus soc:sound: snd_soc_register_card() failed: -22
[    6.390793] snd-rpi-hifiberry-dacplus: probe of soc:sound failed with error -22
```

```
$ uname -a
Linux raspberrypi 6.1.62-v8_16k+ #1 SMP PREEMPT Mon Nov 20 16:06:13 UTC 2023 aarch64 GNU/Linux
```

KonstaT, 2023-11-20:
Thank you. I can confirm that it works after adding the `slave` parameter. Getting audio out of it on Android as well. PR [#5725](https://github.com/raspberrypi/linux/pull/5725) is not needed for this device at least, just adding the slave parameter is enough.

So, the next question is, is the slave parameter needed for all the DACs on Pi 5? Can this be handled somehow in the drivers' overlays (defaults to slave on Pi 5)? Or is this still a workaround for something else?

Also good to know that dtparam=i2s is not needed.

pelwell, 2023-11-20:
Pi 5 (specifically RP1) has slightly different I2S capabilities than those found on the BCM2835-family of processors. It gains the ability to do dual- and quad-lane I2S, but it has separate clock producer (master) and clock consumer (slave) blocks. This means that each overlay has to specify the correct I2S - producer or consumer. For most soundcards this is a fixed property, but a few of them can operate in either mode, which causes problems.

The addition of the "slave" property was a way to solve the problem, but as you have discovered this requires user intervention (and knowledge). A better solution might be to give the overlays for those flexible cards two different i2s-controller properties (or a single property with two label references) - one for each scenario - and modify the driver to use the correct one.

closed 2023-11-21 by KonstaT

pelwell, 2023-11-21:
> Hopefully the drivers/overlays manage the controller setup sometime in the future.

It's on the To-do list.

Rene-Belso, 2023-12-11:
It works for me, adding the "**slave**" in the /boot/comfig.txt. I.e., on a pi5 fully updated 2023-12-11.

dtoverlay=hifiberry-dacplus,slave

Strange that Hifiberry are not super keen to assist with this, or even inform.
Anyway: Thanks to pelwell

MMaciocia, 2024-03-14:
Absolutely fabulous that the addition of the "SLAVE" to dtoverlay fixes the issue for Pi-5, so why the f..k have I not found this fix on the setup page, the blog etc for the devices! This is the most irresponsible behaviour of any hardware supplier that lays claim to any for of quality. First time buying your hardware, also the last!

Nate711, 2024-05-13:
@pelwell Would you be able to help with another audio related issue on the Pi 5? I'm unable to load the overlay for the max98357a on ubuntu 24.04 with kernel 6.8.0-1004-raspi. 
```
pi@ubuntu:~$ sudo dtoverlay max98357a
DTOVERLAY[warn]: no matching platform found
```
I'm entirely unfamiliar with how modules and overlays work so not sure how to continue. Thanks!

pelwell, 2024-05-13:
Yes, but new problem = new Issue