raspberrypi/linux #5727: RPI4: Hardware plane offloading is broken for NV12 (zero-copy video playback)

rmader, 2023-11-21:
### Describe the bug

#### Note: Wayfire, as currently shipped, does not seem to support NV12 offloading. So this issue was only reproduced so far with other compositors, even though I assume that Wayfire will gain support for it eventually.

Context: The RPI4 advertises hardware plane support for NV12 (linear modifier) which is a useful feature to display content from the hardware video decoder or attached cameras without extra copy. Unfortunately this appears to be broken and the only thing displayed is solid green - the equivalent to black/all zero input for NV12.

This has been reproduced on multiple compositors, on this kernel and upstream (see below).

A visual example, showing the `vivid` kernel module test pattern:

[image]

For comparison the pattern composited (also works on Wayfire):
[image]

This happens with different source and destination resolutions, also in fullscreen. I get the exact same behavior on different compositors (Weston and Mutter, both of which work on other hardware), kernels (this downstream and upstream as shipped in Fedora), OSs (see below) and Mesa versions (and double-checked already that the values passed to KMS are sane (minus https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/26283).

So I can say with high confidence that this is most likely a kernel driver issue (or firmware - in any case not on layers above).

### Steps to reproduce the behaviour

I have reproduced the problem in various ways, but the following should make it comparatively easy. If you want to reproduce and struggle at any point, please feel free to ask for more elaborate steps:
1. build `weston` (and the included `weston-simple-dmabuf-v4l`) from `main` - or backport https://gitlab.freedesktop.org/wayland/weston/-/merge_requests/1177 so `weston-simple-dmabuf-v4l` can correctly upload `NV12`
2. enable the `vivid` test pattern as explained by `weston-simple-dmabuf-v4l -h`, i.e.
    - `options vivid node_types=0x1 num_inputs=1 input_types=0x00` > `/etc/modprobe.d/vivid.conf`
    - `modprobe vivid`
    - `v4l2-ctl -d /dev/video0 --set-fmt-video=width=1920,height=1080,pixelformat=NV12`
3. run `weston` as compositor (just on any TTY)
4. open the terminal and run `weston-simple-dmabuf-v4l -f NV12 -d NV12`

Observed result: solid green output whenever "direct scanout"/"hardware plane offloading" is active, otherwise mostly correct pattern (in compositor mode, e.g. when the terminal overlaps the video surface). Flicker between both is also possible due to allocation failures.

Expected result: correct display of the test pattern.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Latest Raspberry Pi OS (32bit) with Weston (`main`/`12.0.94`)
Latest Raspberry Pi OS (64bit) with Weston (`main`/`12.0.94`)
Fedora 39 ([rpmfusion](https://rpmfusion.org/Howto/RaspberryPi) which ships this kernel, 64bit) with Weston (`main`/`12.0.94`) and Mutter (`45.1` and `main` with https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/3177)
Fedora 39 (official upstream kernel), also Weston and Mutter as above

6by9, 2023-11-21:
NV12 output is quite happy through `kmstest -f NV12` (https://github.com/tomba/kmsxx/) or modetest.

I can have a guess at the disjoin. Vivid uses the videobuf2-vmalloc allocator. vc4 requires contiguous buffers.
Exporting the vmalloc dmabuf and trying to import into vc4 should throw an error, but I'm suspecting that's not happening. When vc4 tries to get the appropriate physical address, it fails.

Confirmed - add `allocators=1` to your vivid options to use dma-contig, and it works fine.

Now to try and find where the dmabuf validation is meant to happen in the kernel.....

rmader, 2023-11-21:
> Confirmed - add `allocators=1` to your vivid options to use dma-contig, and it works fine.

Ouch, that works indeed! Thanks - and yeah, would be great to make the kernel fail for the not supported cases :)

6by9, 2023-11-21:
`drm_gem_dma_prime_import_sg_table` that we use via `DRM_GEM_DMA_DRIVER_OPS_WITH_DUMB_CREATE` states that the supplied buffer must be contiguous, and appears to be checking that it is
https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/gpu/drm/drm_gem_dma_helper.c#L469-L470

Now adding debug to see where that goes, and why isn't it being acted on....

6by9, 2023-11-21:
So vmalloc can allocate a contiguous buffer if memory hasn't fragmented, and so the validation can pass.

The `drm_gem_dma_object` appears to have a valid dma_addr that gets programmed into the display hardware, but that is obviously not the address that vivid is writing into. I don't know the kernel memory mapping code well enough to extract more details over the map. vmalloc is giving me dma addresses around 0x00000000f727e000, whilst contig is giving me around 0x00000000d4700000.

More generally the HVS can only get to the bottom 1GB of RAM (excluding Pi5), so vc4 does need a wrapper around `drm_gem_dma_prime_import_sg_table` to check for that bit. (It may happen automagically through dma_buf_attach).

The callstack on `drm_gem_dma_prime_import_sg_table` failing looks like the error should be passed back out to userspace correctly, so there is the question over whether userspace is checking for an error on that:
```
[ 1299.836504] CPU: 3 PID: 869 Comm: weston Tainted: G         C         6.1.62-v8+ #1
[ 1299.836532] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[ 1299.836542] Call trace:
[ 1299.836551]  dump_backtrace+0xfc/0x108
[ 1299.836585]  show_stack+0x20/0x30
[ 1299.836597]  dump_stack_lvl+0x8c/0xb8
[ 1299.836626]  dump_stack+0x18/0x34
[ 1299.836640]  drm_gem_dma_prime_import_sg_table+0xe8/0x100 [drm_dma_helper]
[ 1299.836692]  drm_gem_prime_import_dev+0xac/0x1b0 [drm]
[ 1299.837004]  drm_gem_prime_fd_to_handle+0x1c0/0x240 [drm]
[ 1299.837203]  drm_prime_fd_to_handle_ioctl+0x30/0x48 [drm]
[ 1299.837401]  drm_ioctl_kernel+0xcc/0x188 [drm]
[ 1299.837598]  drm_ioctl+0x21c/0x430 [drm]
[ 1299.837793]  __arm64_sys_ioctl+0xb0/0xf8
[ 1299.837820]  invoke_syscall+0x4c/0x110
[ 1299.837842]  el0_svc_common.constprop.0+0x68/0x128
[ 1299.837858]  do_el0_svc+0x34/0xd0
[ 1299.837874]  el0_svc+0x30/0x98
[ 1299.837891]  el0t_64_sync_handler+0xb8/0xc0
[ 1299.837907]  el0t_64_sync+0x18c/0x190
```
simple-dmabuf-v4l appears to ignore any return code from `zwp_linux_buffer_params_v1_add` which is unlikely to be helping. https://gitlab.freedesktop.org/wayland/weston/-/blob/main/clients/simple-dmabuf-v4l.c?ref_type=heads#L531

rmader, 2023-11-21:
Thanks for the investigation!

> simple-dmabuf-v4l appears to ignore any return code from zwp_linux_buffer_params_v1_add which is unlikely to be helping.

That would mean that the things fail in the compositor in the first place, which doesn't seem to be the case. Note that v3d / the compositors are able to composite the content just fine, it's just KMS which doesn't like it - but doesn't fail on the atomic commits either - neither the `DRM_MODE_ATOMIC_TEST_ONLY` ones (which all common compositors every frame) nor on the actual commit - they succeed just fine. As said I've had no luck seeing anything useful in the logs either, so most likely the kernel / KMS doesn't propagate anything back to userspace.

6by9, 2023-11-21:
If the buffer is non-contiguous, then `ioctl(DRM_IOCTL_PRIME_FD_TO_HANDLE)` will report an error.

If I'm reading the code correctly, then that is also the place where we can validate the dmabuf backing memory for being in the right range, and again will fail `ioctl(DRM_IOCTL_PRIME_FD_TO_HANDLE)`.

I don't know where that fits within the Wayland world, but if it's not checking the return code from that ioctl, then expect badness. AIUI that's a long way before you even think about making an atomic commit with the handle.

rmader, 2023-11-22:
> If the buffer is non-contiguous, then ioctl(DRM_IOCTL_PRIME_FD_TO_HANDLE) will report an error.

Hm, can't see `DRM_IOCTL_PRIME_FD_TO_HANDLE` used in any compositor, relevant mesa code or app. Also: the buffer comes from V4L2 so IIUC the check could only happen once the compositor tries to import things to KMS - during the commit with `DRM_MODE_ATOMIC_TEST_ONLY`.

6by9, 2023-11-22:
Potentially it's wrapped via libdrm as `drmPrimeFDToHandle`. https://gitlab.freedesktop.org/mesa/drm/-/blob/main/xf86drm.c?ref_type=heads#L3391

DRM uses framebuffer handles. 
V4L2 exports a dmabuf, also known in DRM as Prime handles.
Mapping between the two is normally done in a setup phase as it potentially involves costly iommu calls.

Certainly in the kernel logs I only see around 5 buffers being imported early on.
Your `params_listener` `create_succeeded` is closing the dmabuf handles exported from V4L2, so the import has happened before that has been called.
From a purely logical reading of that code, the failure path I'd expect is going to be going through `create_failed` which does a similar closing of all dmabuf handles.

Pi4 using `v3d` has an iommu and can therefore accept non-contiguous buffers. `vc4` HVS doesn't (until Pi5).
Pi0-3 3D is part of the `vc4` driver, and again doesn't have an iommu and therefore requires contiguous buffers. (It also doesn't have native YUV support, so relies on shaders for the conversion).

rmader, 2023-11-22:
> From a purely logical reading of that code, the failure path I'd expect is going to be going through create_failed which does a similar closing of all dmabuf handles.

Interesting, thanks for the explanation! And will double check later. It would be **very** unexpected though if Weston, Mutter and Wayfire/wlroots all accept and correctly render buffers if the buffer creation failed (and the client just ignored the error). I mean - they clearly successfully import the buffers to the GL driver, keep track and release them etc.

All of that shouldn't be possible if the `weston-simple-dmabuf-v4l` couldn't successfully import buffers from v4l2, no?

6by9, 2023-11-22:
I'm looking at the kernel side only, and that appears to be doing what I'm expecting for the code path I'm expecting it to take. As it appears to be the correct codepath when using contiguous source buffers, I'm relatively confident I'm in the right area.

v3d can cope with non-contiguous buffers, so it quite likely to accept the buffer. It's totally irrelevant for vc4.
If GL on Pi0-3 (uses vc4) was accepting the buffer and yet vc4 scanout wasn't, then I'd be more interested.

Throwing a 5 second delay in the kernel `drm_gem_dma_prime_import_sg_table` function, and attaching gdb to weston, on starting weston-simple-dmabuf-v4l I can stop gdb when it pauses.
Vague annotations where debug syms are missing.
```
(gdb) bt
#0  ioctl () at ../sysdeps/unix/sysv/linux/aarch64/ioctl.S:26
#1  0x0000007fb0b56b18 in drmIoctl () at /lib/aarch64-linux-gnu/libdrm.so.2
#2  0x0000007fb0b5a264 in drmPrimeFDToHandle () at /lib/aarch64-linux-gnu/libdrm.so.2
#3  0x0000007faebec6ac in  () at /usr/lib/aarch64-linux-gnu/dri/vc4_dri.so (vc4_bo_open_dmabuf)
#4  0x0000007faee3da6c in  () at /usr/lib/aarch64-linux-gnu/dri/vc4_dri.so (vc4_resource_from_handle)
#5  0x0000007fae00ef60 in  () at /usr/lib/aarch64-linux-gnu/dri/vc4_dri.so 
#6  0x0000007fae00f318 in  () at /usr/lib/aarch64-linux-gnu/dri/vc4_dri.so
#7  0x0000007fae00f428 in  () at /usr/lib/aarch64-linux-gnu/dri/vc4_dri.so
#8  0x0000007fa2c7f4a0 in  () at /lib/aarch64-linux-gnu/libEGL_mesa.so.0
#9  0x0000007fa2c7ba30 in  () at /lib/aarch64-linux-gnu/libEGL_mesa.so.0
#10 0x0000007fa2c6d330 in  () at /lib/aarch64-linux-gnu/libEGL_mesa.so.0 (gr->create_image)
#11 0x0000007fa2efcef8 in import_simple_dmabuf (gr=0x55a28622e0, attributes=0x55a31a45b8) at ../libweston/renderer-gl/gl-renderer.c:2531
#12 0x0000007fa2efd6b0 in import_dmabuf (gr=0x55a28622e0, dmabuf=0x55a31a45a0) at ../libweston/renderer-gl/gl-renderer.c:2729
#13 0x0000007fa2efdc58 in gl_renderer_import_dmabuf (ec=0x55a27ba550, dmabuf=0x55a31a45a0) at ../libweston/renderer-gl/gl-renderer.c:2884
#14 0x0000007fb0e3d380 in weston_compositor_import_dmabuf (compositor=0x55a27ba550, buffer=0x55a31a45a0) at ../libweston/compositor.c:8816
#15 0x0000007fb0e500c8 in params_create_common (client=0x55a3198d80, params_resource=0x55a3054240, buffer_id=0, width=1920, height=1080, format=842094158, flags=0)
    at ../libweston/linux-dmabuf.c:275
#16 0x0000007fb0e50224 in params_create (client=0x55a3198d80, params_resource=0x55a3054240, width=1920, height=1080, format=842094158, flags=0) at ../libweston/linux-dmabuf.c:329
#17 0x0000007fb0a863e0 in  () at /lib/aarch64-linux-gnu/libffi.so.8
#18 0x0000007fb0a85b24 in  () at /lib/aarch64-linux-gnu/libffi.so.8
#19 0x0000007fb0dad1ac in  () at /lib/aarch64-linux-gnu/libwayland-server.so.0
#20 0x0000007fb0da81f0 in  () at /lib/aarch64-linux-gnu/libwayland-server.so.0
#21 0x0000007fb0dab198 in wl_event_loop_dispatch () at /lib/aarch64-linux-gnu/libwayland-server.so.0
#22 0x0000007fb0da89dc in wl_display_run () at /lib/aarch64-linux-gnu/libwayland-server.so.0
#23 0x0000007fb10a0348 in wet_main (argc=1, argv=0x7fffdf9f68, test_data=0x0) at ../compositor/main.c:4203
#24 0x000000556c7c07b4 in main (argc=1, argv=0x7fffdf9f68) at ../compositor/executable.c:33
```
So it does appear that the import is done in the hot path - ho hum.
I can't think why else we'd be following this kernel path, so I am fairly happy this is the import of the NV12 image. import_simple_dmabuf does appear to be given back a valid egl_image, so it gets into the clause at https://gitlab.freedesktop.org/wayland/weston/-/blob/12.0/libweston/renderer-gl/gl-renderer.c?ref_type=heads#L2730

Mesa would be common to all of Weston, Mutter, and Wlroots, so if something is dropping the error code in there it would affect all of them. I'm not saying it is, but it's a potential.

Forcing the import to always fail, weston drops back to GL composition for that plane.

I should add that I built the 12.0 weston branch - main had dependency requirements that I didn't fancy trying to fix.

rmader, 2023-11-22:
Oh right, the ioctl happens in `mesa`->`libdrm` - makes sense :facepalm:

> so I am fairly happy this is the import of the NV12 image. 

:+1: 

> Forcing the import to always fail, weston drops back to GL composition for that plane.

> so if something is dropping the error code in there it would affect all of them. I'm not saying it is, but it's a potential.

So do I understand correctly that we are down to either
1. the kernel not throwing the error in this case
2. `libdrm` or the vc4 mesa driver dropping it

? Sounds reasonable!

Just an idea (please feel free to ignore :P): `vc4` or the kernel don't throw/propagate the error on purpose because they assume the image imported with `vc4` will actually only be used with `v3d` in the compositor (or some app). And what's missing is some flag on the bo (or some other check) to let the KMS driver know that it can't use it / that it's not a scanout buffer.

6by9, 2023-11-22:
The kernel is returning an error - I've logged that at `drm_prime_fd_to_handle_ioctl`.

I know very little about Mesa. It does feel a little strange that the buffer is being imported through gl-renderer.c when it is being rendered directly.

rmader, 2023-11-22:
Ah nice. Ok, I'll dig a bit in `vc4_resource_from_handle()` / `vc4_resource.c` in mesa - the stack trace is super helpful. From a quick look I can't see `vc4` ignoring any error though, see https://gitlab.freedesktop.org/mesa/mesa/-/blob/73d69ef1e6b013f923ec459c789976570baaf447/src/gallium/drivers/vc4/vc4_bufmgr.c#L444-450 / https://gitlab.freedesktop.org/mesa/mesa/-/blob/73d69ef1e6b013f923ec459c789976570baaf447/src/gallium/drivers/vc4/vc4_resource.c#L648-658

> It does feel a little strange that the buffer is being imported through gl-renderer.c when it is being rendered directly.

That's normal for Wayland compositors. They import the image and decide on every frame whether they can offload stuff to KMS or have to composite with GL/VK.

6by9, 2023-11-22:
A brief chat with Phil has given a further line of investigation - swiotlb bounce buffers.

If the buffer is in the wrong part of memory (HVS can only see the bottom 1GB) then the DMA framework may have set up bounce buffers so that the target the HVS gets is in the right region. At the point dma_sync_* calls are made to synchronise caches, it will do a quick copy from the source buffer to the destination.

DRM appears to assume that all buffers are cache coherent, therefore there are no calls to dma_sync_* for cache management. The image data is therefore never copied into the bounce buffer, and you see purest green (YUV 0,0,0) of the memset bounce buffer.
Not calling dma_sync_* does make sense as the DMA operation is ongoing, so if you scrawl new data into the framebuffer, there is the expectation that it will appear nearly immediately on the screen, not that you need to kick something again to copy the image data and then it can appear on screen.

I should be able to dump out the SG lists and confirm if that is the case or not. It'll take some work to determine if there is a sensible fix for it.

6by9, 2023-11-22:
It looks like Phil was right (as usual) and bounce buffers are the cause.
Tracing on swiotlb is a little terse, but gives me 3800 4k pages as being bounced. 1920*1080*1.5 = 3110400 or 759.375 pages. 3800/760 = 5 buffers, which would match the number of buffers allocated by V4L2.

https://dri.freedesktop.org/docs/drm/gpu/drm-mm.html#memory-coherency
So they do appear to assume it's cache coherent, and therefore I'll have a look to see if we can stop it using bounce buffers in this case.

6by9, 2023-11-22:
#5734 should mean that vc4 will not allow the import of dmabufs that require bounce buffers.

I have a suspicion that this should be merged generally into `drm_gem_dma_prime_import_sg_table` as bounce buffers are incompatible with DRM, but that's a bigger discussion to have on dri-devel.

rmader, 2023-11-22:
Awesome, thanks!

rmader, 2023-11-23:
Fixed by https://github.com/raspberrypi/linux/pull/5734

rmader, 2023-11-23:
Thanks for the super quick reaction / investigation / the fix!

closed 2023-11-23 by rmader