raspberrypi/linux #5732: RPi3B+: Bookworm Kernel EDID complaints flood logs

pcollinson, 2023-11-22:
### Describe the bug

I have an ancient flat screen (Dell 1706FPVt) with a DVI interface plugged into an RP3B+ with a DVI to HDMI interface. For bookworm, I've included ```video=HDMI-A-1:1280x1024@60``` on the kernel command line. The screen works fine but every minute I get three messages like:
```
EDID block 0 (tag 0x00) checksum is invalid, remainder is 24
```
written on the console. So researching this, I find that the approved way of adding such devices is to add a compiled appropriate EDID file, provided with the kernel source in ```tools/edid```. So I tried this. The code compiles easily from a Makefile and generates the ```/lib/firmware/edid``` files. However, the only change is that I now get is a message flood saying that it's loading the information, again three times a minute and flooding logs.

Maybe I can stop this output by changing ```/proc/sys/kernel/printk``` .  It seems not.

I've also tried switching between X11 and Wayland, there is not discernible difference.

Now this isn't anything like a show stopper, but it's shortening the life of my storage devices and anything like this is just annoying because it wipes out logs that might be important.

### Steps to reproduce the behaviour

Use vanilla bookworm on an RPI3B+, with a DVI screen connected using a DVI to HDMI adapter. Change the kernel command line to include an appropriate screen setting for the device.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

```
System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fb56ad562991cf3ae5c96ab50983e1deeaefc7b6, stage4

Linux toby 6.1.0-rpi6-rpi-v7 #1 SMP Raspbian 1:6.1.58-1+rpt2 (2023-10-27) armv7l GNU/Linux
Revision	: a020d3
Serial		: 000000006b29969c
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x80000
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Oct 17 2023 15:42:39 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
udev              332700       0    332700   0% /dev
tmpfs              94412     960     93452   2% /run
/dev/sda2       28875676 5874348  21509156  22% /
tmpfs             472044       8    472036   1% /dev/shm
tmpfs               5120       8      5112   1% /run/lock
/dev/sda1         523248  116096    407152  23% /boot/firmware
tmpfs              94408      40     94368   1% /run/user/1001

Filename				Type		Size		Used		Priority
/var/swap                               file		102396		0		-2
/dev/zram0                              partition	262140		3840		100

[... 624 lines cut ...]
48: a0    -- | lo // SD_CLK_R/GPIO48 = SD0_CLK
49: a0    -- | hi // SD_CMD_R/GPIO49 = SD0_CMD
50: a0    -- | hi // SD_DATA0_R/GPIO50 = SD0_DAT0
51: a0    -- | hi // SD_DATA1_R/GPIO51 = SD0_DAT1
52: a0    -- | hi // SD_DATA2_R/GPIO52 = SD0_DAT2
53: a0    -- | hi // SD_DATA3_R/GPIO53 = SD0_DAT3

vcdbg log messages
------------------

Failed to allocate -201253077 bytes for message buffer

dmesg log
---------

[62553.843342] EDID block 0 (tag 0x00) checksum is invalid, remainder is 24
....
Removed a full buffer of replicated messages
....
[82276.179596] EDID block 0 (tag 0x00) checksum is invalid, remainder is 24
```

### Logs

dmesg is currently 1927 lines showing the error above.

popcornmix, 2023-11-22:
Normally the EDID is only re-read when hotplug deasserts/reasserts to indicate display has changed its reported features (or a different display is attached).

I wonder if a hardware fault it triggering this unwantedly?
Can you try adding `vc4.force_hotplug=1` to cmdline.txt (on existing line) which should cause the hotpug line to be ignored.

pcollinson, 2023-11-23:
Thanks for the speedy response.

Adding `vc4.force_hotplug=1` makes the screen stay on - so is an improvement, it's no longer turning off as the system switches from the firmware boot to the kernel boot.

However, it's made no difference to the report
```
EDID block 0 (tag 0x00) checksum is invalid, remainder is 24
```
which still happens 3 times a minute.

I tried adding back in the EDID binaries, installing the compiled code in `/lib/firmware/edid` and adding:

```
drm.edid_firmware=edid/1280x1024.bin
```
into the kernel command line.

I now get:

```
kernel: [drm] Got built-in EDID base block and 0 extensions from "edid/1280x1024.bin" for connector "HDMI-A-1"
```
but it happens 6 times a minute. So I've removed it. However, is it using a cached copy of the data or actually re-reading the file every time?

Just be good to get rid of this printf down to a single repeat.

popcornmix, 2023-11-23:
If you unplug hdmi cable do the 3 (or 6) times a minute messages continue?
The `vc4.force_hotplug=1` should mean the Pi doesn't see the disconnect, so I'd expect it to continue.
If it does, that rules out the display as provoking the message.

The other test is to try a clean install of bookworm (with `vc4.force_hotplug=1`) and check if messages occur.
If they don't then perhaps some installed (but not default) software is triggering it.

pcollinson, 2023-11-23:
OK. It looks like the display is triggering this, sorry I didn't think of that.

Turning the display off - no change. Unplugging when off or on: logs stop. Plugging back in again: logs start again. So it looks like something in this early display is doing it. It must be over 20 years old.

I found I could do a Factory Reset on the display - but that had no effect.

So I'll unplug the screen when not needed.

Thanks for your assistance.

popcornmix, 2023-11-23:
The only thing I can think of that the display is doing to provoke this is toggling the hotplug detect line.
But I'd expect `vc4.force_hotplug=1` to hide that. Just to confirm, that setting is working can you show output of:
```
cat /proc/cmdline
cat /sys/module/vc4/parameters/force_hotplug
```

pcollinson, 2023-11-23:
Hotplug is definitely working....

```
~# cat /proc/cmdline
coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=0  vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=09701a8f-02 rootfstype=ext4 fsck.repair=yes rootwait plymouth.ignore-serial-consoles video=HDMI-A-1:1280x1024@60 vc4.force_hotplug=1 cfg80211.ieee80211_regdom=GB
~# cat /sys/module/vc4/parameters/force_hotplug
1
```

pcollinson, 2023-11-23:
Perhaps the designers wanted to detect DVI or VGA input and switch automatically - which I think it does.

closed 2023-11-24 by pcollinson