raspberrypi/linux #5737: Raspberry OS 12 Bookworm no longer allows SSH over USB-C (Ethernet gadget mode)

Nahuel92, 2023-11-22:
### Describe the bug

Raspberry OS 12 Bookworm no longer allows SSH over USB-C/Ethernet gadget mode.

### Steps to reproduce the behaviour

Hi everyone!

Not sure if this was already reported or if this is the right place to bring this up. If that's not the case, please let me know and I'll close this issue :)

I have a Raspberry Pi 4B that I could connect to via SSH over USB-C with previous Raspberry OS versions by making the following configuration:

### /boot/config.txt

```text
# Adding next line to enable USB-C
dtoverlay=dwc2
```

### /boot/cmdline.txt

```text
# Some stuff...
rootwait modules-load=dwc2,g_ether g_ether.host_addr=aa:bb:cc:dd:ee:ff
```

>**Note:** I made sure that there is only one space after `rootwait` and one space after my new configuration.

### /etc/dhcpcd.conf

```text
# Static IP Config
interface usb0
static ip_address=192.168.2.2
static routers=192.168.2.1
```

But that is not working anymore in Raspberry OS Bookworm. Unfortunately my previous micro SD card stopped working so I noticed this when installing a new one.

It would be great if that configuration can work again on the new Raspberry OS version, as it's super confortable and easy to use a single USB-C cable for both power and network communication with the board.

I'll appreciate any help on this. Thanks in advance!

### Device (s)

Raspberry Pi 4 Mod. B

### System

### cat /etc/rpi-issue

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

### vcgencmd version

Aug 10 2023 15:33:38
Copyright (c) 2021 Broadcom
version 03dc77429335caee083e22ddc8eec09c07f12a7a (clean) (release) (start)

### uname -a

Linux raspberrypi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

### Logs

### ifconfig partial output

eth0: ....
lo: ....
wlan0: ...

(no `usb0` there).

### dmesg | grep usb output

No mention at all of `usb0` (sorry for not pasting logs, I'm writing this on a different machine than my Raspberry board).

### Additional context

The configuration I shared worked fine with a previous Raspberry OS version but that's not the case anymore. I'll try and see if the Raspberry Pi Installer gives me the option to install a previous version to verify that my configuration is the right one and the issue is caused by the new Raspberry OS version.

pelwell, 2023-11-23:
Can you log in via SSH over another interface, e.g. WiFi or Ethernet?

cmitu, 2023-11-23:
> /etc/dhcpcd.conf

But Bookworm doesn't ship with `dhcpcd` anymore, it uses NetworkManager for the network interface configuration. One of the `udev` rules shipped with the the NM package is _/lib/udev/rules.d/85-nm-unmanaged.rules_, which shows:

```
...

# USB gadget device. Unmanage by default, since whatever created it
# might want to set it up itself (e.g. activate an ipv4.method=shared
# connection).
ENV{DEVTYPE}=="gadget", ENV{NM_UNMANAGED}="1"

```
This may indicate that NM doesn't configure/ignores the USB device (`usb0`).

Nahuel92, 2023-11-23:
> Can you log in via SSH over another interface, e.g. WiFi or Ethernet?

Hi. Yes, if I connect my Raspberry Pi to the same WiFi connection I can both ping it and SSH into it.
I'm pretty sure something changed in this new version and Ethernet gadget mode over USB-C stopped working.

Nahuel92, 2023-11-23:
> But Bookworm doesn't ship with `dhcpcd` anymore, it uses NetworkManager for the network interface configuration. One of the `udev` rules shipped with the the NM package is [...]

I'll play around that to see if I can enable it. It's a shame usb0 is not enabled by default like in previous OS versions, though. Thanks for the hint :)

Nahuel92, 2023-11-24:
I was able to make it work thanks to your hint. This is what I did to be able to use SSH over USB-C:

## /boot/config.txt

```properties
# USB-C
dtoverlay=dwc2
```

## /boot/cmdline.txt

```properties
rootwait modules-load=dwc2,g_ether
```

## first_run.sh

Before the `rm ...` command, add this:

```text
cat << EOF > /etc/network/interfaces.d/usb0
auto usb0
allow-hotplug usb0
iface usb0 inet static
    address 192.168.2.2
    netmask 255.255.255.0
    gateway 192.168.2.1
EOF
```

This seems to be a hacky solution, though.

## On Mac host, configure `RNDIS/Ethernet Gadget` as follows:

```text
IPv4: Manually
IP Address: 192.168.2.1
Network Mask: 255.255.255.0
Router: Leave it empty.
DNS Servers: 8.8.8.8 or whatever you prefer
```

aanuroop, 2024-01-02:
Can you please let me know what this first_run.sh is ?  and "Before the rm ... command, add this:" as well ? Thanks in advance.

Nahuel92, 2024-01-03:
> Can you please let me know what this first_run.sh is ? and "_Before the rm ... command, add this_:" as well ? Thanks in advance.

Hi.

You'll find the `first_run.sh` script in the root directory of the micro-SD card immediately after installing Raspberry OS and before booting the OS for the first time (it will be automatically deleted after first boot).

And the `rm...` line is inside that script (it's the line that auto-removes the script after first OS boot).

ThomasKaiser, 2024-02-14:
Late to the game (here in Germany RPi 5 can't be bought by normal people up until few weeks ago).

To restore the former behaviour (activating `dwc2`/`g_ether` ending up with the RPi being accessible via a proper – random but unique – link local address) all that would be needed (by the makers of the distro) is:

  * adding `avahi-autoipd` to list of default packages
  * adding the file `/etc/network/interfaces.d/usb0` with the contents below [1] to the distro

`usb0` remains unconfigured by Network Manager and if `g_ether` isn't loaded the `/etc/network/interfaces.d/usb0` does no harm. But in case `dwc2`/`g_ether` are loaded `avahi-autoipd` simply does the job. RPi5 connected to an USB-C port of my Mac:

    tk@mac-tk ~ % system_profiler SPUSBDataType               
    ...
          Host Controller Driver: AppleT6000USBXHCI
    
            RNDIS/Ethernet Gadget:
    
              Product ID: 0xa4a2
              Vendor ID: 0x0525  (PLX Technology, Inc.)
              Version: 6.01
              Speed: Up to 480 Mb/s
              Manufacturer: Linux 6.1.0-rpi8-rpi-2712 with 1000480000.usb
              Location ID: 0x02100000 / 1
              Current Available (mA): 500
              Current Required (mA): 0
              Extra Operating Current (mA): 0
    ...
    
    tk@mac-tk ~ % ssh rpi5.local
    tk@rpi5.local's password: 
    Linux rpi5 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64
     
    tk@rpi5:~ $ w
     11:03:14 up 1 min,  1 users,  load average: 0.00, 0.01, 0.07
    USER     TTY      FROM             LOGIN@   IDLE   JCPU   PCPU WHAT
    tk       pts/0    fe80::f4d4:88ff: 11:03    1.00s  0.02s   ?    w
       
    tk@rpi5:~ $ nmcli dev
    DEVICE         TYPE      STATE                   CONNECTION           
    wlan0          wifi      connected               Volkartstueberl 5GHz 
    lo             loopback  connected (externally)  lo                   
    p2p-dev-wlan0  wifi-p2p  disconnected            --                   
    eth0           ethernet  unavailable             --                   
    usb0           ethernet  unmanaged               --                   
    
    tk@rpi5:~ $ ip a show dev usb0
    3: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
        link/ether c6:c2:bd:99:f0:b0 brd ff:ff:ff:ff:ff:ff
        inet 169.254.16.182/16 brd 169.254.255.255 scope link usb0:avahi
           valid_lft forever preferred_lft forever
        inet6 fe80::c4c2:bdff:fe99:f0b0/64 scope link 
           valid_lft forever preferred_lft forever

[1] `/etc/network/interfaces.d/usb0` (indentation with tabs):

    allow-hotplug usb0
    iface usb0 inet manual
    	up /usr/sbin/avahi-autoipd --force-bind -D $IFACE
    	down /usr/sbin/avahi-autoipd --kill $IFACE

verxion, 2024-02-15:
@ThomasKaiser - thank you for writing that up - it REALLY makes it sound easy and simple and straightforward.

…unfortunately I couldn’t get it to work.  At all.

Specifically the usb0 device never shows up and never is configured, despite following all of your steps.

My use case is to have the pi5 (as I’ve ALWAYS done with my pi4) be powered and providing ethernet to/from my iPad Pro.  So I’m SSHing into the pi5 from my iPad Pro and yeah, I never get a usb0 device, despite having installed avahi-autoipd and adding your usb0 script.  Note that I’d already had the relevant bits in my config.txt / cmdline.txt from earlier efforts…’’

Any advice to get this to actually work would be GREATLY appreciated!  THANK YOU!  :)

ThomasKaiser, 2024-02-16:
> Any advice to get this to actually work would be GREATLY appreciated

Well, my aim was to point @pelwell and colleagues into the right direction since it's really just installing `avahi-autoipd` and creating `/etc/network/interfaces.d/usb0` with the aforementioned contents. No idea what went wrong in your installation but most probably it's a typo (and end users shouldn't need to care about anything related to this since the OS maintainers should take care)

pelwell, 2024-02-16:
I found a similar thread with a similar conclusion on the forums: https://forums.raspberrypi.com/viewtopic.php?t=357938
Another thread (Thagrol again) focuses on the ZeroConf aspect: https://forums.raspberrypi.com/viewtopic.php?t=358527

@XECDesign What is the preferred method for getting link local IP addresses, now that we're using Network Manager? The nmcli options `ipv4.method link-local ipv4.link-local enabled` are the most obvious, but the lack of an automatic failover makes it painful. 

Is it better to install dhcpcd instead?

verxion, 2024-02-16:
Well, I installed dhcpcd. It didn't work either. I followed all the steps that worked on Pi4 and they don't work (for me) on Pi5...

ThomasKaiser, 2024-02-16:
> Is it better to install dhcpcd instead?

I've to admit that I haven't thought about `dhcpcd` at all 

    tk@rpi5:~ $ apt-cache show avahi-autoipd | grep ^Size
    Size: 40788
    tk@rpi5:~ $ apt-cache show dhcpcd | grep ^Size
    Size: 11088

Unfortunately `dhcpcd-base` will be pulled in by `dhcpcd` as well and so we're well above 600K. Asides that this works too:

    tk@mac-tk ~ % ssh rpi5.local
    Linux rpi5 6.1.0-rpi8-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64
    ...
    
    tk@rpi5:~ $ w
     12:35:10 up 0 min,  1 user,  load average: 0.83, 0.31, 0.11
    USER     TTY      FROM             LOGIN@   IDLE   JCPU   PCPU WHAT
    tk       pts/0    fe80::851:666e:e 12:35    2.00s  0.02s   ?    w
        
    tk@rpi5:~ $ ip a show dev usb0
    3: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
        link/ether 02:84:db:3f:82:06 brd ff:ff:ff:ff:ff:ff
        inet 169.254.228.14/16 brd 169.254.255.255 scope global noprefixroute usb0
           valid_lft forever preferred_lft forever
        inet6 fe80::84:dbff:fe3f:8206/64 scope link 
           valid_lft forever preferred_lft forever
    
    tk@rpi5:~ $ cat /etc/network/interfaces.d/usb0 
    allow-hotplug usb0
    iface usb0 inet dhcp

ThomasKaiser, 2024-02-16:
> Well, I installed dhcpcd. It didn't work either

Since you seem to already suffer from no `usb0` device being created every other step can only fail. If your iPad shows nothing connected via USB then there's nothing to fiddle around with `dhcpd` or something else. Check with `dmesg | grep -E "dwc|usb"` what's going on.

<details>
  <summary>As a reference output from my box:</summary>

    root@rpi5:/home/tk# dmesg | grep -E "dwc|usb"
    [    0.000000] Kernel command line: reboot=w coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=D8:3A:DD:BC:D8:AD vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=c3bda67a-02 rootfstype=ext4 fsck.repair=yes modules-load=dwc2,g_ether rootwait
    [    0.000000] Unknown kernel command line parameters "modules-load=dwc2,g_ether", will be passed to user space.
    [    0.809835] usbcore: registered new interface driver usbfs
    [    0.815348] usbcore: registered new interface driver hub
    [    0.820684] usbcore: registered new device driver usb
    [    0.825786] usb_phy_generic phy: supply vcc not found, using dummy regulator
    [    1.308218] usbcore: registered new device driver r8152-cfgselector
    [    1.314527] usbcore: registered new interface driver r8152
    [    1.320043] usbcore: registered new interface driver lan78xx
    [    1.325730] usbcore: registered new interface driver smsc95xx
    [    1.331578] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
    [    1.337363] dwc_otg: FIQ enabled
    [    1.337365] dwc_otg: NAK holdoff enabled
    [    1.337366] dwc_otg: FIQ split-transaction FSM enabled
    [    1.337369] Module dwc_common_port init
    [    1.337477] usbcore: registered new interface driver uas
    [    1.342818] usbcore: registered new interface driver usb-storage
    [    1.426736] usbcore: registered new interface driver usbhid
    [    1.432328] usbhid: USB HID core driver
    [    2.024210] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
    [    2.032526] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    2.039778] usb usb1: Product: xHCI Host Controller
    [    2.044673] usb usb1: Manufacturer: Linux 6.1.0-rpi8-rpi-2712 xhci-hcd
    [    2.051225] usb usb1: SerialNumber: xhci-hcd.0
    [    2.063711] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
    [    2.072013] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    2.079265] usb usb2: Product: xHCI Host Controller
    [    2.084159] usb usb2: Manufacturer: Linux 6.1.0-rpi8-rpi-2712 xhci-hcd
    [    2.090785] usb usb2: SerialNumber: xhci-hcd.0
    [    2.208425] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
    [    2.216836] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    2.224100] usb usb3: Product: xHCI Host Controller
    [    2.228995] usb usb3: Manufacturer: Linux 6.1.0-rpi8-rpi-2712 xhci-hcd
    [    2.235649] usb usb3: SerialNumber: xhci-hcd.1
    [    2.248125] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
    [    2.256551] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    2.263804] usb usb4: Product: xHCI Host Controller
    [    2.268698] usb usb4: Manufacturer: Linux 6.1.0-rpi8-rpi-2712 xhci-hcd
    [    2.275414] usb usb4: SerialNumber: xhci-hcd.1
    [    2.619889]     modules-load=dwc2,g_ether
    [    2.626163] usb 4-1: new SuperSpeed USB device number 2 using xhci-hcd
    [    2.651229] usb 4-1: New USB device found, idVendor=0781, idProduct=5590, bcdDevice= 1.00
    [    2.659466] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [    2.667150] usb 4-1: Product: Ultra
    [    2.670691] usb 4-1: Manufacturer: SanDisk
    [    2.675027] usb 4-1: SerialNumber: 4C531001400209122575
    [    2.681129] usb-storage 4-1:1.0: USB Mass Storage device detected
    [    2.687405] scsi host0: usb-storage 4-1:1.0
    [    2.740370] dwc2 1000480000.usb: supply vusb_d not found, using dummy regulator
    [    2.769883] dwc2 1000480000.usb: supply vusb_a not found, using dummy regulator
    [    2.994055] dwc2 1000480000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
    [    3.001494] dwc2 1000480000.usb: DWC OTG Controller
    [    3.006397] dwc2 1000480000.usb: new USB bus registered, assigned bus number 5
    [    3.013659] dwc2 1000480000.usb: irq 164, io mem 0x1000480000
    [    3.019463] usb usb5: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
    [    3.027765] usb usb5: New USB device strings: Mfr=3, Product=2, SerialNumber=1
    [    3.035017] usb usb5: Product: DWC OTG Controller
    [    3.039738] usb usb5: Manufacturer: Linux 6.1.0-rpi8-rpi-2712 dwc2_hsotg
    [    3.046465] usb usb5: SerialNumber: 1000480000.usb
    [    4.343562] usb0: HOST MAC fa:b0:9b:48:f3:82
    [    4.347984] usb0: MAC 02:84:db:3f:82:06
    [    4.378965] dwc2 1000480000.usb: bound driver g_ether
    [    4.631680] dwc2 1000480000.usb: new device is high-speed
    [    4.682019] dwc2 1000480000.usb: new device is high-speed
    [    4.703233] dwc2 1000480000.usb: new address 1
    [    5.446883] usbcore: registered new interface driver brcmfmac
</details>

verxion, 2024-02-16:
ThomasKaiser: Thanks for the insights.

I seem to be getting entirely different DWC OTG info than you:

[    0.387987] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    0.388061] dwc_otg: FIQ enabled
[    0.388063] dwc_otg: NAK holdoff enabled
[    0.388066] dwc_otg: FIQ split-transaction FSM enabled
[    0.388070] Module dwc_common_port init

Here is the complete return of your grep command:

```
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:BC:5B:C4 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=4d83a40d-02 rootfstype=ext4 fsck.repair=yes rootwait modules-load=dwc2,g_ether quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
[    0.000000] Unknown kernel command line parameters "splash modules-load=dwc2,g_ether", will be passed to user space.
[    0.040607] usbcore: registered new interface driver usbfs
[    0.040617] usbcore: registered new interface driver hub
[    0.040629] usbcore: registered new device driver usb
[    0.040729] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    0.387826] usbcore: registered new interface driver r8152
[    0.387839] usbcore: registered new interface driver lan78xx
[    0.387850] usbcore: registered new interface driver smsc95xx
[    0.387987] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    0.388061] dwc_otg: FIQ enabled
[    0.388063] dwc_otg: NAK holdoff enabled
[    0.388066] dwc_otg: FIQ split-transaction FSM enabled
[    0.388070] Module dwc_common_port init
[    0.388328] usbcore: registered new interface driver uas
[    0.388340] usbcore: registered new interface driver usb-storage
[    0.403238] usbcore: registered new interface driver usbhid
[    0.403241] usbhid: USB HID core driver
[    0.617987] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    0.617992] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.617996] usb usb1: Product: xHCI Host Controller
[    0.617999] usb usb1: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    0.618002] usb usb1: SerialNumber: xhci-hcd.0
[    0.618387] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[    0.618392] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.618396] usb usb2: Product: xHCI Host Controller
[    0.618399] usb usb2: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    0.618402] usb usb2: SerialNumber: xhci-hcd.0
[    0.677958] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    0.677962] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.677965] usb usb3: Product: xHCI Host Controller
[    0.677969] usb usb3: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    0.677972] usb usb3: SerialNumber: xhci-hcd.1
[    0.678308] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[    0.678313] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.678316] usb usb4: Product: xHCI Host Controller
[    0.678319] usb usb4: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    0.678322] usb usb4: SerialNumber: xhci-hcd.1
[    0.874709]     modules-load=dwc2,g_ether
[    2.964730] usbcore: registered new interface driver brcmfma
```

ThomasKaiser, 2024-02-16:
@verxion please let us stop here hijacking this issue with random support stuff that should go to the forum. I would try to remove the `quiet splash` string from your monstrous `cmdline.txt`, and if after another reboot still no `usb0` device shows up and nothing wrt `dwc2 1000480000.usb` appears in `dmesg` output then open a thread in the support forum.

In case it works simply leave a 'thumbs up' here :)

verxion, 2024-02-16:
Thanks ThomasKaiser, but that didn’t help.  Could you please provide a link to whichever forum you want me to post on?

```
dmesg | grep -E "dwc|usb|g_ether"
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:BC:5B:C4 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=4d83a40d-02 rootfstype=ext4 fsck.repair=yes rootwait modules-load=dwc2,g_ether plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
[    0.000000] Unknown kernel command line parameters "modules-load=dwc2,g_ether", will be passed to user space.
[    0.827451] usbcore: registered new interface driver usbfs
[    0.832967] usbcore: registered new interface driver hub
[    0.838308] usbcore: registered new device driver usb
[    0.843479] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    1.342768] usbcore: registered new interface driver r8152
[    1.348297] usbcore: registered new interface driver lan78xx
[    1.353989] usbcore: registered new interface driver smsc95xx
[    1.359901] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
[    1.365742] dwc_otg: FIQ enabled
[    1.365745] dwc_otg: NAK holdoff enabled
[    1.365747] dwc_otg: FIQ split-transaction FSM enabled
[    1.365751] Module dwc_common_port init
[    1.366013] usbcore: registered new interface driver uas
[    1.371359] usbcore: registered new interface driver usb-storage
[    1.456747] usbcore: registered new interface driver usbhid
[    1.462342] usbhid: USB HID core driver
[    2.062106] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    2.070429] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.077685] usb usb1: Product: xHCI Host Controller
[    2.082582] usb usb1: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    2.089136] usb usb1: SerialNumber: xhci-hcd.0
[    2.101757] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[    2.110063] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.117317] usb usb2: Product: xHCI Host Controller
[    2.122213] usb usb2: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    2.128868] usb usb2: SerialNumber: xhci-hcd.0
[    2.246398] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    2.254837] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.262108] usb usb3: Product: xHCI Host Controller
[    2.267005] usb usb3: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    2.273695] usb usb3: SerialNumber: xhci-hcd.1
[    2.286278] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[    2.294763] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.302019] usb usb4: Product: xHCI Host Controller
[    2.306915] usb usb4: Manufacturer: Linux 6.1.0-rpi7-rpi-2712 xhci-hcd
[    2.313691] usb usb4: SerialNumber: xhci-hcd.1
[    2.681582]     modules-load=dwc2,g_ether
[    4.465691] UDC core: g_ether: couldn't find an available UDC
[    5.516161] usbcore: registered new interface driver brcmfmac
```

ThomasKaiser, 2024-02-17:
https://forums.raspberrypi.com/viewforum.php?f=28

And if you watch closely your `dmesg` output cleaning up your `cmdline.txt` actually helped and you're one step further since now at least `g_ether` driver is loaded just resulting in another error [couldn't find an available UDC](https://www.google.com/search?q=%22couldn%27t+find+an+available+UDC%22+rpi+why%3F) (maybe related to `dwc2` not being loaded so check with `lsmod`, `modprobe` and friends. At least you're far away from upper config layers like `dhcpd` since `usb0` is still missing and testing this stuff with a vanilla Raspberry Pi OS install would be the way to go).

XECDesign, 2024-02-19:
> @XECDesign What is the preferred method for getting link local IP addresses, now that we're using Network Manager? The nmcli options `ipv4.method link-local ipv4.link-local enabled` are the most obvious, but the lack of an automatic failover makes it painful.

Sorry, I'm not sure about this one. I remember looking it up a while ago and seeing people say "yes, you can do this with those options" but no explanation how to get fallback working properly. I haven't tried to get it working myself, so I have no idea.

We used to patch dhcpcd a bit to make it work properly. Now it's just the stock Debian version, so I can't vouch for it.

garytorgersen, 2024-03-09:
Forget g_ether.  Change these:

modules-load=dwc2
Add libcomposite to the end of the /etc/modules file

You will need to do the configuration for libcomposite but you are already creating the network interface so should be simple.

verxion, 2024-03-14:
I wasn’t ever able to use what I found here to get things working. I did my own write up. It works with a USB-C to USB-C cable between iPad and pi5.  I’ve tried the steps multiple times with fresh micro SD cards - works consistently every single time. 

Hopefully this will help some other people out there. 

https://github.com/verxion/RaspberryPi/blob/main/Pi5-ethernet-and-power-over-usbc.md

-Verxion

dimitrovs, 2024-07-24:
Forget USB-C, Raspberry OS 12 Bookworm renders the Raspberry Pi Zero completely unusable as that device does not have any other networking interface.

josafat-daysequerra, 2024-08-03:
I used the guide posted by verxion and got the USB-C device to show up with the correct IP address, I can ping it, however when I try to connect I get the error "Connection Refused" obviously check sshd file and it's set up to listen on all interfaces. Anybody else had this issue?

suttontools, 2024-09-20:
> Forget USB-C, Raspberry OS 12 Bookworm renders the Raspberry Pi Zero completely unusable as that device does not have any other networking interface.

Hi For all those struggling with online guides and the Pi Zero W, use the legacy 32 bit Bullseye based OS and then they work as described.  I don't know how @dimitrovs comment could be better publicised?  I failed to configure the wifi properly (thought I had) so I expected I could  modify the boot files.  Now - days later - I have access via the usb0 interface .
Thanks

kithamon, 2024-10-12:
> Well, my aim was to point @pelwell and colleagues into the right direction since it's really just installing `avahi-autoipd` and creating `/etc/network/interfaces.d/usb0` with the aforementioned contents.

Note for anyone searching this issue: these 2 steps worked perfectly to enable USB networking on my Pi Zero 2W running Bookworm Lite 32 bit.

marko-pi, 2025-01-01:
I use USB-C for diagnostics on my Pi Zero W with Bookworm and the two-step solution from [ThomasKaiser](https://github.com/ThomasKaiser) also worked for me.

TenoTrash, 2025-01-11:
Hello there! I manage to make it work with my laptop and both Pi Zero 2W and my Pi4, mostly following this guide:

https://github.com/verxion/RaspberryPi/blob/main/Pi5-ethernet-and-power-over-usbc.md

But what about if I do want to use the USB tethering from my phone? I have tested with two different phones and the USB shared connection remains unavailable. It do works if I use a USB-A to USB-C cable and power my Raspi 2W or 4 with an external power adapter.

Cheers!

ch4r13, 2025-02-01:
So, How should I fix it ? 
- I don't want to/cannot use Wi-Fi, so everything need to be prepared on MicroSD card, as @marko-pi mentionedin his post @ThomasKaiser and his two-step solution, which means.:

>    adding avahi-autoipd to list of already installed default packages
>    adding the file /etc/network/interfaces.d/usb0 with the contents below [1] to their OS images

I cannot figure out how to.:
- How can I install `avahi-autoipd` ? Should I use `chroot` or is it necessary? Because I can see that it's preinstalled only `avahi-daemon` in `/rootfs/usr/sbin/` 
- I tried to add that file in `/etc/network/interfaces.d/usb0` but it still doesn't work for me.
I'm using [this image](https://downloads.raspberrypi.com/raspios_lite_armhf/images/raspios_lite_armhf-2024-11-19/) So I hope that there isn't any difference.

Also I edited `cmdline.txt` => `modules-load=dwc2,g_ether` and `config.txt` => `dtoverlay=dwc2` on desired lines...

meshosk, 2025-04-07:
@ch4r13 Got same issue and made it working (on pico 2 w) as @ThomasKaiser noted. Just install `avahi-autoipd`, create file `/etc/network/interfaces.d/usb0` with given content. 

The last piece of the puzzle for me was missing `ifupdown` (I'm using official Bookworm 12 lite 64b). I installed it, reboot and it start working.

Kree879, 2025-04-18:
I have flashed Raspberry OS (Bookworm) multiple times to an SD card. Initial setup has Wi-Fi configured. I can then SSH via Wi-Fi. When I connect via cable, my pc detects the RNDIS Gadget device. If I do the OS flash with no Wi-Fi configured, I cannot SSH in via USB cable. The device shows up but the shows "Network Cable Unplugged". I have checked the Config and CMDLINE file and they are both correct. There is an SSH file in root. Cannot get this working.

jdeitmerg, 2025-11-25:
Thanks, @EamonTracey, for the clean solution.

It didn't work unmodified for me on Raspberry Pi OS Bullseye (11.7), 64-bit. NetworkManager would still **not** manage `usb0`. I changed `/etc/udev/rules.d/99-usb0-managed.rules` to

```
ENV{DEVTYPE}=="gadget", ENV{NM_UNMANAGED}="0"
```

This way it works on my system.

macmpi, 2025-11-25:
indeed, default `/usr/lib/udev/rules.d/85-nm-unmanaged.rules` has (in Trixie too):
```
$ grep "NM_UNMANAGED" /usr/lib/udev/rules.d/85-nm-unmanaged.rules
ENV{INTERFACE}=="vboxnet[0-9]*", ENV{NM_UNMANAGED}="1"
ATTR{address}=="00:50:56:*", ENV{INTERFACE}=="vmnet[0-9]*", ENV{NM_UNMANAGED}="1"
ATTR{address}=="00:1c:42:*", ENV{INTERFACE}=="vnic[0-9]*", ENV{NM_UNMANAGED}="1"
ENV{ID_NET_DRIVER}=="veth", ENV{INTERFACE}!="eth[0-9]*", ENV{NM_UNMANAGED}="1"
ENV{DEVTYPE}=="gadget", ENV{NM_UNMANAGED}="1"
```

chr0n1x, 2025-12-04:
hello all, I've followed a bunch of these comments (hopefully not blindly). I've run `rpi-update` successfully & rebooted my device with the following changes

1. `/boot/firmware/config.txt` has `dtoverlay=dwc2` in the `[all]` section
2. added `modules-load=dwc2,g_ether` to `/boot/firmware/cmdline.txt`, no new lines
3. ran the `nmcli` commands that @EamonTracey listed [here](https://github.com/raspberrypi/linux/issues/5737#issuecomment-3067352701)
4. added `ACTION=="add", SUBSYSTEM=="net", KERNEL=="usb0",  ENV{NM_UNMANAGED}="0"` into `/etc/udev/rules.d/99-usb0-managed.rules`

output of `ip l` 
```
» ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
    link/ether 88:a2:9e:24:e2:d0 brd ff:ff:ff:ff:ff:ff
3: usb0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1000
    link/ether e2:b1:78:2d:53:c2 brd ff:ff:ff:ff:ff:ff
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether 88:a2:9e:24:e2:d1 brd ff:ff:ff:ff:ff:ff
```

so `usb0` is there 👍 

`ifconfig` also confirms that it recognizes the interface:

```
nox ~ » ifconfig

# .. other interfaces

usb0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether e2:b1:78:2d:53:c2  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

however, when I connect my rpi5 to my ipad and try to `ssh pi@10.55.0.1` in iSH, I'm getting a `Network unreachable` error.

the iPad itself is allowed to discover devices on my LAN, and when I plug in my rpi5, I `allow` it to connect to the iPad via the iOS prompts.

when _both_ devices are on my wifi/LAN, I can SSH from my ipad into the rpi5. I haven't configured anything on my `sshd_config`, so `Host` is set to `*` by default; I'm assuming that it's able and ready to accept connections on this interface.

Anyone else run into similar issues as me? any direction would be greatly appreciated.