raspberrypi/linux #5743: Pi 5 and soundcards - problems and solutions

pelwell, 2023-11-24:
### Describe the bug

This issue is meant to be a common place to discuss the changes required for I2S support on Pi 5. Existing issues that cover this area are https://github.com/raspberrypi/linux/issues/5724 and https://github.com/raspberrypi/linux/issues/5741, and some of the content below is copied from them.

Pi 5 (specifically RP1) has slightly different I2S capabilities than those found on the BCM2835 family of processors. It gains the ability to do dual- and quad-lane I2S, but it has separate clock producer (master) and clock consumer (slave) blocks. This means that each overlay has to specify the correct I2S controller - producer or consumer. For most soundcards this is a fixed property, but a few of them can operate in either mode, which causes a problem.

The addition of a "slave" property was a way to solve the problem, but this requires user intervention (and knowledge). A better solution might be to give the overlays for those flexible cards two different i2s-controller properties (or a single property with two label references) - one for each scenario - and modify the driver to choose the correct one at runtime.

There can also be problems for Audio HATs with EEPROMs that have embedded overlays, rather than just including the name of an overlay. By counting how many cards use the I2S interface in producer mode and how many in consumer mode, the old label `i2s` was added as an alias for the clock producer block. If this is wrong for a soundcard with an embedded overlay then there is no choice but to suppress the embedded overlay and load one from the OS image instead (unless the EEPROM can be rewritten).

Putting `dtoverlay=` in config.txt before any other dt settings disables the loading of the overlay from the HAT EEPROM. The preferred overlay can then be loaded in the usual way. Alternatively, if all the instances of a soundcard have been programmed with the same UUID then a rule can be added to the `hat_map`, an example of which can be seen in #5742.

### Steps to reproduce the behaviour

blah

### Device (s)

Other

### System

Latest images

pelwell, 2023-11-24:
@gtrainivicious @j-schambacher This is a more open and more efficient place to discuss the required changes than individual issues or private emails.

j-schambacher, 2023-11-24:
@pelwell Thanks Phil, for aligning things here. I will summarize our findings and post here.

pelwell, 2023-11-28:
You've been beaten to it: https://github.com/raspberrypi/linux/issues/5747

pelwell, 2023-11-28:
#5757 is a potential fix for #5747.

j-schambacher, 2023-12-01:
@pelwell
1) Sorry for the delay, I was beaten up with other stuff. I was wondering how we should approach the correct controller/target setup for the new I2S module. It's a 'first come, first wins' solution. It means the first DT definition loads the respective module and pin configuration. Any later module or DT definition fails, especially the former technique where the driver switched BCLK/LRCLK between controller/target. Without additional code for the PI5 this 'autodetect' will never work. Do you agree?
This means (w/o driver change), the potentially best way is your suggestion to use different DT-overlay definitions, which still is a 'challenge' for the existing HATs w/o UUID. But I wonder, why it seems sometimes the controller-module/DT is loaded before the card-DT is loaded requiring the target-module and fails. Is there any DT config which loads the I2S controller module per se?

2) Meanwhile I got the 8 channels running on the DAC side but I need to continue work on the driver side. It somehow interferes with 1) as we need to use again a different DT-overlay including additional pin configs.

j-schambacher, 2023-12-01:
@pelwell I know -off topic, but I don't know where to place it best. It's not a bug or solely related to PI5. You may remember we are working on an amplifier which uses TAS575x in controller mode. Meanwhile, we have the necessary patches in the upstream kernel.
https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?id=736b884a7b68c4eeb66dbf75b97c8ec9b9eeff7f
https://git.kernel.org/pub/scm/linux/kernel/git/broonie/sound.git/commit/?id=1f817805262c2c34142291da376d4932d3c493bc
It has been merged into the 6.7-rc1. What do you suggest how and when we can pull it into the raspberry Linux?

pelwell, 2023-12-01:
We're very happy to take back-ports. Just cut-and-paste that into a new issue and it will be dealt with promptly.

I hope to put together some answers and suggestions in the next day or two, but I agree that setting the pinmux is the main hurdle to choosing the I2S controller at runtime.

pelwell, 2023-12-01:
> Just cut-and-paste that into a new issue and it will be dealt with promptly.

On second thoughts, don't - I'll just do it

pelwell, 2023-12-01:
Done.

clone206, 2024-02-24:
Using i-sabre-q2m overlay with an i2s reclocker HAT is limited to 192KHz sampling rates on the pi 5.

Manual/data sheet for HAT is here: https://github.com/iancanada/DocumentDownload/blob/master/FifoPi/FifoPiQ7/FifoPiQ7Manual.pdf

It specifies that you can use the overlay mentioned above or the generic i2s dac overlay (my understanding is that's the "hifiberry-dac" overlay). I have tried with both overlays and encounter the same sample rate limitation on the pi 5. Also I can't pass DoP through to my DAC over LVDS on the pi 5 whereas I can on the pi 4. Just get low level noise at 176.4KHz PCM rate.

On the pi 4 with the same kernel and overlays the HAT can accept up to 384KHz sample rates and DoP works.

```
clone206@streamer:~ $ uname -a
Linux streamer 6.6.16-v8-16k+ #3 SMP PREEMPT Wed Feb 14 22:08:37 PST 2024 aarch64 GNU/Linux
clone206@streamer:~ $
clone206@streamer:~ $ alsacap
*** Scanning for playback devices ***
Card 0, ID `DAC', name `I-Sabre Q2M DAC'
  Device 0, ID `I-Sabre Q2M DAC i-sabre-codec-dai-0', name `I-Sabre Q2M DAC i-sabre-codec-dai-0', 1 subdevices (1 available)
    2 channels, sampling rate 8000..384000 Hz
    Sample formats: S16_LE, S32_LE
      Subdevice 0, name `subdevice #0'
clone206@streamer:~ $
clone206@streamer:~ $ lsmod |grep sabre
snd_soc_i_sabre_codec    49152  1
regmap_i2c             49152  1 snd_soc_i_sabre_codec
snd_soc_i_sabre_q2m    49152  1
snd_soc_core          327680  5 snd_soc_i_sabre_q2m,vc4,snd_soc_hdmi_codec,designware_i2s,snd_soc_i_sabre_codec
snd_pcm               163840  8 snd_soc_i_sabre_q2m,snd_soc_hdmi_codec,designware_i2s,snd_compress,snd_soc_i_sabre_codec,snd_soc_core,snd_pcm_dmaengine
```

config.txt:
```
dtparam=i2c_arm=on
force_eeprom_read=0
#dtoverlay=i2c0-pi5,baudrate=1000000
#dtparam=i2s=on
#dtparam=spi=on
dtparam=pwr_led_trigger=default-on
dtparam=pwr_led_activelow=off
dtparam=act_led_trigger=default-on
dtparam=act_led_activelow=off
dtparam=eth_led0=14
dtparam=eth_led1=14
dtoverlay=disable-wifi
dtoverlay=disable-bt

# Enable audio (loads snd_bcm2835)
dtparam=audio=off

#dtoverlay=hifiberry-dac
dtoverlay=i-sabre-q2m,slave
#dtoverlay=i2s-dac,slave

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
#camera_auto_detect=1

# Automatically load overlays for detected DSI displays
#display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
#max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1
[... 488 lines cut ...]
[ 8552.208407] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000010] -121
[ 8552.208873] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8842.772392] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8854.222596] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000010] -121
[ 8854.223049] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8897.397952] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8905.091242] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8926.459710] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8933.159147] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8944.539024] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8953.459598] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8962.390989] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 8976.339409] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 9019.899896] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 9022.140564] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[ 9056.973084] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[11405.279146] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[56890.305063] systemd[1407]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set
[82854.749925] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000010] -121
[... cut ...]
[84163.186237] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[86754.204553] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000010] -121
[86754.204760] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
[87187.626885] i-sabre-codec-i2c 1-0048: ASoC: error at snd_soc_component_update_bits on i-sabre-codec-i2c.1-0048 for register: [0x00000021] -121
```

pelwell, 2024-02-25:
You say that it is "limited to 192KHz sampling rates on the pi 5" - how does this limit manifest? What happens when you play a 384kHz stream?

clone206, 2024-02-25:
> You say that it is "limited to 192KHz sampling rates on the pi 5" - how does this limit manifest? What happens when you play a 384kHz stream?

The reclocker HAT fails to lock onto any incoming i2s. It cycles continuously between its two clocks as indicated by its status led's, and in addition, when using an add-on gpio display, it reads "no sig". The connected DAC's display reads "Unlock".

pelwell, 2024-02-25:
That tells us that ALSA and the drivers are all happy with the configuration, and that the failure happens when data starts to flow. I should be able to take a look at the signals using the i2s-dac overlay.

By the way:
```
dtoverlay=i-sabre-q2m,slave
```
The `i-sabre-q2m` overlay only works with clock producer ("master") mode - it doesn't understand any parameters (but the associated error is more of a warning). See `dtoverlay -h i-sabre-q2m`.

pelwell, 2024-02-28:
Just a quick note to say I looked at the signals on a logic analyser today and was able to see the way in which the frequency goes wrong. Hopefully we will have fewer infrastructure disruptions tomorrow and I'll be able to test a fix.

j-schambacher, 2024-02-29:
FYI: while working on the various drivers for our sound cards, I've noted that the clock generation for the 24bit format (S24_LE) could lead to a wrong sample frequency (LRCLK). For a 1kHz sine wave you get 750Hz. I guess the ratio of bit clock and LR clock is calculated incorrectly -> 750Hz/1kHz = 24/32. Could be fixed by fixed-bit-clock-ratio of 64... but it's not the solution. I've tracked it down to the hw_params in the dwc-i2s.c, but without docs I can only guess.

pelwell, 2024-02-29:
Yes, that's exactly what I'm redoing now.

pelwell, 2024-02-29:
With luck, #5999 will solve both of those problems.

pelwell, 2024-02-29:
The PR was aimed at 384kHz. There is no good PLL value for the higher multiples of 44100 - the closest we could get with integer divisors is 355.55kHz for 32-bit stereo, or 352.29kHz for 16-bit stereo - but I can look at the failure mode with 352.8kHz.

pelwell, 2024-02-29:
There is test data at various frequencies here: https://msbtechnology.com/support/bit-perfect-testing/
Do heed the warnings about 0db material!

clone206, 2024-02-29:
> There is test data at various frequencies here: https://msbtechnology.com/support/bit-perfect-testing/ Do heed the warnings about 0db material!

Thanks, will give it a try!

So based on your earlier reply, am I understanding correctly that the pi 5 will never be able to support 352.8KHz i2s?

pelwell, 2024-02-29:
It's worse than I thought - the options are (for 32-bit stereo):
* 50MHz/2/64 = 390.625kHz
* 122880000/5/64 = 384kHz
* 153600000/7/64 = 342.86kHz

> So based on your earlier reply, am I understanding correctly that the pi 5 will never be able to support 352.8KHz i2s?

That's not the case.
1. When running as an I2S clock consumer (i.e. with an external clock source) it can probably do any frequency you like - I would expect an audiophile product to have 2 external crystals - one for 48kHz family and one for the 44.1kHz family, or it's own tuneable PLL source.
2. It is possible to change the PLL frequency to make 352.8kHz possible, but not without losing 384kHz. I'm also not sure how you would switch between the two on the fly.

pelwell, 2024-02-29:
Out of curiosity, is it 16-, 24- or 32-bit audio that you have?

clone206, 2024-02-29:
> It's worse than I thought - the options are (for 32-bit stereo):
>
> * 50MHz/2/64 = 390.625kHz
> * 122880000/5/64 = 384kHz
> * 153600000/7/64 = 342.86kHz
>
>
> That's not the case.
>
> 1. When running as an I2S [...]

Thanks for the clarification. FWIW, I think 352.8KHz is in general more useful for an audiophile as that's the frequency that allows for DoP with dsd128. Also for anybody converting DSD to dxd that frequency would be more useful.

As far as I know, the HAT I'm currently using only works as a slave. It's basically a FIFO/reclocker, though I believe the maker has a similar HAT that works as a master.

Also, responding to your other question, this HAT can be set to convert any incoming bit depth losslessly to 32bit, and that's how I have it set. The actual file I was using to test 352.8KHz PCM is 24 bit.

Thanks again for looking into this. The pi 5 is an amazing piece of engineering, and I swear it actually sounds better with i2s HATs than the pi 4. Hoping it can be my go-to for audiophile sound.

pelwell, 2024-03-01:
I'm sorry - I didn't find a nice solution, then other issues took over and it got put to one side. It's still on the list.

clone206, 2024-03-01:
Confirming 24/384 works with the version I updated to. Also 24/44.1 no longer working.

clone206, 2024-03-03:
For those of us still learning and hoping to perhaps contribute some day, is it possible to do this?

1806336000/80/64 = 352.8kHz
1806336000/147/64 = 192kHz
...
1806336000/640/64 = 44.1kHz

pelwell, 2024-03-04:
It's more complicated than that:
1. There are two intermediate divisors with different ranges to choose from, both integer only. 
2. The final divisor for clk_i2s is only 8 bits, making it all-too-easy to put the lower frequencies out of reach.
I think we're going to have to modify the clock driver to switch between a number of PLL values.

j-schambacher, 2024-03-06:
@pelwell Finally - I hope - I've managed to dig into the DT, I2S module and clk_producer/clk_consumer stuff. Before going further and draw conclusions, I just want to x-check: the main reason for all older hats (providing clocks) not working lies in the formerly required GPIO settings in the HAT EEPROM applying ALT-0 (I2S producer) to the I2S0 module. The only way I've found is to use force_eeprom_read=0 and manually applying the correct overlay. Or, and I think this is the better solution, to provide all the UUIDs, because once the UUID is detected, the rest of HAT-data is completely ignored. Correct?

pelwell, 2024-03-06:
@clone206 #5999 has been updated with code to change the PLL settings on demand. The auto builds have almost finished - another 10 minutes, perhaps - and then its the usual `sudo rpi-update pulls/5999` to install.

j-schambacher, 2024-03-06:
@pelwell 
Here is the HAT dump (from an older DAC+ Pro):
```
# ---------- Dump generated by eepdump handling format version 0x01 ----------
#
# --Header--
# signature=0x69502d52
# version=0x01
# reserved=0
# numatoms=3
# eeplen=1465
# ----------

# Start of atom #0 of type 0x0001 and length 37
# Vendor info
product_uuid ef586afc-2efa-47a0-be2e-95a7d952fe98
product_id 0x0000
product_ver 0x0000
vendor "HiFiBerry"   # length=9
product "DAC+"   # length=4
# End of atom. CRC16=0x8c1e

# Start of atom #1 of type 0x0002 and length 32
# GPIO map info
gpio_drive 0
gpio_slew 0
gpio_hysteresis 0
back_power 0
#        GPIO  FUNCTION  PULL
#        ----  --------  ----
setgpio  2      ALT0     UP
setgpio  3      ALT0     UP
setgpio  18      ALT0     DEFAULT
setgpio  19      ALT0     DEFAULT
setgpio  20      ALT0     DEFAULT
setgpio  21      ALT0     DEFAULT
# End of atom. CRC16=0x9700

# Start of atom #2 of type 0x0003 and length 1360
dt_blob
D0 0D FE ED 00 00 05 4E 00 00 00 38 00 00 04 9C 
00 00 00 28 00 00 00 11 00 00 00 10 00 00 00 00 
.
.
and so on
```

and here is the serial log showing that the GPIO config from the EEPROM is loaded trying to use the pins at I2S@a0000 (which is the I2S0). Later, when the driver tries to change the master/slave behavior it fails because of the GPIO/I2S settings.

```
Read bcm2712-rpi-5-b.dtb bytes    75445 hnd 0x239ba
MESS:00:00:07.116058:0: dtdebug: delete_node(/__local_fixups__)
Read /overlays/overlay_map.dtb bytes     5195 hnd 0x23be7
MESS:00:00:07.142770:0: dtdebug: using platform 'bcm2712'
MESS:00:00:07.150655:0: dtdebug: overlay map loaded
Read /overlays/hat_map.dtb bytes      489 hnd 0x23d43
Read /config.txt bytes     1384 hnd 0x23d42
MESS:00:00:07.232293:0: dtdebug: merge_fragment(/clocks,/fragment@0/__overlay__)
MESS:00:00:07.242099:0: dtdebug: merge_fragment(/clocks/dacpro_osc,/fragment@0/__overlay__/dacpro_osc)
MESS:00:00:07.248332:0: dtdebug:   +prop(compatible)
MESS:00:00:07.253491:0: dtdebug:   +prop(#clock-cells)
MESS:00:00:07.258396:0: dtdebug:   +prop(phandle)
MESS:00:00:07.262842:0: dtdebug: merge_fragment() end
MESS:00:00:07.267184:0: dtdebug: merge_fragment() end
MESS:00:00:07.281721:0: dtdebug: merge_fragment(/axi/pcie@120000/rp1/i2s@a0000,/fragment@1/__overlay__)
MESS:00:00:07.288032:0: dtdebug:   +prop(status)
MESS:00:00:07.293115:0: dtdebug: merge_fragment() end
MESS:00:00:07.306362:0: dtdebug: merge_fragment(/axi/pcie@120000/rp1/i2c@74000,/fragment@2/__overlay__)
MESS:00:00:07.312674:0: dtdebug:   +prop(#address-cells)
MESS:00:00:07.318501:0: dtdebug:   +prop(#size-cells)
MESS:00:00:07.323303:0: dtdebug:   +prop(status)
MESS:00:00:07.331139:0: dtdebug: merge_fragment(/axi/pcie@120000/rp1/i2c@74000/pcm5122@4d,/fragment@2/__overlay__/pcm5122@4d)
MESS:00:00:07.339372:0: dtdebug:   +prop(#sound-dai-cells)
MESS:00:00:07.345417:0: dtdebug:   +prop(compatible)
MESS:00:00:07.350041:0: dtdebug:   +prop(reg)
MESS:00:00:07.354151:0: dtdebug:   +prop(clocks)
```
In the kernel messages you find a lot of deferred probe messages.
I think the problem is the GPIO setting. Is that why the new HAT+ spec does not allow type 2 atoms anymore?
The listing above is for the case if the UUID is not recognized in the hat_map. Once it is found, it works fine. Also 'strange' the HAT UUID seems to be only recognized once there is also the DT-voerlay in the EERPROM defined?! The reason why I did not get it to work earlier.

clone206, 2024-03-07:
> @clone206 #5999 has been updated with code to change the PLL settings on demand. The auto builds have almost finished - another 10 minutes, perhaps - and then its the usual `sudo rpi-update pulls/5999` to install.

Confirming all is well after updating. The HAT locks onto anything I throw at it now. This will make a lot of audiophiles happy. Thank you!

Going to analyze the PR and hopefully learn some things.

clone206, 2024-03-07:
One minor update, I'm still seeing the issue with DoP with the updated PR. This one is curious to me since AFAIK nothing in the chain should be the wiser as the DSD bits are "smuggled" in through normal PCM packets. Still, it's minor as the HAT can directly decode the DoP and pass along the native DSD bitstream to the DAC. But it did work on the pi 4 without needing the HAT's decoder enabled.

https://dsd-guide.com/dop-open-standard

pelwell, 2024-03-07:
> I think the problem is the GPIO setting. Is that why the new HAT+ spec does not allow type 2 atoms anymore?

I don't see how the kernel can be affected by the way the GPIOs and pinmux have been configured. The pinctrl driver doesn't read back the state of the registers to see if a pin is in use - that state is kept in the pinctrl framework, and all pins start out as unclaimed.

> The only way I've found is to use force_eeprom_read=0 and manually applying the correct overlay

Are you saying that putting `dtoverlay=` at the top of config.txt doesn't help?

j-schambacher, 2024-03-07:
`dtoverlay=` works as well. Sorry, didn't mention.

> I don't see how the kernel can be affected by the way the GPIOs and pinmux have been configured. The pinctrl driver doesn't read back the state of the registers to see if a pin is in use - that state is kept in the pinctrl framework, and all pins start out as unclaimed.

Then the question remains, why the kernel can't (re)claim the IOs? I'll do some more testing and let you know.

pelwell, 2024-03-07:
> I'm still seeing the issue with DoP with the updated PR ... it's minor as the HAT can directly decode the DoP and pass along the native DSD bitstream to the DAC.

> Also, I'm getting the occasional dropout with dsd128 (352.8kHz effective PCM rate via DoP),

Can you explain the difference between these two statements? As somebody unfamiliar with the use of digital audio they sound the same. It would help me if you could briefly outline the data flow in these scenarios (and it sounds like the first one has two versions - one where the HAT does the decoding and one where it passes the DoP through).

Unless I get a bad review comment I'll get this PR merged and treat DoP as a separate problem.

pelwell, 2024-03-07:
> dtoverlay= works as well. Sorry, didn't mention.

Thanks - that helps to narrow down the problem, because this line definitely does not affect the GPIO setup.

Would you mind including the entirety of the EEPROM dt_blob data? I'd like to confirm why it fails on a Pi 5.

j-schambacher, 2024-03-07:
Here it is:
```
# ---------- Dump generated by eepdump handling format version 0x01 ----------
#
# --Header--
# signature=0x69502d52
# version=0x01
# reserved=0
# numatoms=3
# eeplen=1465
# ----------

# Start of atom #0 of type 0x0001 and length 37
# Vendor info
product_uuid ef586afc-2efa-47a0-be2e-95a7d952fe98
product_id 0x0000
product_ver 0x0000
vendor "HiFiBerry"   # length=9
product "DAC+"   # length=4
# End of atom. CRC16=0x8c1e

# Start of atom #1 of type 0x0002 and length 32
# GPIO map info
gpio_drive 0
gpio_slew 0
gpio_hysteresis 0
back_power 0
#        GPIO  FUNCTION  PULL
#        ----  --------  ----
setgpio  2      ALT0     UP
setgpio  3      ALT0     UP
setgpio  18      ALT0     DEFAULT
setgpio  19      ALT0     DEFAULT
setgpio  20      ALT0     DEFAULT
setgpio  21      ALT0     DEFAULT
# End of atom. CRC16=0x9700

# Start of atom #2 of type 0x0003 and length 1360
dt_blob
D0 0D FE ED 00 00 05 4E 00 00 00 38 00 00 04 9C 
00 00 00 28 00 00 00 11 00 00 00 10 00 00 00 00 
00 00 00 B2 00 00 04 64 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 01 00 00 00 00 
00 00 00 03 00 00 00 0D 00 00 00 00 62 72 63 6D 
2C 62 63 6D 32 37 30 38 00 00 00 00 00 00 00 01 
66 72 61 67 6D 65 6E 74 40 30 00 00 00 00 00 03 
00 00 00 08 00 00 00 0B 2F 63 6C 6F 63 6B 73 00 
00 00 00 01 5F 5F 6F 76 65 72 6C 61 79 5F 5F 00 
00 00 00 01 64 61 63 70 72 6F 5F 6F 73 63 00 00 
00 00 00 03 00 00 00 15 00 00 00 00 68 69 66 69 
62 65 72 72 79 2C 64 61 63 70 72 6F 2D 63 6C 6B 
00 00 00 00 00 00 00 03 00 00 00 04 00 00 00 17 
00 00 00 00 00 00 00 03 00 00 00 04 00 00 00 24 
00 00 00 01 00 00 00 02 00 00 00 02 00 00 00 02 
00 00 00 01 66 72 61 67 6D 65 6E 74 40 31 00 00 
00 00 00 03 00 00 00 04 00 00 00 2C DE AD BE EF 
00 00 00 01 5F 5F 6F 76 65 72 6C 61 79 5F 5F 00 
00 00 00 03 00 00 00 05 00 00 00 33 6F 6B 61 79 
00 00 00 00 00 00 00 02 00 00 00 02 00 00 00 01 
66 72 61 67 6D 65 6E 74 40 32 00 00 00 00 00 03 
00 00 00 04 00 00 00 2C DE AD BE EF 00 00 00 01 
5F 5F 6F 76 65 72 6C 61 79 5F 5F 00 00 00 00 03 
00 00 00 04 00 00 00 3A 00 00 00 01 00 00 00 03 
00 00 00 04 00 00 00 49 00 00 00 00 00 00 00 03 
00 00 00 05 00 00 00 33 6F 6B 61 79 00 00 00 00 
00 00 00 01 70 63 6D 35 31 32 32 40 34 64 00 00 
00 00 00 03 00 00 00 04 00 00 00 55 00 00 00 00 
00 00 00 03 00 00 00 0B 00 00 00 00 74 69 2C 70 
63 6D 35 31 32 32 00 00 00 00 00 03 00 00 00 04 
00 00 00 66 00 00 00 4D 00 00 00 03 00 00 00 04 
00 00 00 6A 00 00 00 01 00 00 00 03 00 00 00 05 
00 00 00 33 6F 6B 61 79 00 00 00 00 00 00 00 02 
00 00 00 02 00 00 00 02 00 00 00 01 66 72 61 67 
6D 65 6E 74 40 33 00 00 00 00 00 03 00 00 00 04 
00 00 00 2C DE AD BE EF 00 00 00 01 5F 5F 6F 76 
65 72 6C 61 79 5F 5F 00 00 00 00 03 00 00 00 1C 
00 00 00 00 68 69 66 69 62 65 72 72 79 2C 68 69 
66 69 62 65 72 72 79 2D 64 61 63 70 6C 75 73 00 
00 00 00 03 00 00 00 04 00 00 00 71 DE AD BE EF 
00 00 00 03 00 00 00 05 00 00 00 33 6F 6B 61 79 
00 00 00 00 00 00 00 03 00 00 00 04 00 00 00 24 
00 00 00 02 00 00 00 02 00 00 00 02 00 00 00 01 
5F 5F 6F 76 65 72 72 69 64 65 73 5F 5F 00 00 00 
00 00 00 03 00 00 00 21 00 00 00 80 00 00 00 02 
68 69 66 69 62 65 72 72 79 2C 32 34 64 62 5F 64 
69 67 69 74 61 6C 5F 67 61 69 6E 3F 00 00 00 00 
00 00 00 02 00 00 00 01 5F 5F 73 79 6D 62 6F 6C 
73 5F 5F 00 00 00 00 03 00 00 00 23 00 00 00 92 
2F 66 72 61 67 6D 65 6E 74 40 30 2F 5F 5F 6F 76 
65 72 6C 61 79 5F 5F 2F 64 61 63 70 72 6F 5F 6F 
73 63 00 00 00 00 00 03 00 00 00 18 00 00 00 9D 
2F 66 72 61 67 6D 65 6E 74 40 33 2F 5F 5F 6F 76 
65 72 6C 61 79 5F 5F 00 00 00 00 02 00 00 00 01 
5F 5F 6C 6F 63 61 6C 5F 66 69 78 75 70 73 5F 5F 
00 00 00 00 00 00 00 01 66 72 61 67 6D 65 6E 74 
40 32 00 00 00 00 00 01 5F 5F 6F 76 65 72 6C 61 
79 5F 5F 00 00 00 00 01 70 63 6D 35 31 32 32 40 
34 64 00 00 00 00 00 03 00 00 00 04 00 00 00 6A 
00 00 00 00 00 00 00 02 00 00 00 02 00 00 00 02 
00 00 00 01 5F 5F 6F 76 65 72 72 69 64 65 73 5F 
5F 00 00 00 00 00 00 03 00 00 00 04 00 00 00 80 
00 00 00 00 00 00 00 02 00 00 00 02 00 00 00 01 
5F 5F 66 69 78 75 70 73 5F 5F 00 00 00 00 00 03 
00 00 00 3E 00 00 00 A3 2F 66 72 61 67 6D 65 6E 
74 40 31 3A 74 61 72 67 65 74 3A 30 00 2F 66 72 
61 67 6D 65 6E 74 40 33 2F 5F 5F 6F 76 65 72 6C 
61 79 5F 5F 3A 69 32 73 2D 63 6F 6E 74 72 6F 6C 
6C 65 72 3A 30 00 00 00 00 00 00 03 00 00 00 15 
00 00 00 A7 2F 66 72 61 67 6D 65 6E 74 40 32 3A 
74 61 72 67 65 74 3A 30 00 00 00 00 00 00 00 03 
00 00 00 15 00 00 00 AC 2F 66 72 61 67 6D 65 6E 
74 40 33 3A 74 61 72 67 65 74 3A 30 00 00 00 00 
00 00 00 02 00 00 00 02 00 00 00 09 63 6F 6D 70 
61 74 69 62 6C 65 00 74 61 72 67 65 74 2D 70 61 
74 68 00 23 63 6C 6F 63 6B 2D 63 65 6C 6C 73 00 
70 68 61 6E 64 6C 65 00 74 61 72 67 65 74 00 73 
74 61 74 75 73 00 23 61 64 64 72 65 73 73 2D 63 
65 6C 6C 73 00 23 73 69 7A 65 2D 63 65 6C 6C 73 
00 23 73 6F 75 6E 64 2D 64 61 69 2D 63 65 6C 6C 
73 00 72 65 67 00 63 6C 6F 63 6B 73 00 69 32 73 
2D 63 6F 6E 74 72 6F 6C 6C 65 72 00 32 34 64 62 
5F 64 69 67 69 74 61 6C 5F 67 61 69 6E 00 64 61 
63 70 72 6F 5F 6F 73 63 00 66 72 61 67 33 00 69 
32 73 00 69 32 63 31 00 73 6F 75 6E 64 00 
# End of atom. CRC16=0x5dcc

```

j-schambacher, 2024-03-07:
Oh, thanks, Phil! And sorry for the confusion. This means I'll submit ASAP the UUIDs for the hat_map and hopefully even that old stuff gets resolved. I've totally forgotten that problem with the clocks. 
Do you want separate commits per UUID ? Or for now just a list/combined commit?

pelwell, 2024-03-07:
_All_ the downstream DT commits get squashed into one when we port to newer kernels, so one commit for all of them is fine.

clone206, 2024-03-07:
> Can you explain the difference between these two statements? As somebody unfamiliar with the use of digital audio they sound the same. It would help me if you could briefly outline the data flow in [...]

One statement makes note of the fact that I'm unable to pass DoP through to my DAC. I just get low level noise on playback and the DAC's display shows that it's interpreting the stream as PCM.

The other statement makes note of the fact that when the HAT is decoding dsd128 (128 x 44.1K) DoP, I hear dropouts, i.e. discontinuities in the sound. Sometimes there is also a popping sound accompanying the dropout. I don't hear dropouts when the HAT is decoding dsd64 DoP, or at any PCM sample rate.

Just giving as much info as possible about symptoms. The link I posted describes how in DoP the DSD marker occupies the 8 MSBs of each PCM frame, alternating between 2 values every other sample. If I had to guess I would think something is making it difficult for the FIFO HAT to see those markers at the right time, causing buffer underruns.

clone206, 2024-03-19:
Another data point. I tried with another DAC whose display gives a bit more info (and refreshes more quickly) than the DAC I was previously testing with.

Got the same symptoms described above with dsd128 content. But this time I could see that when the dropouts were happening, the DAC's screen briefly reported it was receiving 352.8kHz PCM before switching back to dsd128. Maybe this does indicate that the DoP markers aren't always seen at the right time, perhaps indicating some kind of word select timing issue.

pelwell, 2024-03-19:
But can you account for the fact that 352.8kHz audio sounds OK? Wouldn't timing problems at that level cause samples to be scrambled?

And the waveforms look solid on a logic analyser.

clone206, 2024-03-19:
> But can you account for the fact that 352.8kHz audio sounds OK? Wouldn't timing problems at that level cause samples to be scrambled?
> 
> And the waveforms look solid on a logic analyser.

To be honest I don't have a ton of 352.8kHz content. If anything at that sample rate I bet the popping would be so faint that it could easily be missed. The problem is intermittent enough that I could see how you could get a perfect readout with a small enough sample.

BTW this doesn't happen with the same HATs on the pi 4.