raspberrypi/linux #5747: Possible BCM I2S kernel driver issue for RPi 5

nodemand, 2023-11-25:
### Describe the bug

I'm on a RPi5 4GB running Raspberry Pi OS 64bit Bookworm and my application hangs on ALSA function snd_pcm_readn when I try to read and write to the same device. This occurs on an IQAudio Codec Zero sound card and on a HifiBerry DAC+ADC Pro as well. When I select a USB sound card as the output and one of the others described for input everything works like it should.

Here's my code:
```
#include <alsa/asoundlib.h>
#include <alsa/control.h>
#include <stdio.h>
#include <stdlib.h>
#include <stddef.h>
#include <algorithm>
#include <iterator>

typedef   signed short	drwav_int16;
typedef   signed int	drwav_int32;

#define PCM_DEVICE_IN "plughw:Zero,0"
#define PCM_DEVICE_OUT "plughw:Zero,0"

int pcm, err, result, n;
unsigned int rate, tmp, i=0;

snd_pcm_t			*play;
snd_pcm_t			*record;
snd_pcm_hw_params_t *playparams;
snd_pcm_hw_params_t *recordparams;
snd_pcm_uframes_t	frames;
snd_pcm_sframes_t	recresult;
snd_pcm_format_t	format = SND_PCM_FORMAT_S16_LE;
static snd_output_t *info = NULL;
int                 block = 0;

int buffer_frames 			= 	256;
int bits_per_frame 			= 	16;
const int bits_per_byte		= 	8;
int channels				= 	2;

int bytesize_per_channel  	= 	buffer_frames 
								* bits_per_frame 
								/ bits_per_byte;

int main(int argc, char **argv) {
	if (argc < 3) {
		printf("Usage: %s <sample_rate> <frames>\n", argv[0]);
		return -1;
	}

	rate			= atoi(argv[1]);
	buffer_frames	= atoi(argv[2]);
	
	bytesize_per_channel  	= 	buffer_frames 
								* snd_pcm_format_width(format)  
								/ bits_per_byte;

	drwav_int16 *buffer_in[channels];
	
	
	
	
	/* Open the PCM device in record mode */
	if ((pcm = snd_pcm_open(&record, PCM_DEVICE_IN, SND_PCM_STREAM_CAPTURE, block)) < 0)
		printf("ERROR: Can't open \"%s\" PCM device. %s\n", PCM_DEVICE_IN, snd_strerror(pcm));

	/* Allocate parameters object and fill it with default values*/
	if ((pcm = snd_pcm_hw_params_malloc(&recordparams)) < 0)
		printf("ERROR: Can't allocate hardware parameter structure. %s\n", snd_strerror(pcm));

	if ((pcm = snd_pcm_hw_params_any(record, recordparams)) < 0)
		printf("ERROR: Can't initialize hardware parameter structure. %s\n", snd_strerror(pcm));
[... 775 lines cut ...]
233: a1    -- | hi // HDMI0_SDA/AON_SGPIO1 = HDMI_TX0_BSC_SDA
234: a1    -- | hi // HDMI1_SCL/AON_SGPIO2 = HDMI_TX1_BSC_SCL
235: a1    -- | hi // HDMI1_SDA/AON_SGPIO3 = HDMI_TX1_BSC_SDA
236: a2    -- | hi // PMIC_SCL/AON_SGPIO4 = BSC_M2_SCL
237: a2    -- | hi // PMIC_SDA/AON_SGPIO5 = BSC_M2_SDA

vcdbg log messages
------------------

004157.677: Initial voltage 800000 temp 38379
004355.116: avs_2712: AVS pred 8526 852600 temp 38929
004355.127: vpred 852 mV +0
004981.364: FB framebuffer_swap 1
005009.458: Select resolution HDMI0/2 hotplug 1 max_mode 2
005009.650: BSC_A no ACK
005009.840: BSC_A no ACK
005009.851: HDMI0: Unable to read EDID block 0
005009.864: Select resolution HDMI1/2 hotplug 0 max_mode 2
005606.490: dtb_file 'bcm2712-rpi-5-b.dtb'
007775.758: Starting OS 7775 ms
007778.396: 00000040: -> 00000480
007778.412: 00000030: -> 00100080
007778.428: 00000034: -> 00100080
007778.444: 00000038: -> 00100080
007778.459: 0000003c: -> 00100080

pelwell, 2023-11-28:
There was a bug in the DMA configuration, which I had to add to the 6.1 driver. Things have changed significantly in newer kernels so I need to review my changes in rpi-6.6.y etc. (and possibly back-port some of it), but for now #5757 should fix rpi-6.1.y.

Wait about an hour for the auto-builds to complete, then you'll be able to install a trial kernel by running `sudo rpi-update pulls/5757`.

nodemand, 2023-11-28:
Thank you so much!

gtrainavicius, 2023-11-28:
The full duplex playback and recording on Pisound works with pulls/5757.

nodemand, 2023-11-28:
@pelwell I'm getting the following error in the output:
```
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 6.1.0-rpi4-rpi-2712
WANT_32BIT:0 WANT_64BIT:1 WANT_PI4:1 WANT_PI5:1
 *** Downloading specific artifact revision (this will take a few minutes)
curl  -L https://builds.raspberrypi.com/github/linux/778cd12aef3b18c92a78dae93059acef0feab354/bcmrpi | zcat | tar xf - -C //root/.rpi-firmware --strip-components=2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0

gzip: stdin: unexpected end of file
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0
Invalid artifact specified. Response: 404.
```

pelwell, 2023-11-28:
Give it a while - 20 minutes, say - I pushed an update in response to a comment on the previous version of the patch.

nodemand, 2023-11-28:
okay!

pelwell, 2023-11-28:
I was too pessimistic - it should be ready now.

nodemand, 2023-11-28:
Well, I'm not getting any errors, but where there was sound before when starting `aplay test.wav  -f S16_LE -r48000 -c2 -D plughw:Zero,0` there is nothing now....no sound

pelwell, 2023-11-28:
Is that because you've overwritten your test.wav with silence recorded by arecord?

nodemand, 2023-11-28:
I checked. There is sound coming out of the audio source I used as input for arecord....

pelwell, 2023-11-28:
Did you try playing another file? Have you used the same soundcard and configuration on an older Pi and had valid recordings?

nodemand, 2023-11-28:
I just tried another wav and it won't play back with `aplay` and `mplayer` says: `Audio device got stuck!`

nodemand, 2023-11-28:
Yes it is. It worked right before the patch, just not recording and playback simultaneously...

pelwell, 2023-11-28:
Which Codec Zero alsactl configuration file are you using, the one that used to give you good recordings?

nodemand, 2023-11-28:
hahaha, yes that is exactly the thing I did not run because it generates an error with me and I thought it did not work because of that so I did not run it anymore...

nodemand, 2023-11-28:
it works now!

pelwell, 2023-11-28:
Excellent. I'm merging the patch, and it will appear in future releases.

Thanks for the clear and accurate report.

closed 2023-11-28 by pelwell

nodemand, 2023-11-28:
Only one more thing. I cannot run my application because it depends on libasound2 and libasound-dev and portaudio19-dev. And I'm trying to install libasound-dev,  but it tells me:
```
sudo apt-get install libasound-dev

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libasound2-dev' instead of 'libasound-dev'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libasound2-dev : Depends: libasound2 (= 1.2.8-1+b1) but 1.2.8-1+rpt1 is to be installed
```

pelwell, 2023-11-28:
>  Is this as a result of the patch?

No, it really can't be, therefore this is not the place for your question.

Are you sure that libasound2-dev doesn't work?

Try updating your packages:
```
pi@raspberrypi:~$ apt info libasound2-dev
Package: libasound2-dev
Version: 1.2.8-1+rpt1
Priority: optional
Section: libdevel
Source: alsa-lib
Maintainer: Debian ALSA Maintainers <pkg-alsa-devel@lists.alioth.debian.org>
Installed-Size: 676 kB
Provides: libasound-dev
Depends: libasound2 (= 1.2.8-1+rpt1)
Suggests: libasound2-doc
Homepage: https://www.alsa-project.org/
Download-Size: 110 kB
APT-Manual-Installed: yes
APT-Sources: http://buildbot.pitowers.org:3143 bookworm/main arm64 Packages
Description: shared library for ALSA applications -- development files
 This package contains files required for developing software
 that makes use of libasound2, the ALSA library.
 .
 ALSA is the Advanced Linux Sound Architecture.
```

nodemand, 2023-11-28:
```
apt info libasound2-dev
Package: libasound2-dev
Version: 1.2.8-1+b1
Priority: optional
Section: libdevel
Source: alsa-lib (1.2.8-1)
Maintainer: Debian ALSA Maintainers <pkg-alsa-devel@lists.alioth.debian.org>
Installed-Size: 677 kB
Provides: libasound-dev
Depends: libasound2 (= 1.2.8-1+b1)
Suggests: libasound2-doc
Homepage: https://www.alsa-project.org/
Tag: devel::library, role::devel-lib, works-with::audio
Download-Size: 110 kB
APT-Sources: http://deb.debian.org/debian bookworm/main arm64 Packages
Description: shared library for ALSA applications -- development files
 This package contains files required for developing software
 that makes use of libasound2, the ALSA library.
 .
 ALSA is the Advanced Linux Sound Architecture.
```

nodemand, 2023-11-28:
My application WORKSSSS!!!!

pelwell, 2023-11-28:
Woot!