raspberrypi/linux #5748: Rpi5 sampling rate dac hat

antonellocaroli, 2023-11-26:
### Describe the bug

Set in config.txt to load
`dtoverlay=i-sabre-q2m`

but I don't understand why it shows me a sample rate of 192000
I remember that on Rpi4 it was up to 384000

Where does he get 192000 saple rate from?

```
*** Scanning for playback devices ***
Card 0, ID `DAC', name `I-Sabre Q2M DAC'
  Device 0, ID `I-Sabre Q2M DAC i-sabre-codec-dai-0', name `I-Sabre Q2M DAC i-sabre-codec-dai-0', 1 subdevices (1 available)
    2 channels, sampling rate 8000..192000 Hz
    Sample formats: S16_LE, S32_LE
    Buffer size range from 32 to 131072
    Period size range from 16 to 1024

      Subdevice 0, name `subdevice #0'

```
    
It seems that on anything the maximum samplerate is always 192000
      

### Steps to reproduce the behaviour

Set in config.txt to load
`dtoverlay=i-sabre-q2m`

### Device (s)

Other

### System

6.1.0-rpi4-rpi-2712

### Logs

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.1.0-rpi4-rpi-2712 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000003a000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 256 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 14 pages/cpu s177448 r8192 d43736 u229376
[    0.000000] pcpu-alloc: s177448 r8192 d43736 u229376 alloc=14*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[... 359 lines cut ...]
[    2.852251] pispbe 1000880000.pisp_be: output0 device node registered as /dev/video33
[    2.861742] pispbe 1000880000.pisp_be: output1 device node registered as /dev/video34
[    2.865508] pispbe 1000880000.pisp_be: tdn_output device node registered as /dev/video35
[    2.891555] snd-rpi-i-sabre-q2m soc:sound: snd_soc_register_card() failed: -517
[    2.906855] pispbe 1000880000.pisp_be: stitch_output device node registered as /dev/video36
[    2.941451] pispbe 1000880000.pisp_be: config device node registered as /dev/video37
[    2.956087] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
[    3.012968] snd-rpi-i-sabre-q2m soc:sound: snd_soc_register_card() failed: -517
[    3.127984] i-sabre-codec-i2c 1-0048: ASoC: error at soc_component_read_no_lock on i-sabre-codec-i2c.1-0048 for register: [0x00000001] -121
[    3.127997] snd-rpi-i-sabre-q2m soc:sound: Audiophonics Device ID : FFFFFF87
[    3.128063] i-sabre-codec-i2c 1-0048: ASoC: error at soc_component_read_no_lock on i-sabre-codec-i2c.1-0048 for register: [0x00000002] -121
[    3.128068] snd-rpi-i-sabre-q2m soc:sound: Audiophonics API revision : FFFFFF87
[    3.168612] EXT4-fs (mmcblk0p2): re-mounted. Quota mode: none.
[    3.186572] EXT4-fs (mmcblk0p2): re-mounted. Quota mode: none.
[    4.760692] NET: Registered PF_INET6 protocol family
[    4.761240] Segment Routing with IPv6
[    4.761250] In-situ OAM (IOAM) with IPv6
[    4.894707] 8021q: 802.1Q VLAN Support v1.8
[    4.952408] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    4.952419] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    4.955150] pps pps0: new PPS source ptp0
[    4.955644] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    4.998409] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.055684] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[    9.055706] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

pelwell, 2023-11-27:
Perhaps the author of the DWC I2S driver thought there was no practical reason to use sample rates up to 384kHz? Anyway, see PR #5751 for what should be a fix.

If you wait about an hour you'll be able to run `sudo rpi-update pulls/5751` to install a trial kernel including the change.

antonellocaroli, 2023-11-27:
@pelwell  I modified the file in question and recompiled the kernel, it seems to work:

```
*** Scanning for playback devices ***
Card 0, ID `DAC', name `I-Sabre Q2M DAC'
  Device 0, ID `I-Sabre Q2M DAC i-sabre-codec-dai-0', name `I-Sabre Q2M DAC i-sabre-codec-dai-0', 1 subdevices (1 available)
    2 channels, sampling rate 8000..384000 Hz
    Sample formats: S16_LE, S32_LE
    Buffer size range from 32 to 131072
    Period size range from 16 to 1024

      Subdevice 0, name `subdevice #0'
```