raspberrypi/linux #5753: Pi 5 xHCI controller fails during disk-to-disk copy behind a hub

P33M, 2023-11-27:
### Describe the bug

Doing an rsync between two spinning rust drives attached to the same hub causes a suspected controller lockup. The symptom is 

```
[  194.841355] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[  194.861533] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[  194.861564] xhci-hcd xhci-hcd.0: HC died; cleaning up
```
I.e. for whatever reason the controller goes "quiet".
Further commentary should happen in this issue tracker.

### Steps to reproduce the behaviour

Reproduction here; https://forums.raspberrypi.com/viewtopic.php?t=359625#p2161883

Note that #5060 is specific to the Pi 4, so the same comment cross-posted there doesn't apply.

### Device (s)

Other

### System

Release bookworm. Also we should update the dropdown in the template to include Pi 5...

pjft, 2023-11-27:
Thanks. Removed comment from the Pi4 issue, and leaving here the comment. 
--

I shared some steps that make it replicable 100% of the times for me.

Scenario: 
- Two USB3.0 Hard drives connected to the same USB3.0 hub in a USB3.0 port. In my case, the drives are HDDCOPY and HDDCOPY-OLD - adjust to your mountpoints and hard drives.

OS:

```
Linux rpi5 6.1.0-rpi6-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
```

lsusb -tv 

```
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
        ID 2109:0817 VIA Labs, Inc. 
        |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 5000M
            ID 2109:0817 VIA Labs, Inc. 
            |__ Port 2: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
                ID 1058:25a1 Western Digital Technologies, Inc. Elements / My Passport
            |__ Port 3: Dev 5, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
                ID 1058:25a1 Western Digital Technologies, Inc. Elements / My Passport
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 2109:2817 VIA Labs, Inc. 
        |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
            ID 2109:2817 VIA Labs, Inc. 
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/2p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**Step 1:** create a large test file (10GB - I'm removing it first, but it's unnecessary unless you want to start from scratch).

```
rm /media/pi/HDDCOPY/other/Test/5000.tmp 
dd if=/dev/random of=/media/pi/HDDCOPY/other/Test/5000.tmp bs=100000k count=100 status=progress
```

**Step 2:** delete destination file (in case you previously ran a test that was incomplete. rsync will create different temporary files):

```
rm /media/pi/HDDCOPY-OLD/other/Test/5000.tmp
```

and try to copy the file. All the following commands will inevitably fail:

```
dd if=/media/pi/HDDCOPY/other/Test/5000.tmp of=/media/pi/HDDCOPY-OLD/other/Test/5000.tmp bs=100000k status=progress
```

```
rsync -ah --progress /media/pi/HDDCOPY/other/Test/5000.tmp /media/pi/HDDCOPY-OLD/other/Test/
```

```
cp /media/pi/HDDCOPY/other/Test/5000.tmp /media/pi/HDDCOPY-OLD/other/Test/
```

---

At this stage, I am not 100% confident that it is related to the type of hub, as it can consistently be replicated in both chipsets I have (Realtek and VIA Labs). It does not happen when the two drives are connected to the different USB3.0 ports, nor to the USB2.0 ports. 

I tested this on two separate Raspberry Pi 5, 8GB models.

Thanks. Happy to help.

ajinasokan, 2023-11-27:
I have this exact problem while using a [5G module](https://www.waveshare.com/wiki/SIM8262E-M2_5G_HAT). It connects to network fine but the moment I start a speed test it locks up with the same log as in the issue. Only a reboot brings it back.

Speed tests with this module usually reaches around 1Gbps network throughput. So I'm guessing this issue is happening only when a high speed transfer happens on USB 3.0. I tried both USB 3.0 ports, same issue. Works in USB 2.0 ports, obviously at a lower throughput.

Module works fine in USB 3.0 in Pi 4B.

thmcon, 2023-11-27:
The same for me: in my case I am "only" reading big files from an USB Stick which is plugged into an USB 3  hub which is plugged into one of the USB 3 ports of the raspberry pi 5 8GB. My USB hub also has an RJ45 gigabit lan interface where the file will be transferred over the lan. Which means: reading the file from usb to the raspberry and sending the same file via the same usb port. The memory stick can be read with up to 140 MB/s and the lan switch can handle bandwidth of 1 Gb/s. If I compare the specs of the USB 3 ports of the raspberry pi 5, I would asume, that the max speed of the USB 3 port will be reached.

pjft, 2023-11-27:
Inspired by @ajinasokan 's description, I tried to replicate the issue with just one HDD (copying a file to a different location in the same disk) but couldn't replicate it.

Also, just for sanity checking, I tried this on a Pi400 I had at home and tried to replicate the issue on the following installations:

```
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

```
Linux pi400 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
```

and the same SD card as the Pi5 I ran the tests in.

```
Linux rpi5 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux
```

It did not fail in any of them. So, as @P33M assessed this, this seems to be a Pi5-specific issue, and not related to the other Pi4 issues.

ddscentral, 2023-11-30:
Same issue with a Vantec USB 3.0 to dual Gigabit Ethernet adapter (internally a VIA VL815 hub with 2x RTL8153's wired to it). Usually dies shorty after a reboot, but sometimes does work for a while bit, albeit unstably with a lot of random XHCI errors.
Tried swapping stock Debian kernel 6.1 with Ubuntu 6.5 kernel, will see if this changes anything.

ddscentral, 2023-12-01:
Ok, the adventures continue. Now testing without any hub connected at all.
I have 3 separate USB NICs plugged in. And the exact same issue still occurs, on both 6.1 and 6.5 kernels. 
The same issue also occurs with just 2 NICs plugged in, to USB 3.0 ports.

This issue has nothing to do with hubs.

acavalin, 2023-12-01:
Hello everyone, I'm sharing my experience on this problem:

My setup:
- **(A)** 1 [usb3 hdd drive](https://amzn.eu/d/1eWbq4V) plugged into the Pi5's 1st *usb3* port
- 1 self powered [usb3 hub](https://amzn.eu/d/5kltwHP) plugged into the Pi5's 2nd *usb3* port
- 1 self powered [usb2 hub](https://amzn.eu/d/1QaBQLE) plugged into the Pi5's 1st *usb2* port
- **(B)** 1 [usb3 hdd drive](https://amzn.eu/d/1eWbq4V) plugged into the *usb3 hub*
- **(C)** 1 [usb3 pendrive](https://amzn.eu/d/1dBMDqy)
- `mdadm` software RAID1 setup with (A) and (B)
  ~~~shell
  # cat /proc/mdstat 
  md126 : active raid1 sdb1[2] sda1[0]
        1939733504 blocks super 1.2 [2/2] [UU]
        bitmap: 0/4 pages [0KB], 65536KB chunk
  ~~~

Tests performed:

Case | Result
-----|-------
plug (C) into the *usb3 hub*, copy 2.3GiB file from (C) to (B) | :x: HC died: (B) disappears, (A) still alive (degraded RAID array)
plug (C) into the *usb3 hub*, copy 1.1GiB file from (B) to (C) | :white_check_mark: OK
plug (C) into the *usb2 hub*, copy 2.3GiB file from (C) to (B) | :white_check_mark: OK

<details>
  <summary>view `dmesg` error log with "HC died"</summary>
  
  ```
  xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
  xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
  xhci-hcd xhci-hcd.0: HC died; cleaning up
  sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x03 driverbyte=DRIVER_OK cmd_age=35s
  sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x28 28 00 0a 14 44 20 00 04 00 00
  I/O error, dev sdc, sector 169100320 op 0x0:(READ) flags 0x80700 phys_seg 8 prio class 2
  sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
  sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x28 28 00 0a 14 48 20 00 08 00 00
  I/O error, dev sdc, sector 169101344 op 0x0:(READ) flags 0x80700 phys_seg 16 prio class 2
  usb 1-1: USB disconnect, device number 2
  usb 1-1.2: USB disconnect, device number 4
  usb 2-1: USB disconnect, device number 2
  usb 2-1.3: USB disconnect, device number 3
  usb 1-1.4: USB disconnect, device number 5
  usb 1-2: USB disconnect, device number 3
  md: super_written gets error=-19
  md/raid1:md126: Disk failure on sdb1, disabling device.
  md/raid1:md126: Operation continuing on 1 devices.
  md/raid1:md126: sdb1: rescheduling sector 209452384
  usb 2-1.4: USB disconnect, device number 4
  usb 2-1.4.4: USB disconnect, device number 5
  device offline error, dev sdc, sector 169100320 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
  device offline error, dev sdc, sector 33 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 2
  Buffer I/O error on dev sdc1, logical block 1, lost async page write
  device offline error, dev sdc, sector 168058272 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 2
  Buffer I/O error on dev sdc1, logical block 168058240, lost async page write
  md/raid1:md127: Disk failure on sdb2, disabling device.
  md/raid1:md127: Operation continuing on 1 devices.
  Buffer I/O error on dev dm-0, logical block 457724, async page read
  ```
</details>

acavalin, 2023-12-01:
> Which power supply are you using? An unpowered spinning-rust HDD has a high power requirement.

I'm using the official Pi5 PSU.

I have no problems with the same setup on Pi4 (using official Pi4 PSU), this error arises only when using my brand new Pi5.

(Thanks for the useful question, I updated my previous post)

ddscentral, 2023-12-01:
> Ok, the adventures continue. Now testing without any hub connected at all. I have 3 separate USB NICs plugged in. And the exact same issue still occurs, on both 6.1 and 6.5 kernels. The same issue also occurs with just 2 NICs plugged in, to USB 3.0 ports.
> 
> This issue has nothing to do with hubs.
> 
> USB NICs are one RTL8153 and one or two ASIX AX88179s if that helps.

Forgot to mention, my Pi 5 is being used as a router. Now testing with one older revision ASIX based NIC and one Realtek NIC, all is stable so far with stock 6.1 kernel, no errors or dropouts in the logs.
So the issue is device dependent.

HakanKoseoglu, 2023-12-03:
Issue replicated repeatedly. Copying large files between disks cause it to lock up frequently. In my case the root is also USB meaning I lose everything to debug the issue, since /bin and /usr/bin also become unreadable.

P33M, 2023-12-07:
Moving one device to the other USB controller = no hang.
With two devices on one USB controller, swapping between hub depth of 1 or 2 on either = hangs regardless.
On the analyser a pattern seems to emerge - if the IN endpoint is in a flow-control condition (not transmitting), and the OUT endpoint is bursting, then the IN endpoint becoming available again seems to be the trigger for causing the hang.

But, I get multiple instances of this condition happening elsewhere in the trace (it's normal operation for USB where your drive can't saturate the bus) and no hang happens.

pjft, 2023-12-07:
Correct. It also didn't happen on the USB 2.0 ports, likely because the lower speed doesn't allow the bus to become saturated.

ddscentral, 2023-12-08:
> Correct. It also didn't happen on the USB 2.0 ports, likely because the lower speed doesn't allow the bus to become saturated.
Well, one of my ASIX USB NICs had exactly the same issue (same errors in dmesg) even when connected to USB 2.0 and no hub.

pjft, 2023-12-08:
@ddscentral It might be related, or it might be a different issue altogether, let's see how things develop.

You seem to have tried it with other equivalent devices, it does work. It might be a device-dependent incompatibility in this case, which may display similar symptoms but be caused by a different root cause.

Regarding the ethernet dongle errors, from what I saw in the post history the logs seem to suggest a xHCI failure but very different logs, or do you also get

```
[  194.841355] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[  194.861533] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[  194.861564] xhci-hcd xhci-hcd.0: HC died; cleaning up
```

in dmesg?

pjft, 2023-12-08:
@popcornmix I'm happy to test, is there any side effect to installing this, and/or is there a way to revert or go back to the main/stable updates? Apologies for ignorance.

pjft, 2023-12-08:
And as I was focused on just updating with the test results, I apologize for the haste - I do want to thank @P33M for all the effort and transparency in keeping this issue updated with the findings, as well as making a proposal for a fix. I can only imagine how challenging digging into this kind of issues can be. Thank you!

pelwell, 2023-12-08:
Shall I merge this as at least an improvement?

pjft, 2023-12-08:
So, reporting back:

Even in a real life scenario, with multiple IN and OUT across the drives, it still fails. So there's no real change other than, in a large, consecutive set of bursts, it fails later.

Now, the difference between "stable" and this current change, which may (or may not?) give further clues (even though it might be harder to troubleshoot):
- When doing `rsync --dry-run` between the two drives, in the old code it never crashed and went through the 4TB drives successfully. I just tried it again and it worked.
- In the new code, it consistently crashes halfway through.

Now, running rsync in dry-run mode shouldn't cause the same bursts as a proper file transfer operation (hence why it likely didn't crash before), but now it is crashing. I wonder if there's something not being properly flushed at some point that still causes it to be saturated with the new change?

I'd love to be able to either get more detailed logs on my end for the IN/OUT burst activity, or analyze the behavior if it's something relatively reasonable or practical for me to do. I just don't know how to access that information to troubleshoot further, so I appreciate all you're doing here.

Have a great weekend.

pjft, 2023-12-08:
EDIT: I was mistaken. It still fails in the dry run in stable, just not always and a lot later in the process. Might be related to the amount of differences it actually finds.

P33M, 2023-12-13:
Latest fix works for me with 100GB copied one way, then 100GB copied the other way between UMS devices. Also no issue with UMS -> UAS copy, or UAS -> UMS copy.

To get the v2 fix, do `sudo rpi-update pulls/5788` and reboot.

pjft, 2023-12-13:
Sounds good, will test shortly!

ajinasokan, 2023-12-13:
@P33M I just tried this with my 5G module. It was able to hit 800Mbps+ without freezing! Previously it used to freeze the moment I start the speed test.

pjft, 2023-12-13:
Ok, so that I don't delay my answer as it would be rude, I want to say that:
a) I very much approve of merging this change, and
b) thank you so much for digging into this.

The xhci symptom no longer shows up, so that's a great improvement.

Before we close this, I am getting a narrower symptom that I need to dig into to understand if:
a) it's just coincidental, or
b) it's a new symptom closer to the root cause that was uncovered by this change.

All in all, awesome. Please give me a bit to run some tests with different drives and adapters just to exclude hardware issues on my end.

Thank you!

P33M, 2023-12-13:
We know that this IP is used in other products (that's partially why we selected it for RP1), which comes with the benefit that it's been reasonably battle-hardened. In fact there's about 60 override/debug/feature disable bits that can be set to modify behaviour.

The fix has already been merged so will get wider testing via rpi-update. We will follow up internally to establish what the failure mode is, but that takes time.

pjft, 2023-12-13:
Makes perfect sense, and I appreciate the context.

Let me report back tomorrow on the rest of the tests (they take time), but I'm happy if you want to close this issue and I can create a new one and if I run into anything or just want to share updates.

Thanks once again!

pjft, 2023-12-14:
I confirm that this is no longer reproduceable, now am I getting any errors so it must have been something transient. I'll report back if anything changes, but I'm happy to close this issue.

Thank you once again for looking into this so thoroughly.

closed 2023-12-14 by pelwell

ddscentral, 2023-12-17:
I can confirm the dual gigabit ethernet (VL813 Hub + 2x RTL8153) works fine.

I found some mentions of the DWC3 controller issue which is fixed by setting the "parkmode-disable-ss-quirk"
https://lore.kernel.org/linux-usb/c32007f5-88b9-45c5-b542-b1dc4dbc76ea@baylibre.com/T/
This is dated 2019, so this appears to be a known issue.

UgaBugaOfdeSpeer, 2023-12-31:
> Latest fix works for me with 100GB copied one way, then 100GB copied the other way between UMS devices. Also no issue with UMS -> UAS copy, or UAS -> UMS copy.
> 
> To get the v2 fix, do `sudo rpi-update pulls/5788` and reboot.

I experienced with the rpi-5 issues with my raid-5 (mdadm + usb-storage quirk).
When raid-check was running, I copied some big files to the raid with the result that the raid was not accessible anymore.
Only a reboot could bring the raid-5 back.

Then I read about the issues that you faced.
Accordingly l applied the fix for 5788.
Since then I faced no issues anymore and the raid-5 is, until now, stable.

I have a few question related this fix 
 - When will this fix getting into the distribution and how can this be checked?
 - Should "sudo apt upgrade" not done anymore for now?
 - Which subsystem is updated with this fix

acavalin, 2024-01-04:
> I have a few question related this fix
> 
> * When will this fix getting into the distribution and how can this be checked?
> * Should "sudo apt upgrade" not done anymore for now?
> * Which subsystem is updated with this fix

Hello @UgaBugaOfdeSpeer, on the mentioned issue above pelwell replied to me:

> It will be in the next stable release. There's no hard schedule for that, but it's likely to be before the end of February, and possibly sooner.

and you can `sudo apt upgrade` whenever you want, it will install the latest kernel including this patch when available (before the end of February)

pjft, 2024-01-04:
Thanks. I suppose the question - the way I read it - is more around: "we are now running this patch, so can we do sudo apt upgrade without the risk of reverting this behavior"? 

I'm assuming that this will come in an `rpi-update` package update, and that the next update will have this fix so there are no in-between updates that could revert this, is that it?

dkersh, 2024-01-05:
I'm affected by a similar issue discussed in this topic: https://forums.raspberrypi.com/viewtopic.php?t=361171. However I'm using the latest Libreelec-nightly. 

Is there an equivalent to `sudo rpi-update pulls/5788` for different operating systems?

PowerWiesel, 2024-01-05:
I think I'm also affected by this issue and it seems like `sudo rpi-update pulls/5788` didn't change anything.

I'm running the pi 5 of an nvme to usb adaptor from ugreen. I also enabled max_usb_current=1 just to be safe.
It works as long as I don't do much, also dd'ing a 10 gig test file to the drive worked fine.
But as soon as I install multiple packages via apt it drops out (I'm installing and removing vs code to trigger it).
If it's relevant: it's powered by the 27W PD Pi psu and got the active cooling heatsink installed.

```[  127.963734] sd 0:0:0:0: [sda] tag#29 uas_eh_abort_handler 0 uas-tag 18 inflight: CMD OUT 
[  127.963744] sd 0:0:0:0: [sda] tag#29 CDB: opcode=0x2a 2a 00 01 10 10 28 00 00 08 00
[  133.055315] xhci-hcd xhci-hcd.0: xHCI host not responding to stop endpoint command
[  133.075496] xhci-hcd xhci-hcd.0: xHCI host controller not responding, assume dead
[  133.075533] xhci-hcd xhci-hcd.0: HC died; cleaning up
[  133.075595] usb 2-1: cmd cmplt err -108
[  133.075598] usb 2-1: cmd cmplt err -108
[  133.075601] usb 2-1: cmd cmplt err -108
[  133.075603] usb 2-1: cmd cmplt err -108
[  133.075606] usb 2-1: cmd cmplt err -108
[  133.075608] usb 2-1: cmd cmplt err -108
[  133.075610] usb 2-1: cmd cmplt err -108
[  133.075613] usb 2-1: cmd cmplt err -108
[  133.075615] usb 2-1: cmd cmplt err -108
[  133.075617] usb 2-1: cmd cmplt err -108
[  133.075648] sd 0:0:0:0: [sda] tag#28 uas_eh_abort_handler 0 uas-tag 17 inflight: CMD 
[  133.075653] sd 0:0:0:0: [sda] tag#28 CDB: opcode=0x2a 2a 00 01 10 0f 88 00 00 08 00
[  133.075657] sd 0:0:0:0: [sda] tag#27 uas_eh_abort_handler 0 uas-tag 16 inflight: CMD 
[  133.075659] sd 0:0:0:0: [sda] tag#27 CDB: opcode=0x2a 2a 00 01 10 0e 10 00 00 10 00
[  133.075663] sd 0:0:0:0: [sda] tag#26 uas_eh_abort_handler 0 uas-tag 15 inflight: CMD 
[  133.075665] sd 0:0:0:0: [sda] tag#26 CDB: opcode=0x2a 2a 00 01 10 0d e0 00 00 30 00
[  133.075669] sd 0:0:0:0: [sda] tag#25 uas_eh_abort_handler 0 uas-tag 14 inflight: CMD 
[  133.075671] sd 0:0:0:0: [sda] tag#25 CDB: opcode=0x2a 2a 00 01 10 0d c8 00 00 18 00
[  133.075674] sd 0:0:0:0: [sda] tag#24 uas_eh_abort_handler 0 uas-tag 13 inflight: CMD 
[  133.075676] sd 0:0:0:0: [sda] tag#24 CDB: opcode=0x2a 2a 00 01 10 0d a8 00 00 20 00
[  133.075680] sd 0:0:0:0: [sda] tag#23 uas_eh_abort_handler 0 uas-tag 12 inflight: CMD 
[  133.075682] sd 0:0:0:0: [sda] tag#23 CDB: opcode=0x2a 2a 00 01 10 0d 88 00 00 20 00
[  133.075685] sd 0:0:0:0: [sda] tag#22 uas_eh_abort_handler 0 uas-tag 11 inflight: CMD 
[  133.075687] sd 0:0:0:0: [sda] tag#22 CDB: opcode=0x2a 2a 00 01 10 0d 80 00 00 08 00
[  133.075691] sd 0:0:0:0: [sda] tag#21 uas_eh_abort_handler 0 uas-tag 10 inflight: CMD 
[  133.075693] sd 0:0:0:0: [sda] tag#21 CDB: opcode=0x2a 2a 00 01 10 0c e8 00 00 98 00
[  133.075697] sd 0:0:0:0: [sda] tag#20 uas_eh_abort_handler 0 uas-tag 9 inflight: CMD 
[  133.075698] sd 0:0:0:0: [sda] tag#20 CDB: opcode=0x2a 2a 00 01 10 0c b0 00 00 38 00
[  133.075703] sd 0:0:0:0: [sda] tag#19 uas_eh_abort_handler 0 uas-tag 6 inflight: CMD 
[  133.075704] sd 0:0:0:0: [sda] tag#19 CDB: opcode=0x2a 2a 00 01 10 0c 78 00 00 08 00
[  133.075707] sd 0:0:0:0: [sda] tag#18 uas_eh_abort_handler 0 uas-tag 5 inflight: CMD 
[  133.075709] sd 0:0:0:0: [sda] tag#18 CDB: opcode=0x2a 2a 00 01 10 0b 78 00 00 10 00
[  133.075713] sd 0:0:0:0: [sda] tag#17 uas_eh_abort_handler 0 uas-tag 8 inflight: CMD 
[  133.075715] sd 0:0:0:0: [sda] tag#17 CDB: opcode=0x2a 2a 00 01 10 0c 98 00 00 18 00
[  133.075719] sd 0:0:0:0: [sda] tag#16 uas_eh_abort_handler 0 uas-tag 7 inflight: CMD 
[  133.075720] sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x2a 2a 00 01 10 0c 80 00 00 18 00
[  133.075724] sd 0:0:0:0: [sda] tag#15 uas_eh_abort_handler 0 uas-tag 4 inflight: CMD 
[  133.075726] sd 0:0:0:0: [sda] tag#15 CDB: opcode=0x2a 2a 00 01 10 08 08 00 00 08 00
[  133.075730] sd 0:0:0:0: [sda] tag#14 uas_eh_abort_handler 0 uas-tag 3 inflight: CMD 
[  133.075732] sd 0:0:0:0: [sda] tag#14 CDB: opcode=0x2a 2a 00 01 10 08 00 00 00 08 00
[  133.075735] sd 0:0:0:0: [sda] tag#13 uas_eh_abort_handler 0 uas-tag 2 inflight: CMD 
[  133.075736] sd 0:0:0:0: [sda] tag#13 CDB: opcode=0x2a 2a 00 00 d0 08 18 00 00 08 00
[  133.075741] sd 0:0:0:0: [sda] tag#12 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD 
[  133.075742] sd 0:0:0:0: [sda] tag#12 CDB: opcode=0x2a 2a 00 00 90 88 d0 00 00 08 00
[  133.075746] sd 0:0:0:0: [sda] tag#2 uas_eh_abort_handler 0 uas-tag 21 inflight: CMD 
[  133.075747] sd 0:0:0:0: [sda] tag#2 CDB: opcode=0x2a 2a 00 00 10 12 50 00 00 08 00
[  133.075751] sd 0:0:0:0: [sda] tag#1 uas_eh_abort_handler 0 uas-tag 20 inflight: CMD 
[  133.075753] sd 0:0:0:0: [sda] tag#1 CDB: opcode=0x2a 2a 00 01 d0 08 78 00 00 08 00
[  133.075756] sd 0:0:0:0: [sda] tag#0 uas_eh_abort_handler 0 uas-tag 19 inflight: CMD 
[  133.075758] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x2a 2a 00 01 10 10 30 00 00 08 00
[  133.075764] usb 1-2: USB disconnect, device number 2
[  133.075880] usb 2-1: USB disconnect, device number 2
[  133.075934] sd 0:0:0:0: [sda] tag#3 uas_zap_pending 0 uas-tag 22 inflight: CMD 
[  133.075936] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x35 35 00 00 00 00 00 00 00 00 00
[  133.075949] sd 0:0:0:0: [sda] tag#3 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=30s
[  133.075955] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x35 35 00 00 00 00 00 00 00 00 00
[  133.075963] I/O error, dev sda, sector 5568120 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 2
[... cut ...]
[  218.971768] EXT4-fs warning (device sda2): dx_probe:822: inode #19: lblock 0: comm gmain: error -5 reading directory block
[  218.971914] EXT4-fs warning (device sda2): dx_probe:822: inode #19: lblock 0: comm gmain: error -5 reading directory block
[  222.971638] EXT4-fs warning (device sda2): dx_probe:822: inode #19: lblock 0: comm gmain: error -5 reading directory block
[  222.971785] EXT4-fs warning (device sda2): dx_probe:822: inode #19: lblock 0: comm gmain: error -5 reading directory block
```

pelwell, 2024-01-05:
You can easily install rpi-update on other OSes - go to https://github.com/raspberrypi/rpi-update - but I can't guarantee that it will work.

dkersh, 2024-01-05:
I can't seem to get it working on Libreelec (but it's a read-only file system so there's always going to be a bit of a headache there). I might as well list the outputs of my logs when the fault occurs. The USB SSD drive randomly unmounts and I'm using a vanilla install of Libreelec 12.0 nightly. I'm using the official 27 W power supply with a Pi 5.

`[  590.233378] usb 4-1: cmd cmplt err -71 
[  595.459902] scsi host0: uas_pre_reset: timed out 
[  595.460150] sd 0:0:0:0: [sda] tag#16 uas_zap_pending 0 uas-tag 1 inflight: CMD  
[  595.460154] sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x28 28 00 00 01 06 51 00 00 20 00 
[  595.460166] sd 0:0:0:0: [sda] tag#16 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=5s 
[  595.460174] sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x28 28 00 00 01 06 51 00 00 20 00 
[  595.460177] I/O error, dev sda, sector 67153 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2 
[  595.460216] device offline error, dev sda, sector 67153 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2 
[  595.460230] I/O error, dev loop0, sector 36118 op 0x0:(READ) flags 0x800 phys_seg 2 prio class 2 
[  595.460242] SQUASHFS error: Failed to read block 0x11a2cbc: -5 
[  595.460245] SQUASHFS error: Unable to read fragment cache entry [11a2cbc] 
[  595.460288] SQUASHFS error: Unable to read fragment cache entry [11a2cbc] 
[  595.460290] SQUASHFS error: Unable to read page, block 11a2cbc, size 3563 
[  595.460608] SQUASHFS error: Unable to read fragment cache entry [11a2cbc] 
[  595.460611] SQUASHFS error: Unable to read page, block 11a2cbc, size 3563 
[  595.463291] device offline error, dev sda, sector 0 op 0x1:(WRITE) flags 0x800 phys_seg 0 prio class 2 
[  595.464815] EXT4-fs (sda2): shut down requested (2) 
[  595.464826] Aborting journal on device sda2-8. 
[  595.464877] device offline error, dev sda, sector 1089538 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 2 
[  595.464895] device offline error, dev sda, sector 1089538 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 2 
[  595.464900] Buffer I/O error on dev sda2, logical block 16385, lost sync page write 
[  595.464925] JBD2: I/O error when updating journal superblock for sda2-8. 
[  595.523274] sd 0:0:0:0: [sda] Synchronizing SCSI cache 
[  595.723261] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=DRIVER_OK 
[  595.723557] usb 4-1: Disable of device-initiated U1 failed. 
[  595.723577] usb 4-1: Disable of device-initiated U2 failed. 
[  595.724199] usb 4-1: Enable of device-initiated U1 failed. 
[  595.724328] usb 4-1: Enable of device-initiated U2 failed. 
[  595.920077] usb 4-1: reset SuperSpeed USB device number 2 using xhci-hcd 
[  595.940335] scsi host1: uas 
[  595.941286] scsi 1:0:0:0: Direct-Access     JMicron  Tech             0214 PQ: 0 ANSI: 6 
[  595.944453] sd 1:0:0:0: [sdb] 937703088 512-byte logical blocks: (480 GB/447 GiB) 
[  595.944460] sd 1:0:0:0: [sdb] 4096-byte physical blocks 
[  595.945170] sd 1:0:0:0: [sdb] Write Protect is off 
[  595.945175] sd 1:0:0:0: [sdb] Mode Sense: 53 00 00 08 
[  595.945554] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA 
[  595.945775] sd 1:0:0:0: [sdb] Preferred minimum I/O size 4096 bytes 
[  595.945778] sd 1:0:0:0: [sdb] Optimal transfer size 33553920 bytes not a multiple of preferred minimum block size (4096 bytes) 
[  596.440014]  sdb: sdb1 sdb2 
[  596.440578] sd 1:0:0:0: [sdb] Attached SCSI disk 
[  596.617309] EXT4-fs (sdb2): recovery complete 
[  596.618333] EXT4-fs (sdb2): mounted filesystem 69675956-de9a-48ac-b10c-13fcd7f42e78 r/w with ordered data mode. Quota mode: disabled. `

`LibreELEC (community): nightly-20231231-dfd843f (RPi5.aarch64) 
LibreELEC:~ # journalctl -f 
Jan 03 09:14:40 LibreELEC systemd[1]: systemd-update-utmp-runlevel.service: Deactivated successfully. 
Jan 03 09:14:40 LibreELEC systemd[1]: Finished systemd-update-utmp-runlevel.service. 
Jan 03 09:14:40 LibreELEC systemd[1]: Started kodi.service. 
Jan 03 09:14:40 LibreELEC systemd[1]: Reached target kodi.target. 
Jan 03 09:14:40 LibreELEC systemd[1]: Startup finished in 1.971s (kernel) + 5.817s (userspace) = 7.789s. 
Jan 03 09:14:40 LibreELEC (kodi.sh)[831]: kodi.service: Referenced but unset environment variable evaluates to an empty string: KODI_DEBUG 
Jan 03 09:14:41 LibreELEC kernel: kodi.bin[835]: memfd_create() called without MFD_EXEC or MFD_NOEXEC_SEAL set 
Jan 03 09:15:07 LibreELEC systemd[1]: systemd-hostnamed.service: Deactivated successfully. 
Jan 03 09:15:16 LibreELEC sshd[899]: Accepted password for root from 192.168.50.87 port 50225 ssh2 
Jan 03 09:17:21 LibreELEC sshd[926]: Accepted password for root from 192.168.50.87 port 50228 ssh2 
Jan 03 09:24:23 LibreELEC kernel: usb 4-1: cmd cmplt err -71 
Jan 03 09:24:28 LibreELEC kernel: scsi host0: uas_pre_reset: timed out 
Jan 03 09:24:28 LibreELEC kernel: sd 0:0:0:0: [sda] tag#16 uas_zap_pending 0 uas-tag 1 inflight: CMD  
Jan 03 09:24:28 LibreELEC kernel: sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x28 28 00 00 01 06 51 00 00 20 00 
Jan 03 09:24:28 LibreELEC kernel: sd 0:0:0:0: [sda] tag#16 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=DRIVER_OK cmd_age=5s 
Jan 03 09:24:28 LibreELEC kernel: sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x28 28 00 00 01 06 51 00 00 20 00 
Jan 03 09:24:28 LibreELEC kernel: I/O error, dev sda, sector 67153 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2 
[... cut ...]
Jan 03 09:24:29 LibreELEC kernel: EXT4-fs (sdb2): mounted filesystem 69675956-de9a-48ac-b10c-13fcd7f42e78 r/w with ordered data mode. Quota mode: disabled. 
Jan 03 09:24:29 LibreELEC systemd[1]: Finished udevil-mount@-dev-sdb2.service. 
Jan 03 09:24:30 LibreELEC kodi.sh[835]: [*** LOG ERROR #0001 ***] [2024-01-03 09:24:30] [general] Failed flush to file /storage/.kodi/temp/kodi.log: Input/output error 
Jan 03 09:25:44 LibreELEC sshd[1227]: sshd: no hostkeys available -- exiting. 
Jan 03 09:26:21 LibreELEC sshd[1247]: sshd: no hostkeys available -- exiting.`

Apologies for clogging up the issue, but I think this is a related problem.

ddscentral, 2024-01-05:
LibreELEC uses it's own kernel, not the one supplied by Raspberry Pi. You will have to rebuild the system image with kernel patch from PR #5788 applied yourself or wait until the distro maintainers apply the fix.

For those who don't want to scroll up, the fix is here: https://github.com/raspberrypi/linux/pull/5788/commits/8a0e912ba84880efeb7932d0448b02bf169cbbbd

UgaBugaOfdeSpeer, 2024-01-06:
> I think I'm also affected by this issue and it seems like `sudo rpi-update pulls/5788` didn't change anything.
>
> I'm running the pi 5 of an nvme to usb adaptor from ugreen. I also enabled [...]

Did you try already to use the "usb-storage" iso UAS for your ugreen device?
I had to do this on my raspi-4 (bullseye) and have done this also on the raspi-5 (bookworm)

balupton, 2024-02-14:
Been running a rpi4 with ubuntu server for years with a 4-port usb hub in each usb-3 port, with each usb-hub having 4x usb hard drives attached so 8x usb hdds in total, in a raid btrfs cluster with 100% redundancy. However, I upgraded it to a rpi 5 last week, fresh ubuntu server 23.10 install, and it seems as soon as I start doing anything notable on the cluster, the top usb controller will crash and is not visible to `usbreset`. Only option has been to reboot the device. By the time I get to it, dmesg is already filled with btrfs errors, so I'm not able to see earlier logs (as running with log2ram, so it prunes logs).

Is this fixed pushed to the proposed raspi kernel yet? (that is, installable via: `sudo apt install linux-raspi`

madMAx43v3r, 2024-02-23:
It seems this fix is not yet integrated via `sudo apt upgrade`? I did `sudo rpi-update` to get 6.6.y kernel, will see if that works.

In my case the error happened with a USB NVMe enclosure on the USB 3.0 port (directly attached, no hub). Another HDD was connected also on that USB controller.

EDIT: It failed right away with `6.6.y` kernel, a few minutes after I posted, where before it was running for a few days.

popcornmix, 2024-02-23:
> It seems this fix is not yet integrated via sudo apt upgrade?

apt kernel is:
```
Linux version 6.1.0-rpi8-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25)
```

Looks like the fix was https://github.com/raspberrypi/linux/pull/5788 which was merged into 6.1.68 over a month before the apt kernel was built, so yes, that fix is in apt. If that kernel doesn't fix your issue, I'd guess you have a different issue.

balupton, 2024-02-23:
> which was merged into 6.1.68 over a month before the apt kernel was built, so yes, that fix is in apt. If that kernel doesn't fix your issue, I'd guess you have a different issue.

Got a source?

Repeating my own findings:
https://github.com/raspberrypi/linux/issues/5753#issuecomment-1943553432
The default kernel  of Ubuntu Server 23.10 runs into this, and the linux-raspi kernel only lists it as fixed for noble-proposed:  https://launchpad.net/ubuntu/+source/linux-raspi

cra2710, 2024-03-04:
I had the same issue - Raspberry PI 5, usb-hub RSHTECH RSH-A10, copying from/to disks connected to same hub.

was: 
Linux raspi9 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64

after update:
Linux raspi9 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64

problem is solved.

Thank you very much!

EasyNetDev, 2026-09-02:
Hi all,

I have the same issue with my RPi5 and a 5G modem on USB. I get this error even without any traffic through the modem:
```
[ 1172.234571] xhci-hcd xhci-hcd.1: xHCI host not responding to stop endpoint command
[ 1172.254747] xhci-hcd xhci-hcd.1: xHCI host controller not responding, assume dead
[ 1172.254840] xhci-hcd xhci-hcd.1: HC died; cleaning up
[ 1172.256532] option1 ttyUSB3: usb_wwan_write: submit urb 1 failed: -19
```

I'm running Raspberry Pi OS with Linux Kerne:
```
$ uname -a
Linux MobiRPi5 6.18.39+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.18.39-1+rpt1 (2026-07-29) aarch64 GNU/Linux
```
Even is pretty new Kernel, I still have this issue with the USB controller.

pelwell, 2026-09-02:
How are you powering that Pi 5 and modem? Could you test with a powered hub between the two?

EasyNetDev, 2026-09-02:
> How are you powering that Pi 5 and modem? Could you test with a powered hub between the two?

Nope. I have a [UGREEN PSU 160W](https://eu.ugreen.com/products/ugreen-nexode-pro-160w-gan-fast-charger?_pos=1&_psq=160w&_psid=b42cfa847&_ss=e). I have an [USB Y cable](https://www.aliexpress.com/item/1005006336953084.html?spm=a2g0o.order_list.order_list_main.43.50d41802rkEOH8) to modem. On USB 3.0 cable I cut the power wire and I modem is powered only via USB 2.0 connector.
From this UGREEN PSU I have 2 cables: USB-C1 port powers RPi5 and USB-C3 port powers 5G Modem.
RPi5 never go under voltage. The PSU has a stable 5.1V for both (measured with an USB power meter).

If I try to use normal Y USB cable and power both from RPi5 and from additional USB 2.0 connector, RPi5 complains about over voltage on USB ports. So, USB 3.0 side has only data and USB 2.0 has only power.

I got this issue mostly when I'm booting the RPi5 first time in the morning. I get this error one up to 5-6 times, then works ok over the day.

P33M, 2026-09-02:
Your Y-cable does not separate power. You are effectively connecting two power supplies together through USB VBus, which is definitely not recommended. Does the issue still occur with your peripherals connected to a powered USB hub?

EasyNetDev, 2026-09-02:
> Your Y-cable does not separate power. You are effectively connecting two power supplies together through USB VBus, which is definitely not recommended. Does the issue still occur with your peripherals connected to a powered USB hub?

Check my explanation: I cut the power wire from USB 3.0 side. If I connect the modem only on USB 3.0 is not powering up anymore. I need to power over USB 2.0 connector to get powered up.

Just the GND is common, which it should be.

pelwell, 2026-09-02:
> I got this issue mostly when I'm booting the RPi5 first time in the morning. I get this error one up to 5-6 times, then works ok over the day.

It would have been good to include this in your first message.

EasyNetDev, 2026-09-02:
> > I got this issue mostly when I'm booting the RPi5 first time in the morning. I get this error one up to 5-6 times, then works ok over the day.
> 
> It would have been good to include this in your first message.

Yeah, indeed, I notice this in the last weeks.

pelwell, 2026-09-02:
Hand modification of a cheap USB cable puts you in the "niche use case" category. It's almost certain that the modem is just drawing too much current and malfunctioning on a cold boot.

EasyNetDev, 2026-09-02:
> Hand modification of a cheap USB cable puts you in the "niche use case" category. It's almost certain that the modem is just drawing too much current and malfunctioning on a cold boot.

And for this reason RPi5 XHCI is crashing? I test the same modem on my Ryzen 5900X PC. Worked without this issue for days. With same cable and same PSU.

pelwell, 2026-09-02:
I'm not sure I'd call being unable to complete a transfer "crashing", especially if it recovers without power cycling when asked to do something less power-hungry. Your PC has a much higher power budget, so it's not surprising that it behaves differently.

EasyNetDev, 2026-09-02:
> I'm not sure I'd call being unable to complete a transfer "crashing", especially if it recovers without power cycling when asked to do something less power-hungry. Your PC has a much higher power budget, so it's not surprising that it behaves differently.

It doesn't recover without power cycling :( RPi5.

But in my case the power supplies are separated. So if modem needs more power is draws from PSU directly in both cases. Why in this particular case with RPi5 XHCI crashes, is still a question for me.
Before I separate the power supply over the cable, RPi5 was resetting the VBUS power because modem requested too much power.
I was using also:
```
usb_max_current_enable=1
```

But still after a while RPi5 was complain about power. Now there is no complain about the power. Just this issue.
And the cable I changed is similar with the one from AliExpress, but came with 5G Modem USB case identical to this: https://www.pi-shop.ch/5g-dongle-module
But came with this Y USB cable for more additional power.

I will try the same setup with a Odroid XU4. Unfortunately that one has only 2GB of RAM, but has USB 3.0 ports.

ddscentral, 2026-09-02:
I suspect a hardware (or more precisely - hardware setup) issue though not necessarily Pi 5 specific.
I have similar issues with an electrically similar setup with an external self-powered hard drive box. It just keeps crashing the VLI controller on my Pi 4. Perhaps there is some sort of backfeed or interference coming from that self-powered connected device.
When I had those same storage devices in a box powered by the exact same PSU as Pi's own, I had no such issues.

This likely has nothing to do with the original USB HUB issue which was IIRC a kernel bug.