raspberrypi/linux #5759: Raspberry PI 5 : usb audio issue when playing 176.4 and 352.8 kHz files

chmertes, 2023-11-29:
### Describe the bug

When playing audio files with sample rate of 176.4 or 352.8 kHz, in addition to the music, we can hear a constant sizzling stronger at 352.8 than 176.4.  At all other sample rates (44.1, 48, 88.2, 96, 192 and 384), there are no issues.

This happens with an "Antelope Audio Zodiac Gold" usb DAC connected to the Pi 5 on usb2 or usb3 ports.

Tested with PiOS Bookworm 64 bit and Ubuntu 23.10 64 bit server.

On the same systems, with an "Allo BOSS 1.2" I2C DAC, there are no issues.

On a Pi 4, with DietPi v8.24.1, there are no issues.

### Steps to reproduce the behaviour

Install PiOS 64 bit 
Install and configure MPD to use the Zodiac usb DAC.
Play 176.4 or 352.8 kHz files.

### Device (s)

Other

### System

cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

vcgencmd version
2023/10/30 16:45:10 
Copyright (c) 2012 Broadcom
version 30de0ba5 (release) (embedded)

uname -a
Linux PiOS-RPI5-MPD 6.1.0-rpi6-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

### Logs

Nothing appears in journalctl.

P33M, 2023-12-13:
That's a ridiculously expensive DAC. Also unavailable for sale, which will hamper testing efforts.

Please post a full `sudo lsusb -v && sudo lsusb -t` with the device plugged in, and the output of `dmesg` after playback has hit glitches.

Are there any other DACs that you know of that support this sample rate?

P33M, 2023-12-15:
I have a USB2.0 DAC that can achieve 176.4kHz and 192kHz sample rates. No issues with audio skips or glitches here. Without any further information this won't be investigated further.

chmertes, 2023-12-16:
I only tested with the Zodiac dac.  I don't know if others have the same issue.
But the fact that it works with a P4 and it does not work anymore with a P5 indicates that a regression has been introduced.

Please, see the attached usb_audio.tar.gz file for the result of the lsusb and dmesg commands.

[usb_audio.tar.gz](https://github.com/raspberrypi/linux/files/13692282/usb_audio.tar.gz)

P33M, 2023-12-16:
The only unusual thing I can see is that the Isochronous endpoint used for audio streaming has bmAttributes[3:2] = 'b11 which is Synchronous, i.e. uses SOF packets for timing. As far as I can measure, SOF packets on Pi 5 have the correct interval (125us, ±0.0625us).

The vast majority of DACs use Asynchronous adaptive feedback endpoints, so this will not be commonly encountered.

chmertes, 2024-01-06:
The Zodiac is an asynchronous dac, although the lsusb command returns "Synch Type               Synchronous".  The same value is returned on the Pi 4 where it works fine.

The Zodiac is a 32 bits dac.  Is your dac 24 or 32 bits?  Could the issue be linked to the way the bit depth is managed?