raspberrypi/linux #5764: B101 TC358743 on RPI5

tuyaliang, 2023-12-01:
### Describe the bug

Does Raspberry Pi 5 not support B101(tc358743 hdmi to csi) module? I can't turn on when I plug in B101, the red light keeps on.

### Steps to reproduce the behaviour

Insert B101 to csi0/1
Power on RPI5

### Device (s)

Other

### System

cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

vcgencmd version
2023/10/30 16:45:10
Copyright (c) 2012 Broadcom
version 30de0ba5 (release) (embedded)

uname -a
Linux PiOS-RPI5-MPD 6.1.0-rpi6-rpi-2712 https://github.com/raspberrypi/linux/issues/1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

tuyaliang, 2023-12-01:
> Already discussed https://forums.raspberrypi.com/viewtopic.php?t=359505 https://forums.raspberrypi.com/viewtopic.php?t=359412

But my RPI5 can not boot up when insert B101.

pelwell, 2023-12-01:
[ Corrected from HDMI to CSI, but perhaps even that's wrong ]

Is there anything shown on the HDMI display or serial port?

6by9, 2023-12-01:
That's not a kernel issue.

B101 has a 15pin connector. Pi5 has a 22pin. What adapter have you used to connect the two?
At a guess you've replaced the provided FFC which has contacts on the same side with one of the standard adapters. Almost all other cameras (particularly the Pi ones) use 15pin FFCs with the contacts on opposite sides, and so the adapter cables are expecting that wiring on the camera end. With the B101 flipping it, you're likely to have shorted out the power rails and many other signals.

6by9, 2023-12-01:
The only guaranteed way to connect a third party 15pin camera is via the likes of https://uk.rs-online.com/web/p/raspberry-pi/1363742 (note that there are separate camera and display variants, but you'll have to find an alternate vendor as RS have dropped it) and retaining the original cable.
If you're replacing their cable, be VERY cautious over same or opposite side FFCs.

tuyaliang, 2023-12-01:
> The only guaranteed way to connect a third party 15pin camera is via the likes of https://uk.rs-online.com/web/p/raspberry-pi/1363742 (note that there are separate camera and display variants, but [...]

[image]
The FFC is good for imx219, I have tested.I have also tried both on the same side and on the opposite side.

tuyaliang, 2023-12-01:
I have many B101 boards, but none of them work, I tested on my RPI4B, it is no problem.

tuyaliang, 2023-12-01:
The led of B101 is also not on.

6by9, 2023-12-01:
> The FFC is good for imx219, I have tested.

If the FFC works with an imx219, then I'm 99.99% certain it is wrong for B101.

If you look at the cable supplied with the B101 (just visible in the photo at https://auvidea.eu/images/auvidea/products/bridge/B101/B101_test_setup_RPi_600x350.jpg), the contacts at the 2 ends of the cable are on the same side of the flex.
If you look at the 15-15pin cable supplied with the imx219, the contacts are on opposite sides.

Having the incorrect sided cables effective means pin 1 is swapped with 15, 2 with 14, etc. 
Pin 1 is GND. Pin 15 is 3V3. Swapping those 2 is not going to work, and can potentially cause damage to either device.

> I have also tried both on the same side and on the opposite side.

The connectors on the Pi and B101 only have contacts on one side, so how have you tested same and opposite sides?

Another option for the cable is to use the 22-15 pin FFC you have, join it to a coupler such as https://www.adafruit.com/product/3671, and then use the same sides cable supplied with the B101.

tuyaliang, 2023-12-01:
> If the FFC works with an imx219, then I'm 99.99% certain it is wrong for B101.
>
> If you look at the cable supplied with the B101 (just visible in the photo at [...]

Thanks, I know it,  on one side, RPI5 can not power on, and opposite side, RPI5 can power on, but tc358743 driver not loading.

pelwell, 2023-12-01:
I think @6by9 has explained your problem, and it's not a kernel bug.

6by9, 2023-12-01:
FWIW I have had a B102 probing properly, but as noted in the initially linked threads, Pi5 has to use Media Controller API so simply setting resolutions via V4L2 VIDIOC_S_FMT will not work.

The main supported use case is for raw cameras that run through the Camera Front End (CFE) rather than writing the raw imges direct to memory, so that is the main one tested. With TC358743 wanting to write directly to memory, there appears to be something not quite correctly configured, so there is some follow-up work to do.

tuyaliang, 2023-12-01:
> FWIW I have had a B102 probing properly, but as noted in the initially linked threads, Pi5 has to use Media Controller API so simply setting resolutions via V4L2 VIDIOC_S_FMT will not work.
>
> The [...]

Thanks,I also want to know gstreamer v4lconvert \ v4l2h264dec \ v4l2h264enc does not support on RPI5？

tuyaliang, 2023-12-01:
On RPI4B gst-inspect-1.0  shows:
dji@raspberrypi:~ $ gst-inspect-1.0 | grep v4l2
video4linux2:  v4l2convert: V4L2 Video Converter
video4linux2:  v4l2deviceprovider (GstDeviceProviderFactory)
video4linux2:  v4l2h264dec: V4L2 H264 Decoder
video4linux2:  v4l2h264enc: V4L2 H.264 Encoder
video4linux2:  v4l2jpegdec: V4L2 JPEG Decoder
video4linux2:  v4l2jpegenc: V4L2 JPEG Encoder
video4linux2:  v4l2radio: Radio (video4linux2) Tuner
video4linux2:  v4l2sink: Video (video4linux2) Sink
video4linux2:  v4l2src: Video (video4linux2) Source
video4linux2:  v4l2video18convert: V4L2 Video Converter
video4linux2:  v4l2video31jpegenc: V4L2 JPEG Encoder

But on RPI5B only shows:
dji@raspberrypi:~ $ gst-inspect-1.0 | grep v4l2
video4linux2:  v4l2deviceprovider (GstDeviceProviderFactory)
video4linux2:  v4l2radio: Radio (video4linux2) Tuner
video4linux2:  v4l2sink: Video (video4linux2) Sink
video4linux2:  v4l2src: Video (video4linux2) Source

6by9, 2023-12-01:
Pi5 has only HEVC decode HW acceleration, and that implements the V4L2 stateless decoder instead of the stateful that the v4l2*dec components use.

In theory the ISP backend can be used as a generic format converter/resize that can be used via v4l2convert, however it requires far more configuration which is currently done in userspace with libpisp rather than in the kernel.