raspberrypi/linux #5765: Kernel Compile error 6.1.x

xianrecords2015, 2023-12-01:
### Describe the bug

I downloaded the current kernel release and I am having a compile error issue:

See log

### Steps to reproduce the behaviour

git clone --depth=1 https://github.com/raspberrypi/linux
cd linux
KERNEL=kernel8
make bcm2711_defconfig
make -j4

### Device (s)

Raspberry Pi CM4

### System

Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2

Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux corin1 6.1.64-v8+ #1702 SMP PREEMPT Wed Nov 29 14:23:01 GMT 2023 aarch64 GNU/Linux

### Logs

  DTCO    arch/arm64/boot/dts/overlays/rpi-poe.dtbo
  DTCO    arch/arm64/boot/dts/overlays/rpi-poe-plus.dtbo
In file included from ./arch/arm64/include/asm/lse.h:5,
                 from ./arch/arm64/include/asm/cmpxchg.h:14,
                 from ./arch/arm64/include/asm/atomic.h:16,
                 from ./include/linux/atomic.h:7,
                 from ./include/asm-generic/bitops/atomic.h:5,
                 from ./arch/arm64/include/asm/bitops.h:25,
                 from ./include/linux/bitops.h:68,
                 from ./include/linux/log2.h:12,
                 from kernel/bounds.c:13:
./arch/arm64/include/asm/atomic_ll_sc.h: In function ‘__ll_sc__cmpxchg_double’:
./arch/arm64/include/asm/atomic_ll_sc.h:318:38: error: ‘__uint128_t’ undeclared (first use in this function); did you mean ‘__int128__’?
  318 |  : "=&r" (tmp), "=&r" (ret), "+Q" (*(__uint128_t *)ptr)  \
      |                                      ^~~~~~~~~~~
./arch/arm64/include/asm/atomic_ll_sc.h:325:1: note: in expansion of macro ‘__CMPXCHG_DBL’
  325 | __CMPXCHG_DBL(   ,        ,  ,         )
      | ^~~~~~~~~~~~~
./arch/arm64/include/asm/atomic_ll_sc.h:318:38: note: each undeclared identifier is reported only once for each function it appears in
  318 |  : "=&r" (tmp), "=&r" (ret), "+Q" (*(__uint128_t *)ptr)  \
      |                                      ^~~~~~~~~~~
./arch/arm64/include/asm/atomic_ll_sc.h:325:1: note: in expansion of macro ‘__CMPXCHG_DBL’
  325 | __CMPXCHG_DBL(   ,        ,  ,         )
      | ^~~~~~~~~~~~~
./arch/arm64/include/asm/atomic_ll_sc.h:318:51: error: expected expression before ‘)’ token
  318 |  : "=&r" (tmp), "=&r" (ret), "+Q" (*(__uint128_t *)ptr)  \
      |                                                   ^
./arch/arm64/include/asm/atomic_ll_sc.h:325:1: note: in expansion of macro ‘__CMPXCHG_DBL’
  325 | __CMPXCHG_DBL(   ,        ,  ,         )
      | ^~~~~~~~~~~~~
  DTCO    arch/arm64/boot/dts/overlays/rpi-sense.dtbo
./arch/arm64/include/asm/atomic_ll_sc.h: In function ‘__ll_sc__cmpxchg_double_mb’:
./arch/arm64/include/asm/atomic_ll_sc.h:318:38: error: ‘__uint128_t’ undeclared (first use in this function); did you mean ‘__int128__’?
  318 |  : "=&r" (tmp), "=&r" (ret), "+Q" (*(__uint128_t *)ptr)  \
      |                                      ^~~~~~~~~~~
./arch/arm64/include/asm/atomic_ll_sc.h:326:1: note: in expansion of macro ‘__CMPXCHG_DBL’
  326 | __CMPXCHG_DBL(_mb, dmb ish, l, "memory")
      | ^~~~~~~~~~~~~
./arch/arm64/include/asm/atomic_ll_sc.h:318:51: error: expected expression before ‘)’ token
  318 |  : "=&r" (tmp), "=&r" (ret), "+Q" (*(__uint128_t *)ptr)  \
      |                                                   ^
./arch/arm64/include/asm/atomic_ll_sc.h:326:1: note: in expansion of macro ‘__CMPXCHG_DBL’
  326 | __CMPXCHG_DBL(_mb, dmb ish, l, "memory")
      | ^~~~~~~~~~~~~
  DTCO    arch/arm64/boot/dts/overlays/rpi-sense-v2.dtbo

pelwell, 2023-12-01:
Which platform - hardware and OS - are you running the compile on? If you are running it on a CM4 running the Linux kernel (as you might be), is the OS (not the kernel) 32-bit or 64-bit?

xianrecords2015, 2023-12-01:
CM4 Bulleyes 64 bits

pelwell, 2023-12-02:
You can confirm the OS architecture by running `file $(which ls)`. If you are indeed running the 32-bit OS but with a 64-bit kernel, which is now the default on the Pi 4 family, then you'll need to treat the kernel build as a cross-compile. You may find these instructions helpful: https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compiling-the-kernel