raspberrypi/linux #5768: RaspiKey will not work correct on pi5

semool, 2023-12-03:
**Describe the bug**
When i replace the SDCard with the RaspiKey (32GB) the System will not work.
It comes to CQE Recovery and root gets read only.
The RaspiKey itself works fine in a RPI4 8GB.

**System**
Raspberry Pi5 8GB fully updated Raspberry Pi OS
eeprom 2023/11/20 19:40:17
version 361205c6 (release) (embedded)
6.1.64-v8-16k+ raspberrypi/firmware#1702 SMP PREEMPT Wed Nov 29 14:26:42 GMT 2023 aarch64 GNU/Linux

**Logs**
```
[Do Nov 30 20:19:26 2023] mmc0: running CQE recovery
[Do Nov 30 20:19:26 2023] ------------[ cut here ]------------
[Do Nov 30 20:19:26 2023] mmc0: cqhci: spurious TCN for tag 10
[Do Nov 30 20:19:26 2023] WARNING: CPU: 0 PID: 98 at drivers/mmc/host/cqhci-core.c:787 cqhci_irq+0x394/0x598
[Do Nov 30 20:19:26 2023] Modules linked in: vhost_net tun vhost vhost_iotlb tap drm fuse drm_panel_orientation_quirks backlight dm_mod ip_tables x_tables ipv6
[Do Nov 30 20:19:26 2023] CPU: 0 PID: 98 Comm: kworker/0:1H Not tainted 6.1.63-v8-16k+ raspberrypi/firmware#1700
[Do Nov 30 20:19:26 2023] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[Do Nov 30 20:19:26 2023] Workqueue: kblockd blk_mq_run_work_fn
[Do Nov 30 20:19:26 2023] pstate: 404000c9 (nZcv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[Do Nov 30 20:19:26 2023] pc : cqhci_irq+0x394/0x598
[Do Nov 30 20:19:26 2023] lr : cqhci_irq+0x394/0x598
[Do Nov 30 20:19:26 2023] sp : ffffc00008003d60
[Do Nov 30 20:19:26 2023] x29: ffffc00008003d60 x28: ffff8001001b6540 x27: 0000000000000001
[Do Nov 30 20:19:26 2023] x26: ffff80010075c298 x25: ffffd0000cf79698 x24: ffffd0000d69b132
[Do Nov 30 20:19:26 2023] x23: ffffd0000cf99d98 x22: ffff8001001b6000 x21: ffff80010075c280
[Do Nov 30 20:19:26 2023] x20: 0000000000000002 x19: 000000000000000a x18: 0000000000000001
[Do Nov 30 20:19:26 2023] x17: ffffb001f2928000 x16: ffffc00008000000 x15: ffffffffffffffff
[Do Nov 30 20:19:26 2023] x14: 0000000000000000 x13: 3031206761742072 x12: ffffd0000d54f7d0
[Do Nov 30 20:19:26 2023] x11: 0000000000000003 x10: 0000000000000001 x9 : ffffd0000c0f590c
[Do Nov 30 20:19:26 2023] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : ffffd0000d537790
[Do Nov 30 20:19:26 2023] x5 : ffff8001ffa9aa10 x4 : 0000000000000000 x3 : ffffb001f2928000
[Do Nov 30 20:19:26 2023] x2 : ffff800101313600 x1 : a8b37fea2d466800 x0 : 0000000000000000
[Do Nov 30 20:19:26 2023] Call trace:
[Do Nov 30 20:19:26 2023]  cqhci_irq+0x394/0x598
[Do Nov 30 20:19:26 2023]  sdhci_brcmstb_cqhci_irq+0x64/0x90
[Do Nov 30 20:19:26 2023]  sdhci_irq+0xb8/0x1018
[Do Nov 30 20:19:26 2023]  __handle_irq_event_percpu+0x88/0x280
[Do Nov 30 20:19:26 2023]  handle_irq_event_percpu+0x20/0x68
[Do Nov 30 20:19:26 2023]  handle_irq_event+0x50/0xa8
[Do Nov 30 20:19:26 2023]  handle_fasteoi_irq+0xd0/0x180
[Do Nov 30 20:19:26 2023]  generic_handle_domain_irq+0x34/0x50
[Do Nov 30 20:19:26 2023]  gic_handle_irq+0xa0/0xd8
[Do Nov 30 20:19:26 2023]  call_on_irq_stack+0x24/0x4c
[Do Nov 30 20:19:26 2023]  do_interrupt_handler+0x8c/0x90
[Do Nov 30 20:19:26 2023]  el1_interrupt+0x38/0x70
[Do Nov 30 20:19:26 2023]  el1h_64_irq_handler+0x18/0x28
[Do Nov 30 20:19:26 2023]  el1h_64_irq+0x64/0x68
[Do Nov 30 20:19:26 2023]  _raw_spin_unlock_irqrestore+0x30/0x70
[Do Nov 30 20:19:26 2023]  sdhci_cqe_enable+0x130/0x1a8
[Do Nov 30 20:19:26 2023]  sdhci_brcmstb_cqe_enable+0x9c/0xc0
[Do Nov 30 20:19:26 2023]  cqhci_request+0xd4/0x508
[Do Nov 30 20:19:26 2023]  mmc_cqe_start_req+0x74/0x118
[Do Nov 30 20:19:26 2023]  mmc_blk_mq_issue_rq+0x418/0x7f8
[Do Nov 30 20:19:26 2023]  mmc_mq_queue_rq+0x118/0x2b0
[Do Nov 30 20:19:26 2023]  blk_mq_dispatch_rq_list+0x324/0x8f8
[Do Nov 30 20:19:26 2023]  __blk_mq_sched_dispatch_requests+0xc8/0x178
[Do Nov 30 20:19:26 2023]  blk_mq_sched_dispatch_requests+0x40/0x80
[Do Nov 30 20:19:26 2023]  __blk_mq_run_hw_queue+0x84/0xc0
[Do Nov 30 20:19:26 2023]  blk_mq_run_work_fn+0x28/0x38
[Do Nov 30 20:19:26 2023]  process_one_work+0x208/0x480
[Do Nov 30 20:19:26 2023]  worker_thread+0x50/0x428
[Do Nov 30 20:19:26 2023]  kthread+0x114/0x120
[Do Nov 30 20:19:26 2023]  ret_from_fork+0x10/0x20
[Do Nov 30 20:19:26 2023] ---[ end trace 0000000000000000 ]---
[Do Nov 30 20:19:26 2023] mmc0: running CQE recovery
[Do Nov 30 20:19:26 2023] mmc0: running CQE recovery
[Do Nov 30 20:19:26 2023] mmc0: running CQE recovery
[Do Nov 30 20:19:26 2023] I/O error, dev mmcblk0, sector 1581064 op 0x1:(WRITE) flags 0x800 phys_seg 3 prio class 2
[Do Nov 30 20:19:26 2023] Aborting journal on device mmcblk0p2-8.
[Do Nov 30 20:19:26 2023] mmc0: running CQE recovery
[Do Nov 30 20:19:26 2023] mmc0: running CQE recovery
[Do Nov 30 20:19:26 2023] mmc0: running CQE recovery
[Do Nov 30 20:19:26 2023] systemd-journald[153]: File /var/log/journal/b0d9f83665a145d68a3f5b5b12bfed5e/system.journal corrupted or uncleanly shut down, renaming and replacing.
[Do Nov 30 20:19:26 2023] I/O error, dev mmcblk0, sector 1581056 op 0x1:(WRITE) flags 0x20800 phys_seg 1 prio class 2
[Do Nov 30 20:19:26 2023] Buffer I/O error on dev mmcblk0p2, logical block 131072, lost sync page write
[Do Nov 30 20:19:26 2023] JBD2: I/O error when updating journal superblock for mmcblk0p2-8.
[Do Nov 30 20:19:26 2023] EXT4-fs error (device mmcblk0p2): ext4_journal_check_start:83: comm systemd-random-: Detected aborted journal
[Do Nov 30 20:19:26 2023] EXT4-fs error (device mmcblk0p2): ext4_journal_check_start:83: comm systemd-journal: Detected aborted journal
[Do Nov 30 20:19:26 2023] EXT4-fs (mmcblk0p2): Remounting filesystem read-only
```

DanaGoyette, 2023-12-13:
If I disable a few features by trying some quirk flags, I get it failing to initialize HS200 speed.  If I also disable HS200, I still get some errors.  One quirk I set was to disable CQ, but it didn't seem to take effect at first until I set another bit; maybe I'm off by one?  I added these to /boot/cmdline.txt:

`sdhci.debug_quirks=0x60000 sdhci.debug_quirks2=0x44`

The safest way to test this is by booting from something other than SD, and then just writing from /dev/urandom or /dev/zero to the eMMC.  I'd suggest using a blkdiscard before runs, to trim the whole device and make it blank.

```
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:C5:1A:24 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=a8055fd6-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US drm.debug=0x10e sdhci.debug_quirks=0x60000 sdhci.debug_quirks2=0x44
[    1.445915] sdhci: Secure Digital Host Controller Interface driver
[    1.452146] sdhci: Copyright(c) Pierre Ossman
[    1.456673] sdhci-pltfm: SDHCI platform and OF driver helper
[    2.540338] sdhci-brcmstb 1000fff000.mmc: there is not valid maps for state default
[    2.540779] mmc1: CQHCI version 5.10
[    2.541102] mmc0: CQHCI version 5.10
[    2.590846] mmc0: SDHCI controller on 1000fff000.mmc [1000fff000.mmc] using ADMA 64-bit
[    2.664448] mmc0: new DDR MMC card at address 0001
[    2.669975] mmcblk0: mmc0:0001 BJTD4R 29.1 GiB 
[    2.676275] mmcblk0: mmc0:0001 BJTD4R 29.1 GiB
[    2.681380] mmcblk0boot0: mmc0:0001 BJTD4R 4.00 MiB 
[    2.687710] mmcblk0boot1: mmc0:0001 BJTD4R 4.00 MiB 
[    2.693802] mmcblk0rpmb: mmc0:0001 BJTD4R 4.00 MiB, chardev (243:0)
[    2.730401] mmc1: SDHCI controller on 1001100000.mmc [1001100000.mmc] using ADMA 64-bit
[    2.754653] mmc1: new high speed SDIO card at address 0001
[  113.997408] I/O error, dev mmcblk0, sector 394240 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  113.997426] Buffer I/O error on dev mmcblk0, logical block 12320, lost async page write
[  113.997441] Buffer I/O error on dev mmcblk0, logical block 12321, lost async page write
[  113.997447] Buffer I/O error on dev mmcblk0, logical block 12322, lost async page write
[  113.997452] Buffer I/O error on dev mmcblk0, logical block 12323, lost async page write
[  113.997457] Buffer I/O error on dev mmcblk0, logical block 12324, lost async page write
[  113.997462] Buffer I/O error on dev mmcblk0, logical block 12325, lost async page write
[  113.997467] Buffer I/O error on dev mmcblk0, logical block 12326, lost async page write
[  113.997472] Buffer I/O error on dev mmcblk0, logical block 12327, lost async page write
[  113.997477] Buffer I/O error on dev mmcblk0, logical block 12328, lost async page write
[  113.997483] Buffer I/O error on dev mmcblk0, logical block 12329, lost async page write
[  114.000954] I/O error, dev mmcblk0, sector 404480 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  114.001294] I/O error, dev mmcblk0, sector 405504 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  114.005487] I/O error, dev mmcblk0, sector 406528 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  114.011098] I/O error, dev mmcblk0, sector 407552 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  114.015983] I/O error, dev mmcblk0, sector 408576 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  114.017778] I/O error, dev mmcblk0, sector 409600 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  114.022432] I/O error, dev mmcblk0, sector 410624 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  114.024404] I/O error, dev mmcblk0, sector 411648 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  114.027930] I/O error, dev mmcblk0, sector 412672 op 0x1:(WRITE) flags 0x104000 phys_seg 32 prio class 2
[  121.340382] I/O error, dev mmcblk0, sector 759712 op 0x1:(WRITE) flags 0x4800 phys_seg 32 prio class 2
```

DanaGoyette, 2024-02-01:
Any update on this?  I notice the same issue with another eMMC module, an ODroid eMMC module with their adapter to use it in a micro-SD card slot.  So it's not just the RasPiKey.

pelwell, 2024-02-02:
We've identified a part of the BCM2712 SDHCI driver which, when disabled, allows the RasPiKey to function. We now need to understand why that works, and create a better fix - disabling it permanently is not an option.

padlefot, 2024-02-27:
Affected by this too, tried both raspbian and ubuntu on the rpi5, any updates to this thread? :)

pelwell, 2024-02-28:
I believe this should have been fixed by a change that went into the rpi-6.6.y tree a few weeks ago: https://github.com/raspberrypi/linux/commit/d6b88dc94da4d920fd89d20e235408424a428c45

Do you agree, @P33M?

The next stable build will be from rpi-6.6.y, which means that the next kernel in `apt` will contain the fix. Until then, you can install a _beta_ 6.6 kernel including the fix by running `sudo rpi-update`.

P33M, 2024-02-28:
The PHY now behaves correctly with MMC devices, which should squelch IO errors that were seen by the reported, but my particular card appears to hang during boot (but after many reads have been performed successfully).

martin2day2, 2024-03-08:
> We've identified a part of the BCM2712 SDHCI driver which, when disabled, allows the RasPiKey to function. We now need to understand why that works, and create a better fix - disabling it permanently is not an option.

how can I disable the BCM2712 driver to test if this fix work for me. I have an RP5 and an eMMC with Adapter and also can not start....

pelwell, 2024-03-08:
> you can install a beta 6.6 kernel including the fix by running "sudo rpi-update".

martin2day2, 2024-03-08:
Okay, but how can I do this if I can not boot with the eMMC?

pelwell, 2024-03-08:
An updated image based on the 6.6 kernel is only a matter of weeks away, but if you have a spare SD card and a USB-attached SD card reader then you could:
1. Put an image onto the SD card.
2. Boot from it.
3. Put the RaspiKey into the card reader, and plug it into the Pi.
4. Confirm that the card is mounted on /media/boot and /media/root.
5. Run `sudo ROOT_PATH=/media/root BOOT_PATH=/media/boot rpi-update` (see https://github.com/raspberrypi/rpi-update?tab=readme-ov-file#root_path-and-boot_path).
6. Shutdown, then try booting from the RaspiKey.

m01, 2024-03-10:
Thank you for the fix. I've been able to make a RasPiKey work on a Raspberry 5. I plugged the RasPiKey into a Raspberry Pi 4 and ran `sudo rpi-update`. Do note the warnings that `rpi-update` emits (I hadn't seen the comment just above, otherwise I would've followed those instructions).

haryseldon2, 2024-03-11:
Today I received the update from repository with kernel 6.6 and I can confirm that the emmc boots fine. However, In my case the agnostics test "only" achieves 41 MB/s in secuencial read test. I would expect at least 70 MB/s.
Anyway,  thanks to make it work again.

semool, 2024-03-15:
I have tested already with 4GB and 8GB Pi5. Its running fine. Thx for the fix.