raspberrypi/linux #5770: Zero 2 W wifi bugs with mmc1: Controller never released inhibit bit(s)

llllllllqq, 2023-12-05:
### Describe the bug

Hi, I find at least 2 random bugs, both of them showing mmc1: Controller never released inhibit bit(s) in dmesg first and following some error.

The first is wifi lost. I cannot access the device via wifi, so I have to force plug-out&in power to reboot.
The second is also about wifi, but only error in dmesg. The connection is (maybe) fine.

### Steps to reproduce the behaviour

Both bugs are on the same device and the same kernel/firmware shown in the dmesg.
I'm sure my power supply is enough and stable (It can output at least 5.2v/2a and pass a stress test with USB volt meter showing 5.1v, no under-volt found, throttled=0x0). And the Pi can pass an 1 hour iperf3 bi-direct stress test without any error.

The first bug (not 100% occur, only random):
1. Only connect a SATA SSD (toshiba tr200) via a JMS chip USB to SATA box, and via a USB otg line. (I've tested its max current is 501ma by a 10kHz USB meter, average working current is ~300ma).
2. Use an android app called "foldersync" to upload some big files via sftp to that SSD on Pi.
3. Manually cancel the transfer.
4. Wifi is disappered. And in at least 30 minutes, it won't self-restart, so I force restart.

The second bug (also random, but more often, avg once 2 days):
1. Connect nothing, use the same android app "foldersync" to upload some big files via sftp to SDCARD on Pi.
2. Manually cancel the transfer.
3. Wifi seems normal, but dmesg shows sth error.

### Device (s)

Raspberry Pi Zero 2 W

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

Oct 17 2023 15:42:53
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start_cd)

Linux rpi02w 6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64 GNU/Linux

### Logs

First bug:
```
Dec 02 13:48:12 rpi02w kernel: mmc1: Controller never released inhibit bit(s).
Dec 02 13:48:12 rpi02w kernel: brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -5
Dec 02 13:48:12 rpi02w kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
Dec 02 13:48:12 rpi02w kernel: brcmfmac: brcmf_sdio_rxglom: descriptor len 5 bad: 0
Dec 02 13:48:12 rpi02w kernel: brcmfmac: brcmf_sdio_hdparse: HW header length too long
Dec 02 13:48:12 rpi02w kernel: brcmfmac: brcmf_sdio_rxfail: terminate frame
Dec 02 13:48:12 rpi02w kernel: brcmfmac: brcmf_sdio_rxglom: descriptor len 7 bad: 0
Dec 02 13:48:12 rpi02w kernel: brcmfmac: brcmf_sdio_hdparse: HW header length too long
Dec 02 13:48:12 rpi02w kernel: brcmfmac: brcmf_sdio_rxfail: terminate frame
Dec 02 13:48:18 rpi02w kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 02 13:48:18 rpi02w kernel: ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
Dec 02 13:48:25 rpi02w kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 02 13:48:25 rpi02w kernel: ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
Dec 02 13:48:30 rpi02w kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 02 13:48:30 rpi02w kernel: ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
Dec 02 13:48:37 rpi02w kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 02 13:48:37 rpi02w kernel: ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
Dec 02 13:48:43 rpi02w kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 02 13:48:43 rpi02w kernel: ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
Dec 02 13:48:48 rpi02w kernel: ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 02 13:48:48 rpi02w kernel: ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
......end less repeat....
```
Note: this bug seems to be like https://github.com/raspberrypi/firmware/issues/1723 but I don't know whether it's safe to try over_voltage=2 on my device (my vcgencmd measure_volts is volt=1.2563V)

Second bug:
type a:
```
Dec 04 20:22:35 rpi02w kernel: mmc1: Controller never released inhibit bit(s).
Dec 04 20:22:35 rpi02w kernel: brcmfmac: brcmf_sdio_readframes: read 1108 bytes from channel 2 failed: -5
Dec 04 20:22:35 rpi02w kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
Dec 04 20:22:35 rpi02w kernel: brcmfmac: brcmf_sdio_readframes: brcmf_sdio_readframes: glom superframe w/o descriptor!
Dec 04 20:22:35 rpi02w kernel: brcmfmac: brcmf_sdio_rxfail: terminate frame
```
type b:
```
Dec 02 17:56:17 rpi02w kernel: mmc1: Controller never released inhibit bit(s).
Dec 02 17:56:17 rpi02w kernel: brcmfmac: brcmf_sdio_readframes: read 1372 bytes from channel 2 failed: -5
Dec 02 17:56:17 rpi02w kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
```

I've also uploaded a full dmesg of bug 2, type a.
[dmesg.txt](https://github.com/raspberrypi/linux/files/13555048/dmesg.txt)

pelwell, 2023-12-05:
This sounds like a hardware/power issue. You can try adding `over_voltage=2` to config.txt.

llllllllqq, 2023-12-05:
> This sounds like a hardware/power issue. You can try adding `over_voltage=2` to config.txt.

Thanks for your reply. I've added this config and will keep my eyes on it.
Some additional questions: 
Do you mean all `mmc1: Controller never released inhibit bit(s).` is about power issue? If yes, it'll save a lot time to find whether bugs are triggered because the first bug is very rare.
And after over_voltage=2, is volt=1.3062V safe for my device with only a aluminum heatsink (no fan)?

llllllllqq, 2023-12-05:
I find more information.
These bugs even occur synchronously when I start a sftp session by "foldersync".
```
Dec 04 20:17:01 rpi02w CRON[13185]: pam_unix(cron:session): session closed for user root
Dec 04 20:22:35 rpi02w kernel: mmc1: Controller never released inhibit bit(s).
Dec 04 20:22:35 rpi02w kernel: brcmfmac: brcmf_sdio_readframes: read 1108 bytes from channel 2 failed: -5
Dec 04 20:22:35 rpi02w kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
Dec 04 20:22:35 rpi02w kernel: brcmfmac: brcmf_sdio_readframes: brcmf_sdio_readframes: glom superframe w/o descriptor!
Dec 04 20:22:35 rpi02w kernel: brcmfmac: brcmf_sdio_rxfail: terminate frame
Dec 04 20:22:36 rpi02w sshd[13198]: Accepted publickey for S**** from 192.168.1.1 port 47088 ssh2: ED25519 SHA256:*****
Dec 04 20:22:36 rpi02w sshd[13198]: pam_unix(sshd:session): session opened for user S****(uid=1000) by (uid=0)
Dec 04 20:22:36 rpi02w systemd-logind[440]: New session 94 of user S****.
Dec 04 20:22:36 rpi02w systemd[1]: Started session-94.scope - Session 94 of User S****.
Dec 04 20:22:36 rpi02w sshd[13198]: pam_env(sshd:session): deprecated reading of user environment enabled
Dec 04 20:22:51 rpi02w sshd[13206]: Received disconnect from 192.168.1.1 port 47088:11:
```

pelwell, 2023-12-05:
Can you add "brcmfmac.debug=0x100000" to /boot/firmware/cmdline.txt, reboot and continue your testing? The WiFi firmware should put some diagnostic output into the kernel log, which you can read using "dmesg | grep brcmfmac".

llllllllqq, 2023-12-05:
> Can you add "brcmfmac.debug=0x100000" to /boot/firmware/cmdline.txt, reboot and continue your testing? The WiFi firmware should put some diagnostic output into the kernel log, which you can read using "dmesg | grep brcmfmac".

Ok, thank you.

llllllllqq, 2023-12-05:
With `brcmfmac.debug=0x100000` and `over_voltage=2`, bugs occurred again.

What I was running:
stress -c 2 -i 1 -d 1 (stress the device and the USB to see if undervolt, but not: `throttled=0x0`)
curl download a big file (debian x86_64 iso) to /dev/null (to stress the wifi chip)
use "foldersync" upload 2 big files (202MB & 710MB) to USB disk, without manually cancelling 

When uploaded these 2 files, dmesg showed:

```
[ 2133.914638] mmc1: Controller never released inhibit bit(s).
[ 2133.914786] brcmfmac: mmc_submit_one: CMD53 sg block read failed -5
[ 2133.914820] brcmfmac: brcmf_sdio_rxglom: glom read of 7680 bytes failed: -5
[ 2133.914849] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[ 2287.343373] mmc1: Controller never released inhibit bit(s).
[ 2287.343528] brcmfmac: mmc_submit_one: CMD53 sg block read failed -5
[ 2287.343565] brcmfmac: brcmf_sdio_rxglom: glom read of 8704 bytes failed: -5
[ 2287.343594] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[ 2287.363912] mmc1: Controller never released inhibit bit(s).
[ 2287.364058] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -5
[ 2287.364082] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
[ 2287.366145] brcmfmac: brcmf_sdio_rxglom: descriptor len 6 bad: 0
[ 2287.366260] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[ 2287.366274] brcmfmac: brcmf_sdio_rxfail: terminate frame
[ 2287.367196] brcmfmac: brcmf_sdio_rxglom: descriptor len 9 bad: 0
[ 2287.367314] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[ 2287.367328] brcmfmac: brcmf_sdio_rxfail: terminate frame
```

Note: now it is volt=1.3062V and throttled=0x0

llllllllqq, 2023-12-05:
With brcmfmac.debug=0x100000 and no over_voltage, not using "foldersync" app, but using a standard sftp cli in "OpenSSH_9.5p1, OpenSSL 3.0.12 24 Oct 2023" bugs occurred again *2.

What I was running:
stress -c 1 -i 1 -d 1 (note: I changed to -c 1 to make CPU temp below 80'C)
curl download a big file (debian x86_64 iso) to /dev/null
use sftp cli to upload 1 big files (4000MB data generated by /dev/urandom) to USB disk, without manually cancelling

Then dmesg showed (wifi was not lost, only showed errors):
```
[ 1417.607957] mmc1: Controller never released inhibit bit(s).
[ 1417.608109] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -5
[ 1417.608132] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
[ 1417.610137] brcmfmac: brcmf_sdio_rxglom: descriptor len 5 bad: 0
[ 1417.610237] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[ 1417.610251] brcmfmac: brcmf_sdio_rxfail: terminate frame
[ 1417.612039] brcmfmac: brcmf_sdio_rxglom: descriptor len 6 bad: 0
[ 1417.612218] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[ 1417.612233] brcmfmac: brcmf_sdio_rxfail: terminate frame
```

llllllllqq, 2023-12-06:
Tested with
arm_freq=800
gpu_freq=300
sdram_freq=400
but bugs still exist. All other procedures are same as the penultimate comment above.
```
[36957.150360] mmc1: Controller never released inhibit bit(s).
[36957.150673] brcmfmac: mmc_submit_one: CMD53 sg block read failed -5
[36957.150736] brcmfmac: brcmf_sdio_rxglom: glom read of 4608 bytes failed: -5
[36957.150771] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[36957.151270] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[36957.151291] brcmfmac: brcmf_sdio_rxfail: terminate frame
```

pelwell, 2023-12-06:
That's a useful datapoint. We don't recommend 64-bit on Zero 2 W because of the memory constraints, but that doesn't mean the WiFi shouldn't work.

llllllllqq, 2023-12-11:
> It seems to be an arm64 only bug. I backuped the sdcard and fresh install an armv7 system, and install nearly all same software as in previous arm64 system; after 4 hours testing, no bugs happened. New system info:
> 
> ```
> Raspberry Pi reference 2023-10-10
> Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, fb56ad562991cf3ae5c96ab50983e1deeaefc7b6, stage2
> 
> Oct 17 2023 15:42:53
> Copyright (c) 2012 Broadcom
> version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start_cd)
> 
> Linux rpi02w 6.1.0-rpi6-rpi-v7 #1 SMP Raspbian 1:6.1.58-1+rpt2 (2023-10-27) armv7l GNU/Linux
> ```
> 
> I'll test more with it.

This system works fine during the past 5 days, and the newst 6.1.63 kernel also works fine.

Today I switched to a arm64 kernel (Linux rpi02w 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux) by adding `arm_64bit=1` config (but with armhf packages), and it seems still working fine after a 2 hours test.
I noticed my wifi router perform an auto-select channel, and the only channel number I remember during the bugs is 4 (+8 due to HT40). I will test more with more channels/HT20/arm64 kernel over armhf packages/armhf on armhf.

small question: I wonder whether arm64 kernel loads the same drivers/firmwares as armhf kernel, and whether arm64 kernel consumes more power which may cause the bugs.
note: on armhf kernel, I've tested my power supply can server a 5v1A 2.5' USB HDD well, so maybe it's unlikely to be a power issue?

llllllllqq, 2023-12-12:
More strange test results:

When I use `sudo dd if=/dev/zero of=~/test bs=4M ` to stress my SD card (as system root), the status LED of my USB SATA TO SSD adapter is OFF. If I access it like `ls /mnt/myssd`, the LED often will turn on and `ls` work, but sometimes it will be stuck for a long time till working, and dmesg showing:
```
[  218.709981] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  219.374781] ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110
[  280.949737] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  291.509856] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  295.524233] ieee80211 phy0: brcmf_fill_bss_param: Failed to get bss info (-110)
[  315.030205] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  316.732193] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
```
It looks like the bugs mentioned in this issue, so I posted here.

pelwell, 2023-12-12:
How is your SATA drive and adaptor powered? The Zero 2 has a very limited capacity to power external hardware.

llllllllqq, 2023-12-12:
> How is your SATA drive and adaptor powered? The Zero 2 has a very limited capacity to power external hardware.

Directly from the pi. But I've tested with a USB meter, this adapter + SSD will only comsume 5V 300ma on average and 5V 501ma on peak, and they can pass a very long stress test on pi.

pelwell, 2023-12-12:
> they can pass a very long stress test on pi.

And yet you are getting SDIO problems, which doesn't sound like a pass to me. Your SSD may be able to limp along, but another part of the system is suffering.

If you insist on trying to power an external device from the Zero 2 while the rest of the system is busy then this is what you are going to get. Undervoltage (on at least some part of the board) is not something that can be fixed in software.

llllllllqq, 2023-12-12:
> And yet you are getting SDIO problems, which doesn't sound like a pass to me. Your SSD may be able to limp along, but another part of the system is suffering.
>
> If you insist on trying to power an [...]

Thanks for your advice, I'll use a USB flash drive instead to test this issue further.

llllllllqq, 2023-12-13:
Arm64 kernel on armhf packages with only a USB flash disk (sandisk CZ48, directly powered, max 5V 147ma, avg 5V 80ma), dmesg is like those of many tests before:
```
[ 2371.595553] mmc1: Controller never released inhibit bit(s).                                      
[ 2371.595669] brcmfmac: mmc_submit_one: CMD53 sg block read failed -5                              
[ 2371.595700] brcmfmac: brcmf_sdio_rxglom: glom read of 2048 bytes failed: -5                      
[ 2371.595726] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
```
I noticed that if I let mmc0&mmc1 (or saying SDIO?) busy, this bug seems to occur more frequently.

llllllllqq, 2023-12-15:
I tested `rpi-update next`, and `Linux rpi02w 6.6.5-v8+ #1706 SMP PREEMPT Fri Dec  8 18:43:36 GMT 2023 aarch64 GNU/Linux` still has this bug:
```
[ 1675.506651] mmc1: Controller never released inhibit bit(s).
[ 1675.507753] brcmfmac: brcmf_sdio_readframes: read 1525 bytes from channel 2 failed: -5
[ 1675.507806] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
```

llllllllqq, 2023-12-17:
[iperf.txt](https://github.com/raspberrypi/linux/files/13694641/iperf.txt)
[iperf-r.txt](https://github.com/raspberrypi/linux/files/13694642/iperf-r.txt)

I find the pi has a very high tcp package retry.
On a latest fresh raspbian, both 6.1.63 armhf and 6.1.63 arm64 kernel have this bug.
I tested by `iperf3 -s` on pi connected to 2.4ghz wifi of my router, `iperf3 -c pi_ip -d` on my laptop/android phone/pc connected to 5ghz_wifi /rj45 of my router. As iperf.txt shows, there are 1820 retries in 10s.
I also tested iperf3 on the permutation and combination of laptop/phone/pc on 2.4g/5g/rj45, but no error found, so it's not likely my router's fault.

Does this means my pi's wifi module is broken and related to the bugs in this issue?

llllllllqq, 2023-12-23:
Today, I tried the latest ubuntu server arm64, and the bug still exists. The pi was USB boot on `Linux version 6.5.0-1008-raspi (buildd@bos02-arm64-062) (aarch64-linux-gnu-gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.41) #11-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov 22 19:08:26 UTC 2023 (Ubuntu 6.5.0-1008.11-raspi 6.5.3)` . **No sdcard was inserted (I just wanted to check whether my sdcard is broken that leads to the bug.)** It was a fresh install without non-default software. 

I noticed [PI 4 had similar bugs](https://forums.raspberrypi.com/viewtopic.php?t=309505). 
I also noticed [some people](https://groups.google.com/g/linux-sunxi/c/v6Ktt8lAnw0/m/EKyKHan3H5wJ) also encountered similar bugs when use ssh/scp/sftp on BCM wifi chip on sdio, but they were on other CPU (sunxi).
More links here, I think they might help a little [1](https://groups.google.com/g/linux.debian.kernel/c/UmOltBVxlaA) [2](https://groups.google.com/g/linux-sunxi/c/-GejF5r-HGg) [3](https://groups.google.com/g/linux-sunxi/c/NngIhT-ELVc/m/Q2pt3UdrE-sJ) [4](https://community.nxp.com/t5/i-MX-Processors/Murata-SDIO-wireless-imx8mm-and-Yocto/m-p/918178)

Logs are here. (Sorry, I forgot added brcmfmac.debug=0x100000 to cmdline, but I think it's same as comments above)
```
[  718.588249] mmc1: Controller never released inhibit bit(s).
[  718.594160] brcmfmac: brcmf_sdio_readframes: read 1525 bytes from channel 2 failed: -5
[  718.604209] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[  756.336904] mmc1: Controller never released inhibit bit(s).
[  756.342839] brcmfmac: brcmf_sdio_readframes: read 1525 bytes from channel 2 failed: -5
[  756.351533] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[  762.385556] mmc1: Controller never released inhibit bit(s).
[  762.391492] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -5
[  762.398217] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
[  762.408111] brcmfmac: brcmf_sdio_rxglom: descriptor len 5 bad: 0
[  762.414791] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[  762.421346] brcmfmac: brcmf_sdio_rxfail: terminate frame
[  762.427640] brcmfmac: brcmf_sdio_hdparse: HW superframe header length error
[  762.434803] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[  762.442047] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[  762.448494] brcmfmac: brcmf_sdio_rxfail: terminate frame
[  792.068261] brcmfmac: brcmf_sdio_isr: failed backplane access
[  792.077789] brcmfmac: brcmf_sdio_isr: failed backplane access
[  792.100223] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
[  792.106760] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
[  792.117648] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[  792.125158] brcmfmac: brcmf_sdio_rxfail: terminate frame
[  792.133482] brcmfmac: brcmf_sdio_rxglom: descriptor len 3 bad: 0
[  792.140738] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[  792.147968] brcmfmac: brcmf_sdio_rxfail: terminate frame
[  792.155987] brcmfmac: brcmf_sdio_rxglom: descriptor len 4 bad: 0
[  792.163415] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[  792.170632] brcmfmac: brcmf_sdio_rxfail: terminate frame
```

pelwell, 2023-12-23:
I still have the problem what this is an error I have never encountered, to the extent that a hardware issue seems most likely. Without a means to reproduce it there is little we can do.

llllllllqq, 2023-12-23:
> I still have the problem what this is an error I have never encountered, to the extent that a hardware issue seems most likely. Without a means to reproduce it there is little we can do.

Thanks for your help and comments. I'll ask the seller for a new device, if they can, I'll test the bug with new device. Thanks again.

llllllllqq, 2024-01-08:
I got a new zero2w, and used the same sdcard (armhf raspberry os with arm_64bit=1) for testing, and bugs still exist, so it's not likely a hardware issue.

```
[ 1194.501692] mmc1: Controller never released inhibit bit(s).
[ 1194.501830] brcmfmac: mmc_submit_one: CMD53 sg block read failed -5
[ 1194.501863] brcmfmac: brcmf_sdio_rxglom: glom read of 9216 bytes failed: -5
[ 1194.501889] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[ 1194.502197] brcmfmac: brcmf_sdio_hdparse: HW header length too long
[ 1194.502212] brcmfmac: brcmf_sdio_rxfail: terminate frame
```

The only difference between the two devices I find is default volt. This new device has a higher value 1.2938v, with higher temp when testing.
```
temp=76.8'C
volt=1.2938V
throttled=0x0
```

I'll switch to armhf kernel to see whether bugs exist on the new device with armhf kernel.

nbuchwitz, 2024-01-27:
I see the same error (`mmc1: Controller never released inhibit bit(s).`) on a CM4 based device (RevPi Connect 4) without any particular load / io ops. 

```
[...]
[53007.437237] mmc1: Controller never released inhibit bit(s).
[53007.437307] brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5
[53007.437317] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
[83805.688959] mmc1: Controller never released inhibit bit(s).
[83805.688997] brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5
[83805.689007] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
[145912.200807] mmc1: Controller never released inhibit bit(s).
[145912.200844] brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5
[145912.200854] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
```

The device is running with `6.1.46-rt13-v8`, but does also occur with the rpi stock kernel. I'll check another device and will have a look how this can be reproduced easily. @JamesH65 should have a Connect 4 at hand (from last visit)

llllllllqq, 2024-01-28:
> I see the same error (`mmc1: Controller never released inhibit bit(s).`) on a CM4 (booting from eMMC) based device (RevPi Connect 4) without any particular load / io ops.
> 
> ```
> [...]
> [53007.437237] mmc1: Controller never released inhibit bit(s).
> [53007.437307] brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5
> [53007.437317] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
> [83805.688959] mmc1: Controller never released inhibit bit(s).
> [83805.688997] brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5
> [83805.689007] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
> [145912.200807] mmc1: Controller never released inhibit bit(s).
> [145912.200844] brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5
> [145912.200854] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
> ```
> 
> The device is running with `6.1.46-rt13-v8`, but does also occur with the rpi stock kernel. I'll check another device and will have a look how this can be reproduced easily. @JamesH65 should have a Connect 4 at hand (from last visit)

Hi, to reproduce, you could try ssh to upload a huge file to the pi (via wifi), such as scp local:/dev/urandom to pi:/dev/null. If not lucky, stress wifi by iperf may help.

llllllllqq, 2024-01-31:
Today I buy a new wifi router, wifi7 xiaomi be3600. It's IPQ5xxx CPU, not my old MTK one, and bugs mentioned above still exist (even after I upgraded to 6.1.73 kernel), so I think it's not a router problem.

ceisserer, 2024-06-08:
I experience the same issues on my Raspberry PI 3. Some mmc1 error starting with the "never released inhibit bits" message - running raspberry pi os bookworm 64-bit. However these issues only started to appear after I upgraded from kernel 6.1.0 to 6.6.31 (the machine was running 24/7 for months with 6.1.0 and never reported anything like this - now I get those message every 2-3 days).

Should I worry? Do those errors have the tendency to corrupt my data (mmc) or is it actually about wifi only?

[286922.330014] mmc1: Controller never released inhibit bit(s).
[286922.330232] brcmfmac: mmc_submit_one: CMD53 sg block read failed -5
[286922.330254] brcmfmac: brcmf_sdio_rxglom: glom read of 1536 bytes failed: -5
[286922.330268] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame

[46487.280532] mmc1: Controller never released inhibit bit(s).
[46487.280786] brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5
[46487.280811] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK

aakoskin, 2024-06-28:
I see the bug few times a day on 64-bit Raspberry Pi 3 Model B (kernel 6.6.31+rpt-rpi-v8):

[310243.327942] mmc1: Controller never released inhibit bit(s).
[310243.328371] brcmfmac: brcmf_sdio_readframes: read 1476 bytes from channel 2 failed: -5
[310243.328598] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame

Other than journal spam, it doesn't cause any permanent issue (wifi always recovers) and otherwise the system is running fine with days/week uptime.

HW setup:
Raspberry Pi 3 Model B
Official 7" touchscreen
Official PSU
2 x USB flash stick (rootfs on USB), keyboard & mouse

No undervoltage warnings seen. vcgencmd get_throttled reports that frequency capping has occured at some point (system has 3 days uptime).

aakoskin, 2024-06-29:
The error "mmc1: Controller never released inhibit bit(s)" could be related to frequency capping:

- did a reboot and system was running fine for hours without any error logs and  "vcgencmd get_throttled" was returning 0.

- once I saw the error, get_throttled was reporting 0x20002 and a bit later 0x20000.

Maybe the 10 ms timeout goes somehow wrong when frequency is modified/throttled? My Pi is passively cooled and there's a heat wave currently so overheating is possible.

If it was a hardware/power issue, then why only mmc1/WiFi would be affected? I'm typing this from a system with those error logs and everything seem to work.

wiedehopf, 2024-10-09:
One more data point with DEBUG enabled for the driver:
(over_voltage=2, RPi 3B)

```
[   88.010400] mmc1: Got data interrupt 0x00000020 even though no data operation was in progress.
[   88.010433] mmc-bcm2835: =========== REGISTER DUMP (mmc1)===========
[   88.010439] mmc-bcm2835: Sys addr: 0x003ed45c | Version:  0x00009902
[   88.010446] mmc-bcm2835: Blk size: 0x00007040 | Blk cnt:  0x00000001
[   88.010453] mmc-bcm2835: Argument: 0x21000040 | Trn mode: 0x00000012
[   88.010460] mmc-bcm2835: Present:  0x01ff0a02 | Host ctl: 0x00000002
[   88.010467] mmc-bcm2835: Power:    0x0000000f | Blk gap:  0x00000000
[   88.010474] mmc-bcm2835: Wake-up:  0x00000000 | Clock:    0x00000207
[   88.010481] mmc-bcm2835: Timeout:  0x0000000e | Int stat: 0x00000000
[   88.010487] mmc-bcm2835: Int enab: 0x00ff0133 | Sig enab: 0x00ff0133
[   88.010494] mmc-bcm2835: AC12 err: 0x00000000 | Slot int: 0x00000000
[   88.010500] mmc-bcm2835: Caps:     0x00000000 | Caps_1:   0x00000000
[   88.010507] mmc-bcm2835: Cmd:      0x0000353a | Max curr: 0x00000001
[   88.010513] mmc-bcm2835: Host ctl2: 0x00000000
[   88.010519] mmc-bcm2835: ===========================================
[   88.020674] mmc1: Controller never released inhibit bit(s).
[   88.026284] mmc-bcm2835: =========== REGISTER DUMP (mmc1)===========
[   88.026296] mmc-bcm2835: Sys addr: 0x003ed45c | Version:  0x00009902
[   88.026305] mmc-bcm2835: Blk size: 0x00007040 | Blk cnt:  0x00000001
[   88.026312] mmc-bcm2835: Argument: 0x21000040 | Trn mode: 0x00000012
[   88.026319] mmc-bcm2835: Present:  0x01df0a02 | Host ctl: 0x00000002
[   88.026326] mmc-bcm2835: Power:    0x0000000f | Blk gap:  0x00000000
[   88.026332] mmc-bcm2835: Wake-up:  0x00000000 | Clock:    0x00000207
[   88.026339] mmc-bcm2835: Timeout:  0x0000000e | Int stat: 0x00000000
[   88.026346] mmc-bcm2835: Int enab: 0x00ff0133 | Sig enab: 0x00ff0133
[   88.026353] mmc-bcm2835: AC12 err: 0x00000000 | Slot int: 0x00000000
[   88.026359] mmc-bcm2835: Caps:     0x00000000 | Caps_1:   0x00000000
[   88.026366] mmc-bcm2835: Cmd:      0x0000353a | Max curr: 0x00000001
[   88.026373] mmc-bcm2835: Host ctl2: 0x00000000
[   88.026378] mmc-bcm2835: ===========================================
[   88.026902] brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5
[   88.034632] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
```

This seems to cause a USB short disruption as well (running rtl-sdr ADS-B reception with MLAT loses sync which happens when data on USB is lost).

There is a bunch of these as well but they are probably not related:
```
[   13.023973] mmc1: Got data interrupt 0x00000002 even though no data operation was in progress.
[   13.023997] mmc-bcm2835: =========== REGISTER DUMP (mmc1)===========
[   13.024003] mmc-bcm2835: Sys addr: 0x00061828 | Version:  0x00009902
[   13.024010] mmc-bcm2835: Blk size: 0x00007040 | Blk cnt:  0x00000000
[   13.024017] mmc-bcm2835: Argument: 0x1d200020 | Trn mode: 0x00000032
[   13.024024] mmc-bcm2835: Present:  0x01ff0000 | Host ctl: 0x00000002
[   13.024031] mmc-bcm2835: Power:    0x0000000f | Blk gap:  0x00000000
[   13.024037] mmc-bcm2835: Wake-up:  0x00000000 | Clock:    0x00000207
[   13.024044] mmc-bcm2835: Timeout:  0x0000000e | Int stat: 0x00000000
[   13.024051] mmc-bcm2835: Int enab: 0x02ff000b | Sig enab: 0x02ff000b
[   13.024057] mmc-bcm2835: AC12 err: 0x00000000 | Slot int: 0x00000000
[   13.024064] mmc-bcm2835: Caps:     0x00000000 | Caps_1:   0x00000000
[   13.024070] mmc-bcm2835: Cmd:      0x0000353a | Max curr: 0x00000001
[   13.024077] mmc-bcm2835: Host ctl2: 0x00000000
[   13.024082] mmc-bcm2835: ===========================================
```

Attached the entire dmesg:
[never_released.log](https://github.com/user-attachments/files/17309075/never_released.log)

I've seen this error for several other people running rtl-sdr via USB other pi3 / pi zero 2 w

So the controller sends the interrupt SDHCI_INT_DATA_AVAIL and that's not handled which causes all the following errors it would seem?

redm123, 2025-04-05:
Hi! Just stumbled over this report ... I see the same errors on a Zero 2 W, and it's kinda annoying. 
I don't do any fancy stuff with it, just playing some music via shairport / AirPlay. And when doing this every few seconds I see this messages appearing in `dmesg`.

```
[Fri Apr  4 12:27:42 2025] mmc1: Controller never released inhibit bit(s).
[Fri Apr  4 12:27:42 2025] brcmfmac: brcmf_sdio_readframes: read 1295 bytes from channel 2 failed: -5
[Fri Apr  4 12:27:42 2025] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[Fri Apr  4 12:27:42 2025] brcmfmac: brcmf_sdio_readframes: brcmf_sdio_readframes: glom superframe w/o descriptor!
[Fri Apr  4 12:27:42 2025] brcmfmac: brcmf_sdio_rxfail: terminate frame

[Fri Apr  4 12:27:46 2025] mmc1: Controller never released inhibit bit(s).
[Fri Apr  4 12:27:46 2025] brcmfmac: brcmf_sdio_readframes: read 1402 bytes from channel 2 failed: -5
[Fri Apr  4 12:27:46 2025] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
```

Every now and then I get audio dropouts and then also pinging the devices fails a few times, before it recovers. That's my actual problem here, which led me to trying to debug this. (I'm actually not 100% sure it's related, but there seems a strong correlation to the errors appearing and not appearing...)

I run the Moode Audio distribution. And everything works fine with the Bullseye based releases, 32bit or 64bit. But starting with Bookworm based releases (only 64bit) this problem started to occur. It only happened with playing AirPlay so far. Like not if I play internet streams, or from local NAS. There is also not much CPU load or anything. Also running the afore mentioned `dd if=/dev/zero of=./test.log bs=4M` does not trigger the error.

And it's pretty reproducible, if I go back and forth between the releases. So I kinda would rule out a hardware issue.

The two top processes when playing music via AirPlay are:

```
  1300 root      20   0 1162428  30208  21632 S  14.9   7.1   0:08.08 shairport-sync
    332 root       0 -20       0      0      0 I   2.0   0.0   0:01.87 kworker/u13:1-brcmf_wq/mmc1:0001:1
```
What is that worker doing? Does this mean it's doing disk IO (mmc1)??

only4comments, 2025-04-09:
I confirm this issue and I can reproduce it reliably. 

Running the latest version of Raspberry OS Bookworm 64-bit. 
`Linux rpi 6.12.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.20-1+rpt1~bpo12+1 (2025-03-19) aarch64 GNU/Linux`

HW: 
Raspberry Pi Zero 2 W (with heatsink)
LTE + WIFI modem (connected via RNDIS to USB OTG), power consumption of the modem is ~300 mW (measured with USB power meter)

Power source typically: powerbank (up to 5V@4,5A) / via USB 3.0 hub supporting QC (managed to load it by 20 W just fine, voltage still keeping at 5 V at the powerbank output; voltage on the ports of the hub under load does not drop below 4,9 V)
Another power source tested: USB QuickCharge port of a docking station (this one has a more strict limit of total power output)
No impact.

Never seen a error concerning undervoltage or power-caused throttling in dmesg. 
Never had any stability issues under high CPU load. 

I have `over_voltage=6` in `/boot/firmware/config.txt`  (this is on purpose, to generate slightly more heat in the box, to keep powerbanks on a more comfort temperature).

Usage scenario: 
The RPi is used outdoors in a plastic box, powered by a powerbank and providing a remote control access to a camera (connected via the SoC Broadcom Wi-Fi). Distance .... up to 2 meters typically. 
There is enough available channels in the air (the only second AP in the location is the LTE / Wi-Fi modem, perhaps 20 cm away from the RPi). No other possible sources of interference. 

Problem: 
Wi-Fi connection from the onboard Broadcom is lost. Sometimes after an hour, sometimes after two hours, rarely it can stay connected for a whole night without interruptions. I did not manage these issues to have anything in common. 

Typical errors related to the disconnect: 
`23:46:13 rpi wpa_supplicant[521]: wlan0: CTRL-EVENT-DISCONNECTED bssid=1e:99:4c:93:6f:f8 reason=0 locally_generated=1`

While looking around the logs, even historical ones, I noticed this appearing some time prior to the disconnect: 
```
22:32:27 rpi kernel: mmc1: Controller never released inhibit bit(s).
22:32:27 rpi kernel: brcmfmac: brcmf_sdio_readframes: read 1532 bytes from channel 2 failed: -5
22:32:27 rpi kernel: brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame

after some time: 

23:46:13 rpi wpa_supplicant[521]: wlan0: CTRL-EVENT-DISCONNECTED bssid=1e:99:4c:93:6f:f8 reason=0 locally_generated=1
23:46:13 rpi wpa_supplicant[521]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
23:46:13 rpi wpa_supplicant[521]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=CZ
23:46:13 rpi NetworkManager[518]: <info>  [1741473973.6917] device (wlan0): supplicant interface state: completed -> disconnected
23:46:13 rpi NetworkManager[518]: <info>  [1741473973.7395] device (p2p-dev-wlan0): supplicant management interface state: completed -> disconnected
23:46:13 rpi NetworkManager[518]: <info>  [1741473973.7452] device (wlan0): supplicant interface state: disconnected -> scanning
23:46:13 rpi NetworkManager[518]: <info>  [1741473973.7453] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
23:46:25 rpi wpa_supplicant[521]: wlan0: Failed to initiate sched scan
23:46:28 rpi NetworkManager[518]: <warn>  [1741473988.7408] device (wlan0): link timed out.
```

Note the `reason=0` in the disconnect message... There is no meaning of the reason=0 to be found, this one in particular appears to be "reserved", while other reasons are clearly defined. See https://web.archive.org/web/20231211124800/https://aboutcher.co.uk/2012/07/linux-wifi-deauthenticated-reason-codes/ for a reference.

So I am tempted to blame not the camera (serving as a Wi-Fi AP), as this would have returned a different reason code. 

Having done some search, I have found this post from @pelwell https://github.com/raspberrypi/firmware/issues/1723#issuecomment-2621280178 suggesting that it is a matter of power source. 

My measurements: 
While connected through a hub (with other devices drawing a lot of power from other ports of the hub), running cpuburn-a53 at 1200 MHz with over_voltage=6, the measured values are 1,6A @ 4,9 V. No errors about undervoltage in dmesg. Without the overclock and with over_voltage=3, the power draw is much lower and the measured voltage is around 4,95 V. 

The cpuburn-a53 appears to be eating as much power as possible and in real usage scenario, this situation does not happen at all. 

I can make these errors to pop up in the log, reliably, 100% chance. In my case it is enough to run a long transfer over Wi-Fi (in this case - downloading a 4 GB ISO via Wi-Fi to the SD card). RPi being the only device connected to the powerbank, voltage being stable slightly above 5 V:
 

```
[  220.703811] mmc1: Controller never released inhibit bit(s).
[  220.703939] brcmfmac: mmc_submit_one: CMD53 sg block read failed -5
[  220.703962] brcmfmac: brcmf_sdio_rxglom: glom read of 4608 bytes failed: -5
[  220.704009] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
[  397.848342] mmc1: Controller never released inhibit bit(s).
[  397.848460] brcmfmac: mmc_submit_one: CMD53 sg block read failed -5
[  397.848479] brcmfmac: brcmf_sdio_rxglom: glom read of 6144 bytes failed: -5
[  397.848492] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame
```

It requires the transfer speed to be rather high (around 3 MB/s). When the transfer speed is below 1 MB/s, the error does not appear. 

Surprisingly, the download can even complete, despite the sdio-related errors. Not sure if my disconnects from the camera were in a direct relation to these "inhibit bits" and sdio errors in the past, or totally independent.

n3vu0r, 2025-05-06:
Once a week, I see the same errors like @only4comments on a 64-bit OS which results in Wi-Fi loss until power cycling. I'm doing nothing intense with the PI, it just reads sensors two times per minute and sends the data via Wi-Fi (some 20 bytes). The Raspberry Pi ZERO 2W is connected to an official power supply (I can look it up). The power supply is connected to a big UPS. The Raspberry powers three sensors over its GPIOs: DPS310, SCD41, SPS30.

HarlockP4, 2026-09-04:
I can reproduce what appears to be the same issue on a Raspberry Pi Zero 2 W using a much newer Raspberry Pi kernel.

System:

Linux 6.12.96+rpt-rpi-v8 Debian 1:6.12.96-1+rpt1 kernel build date: 2026-07-24 

Integrated Wi-Fi:

BCM43430/1 Firmware: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7 

I have persistent kernel logging enabled and found at least 9 separate occurrences of:

mmc1: Controller never released inhibit bit(s). 

between 2026-08-20 and 2026-09-04.

Most of them are immediately followed by:

brcmfmac: brcmf_sdio_read_control: read 2048 control bytes failed: -5 brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK 

Some occurrences appear to recover without an observable Wi-Fi disconnection, which is similar to the second behaviour described in this issue.

However, I have also experienced severe system/network hangs.

During one particularly useful captured event on 2026-09-02, NetworkManager was reported blocked in uninterruptible D state for more than 120 seconds and subsequently for more than 241 seconds.

Its kernel stack contained:

__mmc_claim_host sdio_claim_host brcmf_sdio_readshared [brcmfmac] brcmf_sdio_checkdied [brcmfmac] brcmf_sdio_bus_rxctl [brcmfmac] brcmf_proto_bcdc_cmplt [brcmfmac] brcmf_proto_bcdc_query_dcmd [brcmfmac] brcmf_fil_cmd_data [brcmfmac] brcmf_cfg80211_dump_station [brcmfmac] nl80211_dump_station [cfg80211] 

At the same time another kernel worker was also blocked:

task:kworker/... state:D Workqueue: events sdio_irq_work __mmc_claim_host sdio_irq_work 

The event later produced:

ieee80211 phy0: brcmf_cfg80211_dump_station: BRCMF_C_GET_ASSOCLIST failed, err=-110 ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -52 

which looks particularly similar to the failure sequence shown in the original report.

During the same severe incident there were also repeated:

sdhost-bcm2835 3f202000.mmc: timeout waiting for hardware interrupt. 

and some additional processes entered D state while waiting on storage/ext4 I/O.

Therefore I cannot yet say whether the Wi-Fi firmware itself is the root cause, or whether this is a wider MMC/SDIO interaction, but brcmfmac/SDIO is clearly involved in the captured hang.

Other relevant information:

Wi-Fi power saving is disabled. Normal system status reports throttled=0x0. The issue occurs intermittently rather than under a reliably reproducible workload. The system is continuously monitored, so I can collect additional correlated events. The microSD block device (mmcblk0) and the Wi-Fi SDIO path (mmc1 / brcmfmac) are being monitored separately. 

This seems to show that the issue originally reported here is still reproducible on Raspberry Pi's 6.12 kernel in 2026.

I can provide the complete kernel log surrounding the severe event, the complete list of all 9 mmc1 occurrences, boot logs, and additional monitoring data if that would help investigation.