raspberrypi/linux #5783: WiringPi rendered inoperative by latest update!!!

ValGretchev, 2023-12-11:
### Describe the bug

The cpuinfo file in /proc directory was modified and no longer includes the Hardware : BCM2835 line. This causes WiringPi to exit as it can’t find the required information rendering all programs that use the I/O and peripherals inoperative.

### Steps to reproduce the behaviour

Raspberry Pi 4B
[proc-cpuinfo-old.txt](https://github.com/raspberrypi/linux/files/13637311/proc-cpuinfo-old.txt)
[proc-cpuinfo-new.txt](https://github.com/raspberrypi/linux/files/13637312/proc-cpuinfo-new.txt)

### Device (s)

Raspberry Pi 4 Mod. B

### System

Latest Raspberry Pi OS 64-bit

6by9, 2023-12-11:
WiringPi is deprecated by all those who supported it.

Gordon dropped it in 2019.

Those who created the fork at https://github.com/WiringPi/WiringPi put it on life-support, but have dropped it as of Oct 2023 with the comment
> This library was deprecated by its author in August 2019. As of 31st October 2023 nobody has shown an interest in properly maintaining it. Between this, and changes to GPIO in Rasberry Pi OS Bookworm and on the Raspberry Pi 5, this project is going nowhere. It has been archived to more clearly indicate this status.

It's not been shipped with Raspberry Pi OS since Jan 2022 due to it bit-rotting.

I would suggest you look into a library using [libgpio](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/) (eg libgpiod for C, or gpiozero for Python), because it is going to break even bigger time when 6.6 becomes the stable kernel (see #5668, https://github.com/WiringPi/WiringPi/issues/186)

closed 2023-12-11 by 6by9

ValGretchev, 2023-12-11:
> Is there a chance that WiringPi could be adapted to not rely on these modification to /proc/cpuinfo?

Perhaps it could but that job is beyond me. I just need to get my application done and move on. I am 82 years old, and time is a wasting.

6by9, 2023-12-11:
https://github.com/raspberrypi/bookworm-feedback/issues/129 so deliberate change in the kernel to bring it closer to mainline.

Looking at WiringPi, there is already a `#ifdef DONT_CARE_ANYMORE` around the check for the Hardware line, and it isn't defined. So this check hasn't been enabled for 5 years.
The comment that follows the ifdef would probably be appropriate in this situation - https://github.com/WiringPi/WiringPi/blob/master/wiringPi/wiringPi.c#L795-L813

ValGretchev, 2023-12-11:
I see it on line 782 and I will look into it. Thanks for continuing to offer a solution even though you have closed the post. I appreciate it.

Val Gretchev P.Eng.