raspberrypi/linux #5784: Pi 5: i2c: i2c_designware: controller timed out, requiring power cycle to fix

Gadgetoid, 2023-12-11:
### Describe the bug

When using i2c on either of the display connectors to drive an IO expander and touch, I'm getting frequent hard crashes of the i2c controller, requiring a full power cycle to fix:

```
[    6.593324] i2c_designware 1f00088000.i2c: controller timed out
[    6.593343] (NULL device *): regmap_read of register 0x1 failed: -ETIMEDOUT
```

### Steps to reproduce the behaviour

I don't have a minimum repro since I'm connecting and trying to talk to i2c hardware, normally I'd chalk this up as user-error but I'm concerned that a fault like this should be cleared on a soft reboot...

### Device (s)

Other

### System

```
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4
```

```
version c2da2ae7 (release) (embedded)
```

```
Linux pi4 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```

### Logs

Full dmesg output of a failure state:

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.1.0-rpi7-rpi-2712 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000018000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 14 pages/cpu s177448 r8192 d43736 u229376
[    0.000000] pcpu-alloc: s177448 r8192 d43736 u229376 alloc=14*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[... 537 lines cut ...]
[    7.151379] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    7.151389] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    7.151738] Bluetooth: hci0: BCM: Using default device address (43:45:c0:00:1f:ac)
[    7.450912] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    7.450923] Bluetooth: BNEP filters: protocol multicast
[    7.450933] Bluetooth: BNEP socket layer initialized
[    7.454315] Bluetooth: MGMT ver 1.22
[    7.463357] NET: Registered PF_ALG protocol family
[    7.512224] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[    7.512245] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[    7.515110] pps pps0: new PPS source ptp0
[    7.516916] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[    7.529509] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.496973] Bluetooth: RFCOMM TTY layer initialized
[    9.496986] Bluetooth: RFCOMM socket layer initialized
[    9.496996] Bluetooth: RFCOMM ver 1.11
[   12.365945] EXT4-fs (sdb2): mounted filesystem with ordered data mode. Quota mode: none.
[   12.896309] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   13.173343] FAT-fs (sdb1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   17.117551] platform backlight_lcd: deferred probe pending
[   17.117558] platform reg_panel@1: deferred probe pending
[   17.117560] platform reg_touch@1: deferred probe pending
[   17.117563] mipi-dsi 1f00118000.dsi.0: deferred probe pending
[   17.117565] i2c 6-005d: deferred probe pending
```

Gadgetoid, 2023-12-11:
This bug has just happened to me on 6.6.5 too.

pelwell, 2023-12-20:
After the first instance, before I had the logic analyser attached (running a simple program to mirror the levels of GPIOs 40&41 onto GPIOs 6&7), I can't get this to fail. Whether I load the overlay from config.txt or at runtime using "sudo dtoverlay hyperportal10" (*) doesn't seem to make a difference - it's still solid. The only errors I see are:
```
[    7.206607] Goodix-TS 4-005d: Error reading 1 bytes from 0x8140: -121
[    7.263835] Goodix-TS 4-005d: Invalid config (0, 0, 0), using defaults
```
The first of these is because the Goodix controller NAKs the first read, not surprising since it happens less than a millisecond after the reset is released. The retry that happens about 37ms later is successful.
I'm not sure about the second error yet.

It may also be worth saying that I see the backlight turn on, and a flickering band of largely white pixels about 6cm wide across one side (in landscape) of the display.

(*) The overlay needs some sanitisation before the kernel will accept it at runtime, but it ends up looking like this:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <&i2c0if>;
		__overlay__ {
			status = "okay";
		};
	};
		
	fragment@1 {
		target = <&i2c0mux>;
		__overlay__ {
			status = "okay";
		};
	};

	i2cbus: fragment@2 {
		target = <&i2c_csi_dsi>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			pca: pca@56 {
				compatible = "nxp,pca9685-pwm";
				reg = <0x56>;
				#pwm-cells = <2>;
				gpio-controller;
				#gpio-cells = <2>;
				status = "okay";

				gpio-line-names = "PCA_PWM0",
					"PCA_PWM1",
					"PCA_PWM2",
					"PCA_PWM3",
					"PCA_PWM4",
					"PCA_PWM5",
					"PCA_PWM6",
					"PCA_PWM7",
					"PCA_GPIO8_LCD_RST",
					"PCA_PWM9_LCD_BL",
					"PCA_GPIO10_TOUCH_RST",
					"PCA_PWM11",
					"PCA_PWM12",
					"PCA_PWM13",
					"PCA_PWM14",
					"PCA_PWM15";
			};

			gt9271@5d {
				compatible = "goodix,gt9271";
				reg = <0x5d>;
				VDDIO-supply = <&reg_touch>;
				//interrupt-parent = <&pca>;
				//reset-gpios = <&pca 10 0>;
			};
		};
	};

	fragment@3 {
		target-path = "/";
		__overlay__ {
			backlight_lcd: backlight_lcd {
				compatible = "pwm-backlight";
				brightness-levels = <0 4095>;
				num-interpolated-steps = <4096>;
				default-brightness-level = <4096>;
				pwms = <&pca 9 5000000>;
			};

			reg_touch: reg_touch {
				compatible = "regulator-fixed";
				regulator-name = "touch_reg";
				gpio = <&pca 10 0>;
				enable-active-high;
			};

			reg_panel: reg_panel {
				compatible = "regulator-fixed";
				regulator-name = "panel_reg";
				gpio = <&pca 8 0>;
				enable-active-high;
			};
		};
	};

	panel: fragment@4 {
		target = <&dsi1>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			port {
				dsi_out_port: endpoint {
					remote-endpoint = <&panel_dsi_in>;
				};
			};

			panel_dsi: panel@0 {
				compatible = "pimoroni,lcd10";

				reg = <0>;
				label = "hyperportal0";
				//reset-gpios = <&pca 8 1>;
				power-supply = <&reg_panel>;
				port {
					panel_dsi_in: endpoint {
						remote-endpoint = <&dsi_out_port>;
					};
				};
			};
		};
	};

	__overrides__ {
		dsi0 = <&panel>, "target:0=",<&dsi0>,
			<&i2cbus>, "target:0=",<&i2c_csi_dsi0>;
	};
};
```

pelwell, 2023-12-20:
And then it failed hard, with SCL high and SDA held low:
```
pi@raspberrypi:~$ i2cdetect -y 4
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         [  106.864317] i2c_designware 1f00080000.i2c: controller timed out
-- [  107.888296] i2c_designware 1f00080000.i2c: controller timed out
-- [  108.912255] i2c_designware 1f00080000.i2c: controller timed out
-- [  109.936233] i2c_designware 1f00080000.i2c: controller timed out
-- [  110.960209] i2c_designware 1f00080000.i2c: controller timed out
-- [  111.984188] i2c_designware 1f00080000.i2c: controller timed out
-- [  113.008155] i2c_designware 1f00080000.i2c: controller timed out
-- [  114.032134] i2c_designware 1f00080000.i2c: controller timed out
--
10: [  115.056110] i2c_designware 1f00080000.i2c: controller timed out
-- [  116.080086] i2c_designware 1f00080000.i2c: controller timed out
-- [  117.104065] i2c_designware 1f00080000.i2c: controller timed out
-- [  118.128058] i2c_designware 1f00080000.i2c: controller timed out
-- [  119.152032] i2c_designware 1f00080000.i2c: controller timed out
-- [  120.176012] i2c_designware 1f00080000.i2c: controller timed out
-- [  121.199983] i2c_designware 1f00080000.i2c: controller timed out
-- [  122.223957] i2c_designware 1f00080000.i2c: controller timed out
-- [  123.247940] i2c_designware 1f00080000.i2c: controller timed out
-- [  124.271916] i2c_designware 1f00080000.i2c: controller timed out
-- [  125.295901] i2c_designware 1f00080000.i2c: controller timed out
-- [  126.319881] i2c_designware 1f00080000.i2c: controller timed out
-- [  127.343857] i2c_designware 1f00080000.i2c: controller timed out
-- [  128.367843] i2c_designware 1f00080000.i2c: controller timed out
-- [  129.391824] i2c_designware 1f00080000.i2c: controller timed out
-- [  130.415804] i2c_designware 1f00080000.i2c: controller timed out
--
```
Until I flipped the catch on the touchscreen cable and it sprang into life:
```
20: 20 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- 56 -- -- -- -- -- -- 5d -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberrypi:~$ i2cdetect -y 4
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- 56 -- -- -- -- -- -- 5d -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
In other words, I think the touchscreen controller (or perhaps the PWM/GPIO chip) is holding the SDA line low and causing the bus to malfunction. It's hard to tell which, since they are both on the other end of an I2C bus and my bucket has a hole.

mozcelikors, 2024-02-27:
@Gadgetoid Hi, I have a similar issue with i2c_designware on Pi 5 when plugging in a Waveshare 4.3inch Capacitive on a custom kernel 6.1.77 I built from here. I was wondering if you had any update on this.

This doesn't happen on Raspberry Pi OS Bookworm, but only happens when I custom compile the kernel. 
Issue can be viewed here: https://forums.raspberrypi.com/viewtopic.php?t=366281

@pelwell  I also saw contributions by you, Phil, regarding these topics, it would be great to have some insight on the issue I am having, which seems similar here. SCL held high and SDA held low.

Help is greately appreciated. Thanks

pelwell, 2024-02-28:
You're asking me to guess what you have got wrong in your configuration, in a forum which is intended for reporting problems in our kernel, not for general help requests.

You need to put a logic analyser on the I2C lines to see what is going wrong, which may then help you figure out why. But if you are getting the same timeout errors as above then it's likely that the device is holding one or both of the lines low when it shouldn't.

mozcelikors, 2024-02-28:
@pelwell Okay, thanks regardless. Will do.
I understand that this might be the wrong place to ask this, but I have seen many similar questions asked&answered, so I took a shot. Not a problem. Not the first time I received such attitude in this repo :)

JamesH65, 2024-02-28:
> @pelwell Okay, thanks regardless. Will do. I understand that this might be the wrong place to ask this, but I have seen many similar questions asked&answered, so I took a shot. Not a problem. Not the first time I received such attitude in this repo :)

For general help requests, the forums are the right place to ask. forums.raspberrypi.com. You are likely to get a faster response as well, since there are more people there with knowledge than hang around here.

mozcelikors, 2024-03-03:
Problem is fixed when Linux kernel is downgraded to 6.1.63 (official Bookworm kernel version that worked). I guess the fixes in i2c_designware that was introduced after 6.1.63 regarding I2C timings doesn't really get along with some of the devices, causing "i2c timed out" issues.
I haven't tested with official display, but this is the case for Waveshare display. I hope this helps others having the same issue.
Cheers.

steveiliop56, 2024-03-08:
I have the exact same problem with the 4.3 inch screen from waveshare. Most of the times it works but sometimes the display will not show anything and I will just keep getting these i2c timeouts.

pelwell, 2024-03-08:
As I said above:
> it's likely that the device is holding one or both of the lines low when it shouldn't.

If you are also on a Pi 5 then you can check the state of the I2C lines with:
```
pi@raspberrypi:~$ pinctrl 38-41
38: ip    pd | hi // CD0_SDA/GPIO38 = input
39: ip    pd | hi // CD0_SCL/GPIO39 = input
40: a2    pu | hi // CD1_SDA/GPIO40 = SDA4
41: a2    pu | hi // CD1_SCL/GPIO41 = SCL4
```
Run this command next time the display doesn't work. One of the pairs - probably 40&41 - will be SDA and SCL, and all of the signal levels should be `hi` if it's idle.

steveiliop56, 2024-03-08:
Is there any fix for that, I mean how would you know waveshare hardware but maybe there is a way? Or just don't buy waveshare things ever again?

pelwell, 2024-03-08:
If (and it currently is an "if") the hardware is permanently holding one of the I2C lines low, possibly as a result of an incomplete reset, then there is nothing we can do. If it turns out that the device is intentionally using clock stretching and that it is exceeding some time limit programmed into the I2C controller then there may be a possibility to increase the time limit.

steveiliop56, 2024-03-08:
Is there any way to tell the pi to actually reset the screen? Like a "dsi-reinit" thing?

steveiliop56, 2024-03-08:
Also the funny thing is that for some reason even after 50 reboots the display won't reset. In some cases it crashes so bad I have to format the SD card for it to work again.

pelwell, 2024-03-08:
I don't know if there is a generic way to reset a display, but I doubt it. The DSI interface is only for pixel data, the control signals must go via I2C or SPI etc.

I feel the only reliable reset mechanism is with a dedicated reset line driven by a GPIO. The driver for the display must be told which GPIO to use, usually via Device Tree.

steveiliop56, 2024-03-08:
> Also the funny thing is that for some reason even after 50 reboots the display won't reset. In some cases it crashes so bad I have to format the SD card for it to work again.

Hmm what do you think of that? I understand the crash if the controller due to the display but even if I unplug power let it sit there for a bit for the capacitors to run out, still the display won't show anything. It's like it remembers not to work or something.

6by9, 2024-03-08:
Whilst I don't have the Waveshare 4.3" panel, I do have a couple of the others (2.8", 7.9", and 10.1")
They're all fairly similar in design.

You have several largely independent items making up the panel.
- ChipOne ICN6211 DSI to DPI bridge connecting the DSI signals to the panel.
- Microcontroller (MCU) connected over I2C, which provides the backlight PWM, basic power control to the panel, and configuration to the ChipOne ICN6211.
- Touch controller connected to the I2C bus.

There are no commands sent over DSI, just image data.
If you look at [the MCU driver](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/gpu/drm/panel/panel-waveshare-dsi.c), there are very few commands that are sent to it, only for power control and setting backlight PWM level.
If an I2C peripheral decides to hold either SDA or SCL low, then there is relatively little that can be done. There is a recovery mechanism if SDA sticks low, but it still depends on the exact implementation of the peripheral.

I will check the latest kernels against my panels on a Pi5 to see if they also show issues. 

HyperPixel displays are DPI, not DSI, so you lose almost all your GPIOs.

steveiliop56, 2024-03-08:
Yeah it takes all the gpio pins. Well everything leans towards the official display for me or a solution to the waveshare panel. If you have any hardware mod for an x solution that I could try on the screen part I will be happy to do so.

pelwell, 2024-03-08:
> Most of the times it works but sometimes the display will not show anything and I will just keep getting these i2c timeouts.

> even if I unplug power let it sit there for a bit for the capacitors to run out, still the display won't show anything.

The first of those statements makes it sound like an occasional problem, the second makes it sound almost permanent. What's the reality?

steveiliop56, 2024-03-08:
It's kind of both. When the problem appears (usually after a reboot) I can't do anything, if I am lucky after 2-3 reboots or unplugging the cable it works if not I just format the SD card because I can't do anything else. The weird thing is that it doesn't seem to happen at all with a pi4.

6by9, 2024-03-08:
If the panel isn't responding, do some basic diagnostics of why.
- Are there error messages in the kernel logs?
- Does i2cdetect report the MCU and/or touch controller as responding on an I2C level
- Does `pinctrl 38-41` report one or more of the GPIOs stuck low?

Now that you say it, I do vaguely recall warm boots on that panel causing problems on I2C at one point. It's just possible that the order in which signals come up are being misinterpreted as an I2C start by the Waveshare MCU.

steveiliop56, 2024-03-08:
So the problems I have found are a) controller will just keep failing b) `ic2detect -y 1` (if that's relevant) will return nothing (all dashes). I am very confused as to why it works with a format. Like it doesn't work I format the SD card and magically works again.

steveiliop56, 2024-03-08:
Soo it happened again. Here is the output of the `pinctrl` command ```38: ip    pd | hi // CD0_SDA/GPIO38 = input
39: ip    pd | hi // CD0_SCL/GPIO39 = input
40: a2    pu | lo // CD1_SDA/GPIO40 = SDA4
41: a2    pu | hi // CD1_SCL/GPIO41 = SCL4```

pelwell, 2024-03-08:
```
40: a2    pu | lo // CD1_SDA/GPIO40 = SDA4
```
If that state is semi-permanent (you can monitor the line for changes with `pinctrl poll 40`) then that is the cause of the problem.

steveiliop56, 2024-03-08:
Is there any way to reset it? Reboot doesn't do anything...

pelwell, 2024-03-08:
You are asking for help with a third-party display I don't possess.

Any suggestions, @waveshare?

6by9, 2024-03-08:
Oops, thought you were using the Waveshare displays that don't emulate the Pi display, but you're not (it's the 800x480 Pi panel clone). Sorry, no idea on those panels.

If the pictures on https://www.waveshare.com/4.3inch-DSI-LCD.htm are correct, then it's also using the ICN6211. I don't see a MCU on there though, so I don't know what they're doing there. It's also possible that they have updated their code to avoid I2C issues.

steveiliop56, 2024-03-08:
Unfortunately waveshare's product quality is so good that its one leg broke of because apparently you are not supposed to screw it in all they way for it to balance? And the other screw pad just fell off, tried to solder it, no luck. Here is a picture of @waveshare quality:

[image]

*to be honest tho I tried to screw it using a small drill because it wouldn't screw in all they way but I barely clicked the button, it was from these analog drills that turn as much as you press the button. But still it only did one turn, one turn and it broke, I have done this with other displays and they were ok!*

pelwell, 2024-03-08:
I don't want this issue to degenerate into a criticism of Waveshare - it's chatty enough as it is.

steveiliop56, 2024-03-08:
Yeah anyway I don't have anything to test on so thank you for your help.

vhelmer, 2024-03-12:
I'm encountering a similar issue to steveiliop56 with a Raspberry Pi 5 and the Waveshare 7inch DSI LCD (C) https://www.waveshare.com/wiki/7inch_DSI_LCD_(C) on the Raspbian OS Lite.

[image]

After approximately every fifth reboot, the display stops working, and subsequent reboots fail to bring it back up.

Upon failure, the pinctrl output reads:

```
0: a3    pu | hi // ID_SD/GPIO0 = SDA0
1: a3    pu | hi // ID_SC/GPIO1 = SCL0
2: a3    pu | lo // PIN3/GPIO2 = SDA1
3: a3    pu | hi // PIN5/GPIO3 = SCL1
```
The screen resumes working only after physically reconnecting the Raspberry Pi power supply. 

After hours of attempting to troubleshoot how to bring the display up after a reboot that didn't start the screen, I discovered that the only solution is to execute the following command:

```
pinctrl 2-3 op dh
sudo reboot
```

Subsequent reboots then successfully bring the display back. Nonetheless, I assume this is not an ideal solution.

pelwell, 2024-03-13:
```
pinctrl 2-3 op dh
sudo reboot
```
That's an interesting find, and may help others in the same situation. If you have the time, I would be interested to hear if `pinctrl 2-3 ip` is also helpful, and what the output of `pinctrl 2-3` is after making them inputs. This will help to confirm whether the problem is with a device on the bus or whether the I2C controller is malfunctioning.

> Nonetheless, I assume this is not an ideal solution.

No, it's definitely not ideal.

JinShil, 2024-03-14:
I am able to reproduce this symptom using a different slave device (MCP23017, PCA9675, and GP8403).  For me it results repeated "controller timed out" and "i2c_dw_handle_tx_abort: lost arbitration" messages in dmesg.

After running `pinctrl 2-3 ip` and rebooting, SDA remains low, indicating that the slave device is holding data low.  However, running `pinctrl 2-3 op dh` followed by `pinctrl 2-3 a3` seems to get things working again, though only temporarily.  I can also get things working again by power cycling the slave.

vhelmer, 2024-03-14:
I can confirm that SDA1 remains low after executing pinctrl 2-3 ip command following a reboot that did not bring up the display.
 
Here are the results:

```
pi@raspberrypi:~$ pinctrl 2-3
 2: a3    pu | lo // PIN3/GPIO2 = SDA1
 3: a3    pu | hi // PIN5/GPIO3 = SCL1
pi@raspberrypi:~$ pinctrl 2-3 ip
pi@raspberrypi:~$ pinctrl 2-3
 2: ip    pu | lo // PIN3/GPIO2 = input
 3: ip    pu | hi // PIN5/GPIO3 = input
pi@raspberrypi:~$ sudo reboot
pi@raspberrypi:~$ pinctrl 2-3
 2: a3    pu | lo // PIN3/GPIO2 = SDA1
```

neuralassembly, 2024-03-15:
I have a similar problem when AQM0802-based LCD is connected to Pi 5 with recent kernel (6.6.20+).
While executing "i2cdetect -y 1", dmesg showed the following errors many times.

`
[ 189.544399] i2c_designware 1f00074000.i2c: controller timed out
`

In my case, I could avoid this tourble by adding the following configuration in /boot/firmware/config.txt
(100000 is a default value).

`
dtparam=i2c_baudrate=50000
`

I felt that 20000 is more stable. Timing might be severe on Pi 5 with recent kernel.

pelwell, 2024-03-18:
FYI there's a PR (#6050) that may help with this issue. Similar failures are seen when the device fails to ACK data in time, and the PR changes the mark/space ratio on SCL to give the device longer to respond at any given clock speed.

neuralassembly, 2024-03-19:
Dear pelwell,

Thank you for creating the pull request.

I tried "sudo rpi-update pulls/6050".
However, this pull request did not improve the behavior of my device.

When my device stops responding to my program, only the folloiwng error appeared in dmesg.

``
[  161.891681] i2c_designware 1f00074000.i2c: i2c_dw_handle_tx_abort: lost arbitration
``

After that, I tried "i2cdetect -y 1", and the following output appeared.
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: -- -- -- -- -- -- -- -- 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
```

When "--" is shown, the following error appeared.

``
[  248.307349] i2c_designware 1f00074000.i2c: controller timed out
``

So, after "i2cdetect" finishes, the above error appeared 24 times.

pelwell, 2024-03-19:
That's not a failure mode I've encountered.

At this point I think you should open a new issue, explaining the behaviour you were seeing.

steveiliop56, 2024-03-22:
So is the issue fixed with the new 6.6 kernel? Because that's the response I got from waveshare was to try with 6.1 kernel and replace the dtb file with one from here (that was before a week tho).

talksik, 2024-04-21:
@steveiliop56 I just tried copying over the dtbo file from 6.6 `/boot/overlays` into my current system image. It doesn't load the module when querying: `lsmod | grep waveshare`.

I am running into a different type of error. Anyone there to help? 
* I have a yocto custom boot2qt image that enables i2c via meta-raspberrypi settings.
* My config has `dtoverlay=vc4-kms-dsi-waveshare-panel,2_8_inch` as [waveshare's guide says.](https://www.waveshare.com/wiki/2.8inch_DSI_LCD)
* I've tried every different variant of `vc4-fkms-v3d` and `vc4-kms-v3d` and `vc4-kms-v3d-pi4`

Output of `lsmod | grep waveshare`
```
root@b2qt-raspberrypi4-64:~# lsmod | grep waveshare
panel_waveshare_dsi    16384  0
backlight              24576  1 panel_waveshare_dsi
```

Output of `ls /dev/dri`
```
root@b2qt-raspberrypi4-64:~# ls /dev/dri/
by-path     card0       card1       renderD128
```
My entire config.txt file:
```
# flowy note: this is for custom boot2qt yocto image

################################################################################
##  Raspberry Pi Configuration Settings
##
##  Revision 16, 2013/06/22
##
##  Details taken from the eLinux wiki
##  For up-to-date information please refer to wiki page.
##
##  Wiki Location : http://elinux.org/RPiconfig
##
##
##  Description:
##    Details of each setting are described with each section that begins with
##    a double hashed comment ('##')
##    It is up to the user to remove the single hashed comment ('#') from each
##    option they want to enable, and to set the specific value of that option.
##
##  Overclock settings will be disabled at runtime if the SoC reaches temp_limit
##
################################################################################

################################################################################
##  Standard Definition Video Settings
################################################################################

## sdtv_mode
##     defines the TV standard for composite output
##
##     Value    Description
##     -------------------------------------------------------------------------
##     0        Normal NTSC (Default)
##     1        Japanese version of NTSC - no pedestal
##     2        Normal PAL
##     3        Brazilian version of PAL - 525/60 rather than 625/50, different
##              subcarrier
##
#sdtv_mode=0

## sdtv_aspect
##     defines the aspect ratio for composite output
##
##     Value    Description
##     -------------------------------------------------------------------------
##     1        4:3 (Default)
##     2        14:9
##     3        16:9
##
[... 1166 lines cut ...]
Linux b2qt-raspberrypi4-64 6.1.54-v8 #1 SMP PREEMPT Wed Sep 20 14:13:53 UTC 2023 aarch64 GNU/Linux
```

Confirmed that I have waveshare overlay:
```
root@b2qt-raspberrypi4-64:~# ls /boot/overlays/ | grep waveshare
vc4-kms-dsi-waveshare-panel.dtbo
waveshare-can-fd-hat-mode-a.dtbo
waveshare-can-fd-hat-mode-b.dtbo
```

Here is the `dmesg` error that repeats endlessly:
```
...
[  469.247935] Goodix-TS 10-0014: Error writing 1 bytes to 0x814e: -110
[  470.271888] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  470.271936] Goodix-TS 10-0014: Error reading 10 bytes from 0x814e: -110
[  471.295900] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  471.295949] Goodix-TS 10-0014: Error writing 1 bytes to 0x814e: -110
[  472.319893] i2c-bcm2835 fe205000.i2c: i2c transfer timed out
[  472.319944] Goodix-TS 10-0014: Error reading 10 bytes from 0x814e: -110
...
```

Ahh, I'm out of ideas and lost. If anyone can give me any sort of hints that helped them, I'd appreciate it!

6by9, 2024-04-22:
@talksik 2 things:

1. This thread is about Pi5. Seeing as your logs include errors on i2c-10, and your hostname is `b2qt-raspberrypi4-64`, I strongly suspect you're on a Pi4, and therefore all your comments are totally irrelevant to this issue.

2. I'll point you at https://github.com/raspberrypi/linux/blob/rpi-6.6.y/arch/arm/boot/dts/overlays/README#L5146 or `dtoverlay -h vc4-kms-dsi-waveshare-panel`
> Requires vc4-kms-v3d to be loaded.

The original config.txt that you posted is MASSIVE! My memory says there is a maximum size the bootloader will load, so trim it down substantially.

timg236, 2024-04-22:
> @talksik 2 things:
>
> 1. This thread is about Pi5. Seeing as your logs include errors on i2c-10, and your hostname is `b2qt-raspberrypi4-64`, I strongly suspect you're on a Pi4, and therefore all your [...]

On Pi4 the config.txt file should be <= 4KB, otherwise, network boot will truncate it.

On Pi5 there an individual config.txt file (or include) can be up to 16KB,.

talksik, 2024-04-22:
Thanks @6by9 and @timg236 !! I found that there was an issue with enabling camera which caused i2c conflict with the DSI display.

pelwell, 2024-04-22:
I'm not aware of any outstanding Pi 5 I2C problems using the latest kernel, so I'm tentatively closing this issue.

closed 2024-04-22 by pelwell