raspberrypi/linux #5786: Please enable the drivers for Xillybus and XillyUSB as modules

billauer, 2023-12-12:
### Describe the bug

The drivers for Xillybus and XillyUSB are not enabled in the kernel, not even as modules. As a result, people who want to use the PCIe / USB3 interface with Xillybus to interface with an FPGA, need to compile these drivers separately.

These drivers are beneficial for targets that support PCIe and/or USB3 (i.e. BCM2711 and BCM2712).

### Steps to reproduce the behaviour

In the kernel compilation environment:

```
$ make ARCH=arm64 bcm2711_defconfig
#
# configuration written to .config
#
$ grep XILLY .config
# CONFIG_XILLYBUS is not set
# CONFIG_XILLYUSB is not set
```

The same happens with bcm2712_defconfig

The desired output is:
```
CONFIG_XILLYBUS_CLASS=m
CONFIG_XILLYBUS=m
CONFIG_XILLYBUS_PCIE=m
CONFIG_XILLYUSB=m
```

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 400, Other

### System

This applies to all kernel versions currently available.

### Additional context

Xillybus is a solution for communication between programmable hardware (FPGA) and a host using the PCIe bus as well as USB 3.0. It makes the development of a combination between an FPGA and an embedded Linux system considerably easier.

It's further explained in the kernel tree's Documentation/xillybus.txt and at http://www.xillybus.com.

The suggested fix enables the compilation of the driver for Xillybus as a module. This driver has been in the kernel under drivers/char/xillybus/ since kernel v3.18 (and as a staging driver since v3.12) and is beneficial for FPGA hardware developers using the relevant hardware component.

The driver is enabled in several mainstream Linux distributions, e.g. Ubuntu >= 14.04. XillyUSB was enabled in Ubuntu 22.04 and Fedora 35.

Being a kernel module for a PCIe/USB device, it's activated only on a Vendor/Device IDs match, so this change has no possible impact on users who have no direct interest in it. Regardless, the module is stable for several years.

Please consider enabling these drivers for the mentioned targets.

pelwell, 2023-12-12:
Do you want the XILLYBUS_OF option as well? Given that it's so small it seems a shame not to.

pelwell, 2023-12-12:
Added - see https://github.com/raspberrypi/linux/commit/5931a01e0017d719af578fac8475442668807a97.

billauer, 2023-12-12:
Yay! Thanks a lot!

closed 2023-12-12 by pelwell