raspberrypi/linux #5793: Pi 5 with built-in RTC throws error ioctl(4, RTC_UIE_ON, 0): Invalid argument

by, 2023-12-15:
### Describe the bug

hwclock -v throws msg. with error
Waiting for clock tick...
ioctl(4, RTC_UIE_ON, 0): Invalid argument
Waiting in loop for time from /dev/rtc0 to change
...got clock tick
on Pi 5 like on the previous Pis despite the fact that it now has a built-in RTC, so no RTC extension board from other manufacturers.

### Steps to reproduce the behaviour

Run hwclock -v on Pi 5

### Device (s)

Other

### System

cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

vcgencmd version
2023/12/14 16:43:25 
Copyright (c) 2012 Broadcom
version 0f881224 (release) (embedded)

uname -a
Linux aion 6.6.5-v8-16k+ #1706 SMP PREEMPT Fri Dec  8 18:45:36 GMT 2023 aarch64 GNU/Linux

popcornmix, 2023-12-15:
Based on discussion in https://github.com/raspberrypi/linux/issues/3282 perhaps setting `CONFIG_RTC_INTF_DEV_UIE_EMUL` would avoid this?

Testing... yes - it avoids the ioctl complaint:
```
pi@pi5:~ $ sudo hwclock -v
hwclock from util-linux 2.38.1
System Time: 1702640345.932417
Trying to open: /dev/rtc0
Using the rtc interface to the clock.
Assuming hardware clock is kept in UTC time.
Waiting for clock tick...
...got clock tick
Time read from Hardware Clock: 2023/12/15 11:39:06
Hw clock time : 2023/12/15 11:39:06 = 1702640346 seconds since 1969
Time since last adjustment is 1702640346 seconds
Calculated Hardware Clock drift is 0.000000 seconds
2023-12-15 11:39:05.921739+00:00
```

popcornmix, 2023-12-15:
@pelwell any thoughts? The docs do say:

> The emulation code will read the time from the hardware clock several times per second, please enable this option only if you know that you really need it.

pelwell, 2023-12-15:
Aside from the ugly error, does it work without UIE support?

popcornmix, 2023-12-15:
It works for the obvious use cases (reading the time and waking up Pi at an specified time).

@by are you missing any functionality due to this?

by, 2023-12-15:
No, haven't found functional errors yet; as with the Pi 5, RTC became an official product feature, it would be nice to correct the ugly error.

And yes, the kernel switch avoids it, and it would be nice to get it set in the 6.x kernels, similarly to https://github.com/raspberrypi/linux/pull/5057 , if the root cause can't be eliminated, thanks.