raspberrypi/linux #5795: dtoverlay for Hifiberry DAC broken

hackerb9, 2023-12-17:
### Describe the bug

The Hifiberry DAC does not show up as a sound card. It does not even appear on the i2c bus and an oscilloscope shows no data on the relevant pins. This situation appears to be analogous to issue #4898 , in which the overlay for the Hifiberry DAC 2 HD was not working.  

### Relevant section of config.txt

```config
dtoverlay=
dtoverlay=hifiberry-dac
```

(See below for full config.txt.)

### What does i2cdetect -y 1 show? 

I see that no devices are on the i2c bus when using `dtoverlay=hifiberry-dac`. In fact, although hifiberry-dac is supposed to set dtparam=i2c automatically, it does not. To run i2cdetect, I must specify dtparam=i2c explicitly in config.txt in addition to the usual `modprobe i2c-dev`. 

Perhaps of note: knowing that the overlay was fixed for the Hifiberry DAC 2 HD,  I also experimentally tried `dtoverlay=hifiberry-dacplushd`. Although it is the wrong overlay for my card, it does demonstrate that my card is responding on the i2c bus. i2cdetect -y 1 shows 4C as **UU**, in use by driver. My DAC still doesn't work, but it does suggest that the change to the hifiberry-dacplushd overlay by @j-schambacher in pull request #4904 may solve this problem as well. (Removing the second declaration of "clocks" in PLL). 

### Is it a hardware fault? 

Highly unlikely. I have reproduced this problem with two different Raspberry Pi Zeroes and two different DAC hats.

### Steps to reproduce the behaviour

Try `aplay -l` with an i2c DAC hat installed. Note that the DAC is not detected.

### Device (s)

Raspberry Pi Zero, Raspberry Pi Zero W / WH

### System

**OS**: piCore 14.

**Firmware version**:
Apr 25 2023 18:32:05
Copyright (c) 2012 Broadcom
version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453 (clean) (release) (start)

**Kernel version**:  Linux box 6.1.25-piCore #1 Sun Apr 30 13:17:34 EDT 2023 armv6l GNU/Linux

### Logs

### i2cdetect -y 1

<details>

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

</details>

### dmesg

<details>

```log
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.1.25-piCore (tc@piCore) (arm-bcm2708hf-linux-gnueabihf-gcc (crosstool-NG 1.25.0.117_66ac9e6) 12.2.0, GNU ld (crosstool-NG 1.25.0.117_66ac9e6) 2.39) #1 Sun Apr 30 13:17:34 EDT 2023
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[... 369 lines cut ...]
# Linux >= 5.4 changed some internal structures that make it incompatible
# with device-tree overlays of previous versions. Therefore, the hardware
# configuration that is stored in the onboard EEPROM of some DAC cards might
# not load correctly. You can read more at the HiFiBerry blog:
# www.hifiberry.com/blog/configuration-changes-in-linux-5-4 
#force_eeprom_read=0

# DEBUG: Enable i2c explicitly for probing with i2cdetect.
dtparam=i2c_arm=on

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
# Commented out as we're wearing a DAC hat.
#dtparam=audio=on

# Add overlay for HiFi DAC hat. 
dtoverlay=hifiberry-dac
```

</details>

### Additional context

Other overlays of the same vintage likely also need to be updated, such as hifiberry-dacplus and i2c-dac.

pelwell, 2023-12-17:
Install RPiOS on a spare SD card and confirm that it also doesn't work, then we'll look at it.

j-schambacher, 2023-12-17:
The standard Hifiberry DAC does not have any I2C controlled device therefore i2cdetect won't show any DAC device. The aplay -l messages seem strange and may lead to some broken installation. Please follow pelwell's advice and make a clean install.

hackerb9, 2023-12-18:
Please ignore. I misunderstood #4898 to say that an i2s device should be expected to show up on the i2c bus with i2cdetect. That is not the case as my DAC is now working and it still does not show up with i2cdetect. (For anyone else who is using piCore and has the same problem, it was fixed by running `tce-load -wi alsa.tcz`).

pelwell, 2023-12-18:
No problem - thanks for letting us know.