raspberrypi/linux #5801: DSI screen randomly failing on 6.1 kernel on Pi 3B+

lurch, 2023-12-18:
### Describe the bug

I'm running the latest version of Raspberry Pi OS Bookworm 32-bit (but the problem is also present in Bookworm 64-bit), fully up-to-date with `apt`. I've used `raspi-config` to configure it to boot directly to the console instead of to the desktop.
I'm using a Raspberry Pi 3B+ with a Raspberry Pi Touch Display connected (powered according to https://www.raspberrypi.com/documentation/accessories/display.html#powering-the-touch-display ), I'm using an official Raspberry Pi Micro USB PSU, and I've got an Ethernet cable and USB mouse and keyboard connected.

Sometimes, when Linux has finished loading, if I type `kmsprint` it reports:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to find DRM cards
Aborted
```
and if I then try running `startx`, it quickly fails with `(EE) No devices detected.`

Slightly frustratingly, this bug seems semi-random (possibly very timing-dependent) - sometimes after a reboot `kmsprint` will work successfully (and I **can** then run `startx`) and other times after a reboot `kmsprint` will display the error above. The problem only seems to manifest on a Pi 3B+, I wasn't able to reproduce the problem on a Pi5, a Pi4 or a Pi2. I was also unable to reproduce the problem after running `rpi-update` on my Pi 3B+.

@6by9 spent some time looking at the problem with me (after we reverted to the `apt`-installed kernel), and he says the fact that `rpi-update` fixes the problem is just a lucky coincidence (i.e. a quirk that affects the timing), he doesn't think there's any specific fix for the problem in the newer kernel. He says the underlying problem is that the Pi tries to read the Atmel firmware version over I2C, and gets the wrong value. In `dmesg` we have:
```
[    5.315265] rpi_touchscreen_attiny 10-0045: Unknown Atmel firmware revision: 0xe1
```
and Dave thinks this is 1-bit-shifted from the correct value.

If we have one of the "erroneous boots" and then run:
```
sudo rmmod rpi_panel_attiny_regulator
sudo modprobe rpi_panel_attiny_regulator
```
then this tries to re-read the Atmel firmware version again, and (in our limited testing) this seems to **always** work, and `kmsprint` then runs without error again.

We also tried setting `force_turbo=1` in `/boot/firmware/config.txt` to see what impact that has on the problem, and that seems to always fix the problem. (So that's the temporary workaround for now)

Dave hooked up a logic-analyser to the I2C pins on the DSI display, and found that with `force_turbo=1` then the I2C clock runs at 100KHz, but without that setting it runs at 66KHz.

### Steps to reproduce the behaviour

1. Flash `raspios_armhf-2023-12-06/2023-12-05-raspios-bookworm-armhf.img.xz` to an SD card.
2. Don't connect the Raspberry Pi Touch Display yet!
3. Boot up the SD card in the Pi 3B+ and install all apt-updates (at the time of writing, `uanme -a` reports `Linux bookworm32 6.1.0-rpi7-rpi-v7 #1 SMP Raspbian 1:6.1.63-1+rpt1 (2023-11-24) armv7l GNU/Linux`)
4. Use `raspi-config` to configure the Pi to "Console Autologin"
5. Shut down the Pi 3B+, connect the Touch Display
6. Power up the Pi, and run `kmsprint` at the console
7. If it doesn't error, reboot and try step 6 again

### Device (s)

Raspberry Pi 3 Mod. B+

### System

System Information
------------------

Raspberry Pi 3 Model B Plus Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 02b371de1a87bbe70cfcbd498527067a7d2ee6ca, stage5

Linux bookworm32 6.1.0-rpi7-rpi-v7 #1 SMP Raspbian 1:6.1.63-1+rpt1 (2023-11-24) armv7l GNU/Linux
Revision	: a020d3
Serial		: 000000000c46a408
Model		: Raspberry Pi 3 Model B Plus Rev 1.3
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

[... 699 lines cut ...]
[   13.583676] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   14.094206] Bluetooth: hci0: BCM: features 0x2f
[   14.102679] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[   14.102705] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[   14.103208] Bluetooth: hci0: BCM: Using default device address (m.m.m.m)
[   15.720632] lan78xx 1-1.1.1:1.0 eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   15.722839] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   15.722937] NOHZ tick-stop error: local softirq work is pending, handler #08!!!
[   15.857571] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   16.124556] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   16.124578] Bluetooth: BNEP filters: protocol multicast
[   16.124595] Bluetooth: BNEP socket layer initialized
[   16.145465] Bluetooth: MGMT ver 1.22
[   16.176131] NET: Registered PF_ALG protocol family
[   16.219181] cryptd: max_cpu_qlen set to 1000
[   22.414452] Bluetooth: RFCOMM TTY layer initialized
[   22.414489] Bluetooth: RFCOMM socket layer initialized
[   22.414524] Bluetooth: RFCOMM ver 1.11
[   23.513749] platform 1.reg_bridge: deferred probe pending
[   23.513777] platform 1.panel_disp: deferred probe pending
[   23.513789] mipi-dsi 3f700000.dsi.0: deferred probe pending

### Logs

[    5.315265] rpi_touchscreen_attiny 10-0045: Unknown Atmel firmware revision: 0xe1

b1c1l1, 2024-08-29:
Thank you for documenting this issue and workaround!

I am also experiencing this issue with a Raspberry Pi 3B+ and Raspberry Pi Touch Display running `2024-07-04-raspios-bookworm-armhf.img`.  The display would regularly fail to be initialized during boot, and `dmesg` would show the same `Unknown Atmel firmware revision: 0xe1` message.

Enabling `force_turbo=1` seems to have resolved the issue for me.

cfpwastaken, 2024-11-13:
I can confirm this bug and workaround on kernel 6.6

6by9, 2024-11-13:
This has been fixed fully in the latest kernels (post 23rd Sept 2024) as the read that was failing is actually redundant anyway.
That hasn't hit apt yet, but is in rpi-update.

cfpwastaken, 2024-11-13:
interesting. When will it be in a stable release?

6by9, 2024-11-13:
It's just missed the latest release.
We normally bump apt every 2-3 months, or when something useful or critical has been released.