raspberrypi/linux #5808: hdmi power used after power down

nulucidity, 2023-12-23:
### Describe the bug

ive noticed that when the power to the pi (4b running debian 11 32 bit) is turned off after a shutdown, it seems to be using the power from the hdmi input and is trying to reboot.

if i unplug the hdmi from the monitor after doing a shutdown, turning off the power seems to work as expected.

### Steps to reproduce the behaviour

turn off power after shutdown.

### Device (s)

Raspberry Pi 4 Mod. B

### System

debian 11 32 bit

### Logs

n/a

6by9, 2023-12-23:
It is an HDMI source that is meant to supply 5V to the HDMI connector (largely to power the EDID EEPROM), so an HDMI sink should not be applying any power to the 5V line.

I haven't got the Pi4 schematics to hand, but the CM4IO board which generally follows similar principles has a RT9742SNGV power switch on the HDMI 5V line to avoid backpowering.

If there is an issue, then it's not going to have a software fix.

nulucidity, 2023-12-23:
i did some more testing, and the problem occurs when the hdmi is plugged into the hdmi2 slot, but not the hdmi1 slot when used with a 3 port hdmi switch.

also it has a fast blinking green led next to the red power indicator.

[image]