raspberrypi/linux #5809: preempt_rt patched ARM64 kernel will not start on raspberry pi zero w2 or raspberry pi 3b ( BCM2837)

berndbenner, 2023-12-24:
### Describe the bug

I am building custom preempt_rt patched ARM64-kernels for raspberry pi4, cm4 over 2 years and now for raspberry pi 5 with success.

I have tried to build a ARM64-kernel for raspberry pi zero w2 and raspberry pi 3b, but the kernel will hang on start.

Here is the example configuration:
 
*  build from the last stable tag stable_20231123 ( the problem exist also in older 6.1-kernel revisions).   
* The kernel is starting and is working very well  on raspberry pi 4, cm4 and 5.  
* On raspberry pi 3b and raspberry pi zero w2 the kernel will hang on startup with smp-idle problems.

The only difference between the kernels are the initial config generation:
```text
make bcm2711_defconfig  # for rpi4,cm4, rpi3b and zero-w2 
make bcm2712_defconfig  # for rpi5 )
```
and the m-tune compiler flags.

```text
FLAGS+=CFLAGS="-mtune=cortex-a72 -mno-outline-atomics"   # rpi 4,cm4
FLAGS5+=CFLAGS="-mtune=cortex-a76 -mno-outline-atomics"  # rpi5
FLAGSZ+=CFLAGS="-mtune=cortex-a53 -mno-outline-atomics"  # rpi3b and rpiz-w2
```
The problem is only a preempt_rt problem on the BCM2837-CPU.  A custom build kernel without the preempt_rt patch will start like the official kernel8.

[config-6.1.63-preempt_rt-no-start.txt](https://github.com/raspberrypi/linux/files/13760382/config-6.1.63-preempt_rt-no-start.txt)
[config-6.1.63-start.txt](https://github.com/raspberrypi/linux/files/13760384/config-6.1.63-start.txt)
[image]

### Steps to reproduce the behaviour

* pull the kernel ( ex: tag stable_20231123 )
* make bcm2711_defconfig 
* apply the preempt_rt patch ( patch-6.1.64-rt17.patch.gz works with kernel 6.1.63 )
* configure the kernel ( enable 
* build kernel.modules or debian linux-image and/or linux-header packages
* install the kernel and the modules
* try to start the kernel on rpi zero w2 or rpi3b

 

### Device (s)

Raspberry Pi Zero 2 W, Raspberry Pi 3 Mod. B

### System

generated on a raspberry pi zero w2 with custom kernel without preempt_rt patch

Raspberry Pi Zero 2 W Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage2

Linux raspz-w2-bb 6.1.63-zw2-bb-custom #1.1.0 SMP PREEMPT Sat Dec 23 22:41:41 CET 2023 aarch64 GNU/Linux
Revision        : 902120
Serial          : 000000007678b203
Model           : Raspberry Pi Zero 2 W Rev 1.0
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Mar 17 2023 10:52:42
[... 601 lines cut ...]
[   15.012614] bcm2835-codec bcm2835-codec: Device registered as /dev/video18
[   15.012717] bcm2835-codec bcm2835-codec: Loaded V4L2 image_fx
[   15.021445] bcm2835-codec bcm2835-codec: Device registered as /dev/video31
[   15.021576] bcm2835-codec bcm2835-codec: Loaded V4L2 encode_image
[   15.544338] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   15.671820] cfg80211: Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[   15.673845] cfg80211: Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[   15.675968] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   16.123521] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   16.135215] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   16.138157] usbcore: registered new interface driver brcmfmac
[   16.394592] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jun 14 2023 07:27:45 version 7.45.96.s1 (gf031a129) FWID 01-70bd2af7 es7
[   18.722753] Adding 102396k swap on /var/swap.  Priority:-2 extents:5 across:696316k FS
[   20.800647] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   21.347300] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   22.761050] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   23.500126] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   24.144536] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   24.712572] EXT4-fs (sda3): mounted filesystem with ordered data mode. Quota mode: none.
[ 3133.305738] raspberrypi-firmware soc:firmware: failed to allocate a DMA channel
[ 3295.903538] raspberrypi-firmware soc:firmware: failed to allocate a DMA channel

### Additional context

bootcode.bin is on a sd-card, boot/firmware and rootfs is on a fast usb-stick, may be relevant for timing.

nbuchwitz, 2024-01-15:
dwc_otg usb driver splats with rt patch on arm64. Dwc2 driver does work though. Spend some time trying to fix dwc_otg with rt and Arm64, but ended using the dwc2 on older and xhci on newer devices.

berndbenner, 2024-01-22:
The  add of
dtoverlay=dwc2,dr_mode=host
or
dtoverlay=dwc2
to config.txt will not fix the startup issue on a zero w2

After the add the kernel will not show the stack trace, but the startup hangs (endless) after identifiing the mmc card after the message:
mmc1: new high speed SDIO card at address 0001

The non-rt-kernel will also hang at the same position with "dtoverlay=dwc2" in config.txt. I have to comment
it out, to boot. ( only dwc_otg will work )

if I boot the normal non-rt-kernel  the usb-device and the usb-storage device will be detected after
the last message of the rt-boot. Here is a kernel log:
 
```text
Jan 22 22:29:58 raspz-w2-bb kernel: Waiting for root device PARTUUID=a11eb507-02...
Jan 22 22:29:58 raspz-w2-bb kernel: usb 1-1: new high-speed USB device number 2 using dwc_otg
Jan 22 22:29:58 raspz-w2-bb kernel: Indeed it is in host mode hprt0 = 00001101
Jan 22 22:29:58 raspz-w2-bb kernel: mmc0: host does not support reading read-only switch, assuming write-enable
Jan 22 22:29:58 raspz-w2-bb kernel: mmc0: new high speed SD card at address 67c2
Jan 22 22:29:58 raspz-w2-bb kernel: mmcblk0: mmc0:67c2 SD512 473 MiB 
Jan 22 22:29:58 raspz-w2-bb kernel:  mmcblk0: p1
Jan 22 22:29:58 raspz-w2-bb kernel: mmcblk0: mmc0:67c2 SD512 473 MiB
Jan 22 22:29:58 raspz-w2-bb kernel: mmc1: new high speed SDIO card at address 0001
Jan 22 22:29:58 raspz-w2-bb kernel: usb 1-1: New USB device found, idVendor=090c, idProduct=1000, bcdDevice=11.00
Jan 22 22:29:58 raspz-w2-bb kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 22 22:29:58 raspz-w2-bb kernel: usb 1-1: Product: Flash Drive FIT
Jan 22 22:29:58 raspz-w2-bb kernel: usb 1-1: Manufacturer: Samsung
Jan 22 22:29:58 raspz-w2-bb kernel: usb 1-1: SerialNumber: 0371622050001625
Jan 22 22:29:58 raspz-w2-bb kernel: usb-storage 1-1:1.0: USB Mass Storage device detected
Jan 22 22:29:58 raspz-w2-bb kernel: scsi host0: usb-storage 1-1:1.0
Jan 22 22:29:58 raspz-w2-bb kernel: scsi 0:0:0:0: Direct-Access     Samsung  Flash Drive FIT  1100 PQ: 0 ANSI: 6
Jan 22 22:29:58 raspz-w2-bb kernel: sd 0:0:0:0: [sda] 125313283 512-byte logical blocks: (64.2 GB/59.8 GiB)
Jan 22 22:29:58 raspz-w2-bb kernel: sd 0:0:0:0: [sda] Write Protect is off
Jan 22 22:29:58 raspz-w2-bb kernel: sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
Jan 22 22:29:58 raspz-w2-bb kernel: sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
Jan 22 22:29:58 raspz-w2-bb kernel:  sda: sda1 sda2 sda3
Jan 22 22:29:58 raspz-w2-bb kernel: sd 0:0:0:0: [sda] Attached SCSI removable disk
Jan 22 22:29:58 raspz-w2-bb kernel: EXT4-fs (sda2): INFO: recovery required on readonly filesystem
Jan 22 22:29:58 raspz-w2-bb kernel: EXT4-fs (sda2): write access will be enabled during recovery
Jan 22 22:29:58 raspz-w2-bb kernel: EXT4-fs (sda2): recovery complete
Jan 22 22:29:58 raspz-w2-bb kernel: EXT4-fs (sda2): mounted filesystem with ordered data mode. Quota mode: none.
Jan 22 22:29:58 raspz-w2-bb kernel: VFS: Mounted root (ext4 filesystem) readonly on device 8:2.
...
```

The "trick" of only bootcode.bin on the sd-card and other boot stuff,device tree ... and rootfs on a usb-stick will not work on "older" rpi Cortex-A52-devices with the prempt_rt patched kernel by usb-device driver timing issues .

pelwell, 2024-01-23:
We do not support the RT kernel, and we don't recommend running a 64-bit kernel on Zero 2 W because of memory constraints.

berndbenner, 2024-01-23:
Ok Thank you! 
May be, its helpfull to update your custom kernel build and prempt_rt patch instructions to avoid the pitfalls with zero-w2 and rpi3b boards.

closed 2024-01-23 by berndbenner

pelwell, 2024-01-23:
> update your custom kernel build and prempt_rt patch instructions

Which instructions are those?

nbuchwitz, 2024-01-23:
> Ok Thank you! May be, its helpfull to update your custom kernel build and prempt_rt patch instructions to avoid the pitfalls with zero-w2 and rpi3b boards.

Afaik 3b should work. You could try to use the Revolution Pi kernel repo which includes the rt patch with some custom patches tailored towards raspberry pi compute module.