raspberrypi/linux #5813: PI 7" Screen DSI issue on RPI 5 Board

donmahallem, 2023-12-30:
### Describe the bug

The official touchscreen doesn't work on the PI5(stays off, no sign of light or output). With the exact same SD-Card(same installation) on the RPI 3 Board the screen just works.

### Steps to reproduce the behaviour

Attach screen via DSI to either DSI0 or DSI1 to the PI5. Specify in `/boot/firmware/config.txt` as required by the official documentation the DSI output port the screen uses.

```
dtoverlay=vc4-kms-v3d (did even try to specify the -pi5 version)
dtoverlay=vc4-kms-dsi-7inch,dsi0
```

Looking at raspinfo once the correct DSI port is specified the I2C controller times out:

```
[    4.071932] i2c_designware 1f00088000.i2c: controller timed out
[    4.071950] rpi_touchscreen_attiny 6-0045: Failed to read REG_ID reg: -5
[    4.071961] rpi_touchscreen_attiny: probe of 6-0045 failed with error -5
```
This doesn't happen when no DSI cable is connected or the incorrect dsi output for the 7inch screen is specified.

### Device (s)

Other

### System

https://pastebin.com/teagmKfa

6by9, 2024-01-03:
What cable have you used to connect the Pi5 to the display? Pi Zero Camera cables can NOT be used for displays, and tend to result in I2C timeouts as you have due to SCL being tied low. (See https://forums.raspberrypi.com/viewtopic.php?t=362563 for having gone through exactly that loop).

Note that there are separate [camera](https://www.raspberrypi.com/products/camera-cable/) and [display](https://www.raspberrypi.com/products/display-cable/) cables listed on the Pi product page - https://www.raspberrypi.com/products

Vizzyy, 2024-04-20:
Running into similar issues with the official 7" display on my pi5. I'm attempting to completely clone my setup I had on my pi4 onto the 5, but am having to start from scratch as the pi5 requires bookworm. I'm using an official rpi5 display cable -- as others have noted, you can't just use a pi zero camera cable. 

The display powers up and works, so does the touch, but I can't get the `rpi_backlight` python module to work on the pi5.

I noticed that in this directory `/sys/class/backlight/` the screen device shows up at `6-0045` (which is how I found this thread, by googling that string), while on the pi4 it used to be `10-0045`. There's something weirdly different with how the pi5 is registering the display.

6by9, 2024-04-22:
Seeing as you say your display and touch works, it is not the same issue that donmahallem has reported.

The Pi5 has 2 camera/display ports, and routes i2c-4 and i2c-6 to them.
Pi4 has one camera and one display port. i2c-10 is routed to both as image sensors normally have different I2C addresses to the display.

It looks like rpi_backlight makes big assumptions about where the backlight driver will exist - https://github.com/linusg/rpi-backlight/blob/main/rpi_backlight/__init__.py#L31-L35. I see you've already found that repo and commented on https://github.com/linusg/rpi-backlight/issues/60
That is a third party library, so it's largely up to the developers (or others seeing as it is all open source) to add support. It's not an issue in the kernel.

jtbutcher, 2024-07-09:
it now supports an optional argument SYSFS_PATH 

```
usage: rpi-backlight [-h] [--get-brightness] [-b VALUE] [--get-power] [-p VALUE] [-d DURATION] [-B {raspberry-pi,tinker-board,tinker-board-2,microsoft-surface-rt}] [-V] [SYSFS_PATH]

Get/set power and brightness of the official Raspberry Pi 7" touch display.

positional arguments:
  SYSFS_PATH            Optional path to the backlight sysfs, set to :emulator: to use with rpi-backlight-emulator

options:
.......
```

in my case the path was "/sys/class/backlight/6-0045/" so the command would be formatted

`rpi-backlight -p off /sys/class/backlight/6-0045/`

Lozzproblems, 2024-12-01:
I have this issue too, did you ever figure out a fix for this?

6by9, 2024-12-02:
@Lozzproblems 
> I have this issue too, did you ever figure out a fix for this?

@donmahallem never reported back, but the suspicion is that he was using a camera cable not a display one.

@Vizzyy had a different issue in that the display and touch worked, but the backlight path had changed.

So which issue do you have? Or is it something new?

donmahallem, 2024-12-02:
My bad. The FPC was the main culprit and additionally loose contacts. Got it working eventually

6by9, 2024-12-02:
Thanks for the confirmation.

Lozzproblems, 2024-12-02:
I was using a wave share cable, I was trying to make a kind of laptop with the smarti pi touch pro case, I was using the official display and a power splitter, I was using an official power adapter, when I powered it up after assembly, It did not work, the screen showed no sign of life With it all wired up as said in the manual, I plugged the wave share cable from the screen to the dsi0 port on my pi 5