raspberrypi/linux #5816: USB devices only show when a powered one is plugged in

rosecitytransit, 2023-12-31:
### Describe the bug

Not sure if this is the right place for this, but with the 3B+ I have, USB devices only show up in lsusb when a powered one is plugged in.

### Steps to reproduce the behaviour

Plug in USB device (have tried a RTL-SDR as well as an optical mouse): just the USB hub in lsusb
Then plug in a externally powered hard drive adapter I have (ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS578 SATA 6Gb/s): SDR or mouse shows up, though the adapter doesn't unless there's an actual drive connected
Unplug the adapter: just the hub

It seems that either the adapter is reverse supplying power, or only it is commanding the Pi to power the hub.

Upgrading from bullseye to bookworm didn't help

### Device (s)

Raspberry Pi 3 Mod. B+

### System

output attached
[raspinfo.txt](https://github.com/raspberrypi/linux/files/13802095/raspinfo.txt)

### Logs

dmesg appears to be included in raspinfo

### Additional context

If it is a power issue, I am using a Canna Kit DCAR-052A5

pelwell, 2023-12-31:
It's not the right place, which you would know if you'd read the messages you were prompted with.

If you disconnect all the hardware apart from power supply, display and mouse, and the mouse doesn't show up, then it sounds like you have faulty hardware.

closed 2023-12-31 by pelwell

rosecitytransit, 2023-12-31:
But the mouse shows up when the drive adapter is plugged in. Also, I am not using a display.

pelwell, 2024-01-01:
> But the mouse shows up when the drive adapter is plugged in, and the RTL SDR works fine.

But only when external power is supplied. The fault is either the 3B+, or the Canakit supply.

> Also, I am not using a display, or desktop; just SSH.

That's fine - you just need some way to see and interact with a shell.

rosecitytransit, 2024-01-01:
Can power be supplied upstream from a USB device to a host? I would think it can only go from the host to a device.

6by9, 2024-01-02:
> Can power be supplied upstream from a USB device to a host? I would think it can only go from the host to a device, and the drive adapter's external power should only be powering itself. But I know it would good to have an alternative pi power supply to try and/or a powered USB hub.

Backfeeding power from a peripheral is not compliant with the USB spec, but that doesn't stop a number of devices doing so.

The 5V pins of the 4 USB ports of a Pi3B+ are all tied together, and are fed from the Pi 5V rail through a current limiting and reverse powering blocking IC. So applying 5V via one port will power all other devices but won't feed power back into the Pi itself.

Your Pi will work with the supply down to around 3.6V, but that is well outside the range permitted in the USB spec. This is one of the main reasons for having the undervoltage detection on the Pi (red power LED goes out) to make it fairly obvious if you have power supply issues.
It is also possible you've blown the the current limiting IC, but I'd be surprised.