raspberrypi/linux #5817: Kernel error messages just on Pi 5 with Wolfson wm8731 audio codec

rafael2k, 2024-01-02:
### Describe the bug

The sBitx radio uses a Pi 4 and a wm8731 audio codec. With the Pi 5, the kernel shows strange messages (some config.txt option used).

### Steps to reproduce the behaviour

Just boot the radio with a Pi 5. Error messages appear every 10s.
```
[   21.818092] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_bclk_ratio on 1f000a4000.i2s: -22
[   21.818113]  AudioInjector audio: ASoC: error at snd_soc_link_hw_params on AudioInjector audio: -22
[   21.818458]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
[   21.821546] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_bclk_ratio on 1f000a4000.i2s: -22
[   21.821553]  AudioInjector audio: ASoC: error at snd_soc_link_hw_params on AudioInjector audio: -22
[   21.821567]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
[   31.650617] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_bclk_ratio on 1f000a4000.i2s: -22
[   31.650625]  AudioInjector audio: ASoC: error at snd_soc_link_hw_params on AudioInjector audio: -22
[   31.650638]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
[   31.651472] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_bclk_ratio on 1f000a4000.i2s: -22
[   31.651478]  AudioInjector audio: ASoC: error at snd_soc_link_hw_params on AudioInjector audio: -22
[   31.651815]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
[   41.684473] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_bclk_ratio on 1f000a4000.i2s: -22
[   41.684483]  AudioInjector audio: ASoC: error at snd_soc_link_hw_params on AudioInjector audio: -22
[   41.684499]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
[   41.685435] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_bclk_ratio on 1f000a4000.i2s: -22
[   41.685440]  AudioInjector audio: ASoC: error at snd_soc_link_hw_params on AudioInjector audio: -22
[   41.685780]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
```

### Device (s)

Other

### System

RPi 5 only! Pi 4 works like a charm.
Latest Raspberry OS, with updated kernel 6.1.69-v8+ using 4k page variant (kernel8.img).

2023/12/14 16:43:25 
Copyright (c) 2012 Broadcom
version 0f881224 (release) (embedded)

Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2

kernel:
6.1.69-v8+

### Logs

Opening the audio device fails in snd_pcm_hw_params().

rafael2k, 2024-01-02:
Just adding - that in this state the wm8731 is totally unusable.

rafael2k, 2024-01-02:
for the record, the cconfig.txt:
```
kernel=kernel8.img

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=off

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
#camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,noaudio
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1
disable_splash=1

# Run as fast as firmware / board allows
arm_boost=1

dtoverlay=i2c-gpio,bus=2,i2c_gpio_sda=13,i2c_gpio_scl=6

dtoverlay=audioinjector-wm8731-audio
dtoverlay=i2s-mmap

dtoverlay=disable-bt
```
Full dmesg:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.1.69-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1711 SMP PREEMPT Thu Dec 21 14:27:49 GMT 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001a400000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 1024 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[... 489 lines cut ...]
[    9.113439]  bus_probe_device+0xa4/0xb0
[    9.113442]  deferred_probe_work_func+0x90/0xc8
[    9.113445]  process_one_work+0x208/0x480
[    9.113451]  worker_thread+0x50/0x428
[    9.113456]  kthread+0x114/0x120
[    9.113461]  ret_from_fork+0x10/0x20
[    9.113465] ---[ end trace 0000000000000000 ]---
[    9.114261] audioinjector-stereo soc:sound: ASoC: driver name too long 'audioinjector-pi-soundcard' -> 'audioinjector-p'
[    9.116176] audioinjector-stereo soc:sound: successfully loaded
[    9.197209] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
[    9.975949] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   11.402490] macb 1f00100000.ethernet eth0: PHY [1f00100000.ethernet-ffffffff:01] driver [Broadcom BCM54213PE] (irq=POLL)
[   11.402500] macb 1f00100000.ethernet eth0: configuring for phy/rgmii-id link mode
[   11.406358] pps pps0: new PPS source ptp0
[   11.406430] macb 1f00100000.ethernet: gem-ptp-timer ptp clock registered.
[   11.414074] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   16.959338] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   19.242206] tun: Universal TUN/TAP device driver, 1.6
[   20.873742] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_bclk_ratio on 1f000a4000.i2s: -22
[   20.873753]  AudioInjector audio: ASoC: error at snd_soc_link_hw_params on AudioInjector audio: -22
[   20.874621]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
[   20.886061] designware-i2s 1f000a4000.i2s: ASoC: error at snd_soc_dai_set_bclk_ratio on 1f000a4000.i2s: -22
[   20.886070]  AudioInjector audio: ASoC: error at snd_soc_link_hw_params on AudioInjector audio: -22
[   20.886085]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
```

rafael2k, 2024-01-02:
Also, the schematics of the radio (includes the wiring between the wm8731 and the Pi), just in case: https://github.com/afarhan/sbitx/blob/main/sbitx_v3_production.pdf

rafael2k, 2024-01-02:
Also tested with kernel 6.6.8-v8+, same issue.

pelwell, 2024-01-03:
That soundcard driver is using unusual bclk_ratio values, apparently aiming to maintain a constant bclk frequency of 12MHz rather than scaling with the sample size and frequency. But reading the dwc_i2s bclk_ratio handling again and comparing it to that of other I2S blocks shows that:
1. the value would likely have been ignored, and
2. it should only be a factor in clock producer mode.

https://github.com/raspberrypi/linux/pull/5819 should address both of those, and may solve the problem with your soundcard. Wait about 45 minutes for the build/checks to complete, then run `sudo rpi-update pulls/5819` to install a trial kernel (usualy caveats apply).

pelwell, 2024-01-03:
The build is ready now, if you'd like to give it a try.

rafael2k, 2024-01-03:
OK, testing now. Thanks @pelwell !!

rafael2k, 2024-01-03:
Roger that! Loud and clear!

All good, sound card working again. Thanks @pelwell  !

pelwell, 2024-01-03:
Thanks. I should probably do some basic testing of a few other soundcards before merging...

rafael2k, 2024-01-03:
Cool! Do you close this one when you do these checks?
: )

pelwell, 2024-01-03:
Will do.

pelwell, 2024-01-04:
Now tested and merged.

closed 2024-01-04 by pelwell

rafael2k, 2024-02-06:
Should I open this again?
For me, 6.1.74 broke again the wm8731 "snd_soc_audioinjector_pi_soundcard", but now in the Pi 4. 6.1.73 was working fine. I'm getting same kind of errors:
```
[   84.950454] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  138.433994] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[  211.219306] bcm2835-i2s fe203000.i2s: ASoC: error at snd_soc_dai_hw_params on fe203000.i2s: -22
[  211.220038]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
[  215.901118] bcm2835-i2s fe203000.i2s: ASoC: error at snd_soc_dai_hw_params on fe203000.i2s: -22
[  215.902089]  AudioInjector audio: ASoC: error at __soc_pcm_hw_params on AudioInjector audio: -22
```