raspberrypi/linux #5825: can't turn off ethernet leds in bookworm on Pi5

acavalin, 2024-01-04:
**Describe the bug**
Turning off ethernet leds via `/boot/firmware/config.txt` with the latest brand new installed Pi OS 64-bit works on a **Pi4** but not on **Pi5**.

**To reproduce**
Add these lines in `/boot/firmware/config.txt` right after the line `arm_boost=1`:
~~~
dtparam=eth_led0=4
dtparam=eth_led1=4
~~~

**Expected behaviour**: Ethernet leds are both off.

**Actual behaviour**: Ethernet leds are on (`act` blinking).

**System**

* Raspberry Pi 5 Model B Rev 1.0, 8GB RAM
* Which OS and version (`cat /etc/rpi-issue`)?
  ~~~
  Raspberry Pi reference 2023-10-10
  Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2
  ~~~
* Which firmware version (`vcgencmd version`)?
  ~~~
  2023/12/14 16:43:25 
  Copyright (c) 2012 Broadcom
  version 0f881224 (release) (embedded)
  ~~~
* Which kernel version (`uname -a`)?
  ~~~
  Linux raoul 6.1.66-v8-16k+ raspberrypi/firmware#1 SMP PREEMPT Wed Dec 13 15:18:43 UTC 2023 aarch64 GNU/Linux
  ~~~

**Logs**
<details>
  <summary>`dmesg` log</summary>
  
  ```
  [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
  [    0.000000] Linux version 6.1.66-v8-16k+ (runner@fv-az1429-763) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) raspberrypi/firmware#1 SMP PREEMPT Wed Dec 13 15:18:43 UTC 2023
  [    0.000000] random: crng init done
  [    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
  [    0.000000] efi: UEFI not found.
  [    0.000000] Reserved memory: created CMA memory pool at 0x000000001a000000, size 320 MiB
  [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
  [    0.000000] Zone ranges:
  [    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
  [    0.000000]   DMA32    empty
  [    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
  [    0.000000] Movable zone start for each node
  [    0.000000] Early memory node ranges
  [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
  [    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
  [    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
  [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
  [    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
  [    0.000000] psci: probing for conduit method from DT.
  [    0.000000] psci: PSCIv1.1 detected in firmware.
  [    0.000000] psci: Using standard PSCI v0.2 function IDs
  [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
  [    0.000000] psci: SMC Calling Convention v1.2
  [    0.000000] percpu: Embedded 14 pages/cpu s177448 r8192 d43736 u229376
  [    0.000000] pcpu-alloc: s177448 r8192 d43736 u229376 alloc=14*16384
  [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
  [    0.000000] Detected PIPT I-cache on CPU0
  [    0.000000] CPU features: detected: Virtualization Host Extensions
  [    0.000000] CPU features: detected: Hardware dirty bit management
  [    0.000000] CPU features: detected: Spectre-v4
  [    0.000000] CPU features: detected: Spectre-BHB
  [    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[... 711 lines cut ...]
  #   1. https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/178#issuecomment-1386871175
  #   2. https://raspberrypi.stackexchange.com/questions/116677/force-wired-ethernet-speed-to-100-full/140550#140550
  dtparam=eee=off
  dtoverlay=disable-gigabit-ethernet

  # Disable the PWR & Activity LEDs
  dtparam=pwr_led_trigger=default-on
  dtparam=pwr_led_activelow=off
  dtparam=act_led_trigger=none
  dtparam=act_led_activelow=off
  [all]
  ```
</details>

<details>
  <summary>`/boot/firmware/cmdline.txt`</summary>
  
  ```
  console=serial0,115200 console=tty1 root=PARTUUID=a0c4eab0-02 rootfstype=ext4 fsck.repair=yes rootwait ipv6.disable=1 panic=5 consoleblank=0 plymouth.enable=0 dwc_otg.lpm_enable=0 usbhid.quirks=0x0463:0xffff:0x08 printk.time=1 net.ifname
s=0
  ```
</details>

**Additional context**
I'm currently using the patched kernel in [linux#5753](https://github.com/raspberrypi/linux/issues/5753) (`sudo rpi-update pulls/5788`) but the ethernet leds problem were present in the stock firmware too.

pelwell, 2024-01-04:
Fixed by https://github.com/raspberrypi/linux/pull/5826.

If you wait about 45 minutes you'll be able to install a trial kernel (although it's really just the Pi 5 dtb and README that has changed) using sudo rpi-update pulls/5826.

acavalin, 2024-01-04:
Thank you :smiley: 
If I pull this trial kernel, will it have the #5753 patch included?
Sorry for the off topic question :pray: : is it possible to know when the #5753 patch will be included in the production kernel?

pelwell, 2024-01-04:
If I pull this trial kernel, will it have the https://github.com/raspberrypi/linux/issues/5753 patch included?

Yes.

> Sorry for the off topic question 🙏 : is it possible to know when the https://github.com/raspberrypi/linux/issues/5753 patch will be included in the production kernel?

It will be in the next _stable_ release. There's no hard schedule for that, but it's likely to be before the end of February, and possibly sooner.

closed 2024-01-04 by pelwell