raspberrypi/linux #5828: [v3d][KMS/DRM][Raspberry Pi 5] drmModePageFlip() fails when DRM_MODE_PAGE_FLIP_ASYNC flag is set.

bluestang2006, 2024-01-04:
### Describe the bug

Linking a previously corrupted gfx issue when using OpenGL & SDL2 - https://gitlab.freedesktop.org/mesa/mesa/-/issues/10242

```
Calling drmModePageFlip() with the DRM_MODE_PAGE_FLIP_ASYNC flag fails on the Raspberry Pi 5.
This is a major problem since it breaks SDL2 on KMS/DRM on the platform.
We detect the DRM_CAP_ASYNC_PAGE_FLIP cap here:
https://github.com/libsdl-org/SDL/blob/058213366bccf7f62d4c4c6061477d7a69ea336d/src/video/kmsdrm/SDL_kmsdrmvideo.c#L957
...where it's detected as available on both Pi 4 and Pi5...
And then we set the flag here:
https://github.com/libsdl-org/SDL/blob/058213366bccf7f62d4c4c6061477d7a69ea336d/src/video/kmsdrm/SDL_kmsdrmopengles.c#L176
And call drmModePageFlip here with the DRM_CAP_ASYNC_PAGE_FLIP flag here:
https://github.com/libsdl-org/SDL/blob/058213366bccf7f62d4c4c6061477d7a69ea336d/src/video/kmsdrm/SDL_kmsdrmopengles.c#L179
[@apinheiro](https://github.com/apinheiro) Can you please take a look? There are many projects using SDL2 on the Pi4/5.
```

### Steps to reproduce the behaviour

This can easily be duplicated with [Quake3e](https://github.com/ec-/Quake3e) using the opengl1 renderer.

### Device (s)

Other

### System

```
System:
  Host: raspberrypi Kernel: 6.1.0-rpi7-rpi-2712 arch: aarch64 bits: 64
    compiler: gcc v: 12.2.0 Desktop: LXDE v: 0.10.1 wm: Openbox dm: LightDM
    Distro: Debian GNU/Linux 12 (bookworm)
CPU:
  Info: quad core model: N/A variant: cortex-a76 bits: 64 type: MCP
    arch: ARMv8 rev: 1
  Speed (MHz): avg: 2400 min/max: 1000/2400 cores: 1: 2400 2: 2400 3: 2400
    4: 2400 bogomips: 432
  Features: Use -f option to see features
Graphics:
  Device-1: bcm2712-hdmi0 driver: vc4_hdmi v: N/A bus-ID: N/A
    chip-ID: brcm:107c701400
  Device-2: bcm2712-hdmi1 driver: vc4_hdmi v: N/A bus-ID: N/A
    chip-ID: brcm:107c706400
  Display: x11 server: X.Org v: 1.21.1.7 with: Xwayland v: 22.1.9
    compositor: xcompmgr driver: X: loaded: modesetting unloaded: fbdev dri: vc4
    gpu: vc4-drm,vc4_crtc,vc4_dpi,vc4_dsi,vc4_firmware_kms,vc4_hdmi,vc4_hvs,vc4_txp,vc4_v3d,vc4_vec
    display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 96
  Monitor-1: HDMI-A-1 mapped: HDMI-1 model: Gigabyte M27Q res: 1920x1080
    dpi: 82 diag: 684mm (26.9")
  API: OpenGL v: 3.1 Mesa 23.2.1-1~bpo12+rpt2 renderer: V3D 7.1
    direct-render: Yes
```

### Additional context

The solution was posted by one of the Igalia devs:
```
This issue was introduced in [this commit](https://github.com/raspberrypi/linux/commit/5fd6ee7fd084838e09d4e463ae53cd9aaa7fce70) and if you revert it, you will be able to fix the problem. For example, in the 6.7 kernel, this commit was already reverted (https://github.com/raspberrypi/linux/commit/9ef6dbb755b6445888418aa7212fe085543fbbc5). I'll try to let RPi developers know about this info. Thanks for sharing the issue!
```

6by9, 2024-01-04:
We're already discussing it directly with Igalia.

6.7 had reverted the iommu patches due to a kernel API change that hasn't been resolved as yet. Doing so means all image buffers have to come from the CMA heap which will then fragment, so that is a regression.

#5827 is the correct fix - the driver wasn't updating the display list correctly for Pi5, so ended up writing an incorrect address as the image pointer.

bluestang2006, 2024-01-04:
> We're already discussing it directly with Igalia.
>
> 6.7 had reverted the iommu patches due to a kernel API change that hasn't been resolved as yet. Doing so means all image buffers have to come from [...]

thank you! I assume this will be back-ported into 6.1? Feel free to close the issue as necessary.

6by9, 2024-01-17:
Fix and backport have been merged.