raspberrypi/linux #5831: vc4-kms-dsi-waveshare-panel overlay with 7_0_inchC

Romain-Donze, 2024-01-05:
### Describe the bug

within the vc4-kms-dsi-waveshare-panel overlay, it is state that the 7_0_inchC display resolution is 800x480 even though in the README it is 7.0" C 1024x600.

From the waveshare documentation, the 7.0" C is indeed 1024x600 (https://www.waveshare.com/wiki/7inch_DSI_LCD_(C))

### Steps to reproduce the behaviour

With Raspberry Pi OS Lite (64-bit)

config.txt:
dtoverlay=vc4-kms-v3d
dtoverlay=vc4-kms-dsi-waveshare-panel,7_0_inchC

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi OS Lite (64-bit)

pelwell, 2024-01-05:
Any comments, @waveshare?

6by9, 2024-01-05:
Display and touch resolutions can differ - see the 7.9" panel where touch has to be defined as 4096x4096.

https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/vc4-kms-dsi-waveshare-panel-overlay.dts#L93-L94 sets the touch resolution. 
https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/gpu/drm/panel/panel-waveshare-dsi.c#L80-L93 defines the panel resolution.

I don't have the 7.0" C panel to test with.
Do you have one and are observing issues, or is this purely a code review comment?

Romain-Donze, 2024-01-06:
So after tinkering a bit, I found out that my problem was due to the fact that I did'nt set correct params to i2c1 so after adding 

dtoverlay=vc4-kms-dsi-waveshare-panel,7_0_inchC,i2c1 to config.txt it works ok with With Raspberry Pi OS Lite. I now have some issue making it work with a custom yocto build so I created an issue in meta-raspberrypi instead

https://github.com/agherzan/meta-raspberrypi/issues/1260

So I guess that if it is intended to write:

7_0_inchC = <&panel>, "compatible=waveshare,7.0inch-c-panel",
				   <&touch>, "touchscreen-size-x:0=800",
				   <&touch>, "touchscreen-size-y:0=480";

Then we can close this issue

pelwell, 2024-01-06:
If it's working for you, i.e. the display looks correct with no scaling artifacts, and the touchscreen is behaving itself, then I think we're done.

closed 2024-01-06 by pelwell