raspberrypi/linux #5844: bookworm ignores blacklisting of hid_logitech_dj

steinhelten, 2024-01-11:
I need to blacklist hid_logitech_dj for my KVM switch to work.  Unfortunately Bookworm seems to ignore the blacklist for this, even though it is shown by modprobe --showconfig as blacklisted.

I have done this by putting a file containing
blacklist hid_logitech_dj
in the /etc/modprobe.d/ directory, as I have done with previous releases.  After booting, it subsequently appears as blacklisted, but has been loaded anyway!  If I ssh into the system or directly connect a keyboard, and remove the module by hand (modprobe -r hid_logitech_dj), my keyboard and mouse (trackpad) suddenly begin working through the switch.

Please note!  While the rest of the blacklisted modules are in alphabetic order, hid_logitech_dj is NOT.  A high-efficiency linear search of a sorted list may stop when the tested value is greater than the search value - I am just wondering if someone decided to optimize the list search by making it early-terminating without the realization that it may not always be sorted.

```
`
# uname -a
Linux bringebaer 6.1.0-rpi7-rpi-v7 raspberrypi/bookworm-feedback#1 SMP Raspbian 1:6.1.63-1+rpt1 (2023-11-24) armv7l GNU/Linux
# cat /etc/debian_version
12.1
# ls -l /etc/modprobe.d
total 8
-rw-r--r-- 1 root root 17 May 10  2023 blacklist-8192cu.conf
-rw-r--r-- 1 root root 26 Jan  2 16:49 raspi-blacklist.conf
# cat /etc/modprobe.d/raspi-blacklist.conf
blacklist hid_logitech_dj
# modprobe --showconfig | grep blackl
blacklist 8192cu
blacklist arkfb
blacklist aty128fb
blacklist atyfb
blacklist radeonfb
blacklist cirrusfb
blacklist cyber2000fb
blacklist kyrofb
blacklist matroxfb_base
blacklist mb862xxfb
blacklist neofb
blacklist pm2fb
blacklist pm3fb
blacklist s3fb
blacklist savagefb
blacklist sisfb
blacklist tdfxfb
blacklist tridentfb
blacklist vt8623fb
blacklist hid_logitech_dj
# lsmod | grep _dj
hid_logitech_dj        32768  0
# modprobe -r hid_logitech_dj
# lsmod | grep _dj
#
`
```

steinhelten, 2024-01-11:
OK.  Two things - First, by "dongle" I assume you meant the cable to the KVM switch.  If that's not right, I'll have to do this over.  Second, I did this precisely in the order given above, though it seemed a bit odd to me.  I would have thought you'd have wanted me to restore the blacklist file before rebooting, but I did it as you requested, and rebooted before restoring the blacklist file.
[no_blacklist.txt](https://github.com/raspberrypi/linux/files/13907614/no_blacklist.txt)
[blacklist.txt](https://github.com/raspberrypi/linux/files/13907615/blacklist.txt)

steinhelten, 2024-01-11:
If I'm interpretting the contents of those files correctly, it would appear that the blacklist "worked" in that it prevented the driver from being loaded when the KVM switch is connected after booting, and my concerns about rebooting before restoring the file were unfounded.  So, what's difference between it being connected at boot time, or after boot time?

XECDesign, 2024-01-11:
I can't verify right now, but I have a suspicion it might be initramfs shenanigans. 

Does running `sudo update-initramfs -u` fix it?

steinhelten, 2024-01-12:
OK!  That worked!  I booted with the KVM switch attached, and it came up with the keyboard and mouse (trackpad) working, and the driver not loaded.  So... what's the long-term solution?

XECDesign, 2024-01-12:
I think that is the solution. If you looks up how to blacklist modules, the instructions should tell you that that's a necessary step.

Like here - https://wiki.debian.org/KernelModuleBlacklisting

pelwell, 2024-01-12:
Thanks, @XECDesign.

closed 2024-01-12 by pelwell

steinhelten, 2024-01-12:
Interesting.  I don't remember having to do that under buster.  Can you help walk me though the steps for preparing a new system, then?  This is unclear to me, because if I just copy an image onto a new SD card, and boot it, I'm stuck - I can't use the keyboad and mouse, and I can't ssh in because that's off by default.  To get around that this time, I physically disconnected the keyboard from the KVM switch and plugged it in directly to the RPi.  Obviously, I'd prefer to not have to do that each time I create a new system image.

1. image a new SD card
2. mount the SD card (I use an RPI to copy the image, so this should be easy), say, on /mnt
3. echo "blacklist hid_logitech_dj"  > /mnt/etc/modprobe.d/raspi-blacklist.conf
4. then... how do I do a "non-native" initramfs?  OR what's next?

XECDesign, 2024-01-12:
It wasn't a thing under buster or bullseye because we didn't use initramfs then. It's something we just added with bookworm.

Modules can also be blacklisted in cmdline.txt with `module_blacklist=whatever`. I haven't tried it, so I don't know if that helps you. If it does, does that solve your problem?

steinhelten, 2024-01-12:
It just might!  I did the following:

1. commented out the blacklist line in /etc/modprobe.d/raspi-blacklist.conf
2. rebooted
3. confirmed that a) the module was loaded and b) the keyboard and mouse did not work
4. Executed the following:  
```
#!/bin/sh
ed /boot/cmdline.txt <<EOF
s/$/ module_blacklist=hid_logitech_dj/
wq
EOF
```
5. rebooted
6. Verified that the module was NOT loaded, and that the keyboard and mouse worked, which they did

Now, curiously, hid_logitech_dj does NOT appear in the blacklist using "modprobe --showconfig", which might be confusing under some circumstances.  But, by including a version of the above shell script, (including /mnt in the path to cmdline.txt), and including it in my "standup" script, I think I can successfully create viable images.

Thanks!

XECDesign, 2024-01-12:
Heads up that it should be `/boot/firmware/cmdline.txt` (another bookworm change).

/boot/cmdline.txt is a symlink, so it may work, but some tools (like `sed`) will overwrite the symlink with an actual file. I'm guessing in this case it worked because `ed` doesn't do that, or you mounted /boot manually.

steinhelten, 2024-01-12:
Yikes!  It appears that ed was well-behaved in this respect, because I checked and the symlink was preserved. 
 Since the symlink is a relative and not an absolute reference, it would probably work in the non-native case (having it mounted on /mnt), but I'll change my script to use /mnt/boot/firmware/cmdline.txt just to be on the safe side.

Again, thank you.

XECDesign, 2024-01-12:
No worries, glad it's sorted now.

lurch, 2024-01-14:
> and I can't ssh in because that's off by default.

It's easy to enable SSH using [Raspberry Pi Imager](https://www.raspberrypi.com/documentation/computers/getting-started.html#install-using-imager).

steinhelten, 2024-01-15:
Well, while that's a great suggestion in general, I happen to be running FreeBSD on the Pi that I currently use to make the Raspberry Pi OS SD images.

lurch, 2024-01-16:
Ah! In that case, see https://www.raspberrypi.com/documentation/computers/configuration.html#remote-access