raspberrypi/linux #5849: WM8960 failed to configure clock

Romain-Donze, 2024-01-14:
### Describe the bug

So I tried to add support for the WM8960 waveshare hat to my yocto image.

My build is based on Boot2Qt (see here: https://doc.qt.io/Boot2Qt/b2qt-how-to-create-b2qt-image.html)

The wm8960-soundcard does appear in alsamixer but when launching alsamixer, dmesg return a lot of error

```
[   18.736426] wm8960 1-001a: failed to configure clock
[   18.737942] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[   18.739445]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
[   18.742309] wm8960 1-001a: failed to configure clock
[   18.743812] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[   18.745349]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
[   18.748209] wm8960 1-001a: failed to configure clock
[   18.749769] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[   18.751272]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```

### Steps to reproduce the behaviour

config.txt

```
# Enable SPI bus
dtparam=spi=on

# Enable I2C
dtparam=i2c1=on
dtparam=i2c_arm=on

# Enable VC4 Graphics
dtoverlay=vc4-kms-v3d,noaudio,noaudio,noaudio,noaudio,noaudio,noaudio,noaudio
max_framebuffers=2

# Enable USB host mode
dtoverlay=dwc2,dr_mode=host

# Enable the onboard ALSA audio 
dtparam=audio=off 
 
# Automatically load overlays for detected DSI displays 
display_auto_detect=1 
 
# Enable waveshare panel 
dtoverlay=vc4-kms-dsi-waveshare-panel,7_0_inchC,i2c1

dtoverlay=i2s-mmap
dtparam=i2s=on
dtoverlay=wm8960-soundcard
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

Linux b2qt-raspberrypi4-64 6.1.54-v8 #1 SMP PREEMPT Wed Sep 20 14:13:53 UTC 2023 aarch64 GNU/Linux

pelwell, 2024-01-15:
Start by reducing your config.txt to that which is necessary:
* Delete either `dtparam=i2c1=on` or `dtparam=i2c_arm=on` - they are equivalent. You could actually delete both because of the `vc4-kms-dsi-waveshare-panel,i2c` and `dtoverlay=wm8960-soundcard` further down.
* Remove the repeated paramters in `dtoverlay=vc4-kms-v3d,noaudio,noaudio,noaudio,noaudio,noaudio,noaudio,noaudio` - once is enough.
* Delete `dtoverlay=i2s-mmap` - this overlay hasn't existed for years.
* `dtparam=i2s=on` is not necessary because the soundcard overlay automatically enables it.

You may save time by installing a stock RPiOS image onto a spare SD card, using the same config.txt, and running something like `speaker-test -c 2`.

Romain-Donze, 2024-01-15:
Thanks for your answer, I will try to update my config.txt.

One thing I did not mention is that speaker-test give me a clear "Front-left, Front-right". The error occurs only when launching alsamixer or when trying to acces the audio output through my Qt application. Then from my Qt application I am not able to ear anything. (It does work when I am only using the onboard audio so th problem does not come from herr)

pelwell, 2024-01-15:
It should be fine with Yocto - you can always override the paths to the boot and root directories if necessary.

Romain-Donze, 2024-01-15:
So, my first tries led me to this,

With RPiOS lite, everything seems fine aplay shows one card (wm8960-soundcard) and after settings the volume with alsamixer, speaker-test gives me "Front-left Front-right" and dmesg doesn't show any error (so no need for rpi-update there I guess)

With my yocto build however I still cannot make it work (even after correcting config.txt) so I will try the rpi-update to see what's wrong but I will first need to add curl to my distro

config.txt looks like this
```
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=off

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d,noaudio
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]
# Enable waveshare panel 
dtoverlay=vc4-kms-dsi-waveshare-panel,7_0_inchC,i2c1

dtoverlay=wm8960-soundcard
```

Romain-Donze, 2024-01-16:
So after giving-up trying to make rpi-update work with my distro, I directly added your patch to my yocto build and here are the results:

directly after boot
```
root@b2qt-raspberrypi4-64:~# dmesg | grep wm8960
[    5.521038] wm8960_set_dai_sysclk(0, 12288000, 0)
[    5.521103] wm8960_set_dai_sysclk() -> OK
```

then after trying alsamixer:
```
[   79.702325] wm8960_configure_clocking:
[   79.703605] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[   79.704869] wm8960_configure_sysclk(12288000) -> 0, 0, -1
[   79.706379] wm8960 1-001a: failed to configure clock
[   79.707977] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[   79.709286]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```
Repeated multiple time

Also, here is the result of lsmod:
```
Module                  Size  Used by
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  1 garp
llc                    16384  2 stp,garp
usb_f_fs               40960  3
usb_f_rndis            28672  2
u_ether                24576  1 usb_f_rndis
algif_hash             16384  1
aes_arm64              16384  2
algif_skcipher         16384  1
af_alg                 28672  6 algif_hash,algif_skcipher
bnep                   24576  2
libcomposite           69632  12 usb_f_rndis,usb_f_fs
ch341                  20480  0
usbserial              40960  1 ch341
joydev                 28672  0
spidev                 20480  0
rpivid_hevc            49152  0
brcmfmac              331776  0
hci_uart               45056  0
bcm2835_v4l2           45056  0
btbcm                  24576  1 hci_uart
brcmutil               24576  1 brcmfmac
bluetooth             585728  26 hci_uart,btbcm,bnep
v3d                    90112  1
videobuf2_vmalloc      16384  1 bcm2835_v4l2
bcm2835_isp            28672  0
cfg80211              937984  1 brcmfmac
bcm2835_codec          49152  0
bcm2835_mmal_vchiq     36864  3 bcm2835_codec,bcm2835_v4l2,bcm2835_isp
v4l2_mem2mem           40960  2 bcm2835_codec,rpivid_hevc
gpu_sched              49152  1 v3d
videobuf2_dma_contig    20480  3 bcm2835_codec,rpivid_hevc,bcm2835_isp
videobuf2_memops       16384  2 videobuf2_vmalloc,videobuf2_dma_contig
videobuf2_v4l2         32768  5 bcm2835_codec,bcm2835_v4l2,rpivid_hevc,v4l2_mem2mem,bcm2835_isp
videobuf2_common       69632  9 bcm2835_codec,videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,bcm2835_v4l2,rpivid_hevc,v4l2_mem2mem,videobuf2_memops,bcm2835_isp
drm_shmem_helper       24576  1 v3d
videodev              274432  7 bcm2835_codec,videobuf2_v4l2,bcm2835_v4l2,videobuf2_common,rpivid_hevc,v4l2_mem2mem,bcm2835_isp
panel_waveshare_dsi    16384  0
snd_soc_simple_card    20480  0
goodix_ts              28672  0
raspberrypi_hwmon      16384  0
backlight              24576  1 panel_waveshare_dsi
dwc2                  192512  0
snd_soc_simple_card_utils    28672  1 snd_soc_simple_card
ecdh_generic           16384  1 bluetooth
ecc                    36864  1 ecdh_generic
i2c_brcmstb            16384  0
rfkill                 32768  5 bluetooth,cfg80211
roles                  20480  1 dwc2
snd_bcm2835            28672  0
vc_sm_cma              32768  2 bcm2835_mmal_vchiq,bcm2835_isp
spi_bcm2835            24576  0
snd_soc_bcm2835_i2s    20480  2
mc                     61440  7 videodev,bcm2835_codec,videobuf2_v4l2,videobuf2_common,rpivid_hevc,v4l2_mem2mem,bcm2835_isp
uio_pdrv_genirq        16384  0
uio                    24576  1 uio_pdrv_genirq
sch_fq_codel           20480  6
i2c_dev                20480  0
i2c_bcm2708            16384  0
fuse                  139264  1
ipv6                  557056  40
```

P.S just so you know I am building following this: https://github.com/agherzan/meta-raspberrypi/pull/861

pelwell, 2024-01-17:
It looks like ALSA is requesting a combination of channels + sample size + sample rate that the driver thinks it can't support. Please add one more debug line (https://github.com/pelwell/linux/commit/e2ec5a3f7d21a3f9ff6024b41de9abaa0e5fb283), then we should have all the information required to work out if the request is reasonable, and hopefully how to proceed.

Romain-Donze, 2024-01-17:
So here is the result with the new patch

```
[   64.168121] wm8960_hw_params: channels 2, width 16, bclk 1411200, rate/lrclk 44100
[   64.171198] wm8960_configure_clocking:
[   64.172705] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[   64.174016] wm8960_configure_sysclk(12288000) -> 0, 0, -1
[   64.175320] wm8960 1-001a: failed to configure clock
[   64.176699] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[   64.178008]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```

I've gotta say I don't quite understand what's appening

pelwell, 2024-01-17:
I now understand the error messages, but not why it isn't affecting everyone.

The complaint is that there is no integer (*) divisor from the sysclk/mclk of 12288000 to the bclk of 1411200 (which is 44100 * 16 * 2).

I'm guessing that speaker-test uses a different frequency - 32kHz, perhaps.

(*) It's actually even more restrictive than that - there's a list of legal divisors - but you get the idea.

Romain-Donze, 2024-01-18:
Well it works with RPiOS apparently so I guess that's why noone noticed it. 

I have to say still that even speaker-test does not work now. At first, I was including [asound.conf](https://github.com/waveshare/WM8960-Audio-HAT/blob/master/asound.conf) and [wm8960_asound.state](https://github.com/waveshare/WM8960-Audio-HAT/blob/master/wm8960_asound.state) from waveshare repo and with that speaker-test worked but I had the error when running alsamixer or when running my Qt app.

But since I rebuilt my distro to add the debug patches, I did not bother to add asound.conf and asound.state.

So perhaps I can try to add them and see what is going on between those differents states:

- RPiOS (should work by only adding dtoverlay=wm8960-soundcard to config.txt https://github.com/raspberrypi/linux/issues/5849#issuecomment-1892750611)
- custom yocto build
- custom build + asound.conf and asound.state

Romain-Donze, 2024-01-18:
So, there are the results of the differents tests I ran

Simple yocto build: (/etc/asound.conf and /var/lib/alsa/asound.state are empty)
Right after boot
```
root@b2qt-raspberrypi4-64:~# dmesg | grep wm8960
[    5.891741] wm8960_set_dai_sysclk(0, 12288000, 0)
[    5.891804] wm8960_set_dai_sysclk() -> OK
```

After speaker-test -t wav -c 2 -l 1
```
root@b2qt-raspberrypi4-64:~# dmesg | grep wm8960
[  461.480981] wm8960_hw_params: channels 2, width 16, bclk 1411200, rate/lrclk 44100
[  461.482692] wm8960_configure_clocking:
[  461.484000] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[  461.485376] wm8960_configure_sysclk(12288000) -> 0, 0, -1
[  461.486827] wm8960 1-001a: failed to configure clock
[  461.488623] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[  461.490153]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```

After alsamixer (It shows only the master channel and I have to press F6 to select the wm8960-soundcard)
```
root@b2qt-raspberrypi4-64:~# dmesg | grep wm8960
[  510.455464] wm8960_hw_params: channels 2, width 16, bclk 1411200, rate/lrclk 44100
[  510.458511] wm8960_configure_clocking:
[  510.459825] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[  510.461175] wm8960_configure_sysclk(12288000) -> 0, 0, -1
[  510.462482] wm8960 1-001a: failed to configure clock
[  510.463768] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[  510.465107]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```

Yocto build with asound.conf+wm8960_asound.state:
Right after boot
```
root@b2qt-raspberrypi4-64:~# dmesg | grep wm8960
[    6.041760] wm8960_set_dai_sysclk(0, 12288000, 0)
[    6.041830] wm8960_set_dai_sysclk() -> OK
```

After speaker-test -t wav -c 2 -l 1 (no error but no sound)
```
root@b2qt-raspberrypi4-64:~# dmesg | grep wm8960
[   90.295595] wm8960_hw_params: channels 2, width 16, bclk 1536000, rate/lrclk 48000
[   90.296021] wm8960_configure_clocking:
[   90.296058] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[   90.296101] wm8960_configure_sysclk(12288000) -> 0, 0, 7
```

After alsamixer (It shows only the master channel and I have to press F6 to select the wm8960-soundcard) (since I was there I cranked the volume up)
```
root@b2qt-raspberrypi4-64:~# dmesg | grep wm8960
[  193.452224] wm8960_hw_params: channels 2, width 16, bclk 1411200, rate/lrclk 44100
[  193.454346] wm8960_configure_clocking:
[  193.455669] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[  193.456989] wm8960_configure_sysclk(12288000) -> 0, 0, -1
[  193.458336] wm8960 1-001a: failed to configure clock
[  193.459642] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[  193.460954]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```

Rerun speaker-test -t wav -c 2 -l 1 (now I have sound)
```
root@b2qt-raspberrypi4-64:~# dmesg | grep wm8960
[  268.937601] wm8960_hw_params: channels 2, width 16, bclk 1536000, rate/lrclk 48000
[  268.939951] wm8960_configure_clocking:
[  268.941277] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[  268.942659] wm8960_configure_sysclk(12288000) -> 0, 0, 7
```

Barebone RPiOS lite + rpi-update pulls/5850:
Right after boot
```
pi@raspberrypi:~ $ dmesg | grep wm8960
[   16.931132] wm8960_set_dai_sysclk(0, 12288000, 0)
[   16.931152] wm8960_set_dai_sysclk() -> OK
```

After speaker-test -t wav -c 2 -l 1 (Front-left, Front-right loud and clear)
```
pi@raspberrypi:~ $ dmesg | grep wm8960
[   74.656102] wm8960_hw_params: channels 2, width 16, bclk 1536000, rate/lrclk 48000
[   74.660138] wm8960_configure_clocking:
[   74.660153] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[   74.660165] wm8960_configure_sysclk(12288000) -> 0, 0, 7
```

After alsamixer (It directly select the wm8960-soundcard with all its channel)
```
pi@raspberrypi:~ $ dmesg | grep wm8960
nothing more
```

I also have some difference in /proc/asound/devices

Yocto
```
root@b2qt-raspberrypi4-64:~# cat /proc/asound/devices
  0: [ 0]   : control
 16: [ 0- 0]: digital audio playback
 24: [ 0- 0]: digital audio capture
 33:        : timer
```

RPiOS lite
```
pi@raspberrypi:~ $ cat /proc/asound/devices
  2: [ 0- 0]: digital audio playback
  3: [ 0- 0]: digital audio capture
  4: [ 0]   : control
 33:        : timer
```

I really don't know if all of this is usefull but I hope it could help.

pelwell, 2024-01-18:
It is very useful. Firstly, it confirms that sample rate is the fundamental problem, with 48000 working and 44100 not. Secondly, it prompted me to wonder why that might be, and my current thinking is that the audio mixing service - pulse audio or pipewire - is probably sample-rate converting everything to 48kHz. Your Yocto image is probably vanilla ALSA.

One of the configuration files you linked to referred to resampling using the samplerate module (not speex), so you could look enabling that in your build.

Romain-Donze, 2024-01-18:
Also here is the result with my Qt app

```
[  138.146533] wm8960_configure_clocking:
[  138.147843] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[  138.149139] wm8960_configure_sysclk(12288000) -> 0, 0, -1
[  138.150413] wm8960 1-001a: failed to configure clock
[  138.151674] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[  138.152972]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```

Romain-Donze, 2024-01-18:
So what module exactly should I add?

pelwell, 2024-01-18:
The problem you are having is with a standard Linux driver, with third party hardware, and a third party operating system build. I've pointed you in the right direction - get ALSA to do sample rate conversion, or consider adding pulse audio - and that's where my contribution is going to end, unless somebody comes up with a driver patch rise magically fixes the problem.

popcornmix, 2024-01-19:
> It is very useful. Firstly, it confirms that sample rate is the fundamental problem, with 48000 working and 44100 not. Secondly, it prompted me to wonder why that might be, and my current thinking [...]

```
$ grep "default.clock.allowed"  /usr/share/pipewire/pipewire.conf 
    #default.clock.allowed-rates = [ 48000 ]
```
You can enable others here, but it ships with only 48000Hz enabled (mostly as pipewire has a bug of not setting the AES bits which means no hdmi audio at frequencies other than 48000Hz - see https://github.com/raspberrypi/linux/issues/5525)

pelwell, 2024-01-19:
For vanilla ALSA you probably need something like this (untested):
```
pcm.!default {
        type plug
        slave.pcm "dmixer"
}

pcm.dmixer {
        type dmix
        ipc_key 1024
        slave {
                pcm "hw:0,0"
                channels 2
                rate 48000
                format S32_LE
        }
}
```
or perhaps just this:
```
pcm.rate_convert {
    type plug
    slave {
        pcm "hw:0,0"
        rate 48000
    }
}
```
Note that you may need to change the "hw:" numbers to match your setup.

Romain-Donze, 2024-01-20:
I did some more testing with RPiOS since it was working I wanted to find what was different between RPiOS and my yocto build. And it came out that on RPiOS pulseaudio wasn't installed and it is on my yocto build (it is installed by QtMultimedia)

So I installed pulseaudio on RPiOS `sudo apt install pulseaudio` and now I have the same errors directly at boot
```
[   37.043801] wm8960_hw_params: channels 2, width 16, bclk 1411200, rate/lrclk 44100
[   37.044119] wm8960_configure_clocking:
[   37.044122] wm8960_configure_clocking: clk_id 0, freq_in 0, freq_out 12288000
[   37.044125] wm8960_configure_sysclk(12288000) -> 0, 0, -1
[   37.044129] wm8960 1-001a: failed to configure clock
[   37.044132] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[   37.044136]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```

So maybe pulseaudio is breaking things?

Romain-Donze, 2024-01-20:
And so here it is the solution!

Since pulseaudio is installed it was just a matter of configuring it correctly by simply adding `default-sample-rate = 48000` to `~/.pulse/daemon.conf`

I wont close the issue just yet as I will still do some testing

Romain-Donze, 2024-01-21:
So the exact solution used to solve this problem on my distro can be found here

https://github.com/Embi-OS/meta-embi

For those wondering I had to override pulseaudio recipes to edit `daemon.conf`. There is also some more config to add into /boot/config.txt to disable every other audio interface.

closed 2024-01-21 by Romain-Donze

talksik, 2024-06-29:
Thanks for this thread @Romain-Donze . I was able to solve my problem by:

## Brute force trying to solve with various approaches
1. adding the "noaudio" param for `vc4-kms-v3d` in config.txt and making sure i2s was working
2. I also had to do this as noted in [this guide](https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#wm8960-soundcard-support)
```bash
amixer -c1 sset 'Headphone',0 80%,80%
amixer -c1 sset 'Speaker',0 80%,80%
amixer -c1 sset 'Left Input Mixer Boost' toggle
amixer -c1 sset 'Left Output Mixer PCM' toggle
amixer -c1 sset 'Right Input Mixer Boost' toggle
amixer -c1 sset 'Right Output Mixer PCM' toggle
```
3. I had to (I think) add the alsa configuration that @pelwell posted to `/etc/asound.conf` which I believe ensures that 48000 Hz is used. To my knowledge, this prevented the need for the change in `/etc/pulse/daemon.conf`

## One cause of errors
Something to consider is that simply running `amixer` caused my `dmesg` to have the same error:
```
[ 1386.266839] wm8960 1-001a: failed to configure clock
[ 1386.267224] wm8960 1-001a: ASoC: error at snd_soc_dai_hw_params on wm8960-hifi: -22
[ 1386.267987]  bcm2835-i2s-wm8960-hifi: ASoC: error at __soc_pcm_hw_params on bcm2835-i2s-wm8960-hifi: -22
```

However, running `amixer -c0` works fine without outputting errors in `dmesg`. I am a newbie to alsa and linux/embedded in general, so I don't know why this is, but at least the sound works now.

## Test
I plugged in stereo headphones into my audio hat headphone jack and it works crystal clear:
`speaker-test -t wav -c2 -l 2`

## My setup in case it helps someone
I am customizing a yocto image for Qt as well. Qt version 6.6.2. 

Note: I am a noob to yocto as well, just brute forcing.

Here is my config for meta-raspberrypi:
```
KERNEL_MODULE_AUTOLOAD += "snd-soc-wm8960"

RPI_EXTRA_CONFIG = "# \
Enable the onboard ALSA audio \n\
dtparam=audio=off \n\
dtparam=i2s=on \n\
dtparam=i2c_arm=on \n\
dtoverlay=vc4-kms-v3d,noaudio \n\
dtoverlay=vc4-kms-dsi-waveshare-panel,2_8_inch \n\
"
```

My local.conf so that I can test with aplay and arecord and also edit files with vim:
```
IMAGE_INSTALL:append = " curl vim alsa-state alsa-dev alsa-utils alsa-tools alsa-plugins"
```

talksik, 2024-06-29:
One question for @pelwell, how can I change the asound.conf to make arecord work?

```
root@b2qt-raspberrypi4-64:~# arecord test.wav
ALSA lib /usr/src/debug/alsa-lib/1.2.8-r0/src/pcm/pcm_dmix.c:972:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
arecord: main:831: audio open error: Invalid argument
```

talksik, 2024-06-29:
Fiddled a little more and got this for my asound.conf

```
pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixer"
}

pcm.capture {
    type plug
    slave.pcm "dsnooped"
}

pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:0,0"
        channels 2
        rate 48000
        format S32_LE
    }
}

pcm.dsnooped {
    type dsnoop
    ipc_key 2048
    slave {
        pcm "hw:0,0"
        channels 2
        rate 48000
        format S32_LE
    }
}

ctl.!default {
    type hw
    card 0
}
```

Also had to brute force run these to get the audio to be audible otherwise it was recording with too low volume:
```bash
amixer sget 'Capture'
amixer sset 'Capture' 80%

amixer sget 'Left Input Boost Mixer LINPUT1'
amixer sget 'Right Input Boost Mixer RINPUT1'
amixer sset 'Left Input Boost Mixer LINPUT1' 80%
amixer sset 'Right Input Boost Mixer RINPUT1' 80%
``