raspberrypi/linux #5852: V4L2 hangs, prints kernel stack traces on gapless looping

ardera, 2024-01-15:
### Describe the bug

When using a certain gstreamer pipeline and the v4l2 hardware decoder to do gapless looping of a video file, the playback will hang at the first EOS. Terminating the playback using Ctrl+C prints multiple kernel bug reports/stack traces and `driver bug:` messages

### Steps to reproduce the behaviour

```bash
wget https://github.com/flutter/packages/raw/main/packages/video_player/video_player/example/assets/Butterfly-209.mp4 -O butterfly.mp4
gst-launch-1.0 multifilesrc location=butterfly.mp4 loop=true ! queue ! decodebin ! queue ! kmssink
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

```console
hannes@alpaka-pi:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2
hannes@alpaka-pi:~ $ sudo vcgencmd version
Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)
hannes@alpaka-pi:~ $ uname -a
Linux alpaka-pi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
hannes@alpaka-pi:~ $
```

### Logs

`dmesg` output of the last few frames and subsequent Ctrl+C,
with bcm2835-codec `debug=5` and /dev/video10 `debug=0x1f`:
```console
hannes@alpaka-pi:~ $ dmesg
...
[1070921.126048] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 000000000e1bac6d
[1070921.126048] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 000000000e1bac6d
[1070921.126077] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 000000000e1bac6d vbuf->flags 0, seq 0, bytesused 1382400
[1070921.126105] bcm2835-codec bcm2835-codec: device_run: off we go
[1070921.126119] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[1070921.126150] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 000000000e1bac6d
[1070921.126077] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 000000000e1bac6d vbuf->flags 0, seq 0, bytesused 1382400
[1070921.126197] video10: VIDIOC_QBUF: 00:03:34.000000 index=9, type=vid-cap-mplane, request_fd=0, flags=0x00004003, field=none, sequence=0, memory=mmap
[1070921.126105] bcm2835-codec bcm2835-codec: device_run: off we go
[1070921.126119] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[1070921.126252] plane 0: bytesused=1382400, data_offset=0x00000000, offset/userptr=0x40be2000, length=1382400
[1070921.126150] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 000000000e1bac6d
[1070921.126265] timecode=00:00:00 type=0, flags=0x00000000, frames=0, userbits=0x00000000
[1070921.126197] video10: VIDIOC_QBUF: 00:03:34.000000 index=9, type=vid-cap-mplane, request_fd=0, flags=0x00004003, field=none, sequence=0, memory=mmap
[1070921.126551] videodev: v4l2_poll: video10: poll: 00000104 0000003b
[1070921.126252] plane 0: bytesused=1382400, data_offset=0x00000000, offset/userptr=0x40be2000, length=1382400
[1070921.126265] timecode=00:00:00 type=0, flags=0x00000000, frames=0, userbits=0x00000000
[1070921.126551] videodev: v4l2_poll: video10: poll: 00000104 0000003b
[1070921.159365] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 000000001d2b0a73
[1070921.159365] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare: type: 9 ptr 000000001d2b0a73
[1070921.159395] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 000000001d2b0a73 vbuf->flags 0, seq 0, bytesused 1382400
[1070921.159422] bcm2835-codec bcm2835-codec: device_run: off we go
[1070921.159436] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[1070921.159466] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 000000001d2b0a73
[1070921.159513] video10: VIDIOC_QBUF: 00:03:31.000000 index=10, type=vid-cap-mplane, request_fd=0, flags=0x00004003, field=none, sequence=0, memory=mmap
[1070921.159566] plane 0: bytesused=1382400, data_offset=0x00000000, offset/userptr=0x40d34000, length=1382400
[1070921.159579] timecode=00:00:00 type=0, flags=0x00000000, frames=0, userbits=0x00000000
[1070921.159395] bcm2835-codec bcm2835-codec: bcm2835_codec_buf_queue: type: 9 ptr 000000001d2b0a73 vbuf->flags 0, seq 0, bytesused 1382400
[1070921.159422] bcm2835-codec bcm2835-codec: device_run: off we go
[1070921.159436] bcm2835-codec bcm2835-codec: device_run: Submitted op buffer
[1070921.159466] bcm2835-codec bcm2835-codec: device_run: Submitted src 0000000000000000, dst 000000001d2b0a73
[1070921.159513] video10: VIDIOC_QBUF: 00:03:31.000000 index=10, type=vid-cap-mplane, request_fd=0, flags=0x00004003, field=none, sequence=0, memory=mmap
[1070921.159566] plane 0: bytesused=1382400, data_offset=0x00000000, offset/userptr=0x40d34000, length=1382400
[... 291 lines cut ...]
[1070984.057784] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 000000003c8fca72
[1070984.057801] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 000000000f023179
[1070984.057821] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 000000006f33e245
[1070984.057857] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 00000000185f7645
[1070984.060880] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 0000000013c4965b
[1070984.060907] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 0000000095ca3940
[1070984.060925] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 00000000a94f92cb
[1070984.060941] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 00000000f7e7d048
[1070984.060958] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 000000000e1bac6d
[1070984.061001] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 000000001d2b0a73
[1070984.061021] bcm2835-codec bcm2835-codec: bcm2835_codec_buffer_cleanup: ctx:00000000b3d8dcac, vb 000000004d94765e
[1070984.061054] video10: VIDIOC_REQBUFS: count=0, type=vid-cap-mplane, memory=mmap
[1070984.083707] bcm2835-codec bcm2835-codec: bcm2835_codec_release: Releasing instance 00000000b3d8dcac
[1070984.085159] videodev: v4l2_release: video10: release
[1070984.083707] bcm2835-codec bcm2835-codec: bcm2835_codec_release: Releasing instance 00000000b3d8dcac
[1070984.085159] videodev: v4l2_release: video10: release
Freeing pipeline ...
hannes@alpaka-pi:~ $
```

### Additional context

This is a kernel bug, right? i.e. gstreamer should not be able to cause kernel stack dumps?

I think the issue exists in this form since quite some time, I remember trying to do gapless looping in flutter-pi in 2020/2021 or something and having to do normal looping instead because of this hang. Now, I'm having some trouble with normal looping (though not sure it's gstreamer or kernel yet), and while trying to find some steps to reproduce it, I tried this again, saw all the warning / error messages in dmesg and decided to report this.

6by9, 2024-01-15:
What version of GStreamer are you using?

Whilst userspace shouldn't be able to produce kernel warns, this is a condition that is handed by the V4L2 frameworks and cleaned up anyway
https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/media/common/videobuf2/videobuf2-core.c#L2029-L2035

ardera, 2024-01-16:
`gstreamer1.0-plugins-good` shows version `1.22.0-5+rpt1+deb12u1`; I'm on Raspberry Pi OS bookworm (64-bit), all packages are up-to-date.

So the actual hang might also be gstreamer related?