raspberrypi/linux #5864: RP1 VEC: "tv_mode" property is not correctly inherited from command line "video"

njhollinghurst, 2024-01-18:
### Describe the bug

RP1 VEC has a `tv_norm` string parameter which affects
- The set of named DRM modes advertised
- The colour system and pedestal used in some of those modes

(by default it supports all TV modes, prefers NTSC, and guesses NTSC or PAL depending on the active mode).

It can successfully be set as a module parameter from the command line (so there is no urgent problem). However, the more standard `video/tv_mode` command line setting appears to be ignored.

Furthermore, the private enum of tv_norms used by RP1 VEC driver does not quite match DRM's `enum drm_connector_tv_mode`.  It would could cause confusion if an enumerated value were to get passed from the connector to our driver (which we do at least _attempt_ to do, via `drm_object_attach_property()`). The same thing is true for VC4 VEC.

### Steps to reproduce the behaviour

When using RP1 VEC, compare different behaviour between
`video=Composite-1,tv_mode=NTSC-J`
and
`video=Composite-1 drm-rp1-vec.tv_norm=NTSC-J`

### Device (s)

Other

### System

Raspberry Pi 5
rpi-6.1.y as at Thu 18 Jan 11:32:27 GMT 2024

### Additional context

May also be relevant to VC4.

njhollinghurst, 2024-01-22:
On Linux 6.7, that call to `drm_object_attach_property()` seems to crash.
```
[    3.183810] drm-rp1-vec 1f00144000.vec: rp1vec_platform_probe
[    3.190481] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000060
[    3.205211] Mem abort info:
[    3.210970] videodev: Linux video capture interface: v2.00
[    3.219876]   ESR = 0x0000000096000005
[    3.223788]   EC = 0x25: DABT (current EL), IL = 32 bits
[    3.231143]   SET = 0, FnV = 0
[    3.234329]   EA = 0, S1PTW = 0
[    3.237548]   FSC = 0x05: level 1 translation fault
[    3.242910] Data abort info:
[    3.247790]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[    3.253951]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[    3.259068]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[    3.268519] user pgtable: 16k pages, 47-bit VAs, pgdp=00000001018a0000
[    3.275440] [0000000000000060] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[    3.284262] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
[    3.290572] Modules linked in: snd_pcm(+) videobuf2_v4l2 ecc libaes videodev rfkill snd_timer snd drm_rp1_vec(+) v3d videobuf2_common mc spi_bcm2835 gpu_sched drm_dma_helper gpio_keys drm_shmem_helper drm_kms_helper pwm_fan rp1_adc raspberrypi_gpiomem nvmem_rmem uio_pdrv_genirq uio drm i2c_dev drm_panel_orientation_quirks backlight fuse dm_mod ip_tables x_tables ipv6
[    3.323274] CPU: 2 PID: 332 Comm: (udev-worker) Not tainted 6.7.0-rc6-v8-16k+ #2
[    3.330701] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    3.336554] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    3.343545] pc : drm_object_attach_property+0x8/0xc0 [drm]
[    3.349143] lr : rp1vec_platform_probe+0x2fc/0x3e0 [drm_rp1_vec]
[    3.355180] sp : ffffc00080cd3880
[    3.358502] x29: ffffc00080cd3880 x28: ffffc00080cd3d88 x27: 0000000000000007
[    3.365668] x26: 0000000000000001 x25: 0000000000000020 x24: ffff800108077100
[    3.372834] x23: ffffd000521e0030 x22: ffff8001049fc410 x21: ffff80010624f000
[    3.379999] x20: ffff8001078a6080 x19: ffff8001078a67b0 x18: 0000000000000001
[    3.387164] x17: 000000005d936730 x16: ffffd00084d0e410 x15: ffff800108077688
[    3.394331] x14: ffffffffffffffff x13: 0000000000000002 x12: ffff800100a4571a
[    3.401498] x11: 0101010101010101 x10: 000000000000002d x9 : ffffd000521b4a5c
[    3.408665] x8 : ffffd00051fb1cd8 x7 : 00000000c0c0c0c0 x6 : 00000000c0c0c0c0
[    3.415831] x5 : 0000000000000000 x4 : ffff8001078a6978 x3 : ffff8001078a6978
[    3.422996] x2 : 0000000000000007 x1 : 0000000000000000 x0 : ffff8001078a67f0
[    3.430162] Call trace:
[    3.432611]  drm_object_attach_property+0x8/0xc0 [drm]
[    3.437824]  platform_probe+0x70/0xe8
[    3.441498]  really_probe+0x118/0x290
[    3.445172]  __driver_probe_device+0x80/0x140
[    3.449544]  driver_probe_device+0x48/0x130
[    3.453741]  __driver_attach+0x7c/0x138
[    3.457588]  bus_for_each_dev+0x80/0xf0
[    3.461435]  driver_attach+0x2c/0x40
[    3.465021]  bus_add_driver+0xec/0x1f8
[    3.468780]  driver_register+0x68/0x138
[    3.472628]  __platform_driver_register+0x30/0x48
[    3.477346]  rp1vec_platform_driver_init+0x28/0x3ff8 [drm_rp1_vec]
[    3.483556]  do_one_initcall+0x50/0x2a0
[    3.487404]  do_init_module+0x60/0x218
[    3.491164]  load_module+0x1bac/0x1d18
[    3.494923]  __do_sys_init_module+0x194/0x208
[    3.499294]  __arm64_sys_init_module+0x24/0x38
[    3.503751]  invoke_syscall+0x4c/0x118
[    3.507513]  el0_svc_common.constprop.0+0xc8/0xf0
    3.512234]  do_el0_svc+0x24/0x38
[    3.515557]  el0_svc+0x34/0xb8
[    3.518622]  el0t_64_sync_handler+0xc0/0xc8
[    3.522818]  el0t_64_sync+0x190/0x198
[    3.526491] Code: 84d58418 ffffd000 aa1e03e9 d503201f (f9403024)
```
It's early in the function so it's probably one of the dereferences in the first couple of declarations.

njhollinghurst, 2024-01-22:
Yikes -- 6.6 crashes also! Urgent fix required. Should I just remove the offending call (starting from 6.1)?

6by9, 2024-01-22:
Quickest fix is
```
--- a/drivers/gpu/drm/rp1/rp1-vec/rp1_vec.c
+++ b/drivers/gpu/drm/rp1/rp1-vec/rp1_vec.c
@@ -478,7 +478,7 @@ static int rp1vec_platform_probe(struct platform_device *pdev)
        drm_connector_helper_add(&vec->connector, &rp1vec_connector_helper_funcs);
 
        drm_object_attach_property(&vec->connector.base,
-                                  drm->mode_config.tv_mode_property,
+                                  drm->mode_config.legacy_tv_mode_property,
                                   vec->tv_norm);
 
        ret = drm_simple_display_pipe_init(drm,
```

`drm_mode_create_tv_properties_legacy` is documented as being legacy and 
```
 * NOTE: This functions registers the deprecated "mode" connector
 * property to select the analog TV mode (ie, NTSC, PAL, etc.). New
 * drivers must use drm_mode_create_tv_properties() instead.
```
We're a new driver, so for upstream will need to do the job properly.

njhollinghurst, 2024-01-22:
OK, I may even try to fix it properly with the new property. Will it mean we no longer need the "tv_norm" parameter?

The VEC driver never rejects a "foreign" video mode, so non-atomic changes should be safe.  It should also be OK to merge PAL and PAL60, with minor code changes.

It's a pity that the standard enum lacks a "monochrome" option, which we might have liked to add (such an option could also have been a good place to hide/reveal the "vintage" 405- and 819-line TV modes).

njhollinghurst, 2024-01-23:
I think I have a working fix.  Will make a branch and PR for 6.6

As discussed, now I'm planning to do away with the module parameter and OF properties altogether!  To remove a layer of duplication and confusion about the scope of the parameter.  Instead, the command-line "video" argument will set the initial value ("NTSC" by default); this will be used to initialize the DRM connector property, which is the one we care about.

All combinations of DRM mode and TV mode are considered valid (meaningless combinations will default to NTSC[-M] for 60Hz or PAL[-BDGHIK] for 50Hz), to avoid breaking any existing use cases.

6by9, 2024-01-23:
FWIW vc4 VEC has deliberately retained part of the old property via 91112a6fc12cb086cf7a3a350df6a415ba4631ef, which explains some of the weirdness in vc4_vec over which property is being used. Change one, and the other is updated to match (either immediately, or converted back again in get_property).

> It's a pity that the standard enum lacks a "monochrome" option, which we might have liked to add (such an option could also have been a good place to hide/reveal the "vintage" 405- and 819-line TV modes).

There's nothing stopping us adding an extra standard into the enum, and upstreaming it. There have been previous requests for a way to disable the colour burst in the way that `sdtv_disable_colourburst` used to with the firmware driver, and that seems the obvious way to do it.

PR #5885 (2024-02-02): RP1 VEC: Use tv_mode from command line, align enum with drm_connector; tidy
This attempts to fix #5864, targeting Linux 6.6

I narrowly kept the "tv_norm" module parameter  in this draft; but it's somewhat redundant now that we can use video/tv_mode.