raspberrypi/linux #5866: rp1_probe throws lockdep errors if CONFIG_LOCKDEP is enabled

6by9, 2024-01-18:
### Describe the bug

Enable `CONFIG_LOCKDEP` in your kernel build, rebuild, and push to a Pi5.
`rp1_probe` logs lots of lockdep issues, all from the same spot in `irq_domain_set_mapping` due to a mutex not being held [kernel/irq/irqdomain.c:530](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/kernel/irq/irqdomain.c#L530)

@P33M @pelwell You know more about the interrupt hierarchy than I do. Could you have a look please?

```
[    4.254890] ------------[ cut here ]------------
[    4.254899] WARNING: CPU: 0 PID: 170 at kernel/irq/irqdomain.c:530 irq_domain_set_mapping+0x6c/0xa0
[    4.254914] Modules linked in: r8152 spidev brcmfmac_wcc hci_uart raspberrypi_cpufreq btqca btbcm brcmfmac bluetooth brcmutil ecdh_generic onboard_usb_hub(+) cfg80211 bcm2711_thermal crct10dif_ce ecc reset_raspberrypi clk_raspberrypi rfkill raspberrypi_hwmon simpledrm pwm_brcmstb drm_shmem_helper spi_bcm2835 nvmem_rmem drm_kms_helper pcie_brcmstb(+) drm fuse dm_mod backlight ip_tables x_tables ipv6
[    4.254996] CPU: 0 PID: 170 Comm: (udev-worker) Tainted: G        W          6.7.0-v8+ #9
[    4.255002] Hardware name: Raspberry Pi 5 Model B Rev 1.0 (DT)
[    4.255005] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    4.255011] pc : irq_domain_set_mapping+0x6c/0xa0
[    4.255017] lr : irq_domain_set_mapping+0x68/0xa0
[    4.255024] sp : ffff8000811c3060
[    4.255027] x29: ffff8000811c3060 x28: ffff000106d09680 x27: ffffa00085c0f0f0
[    4.255036] x26: 000000000000005d x25: ffffa00085c0ed90 x24: 0000000000000001
[    4.255045] x23: 000000000000005d x22: 0000000000000001 x21: ffff00010b88eb40
[    4.255054] x20: ffff0001000e0000 x19: 00000000000000dc x18: 0000000000000000
[    4.255064] x17: 0000000000000000 x16: 0000000000000000 x15: 00002c8089338de4
[    4.255073] x14: 0000000000000078 x13: 0000000000000001 x12: 0000000000000000
[    4.255082] x11: 0000000000000000 x10: 00000000000000a0 x9 : ffffa00086093c94
[    4.255092] x8 : ffff000106f41b40 x7 : ffffa00085247544 x6 : 000000000000000f
[    4.255101] x5 : 0000000000000000 x4 : 000000000000003c x3 : 0000000000000001
[    4.255110] x2 : ffff600178e6f000 x1 : 0000000000000000 x0 : 0000000000000000
[    4.255119] Call trace:
[    4.255122]  irq_domain_set_mapping+0x6c/0xa0
[    4.255130]  irq_domain_alloc_irqs_locked+0x234/0x29c
[    4.255134]  __irq_domain_alloc_irqs+0x6c/0xc0
[    4.255139]  __msi_domain_alloc_irqs+0x1a0/0x484
[    4.255145]  msi_domain_alloc_locked+0xd8/0xf0
[    4.255150]  msi_domain_alloc_irqs_all_locked+0x60/0x8c
[    4.255155]  pci_msi_setup_msi_irqs+0x2c/0x44
[    4.255162]  __pci_enable_msix_range+0x2f4/0x500
[    4.255167]  pci_alloc_irq_vectors_affinity+0xd0/0x138
[    4.255173]  pci_alloc_irq_vectors+0x14/0x20
[    4.255177]  rp1_probe+0xac/0x384
[    4.255184]  local_pci_probe+0x40/0xbc
[    4.255190]  pci_device_probe+0x1dc/0x1f0
[    4.255196]  really_probe+0x110/0x27c
[    4.255201]  __driver_probe_device+0x78/0x12c
[    4.255205]  driver_probe_device+0x40/0x120
[    4.255209]  __device_attach_driver+0xb8/0xf8
[    4.255213]  bus_for_each_drv+0x84/0xe4
[    4.255219]  __device_attach+0xfc/0x18c
[    4.255224]  device_attach+0x14/0x20
[    4.255227]  pci_bus_add_device+0x64/0xd4
[    4.255232]  pci_bus_add_devices+0x38/0x84
[    4.255236]  pci_bus_add_devices+0x64/0x84
[    4.255240]  pci_host_probe+0x44/0xbc
[    4.255245]  brcm_pcie_probe+0x3c0/0x83c [pcie_brcmstb]
[    4.255259]  platform_probe+0x68/0xd8
[    4.255264]  really_probe+0x110/0x27c
[    4.255268]  __driver_probe_device+0x78/0x12c
[    4.255272]  driver_probe_device+0x40/0x120
[    4.255276]  __driver_attach+0x74/0x128
[    4.255280]  bus_for_each_dev+0x78/0xe0
[    4.255286]  driver_attach+0x24/0x30
[    4.255292]  bus_add_driver+0xe4/0x1e8
[    4.255298]  driver_register+0x60/0x128
[    4.255302]  __platform_driver_register+0x28/0x34
[    4.255307]  brcm_pcie_driver_init+0x20/0x1000 [pcie_brcmstb]
[    4.255318]  do_one_initcall+0x78/0x2f8
[    4.255323]  do_init_module+0x58/0x1ec
[    4.255330]  load_module+0x1b14/0x1c8c
[    4.255337]  init_module_from_file+0x88/0xc8
[    4.255343]  idempotent_init_module+0x188/0x24c
[    4.255350]  __arm64_sys_finit_module+0x64/0xa0
[    4.255357]  invoke_syscall+0x44/0x104
[    4.255364]  el0_svc_common.constprop.0+0xc0/0xe0
[    4.255370]  do_el0_svc+0x1c/0x28
[    4.255376]  el0_svc+0x4c/0xe4
[    4.255384]  el0t_64_sync_handler+0xc0/0xc4
[    4.255389]  el0t_64_sync+0x190/0x194
[    4.255394] irq event stamp: 77202
[    4.255397] hardirqs last  enabled at (77201): [<ffffa000852799f8>] _raw_spin_unlock_irqrestore+0x74/0x78
[    4.255405] hardirqs last disabled at (77202): [<ffffa000852641a0>] el1_dbg+0x28/0x90
[    4.255411] softirqs last  enabled at (77134): [<ffffa00084010afc>] __do_softirq+0x47c/0x4c0
[    4.255416] softirqs last disabled at (76749): [<ffffa0008401685c>] ____do_softirq+0x10/0x1c
[    4.255421] ---[ end trace 0000000000000000 ]---
```

### Steps to reproduce the behaviour

Enable CONFIG_LOCKDEP and observe WARN on Pi5.

### Device (s)

Raspberry Pi 5

### System

I was testing on my mainline upstreaming branch, but has also been observed on rpi-6.6.y and rpi-6.7.y.

pelwell, 2024-01-18:
PR #5869 should resolve this.

6by9, 2024-01-18:
> PR #5869 should resolve this.

It does. Thank you very much.

closed 2024-01-19 by pelwell