raspberrypi/linux #5870: rpi5: USB over-current condition triggered via gpio

nbuchwitz, 2024-01-18:
### Describe the bug

I've been debugging usb port power switching using libusb and rpi5 (another topic/issue). While researching how and if the vbus may be switched with rp1, I came upon the `usb_vbus_pins` definition in 'bcm2712-rpi-5-b.dts'.

Gpio 42 appears to switch the vbus for all USB ports at once, whereas GPIO 43 appears to detect overcurrent.  If I query gpio 43, the Raspberry Pi 5 enters overcurrent protection and all USB ports are disabled. Even though this appears to be correct, it could be a problem because the typical Pi user has access to all gpio pins, and a rogue application can block all USB ports. 

I don't know where the `usb_vbus_pins` are used, but maybe a gpio hog for at least the over current pin can prevent further damage?

### Steps to reproduce the behaviour

Straight to over current: `gpioget 4 43`

or just some switching: `gpioset 4 42=0`

### Device (s)

Raspberry Pi 5

### System

```
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4
```

```
2024/01/05 15:57:40 
Copyright (c) 2012 Broadcom
version 30cc5f37 (release) (embedded)
```

```
Linux pi5 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
```

pelwell, 2024-01-19:
It's interesting that `gpioget 4 43` behaves differently to `gpioget 0 24`. I expected them both to change the fsel/alt function to make them a GPIO, but `gpioget 0 24` is happy to return the pin level while keeping it as a UART RTS pin.

nbuchwitz, 2024-01-19:
Right. A `pinctrl set 43 a2 pu` resets the over current pin as expected and usb is unblocked. Tested some other gpios (also on different chips) and most of thempersisted the fsel. `FAN_PWM` is an exception and does also change it's fsel:

```
nbw@pi5:~/uhubctl $ pinctrl | grep FAN_PWM
45: a0    pd | hi // FAN_PWM/GPIO45 = PWM1_CHAN3
nbw@pi5:~/uhubctl $ gpioget 4 45
0
nbw@pi5:~/uhubctl $ pinctrl | grep FAN_PWM
45: ip    pd | lo // FAN_PWM/GPIO45 = input
```

Could it be, that the other pins (eg. UART RTS) are held by some drivers (gpiod lists them as unused, but at least the BT pins should be used by hci_uart)?

pelwell, 2024-01-19:
I don't think I can put gpio-hogs on these pins without also making them GPIOs, which I don't want to do - they need to have the vbus1 alt function so the RP1 USB hardware can handle them appropriately.

nbuchwitz, 2024-01-19:
Right, a hog can only be output or input ... So there isn't really a solution to prevent this besides a dummy driver / user space application which binds to the gpio label I guess.

Related question: What's the deal with `vbus0` (28/29), `vbus2` (50/51) and `vbus3` (52/53)? Are these meant for individual port vbus control? Thus with access to these gpios one could control each port individually? 50-53 seem unused, while 35/36 unfortunately are used .

pelwell, 2024-01-19:
I think RP1 has a flexible mapping from ports to `vbus<n>` functions for use in other applications, and similarly for `vbus_oc<n>` but as Pi 5 has a single VBUS power switch and overcurrent detector it's all academic.

P33M, 2024-01-19:
Yes, the supportable configurations are 
- Ganged and grouped Vbus/OC
- Grouped Vbus/OC
- Individual Vbus/OC
Individual breaks out each port's overcurrent and vbus_en signal, derived from the respective xhci PORTSC register.
Grouped will multiplex both ports on one controller together, for a single vbus_en/overcurrent pair per controller.
Ganged and grouped multiplexes both controllers together for a single vbus_en/overcurrent pair.

nbuchwitz, 2024-01-19:
Thanks for the clarification, @P33M.

So the hardware design of pi 5 is ganged and thus should set `HUB_CHAR_COMMON_LPSM` (0x0) in `wHubCharacteristics`. Unfortunately this is not the case and the RP1 reports `HUB_CHAR_INDV_PORT_LPSM` (0x1). Is there a way RP1 needs to be strapped? How does the configuration work (esp. which `vbus` is taken for ganged / grouped vbus)?

```
# hub.bus=4, hub.bcdUSB=0x300, hub.port_number=0

(skipped, USB3 duality handling)
--------------------------------------------------------------------------------
# hub.bus=3, hub.bcdUSB=0x200, hub.port_number=0

num_ports=2

w_hub_characteristics=0x9
lpsm: 0x1
common lpsm:  0
individual lpsm:  1
no lpsm:  0
--------------------------------------------------------------------------------
# hub.bus=2, hub.bcdUSB=0x300, hub.port_number=0

(skipped, USB3 duality handling)
--------------------------------------------------------------------------------
# hub.bus=1, hub.bcdUSB=0x200, hub.port_number=0

num_ports=2

w_hub_characteristics=0x9
lpsm: 0x1
common lpsm:  0
individual lpsm:  1
no lpsm:  0
--------------------------------------------------------------------------------
```
 
https://gist.github.com/nbuchwitz/bd7e000fad484dbd7c86b560c2dcc2e9

lategoodbye, 2024-01-20:
> I don't think I can put gpio-hogs on these pins without also making them GPIOs, which I don't want to do - they need to have the vbus1 alt function so the RP1 USB hardware can handle them appropriately.

Yes, gpio-hogs is not the proper solution here. But how about moving the affected pins under the pinctrl of the USB driver like most of the other boards do?

pelwell, 2024-01-20:
What, like this? https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/bcm2712-rpi-5-b.dts#L241

lategoodbye, 2024-01-20:
Yes, this looks like as a good starting point.

Maybe the USB driver needs to claim these pins?
Or there is an issue with the rp1 pinctrl driver. I vaguely remember a strict flag for pinmux.