raspberrypi/linux #5871: Raspberry Pi 5 KMS laggy

the-colonel-dev, 2024-01-19:
### Describe the bug

Hi all, 

On Raspberry Pi 5 Bookworm Lite, using Kernel 6.7 solved the problem of white screen garbage I had running in framebuffer the emulator Dosbox.

Now a new problem appears.

When i run Dosbox from command line, the emulator screen appears correctly, but it seems laggy, I mean when i try to type on keyboard the output is like delayed by some 200/300 millis, in general the refresh of screen seems slow and with some artifact.

This happens with all the Dosboxes versions i tried. It's not a problem of the specific emulator because the same image run on Pi 4 works correctly.

### Steps to reproduce the behaviour

Run Dosbox from command line in Raspberry Pi 5 Bookworm OS Lite

### Device (s)

Raspberry Pi 5

### System

Kernel version 6.7

popcornmix, 2024-01-19:
I suspect the white screen garbage issue has been fixed on 6.1 kernel with https://github.com/raspberrypi/linux/pull/5827 so you can run rpi-update kernel to get a fixed 6.1 kernel. Might be worth switching back to that as a more stable testing environment.

But otherwise, this doesn't sound like a kernel issue, it sounds more like a dosbox issue.

>  It's not a problem of the specific emulator because the same image run on Pi 4 works correctly.

Just to confirm, are you testing exactly the same sdcard, with the same peripherals (e.g. display, usb etc), with the only change swapping the Pi4 for a Pi5? And are you saying the Pi4 has no lag and the Pi5 has significant lag when running dosbox?

the-colonel-dev, 2024-01-19:
I'm closing the issue because switching back to kernel 6.1 everything worked smoothly on Pi 4 and Pi 5.

closed 2024-01-19 by the-colonel-dev

popcornmix, 2024-01-19:
Could you confirm if "sudo rpi-update next" is good or bad?
That gets the 6.6 kernel which is also considered stable (and we are planning to make the default in RPiOS in the near future).

If the issue also affects 6.6, then we would want to investigate more urgently.

the-colonel-dev, 2024-01-19:
I confirm that everything works OK on Pi 4 / Pi 5 updating to kernel version 6.6.
No Lags, no problem of dirty screen update.

popcornmix, 2024-01-19:
Okay - thanks for testing.

I'll note there may be an issue ahead for 6.7 kernel and beyond. It's possible we are missing a downstream patch on that tree, or there has been an upstream regression (which may get resolved in one of the upstream 6.7.x updates).

But that wont' affect RPiOS for a long time.

6by9, 2024-01-19:
Just as a thought, 6.7 is currently missing the iommus due to the kernel API change, so that may have an implication on caching.