raspberrypi/linux #5874: cpuinfo_cur_freq is not reporting the correct frequency

giuliomoro, 2024-01-20:
### Describe the bug

`/sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq` reports the same value as `/sys/devices/system/cpu/cpufreq/policy0/scaling_cur_freq`, even when the actual frequency (as reported by `vcgencmd measure_clock core`) is different from that (e.g.: because of throttling or different limits between the Linux governor and the vc manager).

Here's the relevant [Linux docs](https://www.kernel.org/doc/html/v6.1/admin-guide/pm/cpufreq.html):

> cpuinfo_cur_freq
> Current frequency of the CPUs belonging to this policy as obtained from the hardware (in KHz).
> This is expected to be the frequency the hardware actually runs at. If that frequency cannot be determined, this attribute should not be present.

> scaling_cur_freq
> Current frequency of all of the CPUs belonging to this policy (in kHz).
> In the majority of cases, this is the frequency of the last P-state requested by the scaling driver from the hardware using the scaling interface provided by it, which may or may not reflect the frequency the CPU is actually running at (due to hardware design and other limitations).

So, according to this, `cpuinfo_cur_freq` should return the same value as `vcgencmd measure_clock core` (or whatever the authoritative source is) and if that is not possible, it should not be present.

### Steps to reproduce the behaviour

```
sudo cpufreq-set -g powersave -u 2400MHz -d 2400MHz
```
once the system starts throttling (as reported by `vcgencmd get_throttled` and `vcgencmd measure_clock arm`), the the content of `/sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq` will stay fixed at 2400000. It should instead follow the value returned by `vcgencmd measure_clock arm`.

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

Linux raspberrypi 6.1.65-v8-16k #6 SMP Thu Dec  7 01:20:31 UTC 2023 aarch64 GNU/Linux
Revision        : c04170
Serial          : f4f12a9b9e056981
Model           : Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0xe0000
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

pelwell, 2024-01-22:
Changing the current behaviour in an acceptable way is non-trivial. We can replace the current way of querying the clock frequency with something which actually measures the clock, but this will slow down normal operation and potentially upset the normal operation of the governor. Even removing the cpuinfo_cur_freq property requires a fork of the cpufreq-dt driver, for no real benefit.

closed 2024-01-22 by pelwell

giuliomoro, 2024-01-22:
> for no real benefit.

One could argue that a _real_ benefit is compliance with the specs:

> cpuinfo_cur_freq
> This is expected to be the frequency the hardware actually runs at. If that frequency cannot be determined, this attribute should not be present.