raspberrypi/linux #5876: Unable to set CPU frequency below 1.5GHz

giuliomoro, 2024-01-20:
### Describe the bug

According to  /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies`, these are the CPU frequencies available to the governor:
`1000000 1100000 1200000 1300000 1400000 1500000 1600000 1700000 1800000 1900000 2000000 2100000 2200000 2300000 2400000`
However, setting 
```
sudo cpufreq-set -g powersave -u ${freq}MHz -d ${frq}MHz
```
with `$freq` between 1000 and 1499 always results in `vcgencmd measure_clock arm` to return around 1500MHz. Is it possible to clock the ARM cores on the Pi 5 below 1.5GHz as stated by the governor? How so?

### Steps to reproduce the behaviour

```
for (( freq=1000; freq<=2400; freq+=100 )) ; do
sudo cpufreq-set -g powersave -u ${freq}MHz -d ${freq}MHz
sleep 1
echo expected $freq, got $(vcgencmd measure_clock arm | sed s/.*=// | sed "s/[0-9]\{6,6\}$//") 
done
```
gives
```
expected 1000, got 1500
expected 1100, got 1500
expected 1200, got 1500
expected 1300, got 1500
expected 1400, got 1500
expected 1500, got 1500
expected 1600, got 1600
expected 1700, got 1700
expected 1800, got 1800
expected 1900, got 1900
expected 2000, got 2000
expected 2100, got 2100
expected 2200, got 2200
expected 2300, got 2300
expected 2400, got 2400
```

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

Linux raspberrypi 6.1.65-v8-16k #6 SMP Thu Dec  7 01:20:31 UTC 2023 aarch64 GNU/Linux
Revision        : c04170
Serial          : f4f12a9b9e056981
Model           : Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0xe0000
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

pelwell, 2024-01-22:
Interesting, but that's not what I see:
```
pi@raspberrypi:~$ uname -a                                                                     uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
pi@raspberrypi:~$ sudo cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1500000 1600000 1700000 1800000 1900000 2000000 2100000 2200000 2300000 2400000
```

giuliomoro, 2024-01-22:
not that I can see:

```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on
dtoverlay=spi5-1cs-pi5
#dtoverlay=spi3-1cs-pi5

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

usb_max_current_enable=0
```

popcornmix, 2024-01-22:
As @pelwell says, this is what you should see by default:
```
$ cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1500000 1600000 1700000 1800000 1900000 2000000 2100000 2200000 2300000 2400000 
```
Can you report output of:
```
uname -a
vcgencmd bootloader_version
vcgencmd get_config int
cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
grep . /sys/kernel/debug/clk/fw-clk-arm/*
```

giuliomoro, 2024-01-22:
```
Linux raspberrypi 6.1.65-v8-16k #6 SMP Thu Dec  7 01:20:31 UTC 2023 aarch64 GNU/Linux
2023/10/30 16:45:10
version 30de0ba51efcdfe5aebfb3068e0d5278647611cc (release)
timestamp 1698684310
update-time 0
capabilities 0x0000007f

arm_64bit=1
arm_freq=2400
arm_freq_min=1000
arm_peri_high=1
audio_pwm_mode=2
auto_initramfs=1
avs_temp=43875
camera_auto_detect=1
core_freq=910
core_freq_min=500
disable_commandline_tags=2
disable_fw_kms_setup=1
disable_l2cache=1
disable_overscan=1
display_auto_detect=1
display_default_lcd=-1
display_hdmi_rotate=-1
display_lcd_rotate=-1
dvfs=4
enable_gic=1
enable_uart=-1
force_eeprom_read=1
force_pwm_open=1
framebuffer_depth=16
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq_min=500
hdmi_enable_4kp60=1
hevc_freq=910
hevc_freq_min=500
ignore_lcd=-1
init_uart_clock=0x2dc6c00
isp_freq=910
isp_freq_min=500
mask_gpu_interrupt1=16418
max_framebuffers=2
over_voltage_avs=0x1a388
pause_burst_frames=1
pciex4_reset=1
program_serial_random=1
total_mem=4096
v3d_freq=960
v3d_freq_min=500
vpred=8674
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
1000000 1100000 1200000 1300000 1400000 1500000 1600000 1700000 1800000 1900000 2000000 2100000 2200000 2300000 2400000
/sys/kernel/debug/clk/fw-clk-arm/clk_accuracy:0
/sys/kernel/debug/clk/fw-clk-arm/clk_duty_cycle:1/2
/sys/kernel/debug/clk/fw-clk-arm/clk_enable_count:0
/sys/kernel/debug/clk/fw-clk-arm/clk_flags:CLK_GET_RATE_NOCACHE
/sys/kernel/debug/clk/fw-clk-arm/clk_max_rate:2400000000
/sys/kernel/debug/clk/fw-clk-arm/clk_min_rate:1000000000
/sys/kernel/debug/clk/fw-clk-arm/clk_notifier_count:0
/sys/kernel/debug/clk/fw-clk-arm/clk_phase:0
/sys/kernel/debug/clk/fw-clk-arm/clk_prepare_count:0
/sys/kernel/debug/clk/fw-clk-arm/clk_protect_count:0
/sys/kernel/debug/clk/fw-clk-arm/clk_rate:1000000000
```

popcornmix, 2024-01-22:
> arm_freq_min=1000

I think this is a custom setting you have set. Note that you should be looking in /boot/firmware/config.txt

Report:
```
mount | grep /boot/firmware
grep arm_freq /boot/firmware/config.txt
```

giuliomoro, 2024-01-22:
> Note that you should be looking in /boot/firmware/config.txt

yup that's what I pasted [above](https://github.com/raspberrypi/linux/issues/5876#issuecomment-1903945865).

`mount | grep /boot/firmware`
nothing 
```
$ grep arm_freq /boot/firmware/config.txt
/dev/mmcblk0p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```

pelwell, 2024-01-22:
I presume you meant:
```
$ mount | grep /boot/firmware
/dev/mmcblk0p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
$ grep arm_freq /boot/firmware/config.txt
# Nothing
$
```

popcornmix, 2024-01-22:
I'm confused why you are seeing this. If you have a spare sdcard I'd be curious if you see the same behaviour with a clean install of RPiOS.

giuliomoro, 2024-01-22:
still same thing after flashing a new image with Raspberry Pi Imager:
```
$ cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1000000 1100000 1200000 1300000 1400000 1500000 1600000 1700000 1800000 1900000 2000000 2100000 2200000 2300000 2400000
```

system details: 
```
$ raspinfo
System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

Linux raspberrypi 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
Revision	: c04170
Serial		: f4f12a9b9e056981
Model		: Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0x0
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

Videocore information
---------------------

2023/10/30 16:45:10
Copyright (c) 2012 Broadcom
version 30de0ba5 (release) (embedded)

vc_gencmd_read_response returned -1
error=1 error_msg="Command not registered"

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
udev             1897360       0   1897360   0% /dev
tmpfs             414368    5568    408800   2% /run
/dev/mmcblk0p2  29472740 4600032  23355360  17% /
tmpfs            2071792     288   2071504   1% /dev/shm
tmpfs               5120      48      5072   1% /run/lock
/dev/mmcblk0p1    522232   74832    447400  15% /boot/firmware
tmpfs             414352     160    414192   1% /run/user/1000

Filename				Type		Size		Used		Priority
/var/swap                               file		102384		0		-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20231127
raspberrypi-sys-mods:
  Installed: 20231127
openbox:
  Installed: 3.6.1-10+rpt1
lxpanel:
  Installed: 0.10.1-2+rpt19
pcmanfm:
  Installed: 1.3.2-1+rpt18
rpd-plym-splash:
  Installed: 0.37

Networking Information
----------------------

[... 1200 lines cut ...]
```
 $ grep . /sys/kernel/debug/clk/fw-clk-arm/*
/sys/kernel/debug/clk/fw-clk-arm/clk_accuracy:0
/sys/kernel/debug/clk/fw-clk-arm/clk_duty_cycle:1/2
/sys/kernel/debug/clk/fw-clk-arm/clk_enable_count:0
/sys/kernel/debug/clk/fw-clk-arm/clk_flags:CLK_GET_RATE_NOCACHE
/sys/kernel/debug/clk/fw-clk-arm/clk_max_rate:2400000000
/sys/kernel/debug/clk/fw-clk-arm/clk_min_rate:1000000000
/sys/kernel/debug/clk/fw-clk-arm/clk_notifier_count:0
/sys/kernel/debug/clk/fw-clk-arm/clk_phase:0
/sys/kernel/debug/clk/fw-clk-arm/clk_prepare_count:0
/sys/kernel/debug/clk/fw-clk-arm/clk_protect_count:0
/sys/kernel/debug/clk/fw-clk-arm/clk_rate:1000000000
```

And, as you'd expect from a fresh install:
```
$ grep arm_freq /boot/firmware/config.txt
# nothing
```

```
 $  mount | grep /boot/firmware
/dev/mmcblk0p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```

popcornmix, 2024-01-22:
Your bootloader version is quite old:
```
2023/10/30 16:45:10
version 30de0ba51efcdfe5aebfb3068e0d5278647611cc (release)
timestamp 1698684310
update-time 0
capabilities 0x0000007f
```
I was under the impression that you would get a newer version than that from the normal apt procedure (the rpi-eeprom package).
But you can install a newer version manually (running rpi-update is one way). That may be worth trying.

giuliomoro, 2024-01-22:
> Have you apt upgraded and then rebooted the fresh image?

Now I have:

```
$ vcgencmd bootloader_version
2024/01/05 15:57:40
version 30cc5f37d65b279e820adb1d9840ad3c1cc98229 (release)
timestamp 1704470260
update-time 1705943931
capabilities 0x0000007f
```

And 
```
$ cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies
1500000 1600000 1700000 1800000 1900000 2000000 2100000 2200000 2300000 2400000
```

OK that seems good, thanks for your help.

Now the second part of the question: is it possible to clock the ARM cores on the Pi 5 below 1.5GHz?

giuliomoro, 2024-01-22:
>You could try arm_freq_min=1000 in config.txt.
> It's possible with the newer bootloader this will behave better.

Thanks this seems to work just fine. I tried it as low as 500 and the clock seems to be applied just fine, as confirmed by `vcgencmd measure_clock arm`. Thank for all the help.

closed 2024-01-22 by giuliomoro