raspberrypi/linux #5883: KDE breaks fresh Kali install

GageHoweTamu, 2024-01-23:
### Describe the bug

Raspberry Pi 5, boot from SD card, Kali from Raspberry Pi Imager.

Upon chosing KDE and rebooting, the installation boots normally, then gets stuck at "IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready"

This has happened multiple times, and a reinstall has been the only solution.

### Steps to reproduce the behaviour

sudo apt update
sudo apt install kali-desktop-kde
sudo update-alternatives --config x-session-manager && reboot # I chose ssdm, and x11

### Device (s)

Raspberry Pi 5

### System

fresh install from Kali from Raspberry Pi Imager

pelwell, 2024-01-23:
This area is for problems with the Raspberry Pi Linux kernel. We don't provide support for Kali Linux, its apps and packages. If you identify a problem with our Linux kernel then open a more specific issue here.

Having said that, you may get more clues about where it is stalling by adding `initcall_debug` and `ignore_loglevel` to cmdline.txt, taking care to keep all the settings in a single line.

Master-Koy, 2024-01-24:
> This area is for problems with the Raspberry Pi Linux kernel. We don't provide support for Kali Linux, its apps and packages. If you identify a problem with our Linux kernel then open a more [...]

Hey there, I know this is not the right place to post about my issue but I'm really confused with my situation. 

I'm using Kali-Linux and having a problem while booting.

I have attached a picture showing the issue. Please guide me in whatever way you can.

[image]

closed 2024-07-27 by GageHoweTamu