raspberrypi/linux #5887: Kernel 6.7.1 booting issue

Dark-Sky, 2024-01-24:
### Describe the bug

Not sure but I think it has to do with going into graphics mode.

### Steps to reproduce the behaviour

Use make bcm2711_defconfig and compile the 6.7.1 kernel.

The 6.7.1 will boot and everything seems to be ok. Then if you reboot the system will hang booting up.

[image]

Then when you disconnect and reconnect the power it boots normally after going through fsck. Then when you reboot it hangs again and the whole process is repeated over and over.

Using the same ssd and booting on my pi4 with the 6.7.1 kernel there is no issue. The 2 kernels are compiled at 11e8442cb18e5149225779cc10f1caef75b0b030.

### Device (s)

Raspberry Pi 5

### System

[ray@jellyfin ~]$ vcgencmd version
2024/01/05 15:57:40 
Copyright (c) 2012 Broadcom
version 30cc5f37 (release) (embedded)

Kernel 6.7.1

### Logs

Only time I can get to the logs is when it fully boots.

dmesg: https://pastebin.com/nHRpp8Cn
xorg-log: https://pastebin.com/L7Ugym7W

6by9, 2024-01-24:
If you tell the system not to boot into any window manager then does it boot successfully?

I have a Pi5 booted with that kernel, and it seems happy at both console and booted into Wayfire.

popcornmix, 2024-01-24:
A 6.7.1 kernel here with sdcard + nfs root on RPiOS is rebooting to desktop okay on a Pi5.
So I think there are more factors involved than just 6.7.1 + Pi5 doesn't reboot properly.

I'll keep an eye for any issues.
Are you seeing this on RPiOS?
Do you also see it with an sdcard boot?

Dark-Sky, 2024-01-24:
I installed the 6.7.1 kernel on a sdcard with KDE Plasma (my ssd has XFCE) and it is doing the same thing. It hangs with the message "Reached Graphical interface".  There were no errors showing up in the terminal before it hung up.

I disabled SDDM and it booted to the login prompt just fine and was able to log in.

Master-Koy, 2024-01-24:
> A 6.7.1 kernel here with sdcard + nfs root on RPiOS is rebooting to desktop okay on a Pi5. So I think there are more factors involved than just 6.7.1 + Pi5 doesn't reboot properly.
> 
> I'll keep an eye for any issues. Are you seeing this on RPiOS? Do you also see it with an sdcard boot?

Hi, please help me with my issue also.

https://github.com/Master-Koy/-Boot-Error-Stuck-at-Boot-in-Kali-Linux-in-Vmware

pelwell, 2024-01-24:
<deleted off-topic message>

Dark-Sky, 2024-01-24:
I forgot to mention with the sdcard test above with KDE I could never get it to boot into the DE or never saw the login screen.

Dark-Sky, 2024-01-25:
More tests today with the KDE sdcard install. I was able to get to the logs when it fails to load the login screen:

xorg-log Pi5: https://pastebin.com/npRcpQbX
dmesg Pi5: https://pastebin.com/6BGetDDU
journalctl Pi5: https://pastebin.com/CkjgAHfw

I then put the same sdcard in my pi4 with the pi4 kernel using the same 6.7.1  kernel version and it booted up right away with out issue.

I have a VIZIO tv for a monitor. With some kernels in the past I had to load a edid.dat file in cmdline .txt and do some custom monitor setting in config.txt to make things work right. Honestly this exhibits some the same issues with this pi5 kernel but adding this makes no difference.

popcornmix, 2024-01-25:
I wonder if you could be hitting a form of [this issue](https://forums.raspberrypi.com/viewtopic.php?t=358853)?
Can you try manually adding `/etc/X11/xorg.conf.d/ 99-vc4.conf`:
```
Section "OutputClass"
  Identifier "vc4"
  MatchDriver "vc4"
  Driver "modesetting"
  Option "PrimaryGPU" "true"
EndSection
```

Dark-Sky, 2024-01-25:
> I wonder if you could be hitting a form of [this issue](https://forums.raspberrypi.com/viewtopic.php?t=358853)? Can you try manually adding `/etc/X11/xorg.conf.d/99-vc4.conf`:
>
> ```
> Section [...]

It is already there.

Dark-Sky, 2024-01-26:
> Are you seeing this on RPiOS?

Today I installed RPiOS and did all the updates . Then I installed 6.7.1 and rebooted. All I got was a black screen with a blinking cursor. The I loaded my edid.dat file in cmdline.txt and specified perimeters for my vizio tv in config.txt and I still got the black screen and a blinking cursor.

I then switched TTY's and ran startx and the desktop loaded up right away and all seemed ok within the desktop.

Dark-Sky, 2024-01-26:
I have narrowed it down to this. For some reason the 6.7.1 kernel fails to start SDDM and lightdm display managers. I finally saw an error pop up saying "Failed to start lightdm display manager". If I disable the display managers and login at the login prompt and use startx the desktop loads up with no problem.

Since you all have no issue this leaves me with the conclusion that my vizio tv may be causing the issue but I can not test as it is the only monitor I have right now.

6by9, 2024-01-30:
Is the vc4 DRM driver actually loading then, or is your system dropping back to fbdev?

Checking for the existence of `/dev/dri/card0` and `/dev/dri/card1` (one will be v3d and one vc4) would be the quickest route to check. Looking at `/sys/kernel/debug/device_component/*` for unbound devices would be the next step, and for deferred devices in `/sys/kernel/debug/devices_deferred`.

Dark-Sky, 2024-01-30:
@6by9 This is when It boots into the desktop like it is supposed to:

```
[ray@jellyfin ~]$ ls /dev/dri/card0
/dev/dri/card0
[ray@jellyfin ~]$ ls /dev/dri/card1
/dev/dri/card1
[ray@jellyfin ~]$ sudo ls /sys/kernel/debug/device_component/*
ls: cannot access '/sys/kernel/debug/device_component/*': No such file or directory
[ray@jellyfin ~]$ sudo ls /sys/kernel/debug/device_component/
axi:gpu
[ray@jellyfin ~]$ sudo ls /sys/kernel/debug/devices_deferred
/sys/kernel/debug/devices_deferred
```

Dark-Sky, 2024-01-30:
> cat would be more useful than ls:

```
[ray@jellyfin ~]$ sudo cat /sys/kernel/debug/device_component/axi:gpu
aggregate_device name                                  status
-------------------------------------------------------------
axi:gpu                                                 bound

device name                                            status
-------------------------------------------------------------
107c580000.hvs                                          bound
107c701400.hdmi                                         bound
107c706400.hdmi                                         bound
107c500000.mop                                          bound
107c501000.moplet                                       bound
107c410000.pixelvalve                                   bound
107c411000.pixelvalve                                   bound
[ray@jellyfin ~]$ 
[ray@jellyfin ~]$ sudo cat /sys/kernel/debug/devices_deferred
[ray@jellyfin ~]$
```

popcornmix, 2024-01-30:
Looks the same as I get on a working (RPiOS 6.6 kernel) system.

Dark-Sky, 2024-01-30:
I just installed and tested 6.7.2 and noticed there were no micro blanking at the Window Manager login screen so I uninstalled `xf86-video-fbdev` and rebooted and there was no issue even after rebooting several times. So looks like the issue went away in 6.7.2.

popcornmix, 2024-01-30:
Based on that, possibly this? https://github.com/raspberrypi/linux/pull/5833

It's not immediately obvious how that caused your boot failure (perhaps a lack of cma?),
but that a slightly relevant change between 6.7.1 and 6.7.2

Dark-Sky, 2024-01-30:
I had cma set to 512 and no gpu_mem specified in config.txt.

closed 2024-01-31 by Dark-Sky