raspberrypi/linux #5893: incorrect sound in speaker-test with format S24_3LE and S32_LE

qrp73, 2024-01-25:
### Describe the bug

When using speaker-test command with sample format S24_3LE or S32_LE, the displayed message doesn't match with actually playing sound

### Steps to reproduce the behaviour

1) Insert USB sound card which supports S24_3LE or S32_LE with 384000 Hz sample rate

2) execute this command and make sure that the sound match with displayed message (as expected):
```
speaker-test -t sine -F S16_LE -r 384000 -c 2
```

3) now execute this command:
```
speaker-test -t sine -F S24_3LE -r 384000 -c 2
```

Expected result: the sound match with displayed message

Actual result: when message "0 - Front Left" is displayed, the sound is played on left channel and then on right channel. Then when message "0 - Front Left" is repeated, the sound is still played on right channel. The bug...

The same issue with this command:
```
speaker-test -t sine -F S32_LE -r 384000 -c 2
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 40f37458ae7cadea1aec913ae10b5e7008ebce0a, stage4

$ vcgencmd version
Dec  8 2023 12:36:15 
Copyright (c) 2012 Broadcom
version e02d33b3122450accf9dea471a177d3b5623f5ad (clean) (release) (start)

$ uname -a
Linux raspi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

### Additional context

I'm using GraveAudio DA06 CX31993 sound card for USB bought on aliexpress.

Also I notice that when I running speaker-test with sample rate 384000, the sound card hardware is running at 48000 Hz:
```
$ cat /proc/asound/card2/stream0
Synaptics CX31993 384Khz HIFI AUDIO at usb-0000:01:00.0-1.4.4, high speed : USB Audio

Playback:
  Status: Running
    Interface = 1
    Altset = 3
    Packet Size = 72
    Momentary freq = 48000 Hz (0x6.0000)
  Interface 1
    Altset 1
    Format: S16_LE
    Channels: 2
    Endpoint: 0x01 (1 OUT) (SYNC)
    Rates: 44100, 48000, 96000, 192000, 384000
    Data packet interval: 125 us
    Bits: 16
    Channel map: FL FR
  Interface 1
    Altset 2
    Format: S24_3LE
    Channels: 2
    Endpoint: 0x01 (1 OUT) (SYNC)
    Rates: 44100, 48000, 96000, 192000, 384000
    Data packet interval: 125 us
    Bits: 24
    Channel map: FL FR
  Interface 1
    Altset 3
    Format: S32_LE
    Channels: 2
    Endpoint: 0x01 (1 OUT) (SYNC)
    Rates: 44100, 48000, 96000, 192000, 384000
    Data packet interval: 125 us
    Bits: 32
    Channel map: FL FR

Capture:
  Status: Stop
  Interface 2
    Altset 1
    Format: S16_LE
    Channels: 1
    Endpoint: 0x81 (1 IN) (SYNC)
    Rates: 44100, 48000, 96000
    Data packet interval: 125 us
    Bits: 16
    Channel map: FL
  Interface 2
    Altset 2
    Format: S24_3LE
    Channels: 1
    Endpoint: 0x81 (1 IN) (SYNC)
    Rates: 44100, 48000, 96000
    Data packet interval: 125 us
    Bits: 24
    Channel map: FL
```

But I'm not sure if this is a bug or there is needs to configure sample rate somewhere? I tried to setup 384000 Hz in /etc/pulse/daemon.conf but it doesn't helps, it still working at 48000 Hz... Any idea on how to fix it?

popcornmix, 2024-01-25:
I assume you are on bookworm?

If so pipewire will be enabled by default and will control output samplerate (as it potentially mixes audio from many different clients).

Look here:
```
cat /usr/share/pipewire/pipewire.conf
...
default.clock.allowed-rates = [ 48000 ]
```
and try adding additional samplerates.

qrp73, 2024-01-25:
> Look here:
> cat /usr/share/pipewire/pipewire.conf
> default.clock.allowed-rates = [ 48000 ]

Thanks a lot it helps! 
I uncommented this line and added new sample rates supported by my card:
```
default.clock.allowed-rates = [ 44100, 48000, 96000, 192000, 384000 ]
```
after reboot the card sample rate is automatically selected depends on requested sample rate. Now RtAudio works with proper sample rate with no resampling.

It was real headache to find how to fix it :)

qrp73, 2024-01-26:
Just tested original issue (with not matching displaying message and playing sound) and it seems that this is an issue of speaker-test, because the same issue happens on ubuntu running on intel x86-64 platform. 

Since it don't related to raspberry pi, I will close this issue.

closed 2024-01-26 by qrp73