raspberrypi/linux #5896: pcie-32bit-dma overlay does not work on Pi 5's external PCIe interface

RodrigoRosmaninho, 2024-01-25:
### Describe the bug

This is the same issue as #4197 (that was fixed by #4216).
However, that previous fix (the pcie-32bit-dma dtb overlay) is only compatible with CM4 and has no effect on Pi 5.

I tested this on a Raspberry Pi 5 connected to Pineberry Pi's "Hat AI!", to which I plugged in a MikroTik R11e-5HnD (Atheros AR9580) wireless card.

### Steps to reproduce the behaviour

Connect a PCIe device that can only use 32bit DMA addresses in order to function correctly, such as a wireless network card using Atheros AR9580.

### Device (s)

Raspberry Pi 5

### System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4

Linux opportunity 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

### Logs

The symptoms are exactly the same as seen here: https://github.com/geerlingguy/raspberry-pi-pcie-devices/issues/313

RodrigoRosmaninho, 2024-01-25:
I have created a PR (#5897) to add a new pcie-32bit-dma-overlay-pi5 dts overlay that fixes this issue.
I've confirmed that this works correctly on my setup (described above).

pelwell, 2024-01-25:
Closing this as unnecessary - the PR is sufficient.

closed 2024-01-25 by pelwell