raspberrypi/linux #5909: configs: enable CONFIG_NVME_HWMON

aboulfad, 2024-01-29:
### Describe the bug

NVMe HWMON support allows reporting the current NVMe drive temperature sensor(s) and min/max thresholds via this kernel infrastructure without the need for any utilities, no root requirement. This would be highly desirable for rpi5.

Please add, if possible the following in the kernel configs:
[CONFIG_NVME_HWMON=y](url=https://cateee.net/lkddb/web-lkddb/NVME_HWMON.html)

### Steps to reproduce the behaviour

`
grep CONFIG_NVME_HWMON /boot/config-6.1.0-rpi7-rpi-2712 
# CONFIG_NVME_HWMON is not set
`

### Device (s)

Raspberry Pi 5

### System

cat /etc/rpi-issue 
Raspberry Pi reference 2023-12-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 2acf7afcba7d11500313a7b93bb55a2aae20b2d6, stage2

vcgencmd version
2024/01/05 15:57:40 
Copyright (c) 2012 Broadcom
version 30cc5f37 (release) (embedded)

uname -a
Linux rpi5 6.1.0-rpi7-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

bit-bash, 2024-01-29:
When the "official" NVMe HAT (or maybe HAB, fingers crossed) comes out, I would be interested in this as well.
Based on different cooling setups, cases, and environments, it would be nice to check the NVMe status during a new system break in period.

aboulfad, 2024-01-29:
I compiled a kernel with that config and it works with the nvme base I use (pimoroni) & Crucial nvme. 
It shouldn’t depend on the official hat release. It’s a standard Linux kernel config that I think was added in 5.5, it’s odd that it wasn’t added by default given rpi5 pcie nvme support.

pelwell, 2024-01-30:
> it’s odd that it wasn’t added by default given rpi5 pcie nvme support.

Are we weird for not being that interested in the temperature of an NVME drive?

It's a shame that it's a built-in and not a module, but it seems to add less than a kilobyte to the uncompressed kernel. That's now in the rpi-6.6.y kernel defconfigs (rpi-6.1.y is essentially on life-support) - see 72b7716fcc788.

aboulfad, 2024-01-30:
Thank you for the addition & I don't understand the  rpi-6.1.y comment, isn't that the kernel being shipped/used with rpi5 ? (excuse my kernel ignorance) or do I need to update.

PS: @pelwell my apologies for poor choice of words (odd), meant to say it's peculiar or rather intriguing.