raspberrypi/linux #5911: rtw88 and rtw89 usb wireless modules are not enabled in kernel

arekm, 2024-01-30:
### Describe the bug

The raspberry pi 4 wifi driver/firmware is known to be buggy and no one is actively working on fixing it (looking at bug #3849 comments).

So we need to be able to use USB cards as alternatives.

Unfortunately kernel lacks support for some of these. Testing "next" 6.6.13 kernel.

Firmware files for these cards are already available in pi os:

````
root@octopi-prusa:~# dpkg -L firmware-realtek |grep -i rtw
/lib/firmware/rtw88
/lib/firmware/rtw88/rtw8723d_fw.bin
/lib/firmware/rtw88/rtw8821c_fw.bin
/lib/firmware/rtw88/rtw8822b_fw.bin
/lib/firmware/rtw88/rtw8822c_fw.bin
/lib/firmware/rtw88/rtw8822c_wow_fw.bin
/lib/firmware/rtw89
/lib/firmware/rtw89/rtw8852a_fw.bin
/lib/firmware/rtw89/rtw8852b_fw.bin
/lib/firmware/rtw89/rtw8852c_fw.bin
````

but not drivers

````
root@octopi-prusa:~# zcat /proc/config.gz |grep RTW
# CONFIG_RTW88 is not set
# CONFIG_RTW89 is not set
root@octopi-prusa:~# lsusb|grep -i realt
Bus 001 Device 006: ID 0bda:b812 Realtek Semiconductor Corp. RTL88x2bu [AC1200 Techkey]
````

Please enable these in kernel config as there is no other alternative it seems than using usb wifi cards.

### Steps to reproduce the behaviour

1) boot 6.6 kernel
````
# uname -a
Linux octopi-prusa 6.6.13-v8+ #1724 SMP PREEMPT Mon Jan 22 13:09:57 GMT 2024 aarch64 GNU/Linux
````

2) 
````
root@octopi-prusa:~# modinfo rtw88
modinfo: ERROR: Module rtw88 not found.
````

### Device (s)

Raspberry Pi 4 Mod. B

### System

root@octopi-prusa:~# cat /etc/rpi-issue
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage2
root@octopi-prusa:~# vcgencmd version
Dec  8 2023 12:37:10
Copyright (c) 2012 Broadcom
version e02d33b3122450accf9dea471a177d3b5623f5ad (clean) (release) (start_x)
root@octopi-prusa:~# uname -a
Linux octopi-prusa 6.6.13-v8+ #1724 SMP PREEMPT Mon Jan 22 13:09:57 GMT 2024 aarch64 GNU/Linux

pelwell, 2024-01-30:
It's not quite that simple. RTW88, despite what the Kconfig file says, does not create rtw88.ko - it's only a gate for a number of different RTW88* drivers, some of which are USB-, some PCIe- and some SDIO-based. RTW89 is similar, except that the devices are only PCIe-based.

I propose to enable only the USB-based devices, requiring the following defconfig changes:
```diff
+CONFIG_RTW88=m
+CONFIG_RTW88_8822BU=m
+CONFIG_RTW88_8822CU=m
+CONFIG_RTW88_8723DU=m
+CONFIG_RTW88_8821CU=m
```
That results in 10(!) modules, totalling 1.1MB, but fortunately they compress down to 330KB (for each build).

See https://github.com/raspberrypi/linux/pull/5913. After about 45 minutes you'll be able to run `sudo rpi-update pulls/5913` to install a trial kernel including the modules (beta software, backup your data, etc.).

arekm, 2024-01-30:
Makes sense.

Works, thanks!

````
root@octopi-prusa:~# uname -a
Linux octopi-prusa 6.6.14-v8+ #1 SMP PREEMPT Tue Jan 30 17:42:18 UTC 2024 aarch64 GNU/Linux
root@octopi-prusa:~# dmesg|grep -i rtw
[    9.072599] rtw_8822bu 1-1.4:1.0: Firmware version 27.2.0, H2C version 13
[    9.834148] usbcore: registered new interface driver rtw_8822bu
root@octopi-prusa:~# ethtool -i wlan1
driver: rtw_8822bu
version: 6.6.14-v8+
firmware-version: N/A
expansion-rom-version:
bus-info: 1-1.4:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
````

iperf 120s result:

````
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-120.00 sec  2.31 GBytes   165 Mbits/sec    6             sender
[  5]   0.00-120.05 sec  2.31 GBytes   165 Mbits/sec                  receiver
````

closed 2024-01-30 by pelwell